pyusb/pyusb #325: self.dev.set_configuration() Git is the latest version

WanManito, 2020-09-10:
In 1.0.2, I run 'usb.util. Dispose_Resources (self.dev)' to release the resource,Then connect again without any problems
In the latest version of Git,Reconnection after the above operation will die at "self.dev.set_Configuration ()",
After testing, it should be me that releases the resource, and the child thread is still waiting "self.dev.read(self.inpoint, 64)",
An additional error exception is then fired,Then connect again at set Configuration and it gets stuck,Because in 1.0.2, this works fine,Just switching to the latest version creates this problem

jonasmalacofilho, 2020-09-12:
> and the child thread is still waiting "self.dev.read(self.inpoint, 64)",

Calling `dispose_resources` on one thread while another tries to read from the device will lead to unexpected or undefined results.  We make no assurances that `Device` instances can be shared between threads.

> An additional error exception is then fired,

Can you provide both exceptions?

Additionally, were you already ignoring an exception to make 1.0.2 "work"?

---

Please see if you can `git bisect` the problem to a specific commit that introduces this change in behavior.

WanManito, 2020-09-14:
[image]
Hello, this is my application mode. I run the "close function" and can't connect again

jonasmalacofilho, 2020-09-16:
> An **additional** error exception is then fired,

Can you provide both exceptions?

And were you already ignoring an exception to make 1.0.2 "work"?

---

Please also see if you can `git bisect` the problem to a specific commit that introduces this change in behavior.

WanManito, 2020-09-17:
Hello, I guess the above problem in version 1.1 is due to multithreading programming. The figure above is the way I implemented it. I don't know why. The following figure is the additional exception
[image]

WanManito, 2020-09-22:
Could you give me some Pointers

WanManito, 2020-09-22:
Could you give me some Pointers @jonasmalacofilho

jonasmalacofilho, 2020-09-22:
Ideally I would have the reader thread "own" the device handle, and add an abstraction to allow other threads to ask the reader to: (a) stop trying to read; and (b) unclaim the interface and otherwise release the device.

The specifics on how to implement this will vary a bit depending on what else your software has to do; I also don't know of the top of my head the best way to implement this using libusb synchronous APIs (which are the only ones PyUSB supports right now).

---

Another approach is to simply get a new handle after the problem happens (so that you don't get stuck in `set_configuration`).

Again, I think you really need should solve your concurrency (because it is _not_ sound to access the handle from multiple threads), but maybe keeping this other approach in mind may help if you need a hack to overcome our limitation of only supporting the synchronous APIs.

WanManito, 2020-09-23:
@jonasmalacofilho Thank you for your reply. I get the dev object in the child thread and release the resource when the thread stops

closed 2020-09-28 by jonasmalacofilho