pyusb/pyusb #326: usb.core.find Running time

WanManito, 2020-09-14:
[image]
[image]
I seem to feel that the "find" function takes more time to update to the latest version. I don't know if it's my illusion

jonasmalacofilho, 2020-09-16:
Cannot reproduce on my (real) Linux machine:

```
$ ./timeit.sh
Using PyUSB 1.0.2
50000000 loops, best of 5: 4.64 nsec per loop
```

```
$ ./timeit.sh
Using PyUSB 1.1.0.post2+g6321b838432f
50000000 loops, best of 5: 4.63 nsec per loop
```

I will try this again on a (real) Windows machine and a couple of other Linux machines.

At the same time, please also use the script bellow to check whether it's just your "illusion" (run it once under PyUSB 1.0.2, and another time under 1.1.0 or git).

---

```
$ cat timeit.sh
#!/bin/bash

python -c "import usb; print(f'Using PyUSB {usb.__version__}')"
python -m timeit -s "import usb; list(usb.core.find(find_all=True))"
```

WanManito, 2020-09-17:
[image]
It's all of a sudden normal again, I don't know what's going on

closed 2020-09-28 by jonasmalacofilho