pyusb/pyusb #327: Clean up pyusb website: https://pyusb.github.io/pyusb/

mcuee, 2020-09-17:
A few clean up is needed.

1) libusb-1.0 link needs to change to "https://libusb.info/"
2) libusbx needs to be removed.
3) better to remove openusb as well.

mcuee, 2020-09-17:
I am not familiar with github pages yet. It seems to be a special repo and I can not see it.

For libusb, there is a git repo under libusb organization.
https://github.com/libusb/libusb.github.com

mcuee, 2020-09-17:
Another thing, it still mentions Python 2.4. I am not so sure if we want to support that. Probably we can change the sentence to "but most things should still work with Python 2.7".

jonasmalacofilho, 2020-09-17:
Agreed, it needs some cleaning up.  It and the README in this repository also need to more closely match.

I'll do that in the next few days.

clach04, 2020-09-28:
> 
> 
> I am not familiar with github pages yet. It seems to be a special repo and I can not see it.
> 
> For libusb, there is a git repo under libusb organization.
> https://github.com/libusb/libusb.github.com

Content in gh-pages branch https://github.com/pyusb/pyusb/tree/gh-pages is where the site https://pyusb.github.io/pyusb/ gets its source. PRs can be opened against that branch :)

mcuee, 2020-09-30:
I see. Thanks. I have just updated a bit on the github pages.

mcuee, 2020-09-30:
Close this for now. We can continue to update the git pages.

closed 2020-09-30 by mcuee