pyusb/pyusb #328: pyusb 1.1.0 installation fails with Python 3

DrumFrog, 2020-09-17:
Installing under Python 3.8.2 (Ubuntu 20.04) or 3.7.2 (Windows 7) throws not successful:

```
pip install pyusb==1.1.0
Collecting pyusb==1.1.0
  Downloading https://artifactory***.com/artifactory/api/pypi/pypi/packages/packages/12/9b/8f5be839753667c39fe522162bea7f8121f28ba49c5ad1e5681681967c79/pyusb-1.1.0.tar.gz (78 kB)
     |████████████████████████████████| 78 kB 2.6 MB/s
  Installing build dependencies ... error
  ERROR: Command errored out with exit status 1:
   command: /path/to/python/bin/python /path/to/python/lib/python3.8/site-packages/pip install --ignore-installed --no-user --prefix /tmp/pip-build-env-nj74sse1/overlay --no-warn-script-location --no-binary :none: --only-binary :none:  -- 'setuptools>=30.3.0' wheel setuptools_scm
       cwd: None
  Complete output (39 lines):
  Traceback (most recent call last):
    File "/usr/lib/python3.8/runpy.py", line 193, in _run_module_as_main
      return _run_code(code, main_globals, None,
    File "/usr/lib/python3.8/runpy.py", line 86, in _run_code
      exec(code, run_globals)
    File "/path/to/python/lib/python3.8/site-packages/pip/__main__.py", line 23, in <module>
      from pip._internal.cli.main import main as _main  # isort:skip # noqa
    File "/path/to/python/lib/python3.8/site-packages/pip/_internal/cli/main.py", line 5, in <module>
      import locale
    File "/usr/lib/python3.8/locale.py", line 16, in <module>
      import re
    File "/usr/lib/python3.8/re.py", line 143, in <module>
      class RegexFlag(enum.IntFlag):
  AttributeError: module 'enum' has no attribute 'IntFlag'
  Error in sys.excepthook:
  Traceback (most recent call last):
    File "/usr/lib/python3/dist-packages/apport_python_hook.py", line 62, in apport_excepthook
      if not enabled():
    File "/usr/lib/python3/dist-packages/apport_python_hook.py", line 24, in enabled
      import re
    File "/usr/lib/python3.8/re.py", line 143, in <module>
      class RegexFlag(enum.IntFlag):
  AttributeError: module 'enum' has no attribute 'IntFlag'

  Original exception was:
  Traceback (most recent call last):
    File "/usr/lib/python3.8/runpy.py", line 193, in _run_module_as_main
      return _run_code(code, main_globals, None,
    File "/usr/lib/python3.8/runpy.py", line 86, in _run_code
      exec(code, run_globals)
    File "/path/to/python/lib/python3.8/site-packages/pip/__main__.py", line 23, in <module>
      from pip._internal.cli.main import main as _main  # isort:skip # noqa
    File "/path/to/python/lib/python3.8/site-packages/pip/_internal/cli/main.py", line 5, in <module>
      import locale
    File "/usr/lib/python3.8/locale.py", line 16, in <module>
      import re
    File "/usr/lib/python3.8/re.py", line 143, in <module>
      class RegexFlag(enum.IntFlag):
  AttributeError: module 'enum' has no attribute 'IntFlag'
  ----------------------------------------
ERROR: Command errored out with exit status 1: /path/to/python/bin/python /path/to/python/lib/python3.8/site-packages/pip install --ignore-installed --no-user --prefix /tmp/pip-build-env-nj74sse1/overlay --no-warn-script-location --no-binary :none: --only-binary :none: -- 'setuptools>=30.3.0' wheel setuptools_scm Check the logs for full command output.pip install pyusb==1.1.0
Collecting pyusb==1.1.0
  Downloading https://artifactory***.com/artifactory/api/pypi/pypi/packages/packages/12/9b/8f5be839753667c39fe522162bea7f8121f28ba49c5ad1e5681681967c79/pyusb-1.1.0.tar.gz (78 kB)
     |████████████████████████████████| 78 kB 2.6 MB/s
  Installing build dependencies ... error
  ERROR: Command errored out with exit status 1:
   command: /path/to/python/bin/python /path/to/python/lib/python3.8/site-packages/pip install --ignore-installed --no-user --prefix /tmp/pip-build-env-nj74sse1/overlay --no-warn-script-location --no-binary :none: --only-binary :none:  -- 'setuptools>=30.3.0' wheel setuptools_scm
       cwd: None
  Complete output (39 lines):
  Traceback (most recent call last):
    File "/usr/lib/python3.8/runpy.py", line 193, in _run_module_as_main
      return _run_code(code, main_globals, None,
    File "/usr/lib/python3.8/runpy.py", line 86, in _run_code
      exec(code, run_globals)
    File "/path/to/python/lib/python3.8/site-packages/pip/__main__.py", line 23, in <module>
      from pip._internal.cli.main import main as _main  # isort:skip # noqa
    File "/path/to/python/lib/python3.8/site-packages/pip/_internal/cli/main.py", line 5, in <module>
      import locale
    File "/usr/lib/python3.8/locale.py", line 16, in <module>
      import re
    File "/usr/lib/python3.8/re.py", line 143, in <module>
      class RegexFlag(enum.IntFlag):
  AttributeError: module 'enum' has no attribute 'IntFlag'
  Error in sys.excepthook:
  Traceback (most recent call last):
    File "/usr/lib/python3/dist-packages/apport_python_hook.py", line 62, in apport_excepthook
      if not enabled():
    File "/usr/lib/python3/dist-packages/apport_python_hook.py", line 24, in enabled
      import re
    File "/usr/lib/python3.8/re.py", line 143, in <module>
      class RegexFlag(enum.IntFlag):
  AttributeError: module 'enum' has no attribute 'IntFlag'

  Original exception was:
  Traceback (most recent call last):
    File "/usr/lib/python3.8/runpy.py", line 193, in _run_module_as_main
      return _run_code(code, main_globals, None,
    File "/usr/lib/python3.8/runpy.py", line 86, in _run_code
      exec(code, run_globals)
    File "/path/to/python/lib/python3.8/site-packages/pip/__main__.py", line 23, in <module>
      from pip._internal.cli.main import main as _main  # isort:skip # noqa
    File "/path/to/python/lib/python3.8/site-packages/pip/_internal/cli/main.py", line 5, in <module>
      import locale
    File "/usr/lib/python3.8/locale.py", line 16, in <module>
      import re
    File "/usr/lib/python3.8/re.py", line 143, in <module>
      class RegexFlag(enum.IntFlag):
  AttributeError: module 'enum' has no attribute 'IntFlag'
  ----------------------------------------
ERROR: Command errored out with exit status 1: /path/to/python/bin/python /path/to/python/lib/python3.8/site-packages/pip install --ignore-installed --no-user --prefix /tmp/pip-build-env-nj74sse1/overlay --no-warn-script-location --no-binary :none: --only-binary :none:  -- 'setuptools>=30.3.0' wheel setuptools_scm Check the logs for full command output.
```

jonasmalacofilho, 2020-09-17:
This doesn't seem to be related to PyUSB at all: the errors are in the `re` module.

```
    File "/usr/lib/python3.8/re.py", line 143, in <module>
      class RegexFlag(enum.IntFlag):
  AttributeError: module 'enum' has no attribute 'IntFlag'
```

Additionally, installing with `pip` works on my machines (Python 3.8.5 on ArchLinux, Python 3.8.3 on Windows) and in our CI.

Perhaps your Python installation is broken?

jonasmalacofilho, 2020-09-28:
Cannot reproduce even with the reported Ubuntu and Python versions:

```
$ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 20.04.1 LTS
Release:	20.04
Codename:	focal

$ uname -rv
5.4.0-45-generic #49-Ubuntu SMP Wed Aug 26 13:38:52 UTC 2020

$ python3 --version
Python 3.8.2

$ pip3 install pyusb
Collecting pyusb
  Downloading pyusb-1.1.0.tar.gz (78 kB)
     |████████████████████████████████| 78 kB 5.7 MB/s 
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
    Preparing wheel metadata ... done
Building wheels for collected packages: pyusb
  Building wheel for pyusb (PEP 517) ... done
  Created wheel for pyusb: filename=pyusb-1.1.0-py3-none-any.whl size=58009 sha256=89215af91863d2b638acfa55837cfc03f5e2bbd3106de6a0b5781d7b2e0a8e44
  Stored in directory: /root/.cache/pip/wheels/c5/ab/d8/68f0e719235c5a5f5b0ddb3ef12f3cbe7e2c2706ff56f3f26b
Successfully built pyusb
Installing collected packages: pyusb
Successfully installed pyusb-1.1.0
```

My previous comment has also been left unanswered, so I'm assuming the issue is really _unrelated_ to PyUSB (and, possibly, already solved).  But feel free to add more details if applicable.

closed 2020-09-28 by jonasmalacofilho

DrumFrog, 2020-09-29:
sorry for not answering - I'll check once more

twslankard, 2020-11-18:
This issue appears to be due to an incompatibility between `enum34` which backports Python 3.4's built-in `enum`.

When installing `pyusb` in an environment that has `enum34` already installed, I get the error described in this issue. After removing `enum34`, installation succeeds.

twslankard, 2020-11-18:
Here are steps to reproduce:

```
user@example:~/test$ virtualenv -ppython3.8 venv
created virtual environment CPython3.8.5.final.0-64 in 188ms
  creator CPython3Posix(dest=[redacted], clear=False, global=False)
  seeder FromAppData(download=False, packaging=latest, ipaddr=latest, html5lib=latest, pep517=latest, wheel=latest, urllib3=latest, webencodings=latest, contextlib2=latest, chardet=latest, distlib=latest, appdirs=latest, msgpack=latest, idna=latest, distro=latest, lockfile=latest, pkg_resources=latest, pyparsing=latest, requests=latest, setuptools=latest, progress=latest, six=latest, pip=latest, colorama=latest, certifi=latest, retrying=latest, pytoml=latest, CacheControl=latest, via=copy, app_data_dir=[redacted])
  activators BashActivator,CShellActivator,FishActivator,PowerShellActivator,PythonActivator,XonshActivator

user@example:~/test$ . venv/bin/activate

(venv) user@example:~/test$ pip uninstall enum34
Found existing installation: enum34 1.1.10
Uninstalling enum34-1.1.10:
  Would remove:
    [redacted]/test/venv/lib/python3.8/site-packages/enum/*
    [redacted]/test/venv/lib/python3.8/site-packages/enum34-1.1.10.dist-info/*
Proceed (y/n)? y
  Successfully uninstalled enum34-1.1.10

(venv) user@example:~/test$ pip install pyusb
Collecting pyusb
  Using cached pyusb-1.1.0.tar.gz (78 kB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
    Preparing wheel metadata ... done
Building wheels for collected packages: pyusb
  Building wheel for pyusb (PEP 517) ... done
  Created wheel for pyusb: filename=pyusb-1.1.0-py3-none-any.whl size=58009 sha256=f355c1bc6799d5fceca86d98742d5cd628e6e4923de1cd265af5c3df132bbaa7
  Stored in directory: [redacted]/.cache/pip/wheels/c5/ab/d8/68f0e719235c5a5f5b0ddb3ef12f3cbe7e2c2706ff56f3f26b
Successfully built pyusb
Installing collected packages: pyusb
Successfully installed pyusb-1.1.0
```

jonasmalacofilho, 2020-11-18:
I still cannot reproduce the problem (albeit with a different Python version), but it makes sense.

The `enum34` package may replace Python's standard library `enum`, but only with the types that existed in Python 3.4.  This means it's missing `IntFlag`, which breaks anything that tries to use it, in this case Python's standard `re` module.