pyusb/pyusb #333: show_devices() segfaults when run twice

btigner, 2020-10-09:
OS FreeBSD 12.1 AMD64
Python 3.7.9
pyusb 1.0.0
code:
python 
>>> from usb.core import show_devices
>>> show_devices()
DEVICE ID 0403:6014 on Bus 000 Address 010, Specified at interface
DEVICE ID 0403:6015 on Bus 000 Address 009, Specified at interface
DEVICE ID 093a:2510 on Bus 000 Address 008, Specified at interface
DEVICE ID 05e3:0608 on Bus 000 Address 007, Hub
DEVICE ID 1462:7c37 on Bus 000 Address 006, Specified at interface
DEVICE ID 413c:2107 on Bus 000 Address 005, Specified at interface
DEVICE ID 0000:0000 on Bus 000 Address 004, Hub
DEVICE ID 0000:0000 on Bus 000 Address 003, Hub
DEVICE ID 0000:0000 on Bus 000 Address 002, Hub
DEVICE ID 0000:0000 on Bus 000 Address 001, Hub

>>> show_devices()
Segmentation fault (core dumped)

This happens every time. I thought it was a problem with pyftdi, but I now think the 
problem is in pyusb.

Any suggestions, or help would be appreciated.

btigner, 2020-10-09:
Sorry, I mistyped the version of pyusb, it is 1.1.0 , not 1.0.0. my apologies.

jonasmalacofilho, 2020-10-09:
Can you run this with `PYUSB_DEBUG=debug LIBUSB_DEBUG=4`?

btigner, 2020-10-16:
root@X4:/home/barry # set PYUSB_DEBUG=debug
root@X4:/home/barry # set LIBUSB_DEBUG=4
root@X4:/home/barry # python
Python 3.7.9 (default, Oct  3 2020, 01:29:35) 
[Clang 8.0.1 (tags/RELEASE_801/final 366581)] on freebsd12
Type "help", "copyright", "credits" or "license" for more information.
>>> from usb.core import show_devices
>>> show_devices()
DEVICE ID 093a:2510 on Bus 000 Address 008, Specified at interface
DEVICE ID 05e3:0608 on Bus 000 Address 007, Hub
DEVICE ID 1462:7c37 on Bus 000 Address 006, Specified at interface
DEVICE ID 413c:2107 on Bus 000 Address 005, Specified at interface
DEVICE ID 0000:0000 on Bus 000 Address 004, Hub
DEVICE ID 0000:0000 on Bus 000 Address 003, Hub
DEVICE ID 0000:0000 on Bus 000 Address 002, Hub
DEVICE ID 0000:0000 on Bus 000 Address 001, Hub

>>> show_devices()
Segmentation fault (core dumped)

I am running this in a csh shell.
pyusb 1.1.0
linux_libusb-13.0r358841_4

jonasmalacofilho, 2020-10-16:
Setting variables doesn't automatically _export_ them to sub processes.

You need to either

```
$ export PYUSB_DEBUG=debug
$ export LIBUSB_DEBUG=4
$ python -c "from usb.core import show_devices; print(show_devices()); print(show_devices())"
```

or, for a one-liner,

```
$ PYUSB_DEBUG=debug LIBUSB_DEBUG=4 python -c "from usb.core import show_devices; print(show_devices()); print(show_devices())"
```

Note: I also replaced your manual steps with a `python -c ...` invocation, but that doesn't change anything.

---

Why did you close the issue?

btigner, 2020-10-16:
I accidentally closed it and did not know how to undo that error.

I assume you are using the bash shell so I will try it tonight.

Barry

btigner, 2020-10-16:
ok, here it is...

[root@X4 /home/barry]# export PYUSB_DEBUG=debug
[root@X4 /home/barry]# export LIBUSB_DEBUG=4
[root@X4 /home/barry]# python -c "from usb.core import show_devices; print(show_devices()); print(show_devices())"
2020-10-16 17:56:00,804 ERROR:usb.backend.libusb1:Error loading libusb 1.0 backend
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/site-packages/usb/backend/libusb1.py", line 962, in get_backend
    _setup_prototypes(_lib)
  File "/usr/local/lib/python3.7/site-packages/usb/backend/libusb1.py", line 311, in _setup_prototypes
    lib.libusb_get_parent.argtypes = [c_void_p]
  File "/usr/local/lib/python3.7/ctypes/__init__.py", line 377, in __getattr__
    func = self.__getitem__(name)
  File "/usr/local/lib/python3.7/ctypes/__init__.py", line 382, in __getitem__
    func = self._FuncPtr((name_or_ordinal, self))
AttributeError: Undefined symbol "libusb_get_parent"
2020-10-16 17:56:00,815 ERROR:usb.libloader:'OpenUSB library' could not be found
2020-10-16 17:56:00,815 ERROR:usb.backend.openusb:Error loading OpenUSB backend
2020-10-16 17:56:00,830 INFO:usb.core:find(): using backend "usb.backend.libusb0"
2020-10-16 17:56:00,830 DEBUG:usb.backend.libusb0:_LibUSB.enumerate_devices()
2020-10-16 17:56:00,830 DEBUG:usb.backend.libusb0:_LibUSB.get_device_descriptor(<usb.backend.libusb0._usb_device object at 0x801bf0f80>)
2020-10-16 17:56:00,830 DEBUG:usb.backend.libusb0:_LibUSB.get_device_descriptor(<usb.backend.libusb0._usb_device object at 0x801c7f0e0>)
2020-10-16 17:56:00,831 DEBUG:usb.backend.libusb0:_LibUSB.get_device_descriptor(<usb.backend.libusb0._usb_device object at 0x801c7f320>)
2020-10-16 17:56:00,831 DEBUG:usb.backend.libusb0:_LibUSB.get_device_descriptor(<usb.backend.libusb0._usb_device object at 0x801c7f440>)
2020-10-16 17:56:00,831 DEBUG:usb.backend.libusb0:_LibUSB.get_device_descriptor(<usb.backend.libusb0._usb_device object at 0x801c7f560>)
2020-10-16 17:56:00,831 DEBUG:usb.backend.libusb0:_LibUSB.get_device_descriptor(<usb.backend.libusb0._usb_device object at 0x801c7f680>)
2020-10-16 17:56:00,831 DEBUG:usb.backend.libusb0:_LibUSB.get_device_descriptor(<usb.backend.libusb0._usb_device object at 0x801c7f7a0>)
2020-10-16 17:56:00,831 DEBUG:usb.backend.libusb0:_LibUSB.get_device_descriptor(<usb.backend.libusb0._usb_device object at 0x801c7f8c0>)
DEVICE ID 093a:2510 on Bus 000 Address 008, Specified at interface
DEVICE ID 05e3:0608 on Bus 000 Address 007, Hub
DEVICE ID 1462:7c37 on Bus 000 Address 006, Specified at interface
DEVICE ID 413c:2107 on Bus 000 Address 005, Specified at interface
DEVICE ID 0000:0000 on Bus 000 Address 004, Hub
DEVICE ID 0000:0000 on Bus 000 Address 003, Hub
DEVICE ID 0000:0000 on Bus 000 Address 002, Hub
DEVICE ID 0000:0000 on Bus 000 Address 001, Hub

2020-10-16 17:56:00,831 DEBUG:usb.backend.libusb1:_LibUSB.__init__(<CDLL 'libusb.so.3', handle 801b79c00 at 0x8015d7b10>)
2020-10-16 17:56:00,831 INFO:usb.core:find(): using backend "usb.backend.libusb1"
2020-10-16 17:56:00,831 DEBUG:usb.backend.libusb1:_LibUSB.enumerate_devices()
Segmentation fault (core dumped)

jonasmalacofilho, 2020-10-16:
There are three things wrong here, at least.

```
# first run
2020-10-16 17:56:00,804 ERROR:usb.backend.libusb1:Error loading libusb 1.0 backend
AttributeError: Undefined symbol "libusb_get_parent"
2020-10-16 17:56:00,815 ERROR:usb.libloader:'OpenUSB library' could not be found
2020-10-16 17:56:00,815 ERROR:usb.backend.openusb:Error loading OpenUSB backend
2020-10-16 17:56:00,830 INFO:usb.core:find(): using backend "usb.backend.libusb0"
```

- The libusb1 shim you're using is missing a symbol.

```
# second run
2020-10-16 17:56:00,831 DEBUG:usb.backend.libusb1:_LibUSB.enumerate_devices()
```

- In the second run PyUSB is trying to use the libusb1 backend despite the previous failure to load it.  For some reason it now succeeds to load.

```
# second run
2020-10-16 17:56:00,831 DEBUG:usb.backend.libusb1:_LibUSB.enumerate_devices()
Segmentation fault (core dumped)
```

- No idea what's happening there, because the libusb1 shim doesn't appear to respond to the same debugging options.  Do you know if the underlying FreeBSD libusb has any debugging/tracing feature that you can enable?  Otherwise you may need to get a backtrace for this segfault.

btigner, 2020-10-17:
I don't know if this is what I should have done.
I downloaded libusb1 from github, ran autogen, and configure , and make install 
Making install in libusb
 .././install-sh -c -d '/usr/local/lib'
 /bin/sh ../libtool   --mode=install /usr/bin/install -c   libusb-1.0.la '/usr/local/lib'
libtool: install: /usr/bin/install -c .libs/libusb-1.0.so.0.2.0 /usr/local/lib/libusb-1.0.so.0.2.0
libtool: install: (cd /usr/local/lib && { ln -s -f libusb-1.0.so.0.2.0 libusb-1.0.so.0 || { rm -f libusb-1.0.so.0 && ln -s libusb-1.0.so.0.2.0 libusb-1.0.so.0; }; })
libtool: install: (cd /usr/local/lib && { ln -s -f libusb-1.0.so.0.2.0 libusb-1.0.so || { rm -f libusb-1.0.so && ln -s libusb-1.0.so.0.2.0 libusb-1.0.so; }; })
libtool: install: /usr/bin/install -c .libs/libusb-1.0.lai /usr/local/lib/libusb-1.0.la
libtool: install: /usr/bin/install -c .libs/libusb-1.0.a /usr/local/lib/libusb-1.0.a
libtool: install: chmod 644 /usr/local/lib/libusb-1.0.a
libtool: install: ranlib /usr/local/lib/libusb-1.0.a
 .././install-sh -c -d '/usr/local/include/libusb-1.0'
 /usr/bin/install -c -m 644 libusb.h '/usr/local/include/libusb-1.0'
 ./install-sh -c -d '/usr/local/lib/pkgconfig'
 /usr/bin/install -c -m 644 libusb-1.0.pc '/usr/local/lib/pkgconfig'

I rebooted and ran your one liner, now it displays nothing.
[root@X4 /home/barry]# PYUSB_DEBUG=debug LIBUSB_DEBUG=4 python -c "from usb.core import show_devices; print(show_devices()); print(show_devices())"
2020-10-17 08:23:37,316 DEBUG:usb.backend.libusb1:_LibUSB.__init__(<CDLL 'libusb-1.0.so.0', handle 801b79c00 at 0x8015d7bd0>)
[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.000007] [00a3a000] libusb: debug [libusb_init] created default context
[ 0.000010] [00a3a000] libusb: debug [libusb_init] libusb v1.0.23.11568
[ 0.000019] [00a3a000] libusb: debug [usbi_add_event_source] add fd 3 events 1
2020-10-17 08:23:37,316 INFO:usb.core:find(): using backend "usb.backend.libusb1"
2020-10-17 08:23:37,316 DEBUG:usb.backend.libusb1:_LibUSB.enumerate_devices()
[ 0.000231] [00a3a000] libusb: debug [libusb_get_device_list]  

2020-10-17 08:23:37,316 INFO:usb.core:find(): using backend "usb.backend.libusb1"
2020-10-17 08:23:37,317 DEBUG:usb.backend.libusb1:_LibUSB.enumerate_devices()
[ 0.000384] [00a3a000] libusb: debug [libusb_get_device_list]  

2020-10-17 08:23:37,317 DEBUG:usb.backend.libusb1:_LibUSB._finalize_object()
[ 0.000567] [00a3a000] libusb: debug [libusb_exit]  
[ 0.000569] [00a3a000] libusb: debug [libusb_exit] destroying default context
[ 0.000571] [00a3a000] libusb: debug [usbi_remove_event_source] remove fd 3

If I make uninstall, everything goes back to the original prolbem.

jonasmalacofilho, 2020-10-17:
Regarding libusb1 vs. FreeBSD's linux_libusb, you should file a bug report to the latter about it missing `libusb_get_parent` (available since 1.0.12/2012, non-experimental since 1.0.16/2013).<sup>[[1]](https://github.com/libusb/libusb/commit/cfb8610242394d532778a483570089c2bed52c84#diff-c1f9bc250077d41456a3e580fca0ddf5d8c25b741bff6d9b9505990a8b70b254R709-R725)[[2]](https://github.com/libusb/libusb/commit/b4c18fac65a594502eec5edd2611d5953e7950f7#diff-c1f9bc250077d41456a3e580fca0ddf5d8c25b741bff6d9b9505990a8b70b254L795-R804)</sup>

What I would like you to test, and it may fix the segfault, is to run the one-liner with PyUSB from the latest git HEAD from this repository.  You can use pip for that, or you can manually clone the repository and run the one liner the project root.  Regardless of the method, please also run `python -c "import usb; print(usb.__version__)"` so we can check that the correct version is in use.

jonasmalacofilho, 2020-10-24:
@btigner, can you provide any additional feedback?

mcuee, 2020-10-26:
@btigner Please also take note FreeBSD libusb implementation is not under the normal libusb project. You should file a bug report to FreeBSD libusb.

mcuee, 2020-10-26:
From libusb wiki:
https://github.com/libusb/libusb/wiki
```Note: FreeBSD 8 and above include a FreeBSD-specific reimplementation of the libusb API, so your applications will probably work there too. The source code for this library can be found here https://svnweb.freebsd.org/base/head/lib/libusb/.
```

jonasmalacofilho, 2020-10-28:
Closing after 10 days of inactivity.

For the record, I'm fairly confident that the segfault was caused by the bug fixed in c8d48cdb9165.

closed 2020-10-28 by jonasmalacofilho