pyusb/pyusb #334: Reading from USB mouse on Windows

glaukon-ariston, 2020-10-23:
I am trying to read from a USB mouse (on the Interrupt IN Endpoint) on Windows and I am not getting any data back (as I move the mouse the pointer on the screen moves as well). Although I am passing `timeout = 10000` to the `dev.read()` the function returns immediately (see the timestamps in the provided log).

```python
#!/usr/bin/python
# -*- coding: utf-8 -*-
import os
os.environ['PYUSB_DEBUG'] = 'debug'
import usb.core
import usb.util

dev = usb.core.find(idVendor=0x1EA7, idProduct=0x0064)
if dev is None:
    raise ValueError('Device not found')
print(dev)
cfg = usb.util.find_descriptor(dev, bConfigurationValue=1)
dev.set_configuration(cfg)
try:
    dev.set_interface_altsetting(interface = 0, alternate_setting = 0)
except USBError:
    pass
iface = cfg[(0,0)]
endpoint = iface[0]
# print(endpoint)
while True:
    try:
        data_list = dev.read(endpoint.bEndpointAddress, endpoint.wMaxPacketSize, timeout = 10000).tolist()
        print(" ".join("{:02x}".format(ord(c)) for c in data_list))
    except usb.core.USBError as error:
        if error.args == ('Operation timed out',):
            print('Timeout', end=' ')
            continue
    # For keyboard interrupt
    except KeyboardInterrupt:
        print("Read interrupted by user. Exiting.")
        exit(-1)
```

The output is as follows:

```
*** Python 3.7.2 (tags/v3.7.2:9a3ffc0492, Dec 23 2018, 23:09:28) [MSC v.1916 64 bit (AMD64)] on win32. ***
>>> 
*** Remote Interpreter Reinitialized  ***
>>> 
[Dbg]>>> 
2020-10-23 09:48:06,046 DEBUG:usb.backend.libusb1:_LibUSB.__init__(<WinDLL 'C:\app\electro\rtl_433\libusb-1.0.dll', handle 6b600000 at 0x1f39089e240>)
2020-10-23 09:48:06,140 INFO:usb.core:find(): using backend "usb.backend.libusb1"
2020-10-23 09:48:06,265 DEBUG:usb.backend.libusb1:_LibUSB.enumerate_devices()
2020-10-23 09:48:06,496 DEBUG:usb.backend.libusb1:_LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x000001F39094ACF8>)
2020-10-23 09:48:06,620 DEBUG:usb.backend.libusb1:_LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x000001F39094AEB8>)
2020-10-23 09:48:06,744 DEBUG:usb.backend.libusb1:_LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x000001F39094A2B0>)
2020-10-23 09:48:06,870 DEBUG:usb.backend.libusb1:_LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x000001F39094AEB8>)
2020-10-23 09:48:06,996 DEBUG:usb.backend.libusb1:_LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x000001F39094AEF0>)
2020-10-23 09:48:07,119 DEBUG:usb.backend.libusb1:_LibUSB.open_device(<usb.backend.libusb1._Device object at 0x000001F39094AEF0>)
2020-10-23 09:48:07,246 DEBUG:usb.backend.libusb1:_LibUSB.ctrl_transfer(<usb.backend.libusb1._DeviceHandle object at 0x000001F390888320>, 128, 6, 768, 0, array('B', [0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]), 1000)
2020-10-23 09:48:07,370 DEBUG:usb.backend.libusb1:_LibUSB.ctrl_transfer(<usb.backend.libusb1._DeviceHandle object at 0x000001F390888320>, 128, 6, 769, 1033, array('B', [0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]), 1000)
2020-10-23 09:48:07,498 DEBUG:usb.backend.libusb1:_LibUSB.get_configuration_descriptor(<usb.backend.libusb1._Device object at 0x000001F39094AEF0>, 0)
2020-10-23 09:48:07,628 DEBUG:usb.backend.libusb1:_LibUSB.get_interface_descriptor(<usb.backend.libusb1._Device object at 0x000001F39094AEF0>, 0, 0, 0)
2020-10-23 09:48:07,752 DEBUG:usb.backend.libusb1:_LibUSB.get_configuration_descriptor(<usb.backend.libusb1._Device object at 0x000001F39094AEF0>, 0)
2020-10-23 09:48:07,876 DEBUG:usb.backend.libusb1:_LibUSB.get_endpoint_descriptor(<usb.backend.libusb1._Device object at 0x000001F39094AEF0>, 0, 0, 0, 0)
2020-10-23 09:48:08,001 DEBUG:usb.backend.libusb1:_LibUSB.get_interface_descriptor(<usb.backend.libusb1._Device object at 0x000001F39094AEF0>, 0, 0, 0)
2020-10-23 09:48:08,126 DEBUG:usb.backend.libusb1:_LibUSB.get_configuration_descriptor(<usb.backend.libusb1._Device object at 0x000001F39094AEF0>, 0)
2020-10-23 09:48:08,251 DEBUG:usb.backend.libusb1:_LibUSB.get_interface_descriptor(<usb.backend.libusb1._Device object at 0x000001F39094AEF0>, 0, 1, 0)
2020-10-23 09:48:08,366 DEBUG:usb.backend.libusb1:_LibUSB.get_configuration_descriptor(<usb.backend.libusb1._Device object at 0x000001F39094AEF0>, 0)
DEVICE ID 1ea7:0064 on Bus 001 Address 003 =================
 bLength                :   0x12 (18 bytes)
 bDescriptorType        :    0x1 Device
 bcdUSB                 :  0x110 USB 1.1
 bDeviceClass           :    0x0 Specified at interface
 bDeviceSubClass        :    0x0
[... 39 lines cut ...]
2020-10-23 09:48:09,002 DEBUG:usb.backend.libusb1:_LibUSB.get_configuration_descriptor(<usb.backend.libusb1._Device object at 0x000001F39094AEF0>, 0)
2020-10-23 09:48:09,127 DEBUG:usb.backend.libusb1:_LibUSB.claim_interface(<usb.backend.libusb1._DeviceHandle object at 0x000001F390888320>, 0)
2020-10-23 09:48:09,252 DEBUG:usb.backend.libusb1:_LibUSB.set_interface_altsetting(<usb.backend.libusb1._DeviceHandle object at 0x000001F390888320>, 0, 0)
2020-10-23 09:48:09,370 DEBUG:usb.backend.libusb1:_LibUSB.get_interface_descriptor(<usb.backend.libusb1._Device object at 0x000001F39094AEF0>, 0, 0, 0)
2020-10-23 09:48:09,502 DEBUG:usb.backend.libusb1:_LibUSB.get_configuration_descriptor(<usb.backend.libusb1._Device object at 0x000001F39094AEF0>, 0)
2020-10-23 09:48:09,627 DEBUG:usb.backend.libusb1:_LibUSB.get_endpoint_descriptor(<usb.backend.libusb1._Device object at 0x000001F39094AEF0>, 0, 0, 0, 0)
2020-10-23 09:48:09,752 DEBUG:usb.backend.libusb1:_LibUSB.get_interface_descriptor(<usb.backend.libusb1._Device object at 0x000001F39094AEF0>, 0, 0, 0)
2020-10-23 09:48:09,870 DEBUG:usb.backend.libusb1:_LibUSB.get_configuration_descriptor(<usb.backend.libusb1._Device object at 0x000001F39094AEF0>, 0)
2020-10-23 09:48:09,997 DEBUG:usb.backend.libusb1:_LibUSB.get_configuration_descriptor(<usb.backend.libusb1._Device object at 0x000001F39094AEF0>, 0)
2020-10-23 09:48:10,120 DEBUG:usb.backend.libusb1:_LibUSB.get_interface_descriptor(<usb.backend.libusb1._Device object at 0x000001F39094AEF0>, 0, 0, 0)
2020-10-23 09:48:10,245 DEBUG:usb.backend.libusb1:_LibUSB.get_configuration_descriptor(<usb.backend.libusb1._Device object at 0x000001F39094AEF0>, 0)
2020-10-23 09:48:10,381 DEBUG:usb.backend.libusb1:_LibUSB.get_endpoint_descriptor(<usb.backend.libusb1._Device object at 0x000001F39094AEF0>, 0, 0, 0, 0)
2020-10-23 09:48:10,506 DEBUG:usb.backend.libusb1:_LibUSB.get_interface_descriptor(<usb.backend.libusb1._Device object at 0x000001F39094AEF0>, 0, 0, 0)
2020-10-23 09:48:10,631 DEBUG:usb.backend.libusb1:_LibUSB.get_configuration_descriptor(<usb.backend.libusb1._Device object at 0x000001F39094AEF0>, 0)
2020-10-23 09:48:10,756 DEBUG:usb.backend.libusb1:_LibUSB.intr_read(<usb.backend.libusb1._DeviceHandle object at 0x000001F390888320>, 129, 0, array('B', [0, 0, 0, 0, 0, 0, 0, 0]), 10000)
[... cut ...]
2020-10-23 09:48:11,256 DEBUG:usb.backend.libusb1:_LibUSB.intr_read(<usb.backend.libusb1._DeviceHandle object at 0x000001F390888320>, 129, 0, array('B', [0, 0, 0, 0, 0, 0, 0, 0]), 10000)
...[more of the same]...
2020-10-23 09:48:19,151 DEBUG:usb.backend.libusb1:_LibUSB.intr_read(<usb.backend.libusb1._DeviceHandle object at 0x000001F390888320>, 129, 0, array('B', [0, 0, 0, 0, 0, 0, 0, 0]), 10000)
2020-10-23 09:48:19,276 DEBUG:usb.backend.libusb1:_LibUSB.intr_read(<usb.backend.libusb1._DeviceHandle object at 0x000001F390888320>, 129, 0, array('B', [0, 0, 0, 0, 0, 0, 0, 0]), 10000)
Read interrupted by user. Exiting.
Exit code:  -1
>>> ```

glaukon-ariston, 2020-10-23:
Just read this on the [libusb](https://github.com/libusb/libusb/wiki/Windows#driver-installation) wiki under `How to use libusb on Windows|Known Restrictions`:

_HID keyboards and mice cannot be accessed using the native HID driver as Windows reserves exclusive access to them._
 
Could this be the reason I am not able to read data from my USB mouse? Can I install another driver?

jonasmalacofilho, 2020-10-23:
It is probably related to that security feature.

You could replace the driver with Zadig, but this would make the mouse unusable to all other applications (including the Windows graphical shell itself).

glaukon-ariston, 2020-10-23:
Another interesting info from the [libusb ](https://github.com/libusb/libusb/wiki/Windows#driver-installation) wiki:

_Windows HID API does not support custom Control Transfer, everything needs to be done through report. Please check out the discussion [here](http://janaxelson.com/forum/index.php?topic=1802.0)._

I put `LIBUSB_DEBUG=4` but didn't see anything different in the output.

jonasmalacofilho, 2020-10-24:
> I put `LIBUSB_DEBUG=4` but didn't see anything different in the output.

Oh, you probably need to set this in your shell (Command Prompt, PowerShell, etc.), not within the Python interpreter.

glaukon-ariston, 2020-10-24:
> > I put `LIBUSB_DEBUG=4` but didn't see anything different in the output.
> 
> Oh, you probably need to set this in your shell (Command Prompt, PowerShell, etc.), not within the Python interpreter.

I have now put it in the user's environment variables and restarted the Windows. I am still not seeing any change in the output. I am using PyScripter to run the Python script.

jonasmalacofilho, 2020-10-24:
That's weird.  Where did `C:\app\electro\rtl_433\libusb-1.0.dll` come from?  Maybe it was compiled without any logging functions?

glaukon-ariston, 2020-10-25:
> That's weird. Where did `C:\app\electro\rtl_433\libusb-1.0.dll` come from? Maybe it was compiled without any logging functions?

That's from the installation of the `rtl_433`. I also downloaded `libusb-1.0.dll` (libusb-1.0.23\MS64\dll\libusb-1.0.dll in the archive) from the libusb website and got the same result (not debug output from libusb-1.0.dll).

jonasmalacofilho, 2020-10-25:
In that case the environment variable is not reaching the correct shell for some reason.

I don't know how PyScripter runs, but in a command prompt window you should be able to simply:

```cmd
set PYUSB_DEBUG=debug
set LIBUSB_DEBUG=4
python <your program>
```

That is, as long `python` is in your `PATH`.

glaukon-ariston, 2020-10-27:
I did run in from the command line now and that made it show the debug messages from libusb. I couldn't pipe them to a file so I presume the debug message don't go neither to stdout nor to stderr and that was the reason I couldn't see them in PyScripter's output window. 

Anyway, it says `HID transfer failed: [5] Access is denied` at one point in the interrupt endpoint read. I guess, that confirms the previously mentioned info that Windows reserves exclusive access to mouse and keyboard devices.

```
2020-10-27 12:10:44,042 DEBUG:usb.backend.libusb1:_LibUSB.intr_read(<usb.backend.libusb1._DeviceHandle object at 0x00000297D0C715C0>, 129, 0, array('B', [90, 90, 90, 90, 90, 90, 90, 90]), 10000)
[ 0.184786] [00002f70] libusb: debug [libusb_alloc_transfer] transfer 00000297CEDDBE58
[ 0.184786] [00002f70] libusb: debug [libusb_submit_transfer] transfer 00000297CEDDBE58
[ 0.184786] [00002f70] libusb: debug [usbi_add_pollfd] add fd 2 events 1
[ 0.184786] [00002f70] libusb: debug [hid_submit_bulk_transfer] matched endpoint 81 with interface 0
[ 0.184786] [00002f70] libusb: debug [hid_submit_bulk_transfer] reading 8 bytes (report ID: 0x00)
[ 0.184786] [00002f70] libusb: error [hid_submit_bulk_transfer] HID transfer failed: [5] Access is denied.
[ 0.184786] [00002f70] libusb: debug [usbi_remove_pollfd] remove fd 2
[ 0.184786] [00002f70] libusb: debug [libusb_free_transfer] transfer 00000297CEDDBE58
```

The full log follows:
```
C:\app\RE\USB>python mouse_read_mwe.py
2020-10-27 12:10:43,857 DEBUG:usb.backend.libusb1:_LibUSB.__init__(<WinDLL 'C:\app\dev\Python37\libusb-1.0.dll', handle 7ffc2af00000 at 0x297d0d223c8>)
[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.000000] [00002f70] libusb: debug [libusb_init] created default context
[ 0.000000] [00002f70] libusb: debug [libusb_init] libusb v1.0.23.11397
[ 0.000000] [00002f70] libusb: debug [windows_init_dlls] Will use CancelIoEx for I/O cancellation
[ 0.000000] [00002f70] libusb: debug [get_windows_version] Windows 10 64-bit
[ 0.000000] [00002f70] libusb: debug [windows_init_clock] hires timer available (Frequency: 10000000 Hz)
[ 0.000000] [00002f70] libusb: debug [windows_init_clock] timer thread will run on core #0
[ 0.000000] [00002f70] libusb: debug [htab_create] using 1021 entries hash table
[ 0.006509] [00002f70] libusb: info [winusbx_init] libusbK DLL is not available, will use native WinUSB
[ 0.006509] [00002f70] libusb: debug [winusbx_init] initalized sub API libusbK
[ 0.006509] [00002f70] libusb: debug [winusbx_init] initalized sub API libusb0
[ 0.006509] [00002f70] libusb: debug [winusbx_init] initalized sub API WinUSB
[ 0.006509] [00002f70] libusb: info [windows_init] UsbDk backend is not available
[ 0.006509] [00002f70] libusb: debug [usbi_add_pollfd] add fd 0 events 1
2020-10-27 12:10:43,864 INFO:usb.core:find(): using backend "usb.backend.libusb1"
2020-10-27 12:10:43,864 DEBUG:usb.backend.libusb1:_LibUSB.enumerate_devices()
[ 0.006509] [00002f70] libusb: debug [libusb_get_device_list]
[ 0.006509] [00002f70] libusb: debug [winusb_get_device_list] allocating new device for session [7F]
[ 0.006509] [00002f70] libusb: debug [winusb_get_device_list] allocating new device for session [80]
[ 0.022152] [00002f70] libusb: debug [winusb_get_device_list] allocating new device for session [81]
[ 0.022152] [00002f70] libusb: debug [get_api_type] driver(s): BTHUSB
[ 0.022152] [00002f70] libusb: debug [get_api_type] lower filter driver(s): ibtusb
[ 0.022152] [00002f70] libusb: debug [winusb_get_device_list] allocating new device for session [31]
[ 0.022152] [00002f70] libusb: debug [get_api_type] driver(s): HidUsb
[ 0.022152] [00002f70] libusb: debug [get_api_type] matched driver name against HID API
[ 0.022152] [00002f70] libusb: debug [winusb_get_device_list] allocating new device for session [68]
[ 0.022152] [00002f70] libusb: debug [get_api_type] driver(s): USBSTOR
[ 0.022152] [00002f70] libusb: debug [winusb_get_device_list] allocating new device for session [69]
[ 0.022152] [00002f70] libusb: debug [get_api_type] driver(s): usbccgp
[ 0.022152] [00002f70] libusb: debug [get_api_type] matched driver name against Composite API
[ 0.022152] [00002f70] libusb: debug [winusb_get_device_list] allocating new device for session [6A]
[ 0.022152] [00002f70] libusb: debug [enumerate_hcd_root_hub] assigning HCD 'PCI\VEN_8086&DEV_9C31&SUBSYS_221417AA&REV_04\3&B1BFB68&0&A0' bus number 1
[ 0.022152] [00002f70] libusb: debug [enumerate_hcd_root_hub] assigning HCD 'PCI\VEN_8086&DEV_9C26&SUBSYS_221417AA&REV_04\3&B1BFB68&0&E8' bus number 2
[ 0.022152] [00002f70] libusb: debug [winusb_get_device_list] found existing device for session [80]
[ 0.090489] [00002f70] libusb: debug [init_device] found 1 configurations (active conf: 1)
[ 0.090489] [00002f70] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 25 bytes)
[ 0.090489] [00002f70] libusb: debug [init_device] (bus: 2, addr: 1, depth: 1, port: 1): 'USB\VID_8087&PID_8000\5&BA8841C&0&1'
[ 0.090489] [00002f70] libusb: debug [winusb_get_device_list] found existing device for session [6A]
[ 0.090489] [00002f70] libusb: debug [init_device] found 1 configurations (active conf: 1)
[ 0.090489] [00002f70] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 1086 bytes)
[... cut ...]
[ 0.184786] [00002f70] libusb: debug [hid_submit_bulk_transfer] reading 8 bytes (report ID: 0x00)
[ 0.184786] [00002f70] libusb: error [hid_submit_bulk_transfer] HID transfer failed: [5] Access is denied.
[ 0.184786] [00002f70] libusb: debug [usbi_remove_pollfd] remove fd 2
[ 0.184786] [00002f70] libusb: debug [libusb_free_transfer] transfer 00000297CEDDBE58
2020-10-27 12:10:44,042 DEBUG:usb.backend.libusb1:_LibUSB.intr_read(<usb.backend.libusb1._DeviceHandle object at 0x00000297D0C715C0>, 129, 0, array('B', [90, 90, 90, 90, 90, 90, 90, 90]), 10000)
```

jonasmalacofilho, 2020-10-27:
> I guess, that confirms the previously mentioned info that Windows reserves exclusive access to mouse and keyboard devices.

Yeap (unfortunately); there's also:

```
[ 0.122282] [00002f70] libusb: warning [hid_open] could not open HID device in R/W mode (keyboard or mouse?) - trying without
```

At this point, I think there's nothing left but replacing the kernel driver.  That is, if you don't mind taking the device away from Windows.

---

The hang up on `intr_read` could be a bug on libusb: I would expect it to abort after the previous "Access is denied" errors.  I'll open an issue upstream (will link it back here); can you follow it and see if you can provide any additional info for their developers?

jonasmalacofilho, 2020-10-27:
Actually, did you forcefully quit the application at that point, or was it stuck in the libusb call?  Put another way, was the application stuck in PyUSB (endlessly looping over `usb.backend.libusb1:_LibUSB.intr_read`) or libusb?

closed 2020-10-27 by jonasmalacofilho

glaukon-ariston, 2020-10-27:
No more questions for now. Thank you for your help.