pyusb/pyusb #335: usb.core.USBError: [Errno 19] No such device (it may have been disconnected)

yjw-alex, 2020-10-24:
**Please help me**
**Here are the codes**
`import usb.core
import usb.util

import my_config

class MasterDongle(object):
    def __init__(self):
        usb_dev = usb.core.find(idVendor=my_config.USB_VID_826X_MASTER_DONGLE,
                                idProduct=my_config.USB_PID_826X_MASTER_DONGLE)
        if usb_dev is None:
            raise ValueError('Device not found')
        print(usb_dev)
        # for usb_dev in usb_dev_all:
        print('USB: VendorID=' + hex(usb_dev.idVendor) + ' & ProductID=' + hex(usb_dev.idProduct))
        usb_dev.set_configuration()
        # usb_dev.read(0x88, 54)
        # cfg = usb_dev.get_active_configuration()
    def open(self):
        pass

    def close(self):
        pass

master_dongle = MasterDongle()`
**Here are the Logs**
DEVICE ID 248a:881d on Bus 003 Address 009 =================
 bLength                :   0x12 (18 bytes)
 bDescriptorType        :    0x1 Device
 bcdUSB                 :  0x110 USB 1.1
 bDeviceClass           :    0x0 Specified at interface
 bDeviceSubClass        :    0x0
 bDeviceProtocol        :    0x0
 bMaxPacketSize0        :    0x8 (8 bytes)
 idVendor               : 0x248a
 idProduct              : 0x881d
 bcdDevice              :  0x100 Device 1.0
 iManufacturer          :    0x1 Error Accessing String
 iProduct               :    0x2 Error Accessing String
 iSerialNumber          :    0x3 Error Accessing String
 bNumConfigurations     :    0x1
  CONFIGURATION 1: 50 mA ===================================
   bLength              :    0x9 (9 bytes)
   bDescriptorType      :    0x2 Configuration
   wTotalLength         :   0x52 (82 bytes)
   bNumInterfaces       :    0x3
   bConfigurationValue  :    0x1
   iConfiguration       :    0x0
   bmAttributes         :   0xa0 Bus Powered, Remote Wakeup
   bMaxPower            :   0x19 (50 mA)
    INTERFACE 0: Printer ===================================
     bLength            :    0x9 (9 bytes)
     bDescriptorType    :    0x4 Interface
     bInterfaceNumber   :    0x0
     bAlternateSetting  :    0x0
     bNumEndpoints      :    0x2
     bInterfaceClass    :    0x7 Printer
     bInterfaceSubClass :    0x1
     bInterfaceProtocol :    0x2
     iInterface         :    0x0
      ENDPOINT 0x88: Bulk IN ===============================
       bLength          :    0x7 (7 bytes)
       bDescriptorType  :    0x5 Endpoint
       bEndpointAddress :   0x88 IN
       bmAttributes     :    0x2 Bulk
       wMaxPacketSize   :   0x40 (64 bytes)
       bInterval        :    0x0
      ENDPOINT 0x5: Bulk OUT ===============================
       bLength          :    0x7 (7 bytes)
       bDescriptorType  :    0x5 Endpoint
       bEndpointAddress :    0x5 OUT
       bmAttributes     :    0x2 Bulk
       wMaxPacketSize   :   0x40 (64 bytes)
       bInterval        :    0x0
    INTERFACE 1: Human Interface Device ====================
     bLength            :    0x9 (9 bytes)
     bDescriptorType    :    0x4 Interface
     bInterfaceNumber   :    0x1
     bAlternateSetting  :    0x0
     bNumEndpoints      :    0x1
     bInterfaceClass    :    0x3 Human Interface Device
     bInterfaceSubClass :    0x1
     bInterfaceProtocol :    0x1
     iInterface         :    0x0
      ENDPOINT 0x81: Interrupt IN ==========================
       bLength          :    0x7 (7 bytes)
       bDescriptorType  :    0x5 Endpoint
       bEndpointAddress :   0x81 IN
       bmAttributes     :    0x3 Interrupt
       wMaxPacketSize   :    0x8 (8 bytes)
       bInterval        :    0xa
    INTERFACE 2: Human Interface Device ====================
     bLength            :    0x9 (9 bytes)
     bDescriptorType    :    0x4 Interface
     bInterfaceNumber   :    0x2
     bAlternateSetting  :    0x0
     bNumEndpoints      :    0x1
     bInterfaceClass    :    0x3 Human Interface Device
     bInterfaceSubClass :    0x1
     bInterfaceProtocol :    0x2
     iInterface         :    0x0
      ENDPOINT 0x82: Interrupt IN ==========================
       bLength          :    0x7 (7 bytes)
       bDescriptorType  :    0x5 Endpoint
       bEndpointAddress :   0x82 IN
       bmAttributes     :    0x3 Interrupt
       wMaxPacketSize   :    0x8 (8 bytes)
       bInterval        :    0x4
USB: VendorID=0x248a & ProductID=0x881d
Traceback (most recent call last):
  File "master_dongle.py", line 38, in <module>
    master_dongle = MasterDongle()
  File "master_dongle.py", line 28, in __init__
    usb_dev.set_configuration()
  File "C:\Users\edward\anaconda3\envs\Py37Qt5\lib\site-packages\usb\core.py", line 903, in set_configuration
    self._ctx.managed_set_configuration(self, configuration)
  File "C:\Users\edward\anaconda3\envs\Py37Qt5\lib\site-packages\usb\core.py", line 113, in wrapper
    return f(self, *args, **kwargs)
  File "C:\Users\edward\anaconda3\envs\Py37Qt5\lib\site-packages\usb\core.py", line 158, in managed_set_configuration
    self.managed_open()
  File "C:\Users\edward\anaconda3\envs\Py37Qt5\lib\site-packages\usb\core.py", line 113, in wrapper
    return f(self, *args, **kwargs)
  File "C:\Users\edward\anaconda3\envs\Py37Qt5\lib\site-packages\usb\core.py", line 131, in managed_open
    self.handle = self.backend.open_device(self.dev)
  File "C:\Users\edward\anaconda3\envs\Py37Qt5\lib\site-packages\usb\backend\libusb1.py", line 804, in open_device
    return _DeviceHandle(dev)
  File "C:\Users\edward\anaconda3\envs\Py37Qt5\lib\site-packages\usb\backend\libusb1.py", line 652, in __init__
    _check(_lib.libusb_open(self.devid, byref(self.handle)))
  File "C:\Users\edward\anaconda3\envs\Py37Qt5\lib\site-packages\usb\backend\libusb1.py", line 604, in _check
    raise USBError(_strerror(ret), ret, _libusb_errno[ret])
usb.core.USBError: [Errno 19] No such device (it may have been disconnected)

jonasmalacofilho, 2020-10-25:
This could be caused by the presence of HID interfaces: not only will libusb be limited by the underlying HID driver, but these interfaces could also be claimed (although in that case I would expect a different error).

I would try to avoid setting the configuration in the first place: [FAQ/How (not) to call set_configuration() [...]](https://github.com/pyusb/pyusb/blob/master/docs/faq.rst#how-not-to-call-set_configuration-on-a-device-already-configured-with-the-selected-configuration).

Besides this possible workaround, can you rerun this code with `PYUSB_DEBUG=debug` and `LIBUSB_DEBUG=4`?

```cmd
:: assuming windows command prompt
set PYUSB_DEBUG=debug
set LIBUSB_DEBUG=4
python <your program>
```

yjw-alex, 2020-10-25:
### Here are the logs after 'set PYUSB_DEBUG=debug' and 'set LIBUSB_DEBUG=4'
2020-10-25 11:07:04,315 DEBUG:usb.backend.libusb1:_LibUSB.__init__(<WinDLL 'C:\Windows\system32\libusb-1.0.dll', handle 7ffc3b3d0000 at 0x1478dd2ce48>)
[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.001000] [0000252c] libusb: debug [libusb_init] created default context
[ 0.002009] [0000252c] libusb: debug [libusb_init] libusb v1.0.23.11397
[ 0.003011] [0000252c] libusb: debug [windows_init_dlls] Will use CancelIoEx for I/O cancellation
[ 0.004006] [0000252c] libusb: debug [get_windows_version] Windows 10 64-bit
[ 0.004006] [0000252c] libusb: debug [windows_init_clock] hires timer available (Frequency: 10000000 Hz)
[ 0.005594] [0000252c] libusb: debug [windows_init_clock] timer thread will run on core #0
[ 0.007208] [0000252c] libusb: debug [htab_create] using 1021 entries hash table
[ 0.011198] [0000252c] libusb: info [winusbx_init] libusbK DLL is not available, will use native WinUSB
[ 0.013209] [0000252c] libusb: debug [winusbx_init] initalized sub API libusbK
[ 0.013209] [0000252c] libusb: debug [winusbx_init] initalized sub API libusb0
[ 0.014219] [0000252c] libusb: debug [winusbx_init] initalized sub API WinUSB
[ 0.016208] [0000252c] libusb: info [windows_init] UsbDk backend is not available
[ 0.016208] [0000252c] libusb: debug [usbi_add_pollfd] add fd 0 events 1
2020-10-25 11:07:04,334 INFO:usb.core:find(): using backend "usb.backend.libusb1"
2020-10-25 11:07:04,335 DEBUG:usb.backend.libusb1:_LibUSB.enumerate_devices()
[ 0.019202] [0000252c] libusb: debug [libusb_get_device_list]
[ 0.053200] [0000252c] libusb: debug [winusb_get_device_list] allocating new device for session [73]
[ 0.053200] [0000252c] libusb: debug [winusb_get_device_list] allocating new device for session [74]
[ 0.054219] [0000252c] libusb: debug [winusb_get_device_list] allocating new device for session [75]
[ 0.055219] [0000252c] libusb: debug [winusb_get_device_list] allocating new device for session [76]
[ 0.056201] [0000252c] libusb: debug [winusb_get_device_list] allocating new device for session [77]
[ 0.056201] [0000252c] libusb: debug [winusb_get_device_list] allocating new device for session [78]
[ 0.058215] [0000252c] libusb: debug [get_api_type] driver(s): HidUsb
[ 0.058215] [0000252c] libusb: debug [get_api_type] lower filter driver(s): AppleDFR
[ 0.059199] [0000252c] libusb: debug [get_api_type] matched driver name against HID API
[ 0.060201] [0000252c] libusb: debug [winusb_get_device_list] allocating new device for session [56]
[ 0.061207] [0000252c] libusb: debug [get_api_type] driver(s): usbccgp
[ 0.061207] [0000252c] libusb: debug [get_api_type] matched driver name against Composite API
[ 0.062196] [0000252c] libusb: debug [winusb_get_device_list] allocating new device for session [57]
[ 0.063203] [0000252c] libusb: debug [get_api_type] driver(s): HidUsb
[ 0.066215] [0000252c] libusb: debug [get_api_type] matched driver name against HID API
[ 0.067202] [0000252c] libusb: debug [winusb_get_device_list] allocating new device for session [58]
[ 0.068201] [0000252c] libusb: debug [get_api_type] driver(s): WINUSB
[ 0.068201] [0000252c] libusb: debug [get_api_type] matched driver name against WinUSB
[ 0.068201] [0000252c] libusb: debug [winusb_get_device_list] allocating new device for session [59]
[ 0.069196] [0000252c] libusb: debug [get_api_type] driver(s): usbccgp
[ 0.069196] [0000252c] libusb: debug [get_api_type] matched driver name against Composite API
[ 0.069196] [0000252c] libusb: debug [winusb_get_device_list] allocating new device for session [5A]
[ 0.070196] [0000252c] libusb: debug [get_api_type] driver(s): USBSTOR
[ 0.070196] [0000252c] libusb: debug [winusb_get_device_list] allocating new device for session [5B]
[ 0.071196] [0000252c] libusb: debug [get_api_type] driver(s): usbccgp
[ 0.071196] [0000252c] libusb: debug [get_api_type] matched driver name against Composite API
[ 0.071196] [0000252c] libusb: debug [winusb_get_device_list] allocating new device for session [5C]
[ 0.072205] [0000252c] libusb: debug [get_api_type] driver(s): usbccgp
[ 0.072205] [0000252c] libusb: debug [get_api_type] matched driver name against Composite API
[ 0.072205] [0000252c] libusb: debug [winusb_get_device_list] allocating new device for session [5D]
[ 0.073196] [0000252c] libusb: debug [get_api_type] driver(s): HidUsb
[ 0.073196] [0000252c] libusb: debug [get_api_type] matched driver name against HID API
[ 0.074196] [0000252c] libusb: debug [winusb_get_device_list] allocating new device for session [5E]
[ 0.074196] [0000252c] libusb: debug [enumerate_hcd_root_hub] assigning HCD 'PCI\VEN_8086&DEV_8A13&SUBSYS_00000000&REV_03\3&11583659&0&68' bus number 1
[ 0.074196] [0000252c] libusb: debug [enumerate_hcd_root_hub] assigning HCD 'PCI\VEN_106B&DEV_1801&SUBSYS_1801106B&REV_01\4&2BC937B7&0&01E4' bus number
2
[ 0.075196] [0000252c] libusb: debug [enumerate_hcd_root_hub] assigning HCD 'PCI\VEN_8086&DEV_38ED&SUBSYS_72708086&REV_10\3&11583659&0&A0' bus number 3
[ 0.078226] [0000252c] libusb: debug [winusb_get_device_list] found existing device for session [77]
[ 0.079197] [0000252c] libusb: debug [init_device] found 1 configurations (active conf: 1)
[ 0.079197] [0000252c] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 31 bytes)
[... cut ...]
[ 1.619769] [0000252c] libusb: debug [libusb_unref_device] destroy device 3.20
[ 1.620777] [0000252c] libusb: debug [libusb_unref_device] destroy device 3.0
2020-10-25 11:07:05,937 DEBUG:usb.backend.libusb1:_LibUSB._finalize_object()
[ 1.621767] [0000252c] libusb: debug [libusb_exit]
[ 1.621767] [0000252c] libusb: debug [libusb_exit] destroying default context
[ 1.622764] [0000252c] libusb: debug [usbi_remove_pollfd] remove fd 0
[ 1.624764] [00002510] libusb: debug [windows_clock_gettime_threaded] timer thread quitting

jonasmalacofilho, 2020-10-25:
Relevant line:

> ```
> [ 1.577761] [0000252c] libusb: warning [hid_open] could not open HID device in R/W mode (keyboard or mouse?) - trying without
> ```

Setting the configuration on this device will not work on Windows.  But like I mentioned above, you can avoid setting it (just checking that it's already set), which should work since there's only one.

closed 2020-10-25 by jonasmalacofilho

yjw-alex, 2020-10-25:
I have avoided setting it, but calling other funcs of the object usb_dev like read(), get_active_configuration() still comes out this issue.

jonasmalacofilho, 2020-10-25:
That makes on these restricted HID interfaces, but you may be able to read/write to the Printer interface.

You can also replace the kernel drivers on the HID interfaces, particularly if these devices aren't really mice/keyboards (or if you otherwise don't care that system stops receiving input reports from it).

yjw-alex, 2020-10-26:
Thank you very much~~