pyusb/pyusb #336: question: URB_CONTROL out without setup data

xhpohanka, 2020-11-03:
Hello,

I want to use this control message which I see in wireshark log
```
223	0.724765	host	1.3.0	USB	28	URB_CONTROL out
...
URB Function: URB_FUNCTION_SELECT_CONFIGURATION (0x0000)
URB transfer type: URB_CONTROL (0x02)
Packet Data Length: 0
Control transfer stage: Data (1)
```

```
0000   1c 00 60 74 7d 4a 07 ba ff ff 00 00 00 00 00 00
0010   00 01 00 03 00 00 02 00 00 00 00 01
```

If I understand it correctly it is similar like message from `dev.ctrl_transfer()` but without any Setup data.

Is it possible with pyusb?

jonasmalacofilho, 2020-11-04:
You can, but first you would need to decode these bytes into the fields of a setup packet:

https://github.com/pyusb/pyusb/blob/c8d48cdb9165d60e8b11b7d89b6ed80b4c56ee98/usb/core.py#L1031-L1032

The software you're using to view these captures should be able to do that for you.

You may also want to take a look at the relevant parts of the USB specification (for example, section 9.3 of the USB 3.2 spec).

---

All that aside, this transfer appears to set the active configuration.  If that's really the case, you should instead use [`Device.set_configuration`](https://github.com/pyusb/pyusb/blob/c8d48cdb9165d60e8b11b7d89b6ed80b4c56ee98/usb/core.py#L894-L902):

> You should always use this function rather than formulating your own SET_CONFIGURATION control request. This is because the underlying operating system needs to know when such changes happen.
> —[`libusb_set_configuration()`](http://libusb.sourceforge.net/api-1.0/group__libusb__dev.html#ga785ddea63a2b9bcb879a614ca4867bed)

closed 2020-11-08 by jonasmalacofilho

xhpohanka, 2020-11-09:
thank you for clarification