pyusb/pyusb #337: setup.py failure

sonistefregy, 2020-11-30:
### Problem
When attempting to install pyusb v1.1.0 as a dependency offline, pip is unable to find a version for installation, which causes the install to fail.

### Reproducing error
The following is run under Ubuntu 18.04:
```bash
$ python3 -m venv venv && cd venv && source bin/activate
$ python --version
Python 3.6.9
$ pip install -U setuptools pip && pip list
...
Package       Version
------------- -------
pip           20.3
pkg-resources 0.0.0
setuptools    50.3.2
$ pip download setuptools_scm pyusb
...
Saved ./pyusb-1.1.0.tar.gz
Saved ./setuptools_scm-4.1.2-py2.py3-none-any.whl
Saved ./setuptools-50.3.2-py3-none-any.whl
Successfully downloaded pyusb setuptools-scm setuptools
$ pip install pyusb-1.1.0.tar.gz --no-index -f .
Looking in links: .
Processing ./pyusb-1.1.0.tar.gz
  Installing build dependencies ... error
  ERROR: Command errored out with exit status 1:
   command: /home/user/venv/bin/python3 /home/user/venv/lib/python3.6/site-packages/pip install --ignore-installed --no-user --prefix /tmp/pip-build-env-sfmrtu9a/overlay --no-warn-script-location --no-binary :none: --only-binary :none: --no-index --find-links . -- 'setuptools>=30.3.0' wheel setuptools_scm
       cwd: None
  Complete output (3 lines):
  Looking in links: .
  ERROR: Could not find a version that satisfies the requirement wheel
  ERROR: No matching distribution found for wheel
  ----------------------------------------
ERROR: Command errored out with exit status 1: /home/user/venv/bin/python3 /home/user/venv/lib/python3.6/site-packages/pip install --ignore-installed --no-user --prefix /tmp/pip-build-env-sfmrtu9a/overlay --no-warn-script-location --no-binary :none: --only-binary :none: --no-index --find-links . -- 'setuptools>=30.3.0' wheel setuptools_scm Check the logs for full command output.
```

### Comments

I'm not sure if this is a `setuptools_scm` problem or a `pyusb` problem, but wanted to bring the issue to your attention as pyusb has switched to setuptools_scm.

jonasmalacofilho, 2020-11-30:
Related: pypa/pip#5696

jonasmalacofilho, 2020-11-30:
I can reproduce the problem, and the issue linked above explains what is happening.

However I haven't been able to get `--no-build-isolation` and `--no-index` to play nice in this particular case.

@ap-- do you happen to alread know how to fix this?

ap--, 2020-11-30:
Hmmm....
I will need to reproduce here, to figure out what's going on.

But in case this worked before, and just started occurring today:
Today is one of those wonderful days, where pip got updated with potentially breaking changes:

```

ap--, 2020-11-30:
Also, it says it's missing `wheel` in the local index. I would check and see if `pip install wheel` is breaking too.

jonasmalacofilho, 2020-11-30:
I'm also able to reproduce the problem with earlier (20.2, 20.1) versions of pip.

That said,

```
$ pip download setuptools_scm pyusb wheel
                                    ^^^^^
```

is enough to make the issue go away.

Is this something we need to list somewhere (maybe [`setup_requires`](https://github.com/pyusb/pyusb/blob/c8d48cdb9165d60e8b11b7d89b6ed80b4c56ee98/setup.cfg#L5))?

ap--, 2020-11-30:
It's in `pyproject.toml` and if I am not mistaken a dependency for isolated pep517 builds.

So, I do think, this is up to the user because they chose to manage their own local index.
But I will have another look in the next days if there's additional things that can be done.

jonasmalacofilho, 2020-11-30:
That makes sense.

And thank you for looking into this!

sonistefregy, 2020-11-30:
Can confirm -- PEBCAK issue with missing the `wheel` dependency.

```bash
$ pip download setuptools_scm pyusb wheel
...
Saved ./pyusb-1.1.0.tar.gz
Saved ./setuptools_scm-4.1.2-py2.py3-none-any.whl
Saved ./wheel-0.35.1-py2.py3-none-any.whl
Saved ./setuptools-50.3.2-py3-none-any.whl
Successfully downloaded pyusb setuptools-scm wheel setuptools
$ pip install pyusb-1.1.0.tar.gz --no-index -f .
...
Successfully installed pyusb-1.1.0
```

closed 2020-11-30 by sonistefregy

ap--, 2020-12-04:
@jonasmalacofilho, just to follow up on this:

Basically if I would rephrase this issue, it would be: "What's the best way to prevent errors when a user wants to keep an offline package cache"

And i think there's 3 possible things to improve this:

1. the first google results basically say do `pip download -r requirements.txt ...`. So that could be added and include `setuptools_scm` and `wheel` etc... but it's of course another file to keep in sync
2. there could be some docs added somewhere to actually use:
    ```console
    pip install wheel
    pip wheel -w /path/to/my/wheelhouse pyusb` 
    ```
    Which builds the wheel and wont require any online connection later.
3. provide a pyusb wheel on pypi. Then `pip download` should prefer the wheel too (if not requested otherwise) and not require at least the `setuptools_scm` dependency during offline install

Doing 3 is easily done when pushing the sdist to pypi and will probably solve any package cache related issues.

Cheers,
Andreas :smiley:

jonasmalacofilho, 2020-12-05:
@ap-- 

Yeah, 3 is probably a good enough solution.

Thanks for looking into this.