pyusb/pyusb #340: usb.core.NoBackendError: No backend available

hiteshkhanna31, 2020-12-08:
I am trying to start using Pyusb library to implement a USB logger for my STM32 device.

I started following the tutorial but got some error while checking the usb device connected to the PC.

import usb
import usb.util
import usb.core

dev = usb.core.find(idVendor=0x483, idProduct=0x572B)
print(dev)

Ended up with this error:
    raise NoBackendError('No backend available')
usb.core.NoBackendError: No backend available

I tried following up for the issue on this git but couldn't find any proper solution.
Appreciate any help on this

jonasmalacofilho, 2020-12-08:
Hi,

You need to install LibUSB (preferably 1.x).

You can find a bit more information in the [FAQ](https://github.com/pyusb/pyusb/blob/master/docs/faq.rst).

hiteshkhanna31, 2020-12-08:
> Hi,
> 
> You need to install LibUSB (preferably 1.x).
> 
> You can find a bit more information in the [FAQ](https://github.com/pyusb/pyusb/blob/master/docs/faq.rst).

hiteshkhanna31, 2020-12-08:
@jonasmalacofilho I have already installed the library  LibUSB.
Please find the screenshot of the library installed on my PYcharm environment. 
[image]

jonasmalacofilho, 2020-12-08:
These are (different) Python wrappers. What you need is the actual LibUSB dynamic library:

https://github.com/libusb/libusb

hiteshkhanna31, 2020-12-08:
I am not sure what is the difference between those.
I have downloaded the library from github and tried to follow the Installation Instructions for Windows using Microsoft Visual Studio. But it did not understand the rest of the procedure given there.

Please help on this and as i am using pycharm for python coding. 

[image]

jonasmalacofilho, 2020-12-08:
I find it simpler to place the precompiled DLLs (part of the `libusb-<version>.7z` files in the releases tab) into the system (e.g. `C:\Windows\System32` for 64-bit programs, or `C:\Windows\SysWOW64` for 32-bit ones) or python (`C:\Python39` or your PyCharm root) directories.

hiteshkhanna31, 2020-12-08:
I am now getting confused . I have downloaded the libusb-1.0.23.7z file from [https://github.com/libusb/libusb/releases/tag/v1.0.23 ](url) 
and i have extracted the folder and found there are many libusb-1.0.dll in different folder (for example MinGW64,MS64) 
From which folder shall i select the file and keep in Pycharm root folder.

jonasmalacofilho, 2020-12-10:
Try `MS64` first.

mcuee, 2020-12-11:
All version of libusb binary DLL should probably work for the purpose of pyusb. MS64 is probably the best to try with official Python distribution which is built with Visual Studio.

But in case the libusb binary do not work for you, please build using the source codes.

mcuee, 2020-12-24:
Close this as well.

closed 2020-12-24 by mcuee

Qbicz, 2024-02-28:
Leaving it here for people googling the solution for `No backend available` in pyusb on **Embedded Linux**.

On Linux, ctypes behavior is described here: https://docs.python.org/3/library/ctypes.html#finding-shared-libraries

For a Linux device without such capability, you can specify the library location like this, with `find_library`: https://github.com/pyusb/pyusb/blob/8205fc3062d59aa63b2da1cb300f59f95c5dcaaa/docs/tutorial.rst#specifying-libraries-by-hand