pyusb/pyusb #341: Timeout error writing to USB device

theDweeb, 2020-12-11:
Hello, thank you for this awesome package. One issue I am having is writing to my USB device. I can successfully read from the USB device multiple times, but I get a timeout error when trying to write to the device. Using Total Phase's Beagle 12 USB sniffer I can see activity when receiving data, but no activity at all when trying to send data.

below is a print out of the device's IN endpoint and OUT endpoint, which match characteristics described in my USB firmware.

```
in EP:       ENDPOINT 0x81: Bulk IN ===============================
       bLength          :    0x7 (7 bytes)
       bDescriptorType  :    0x5 Endpoint
       bEndpointAddress :   0x81 IN
       bmAttributes     :    0x2 Bulk
       wMaxPacketSize   :   0x10 (16 bytes)
       bInterval        :    0x0
out EP:       ENDPOINT 0x1: Bulk OUT ===============================
       bLength          :    0x7 (7 bytes)
       bDescriptorType  :    0x5 Endpoint
       bEndpointAddress :    0x1 OUT
       bmAttributes     :    0x2 Bulk
       wMaxPacketSize   :    0x8 (8 bytes)
       bInterval        :    0x0
Traceback (most recent call last):
  File "c:/Users/Actev/Desktop/Actev_Software/Actev_Application/UVD200_TF.py", line 331, in <module>
    test_fixture.dev.write(0x1, 'msg', 100)
  File "C:\Users\Actev\Desktop\Actev_Software\Actev_Application\venv\lib\site-packages\usb\core.py", line 982, in write
    self.__get_timeout(timeout)
  File "C:\Users\Actev\Desktop\Actev_Software\Actev_Application\venv\lib\site-packages\usb\backend\libusb0.py", line 549, in bulk_write
    data, timeout)
  File "C:\Users\Actev\Desktop\Actev_Software\Actev_Application\venv\lib\site-packages\usb\backend\libusb0.py", line 659, in __write
    timeout
  File "C:\Users\Actev\Desktop\Actev_Software\Actev_Application\venv\lib\site-packages\usb\backend\libusb0.py", line 447, in _check
    raise USBError(errmsg, ret)
usb.core.USBError: [Errno None] b'libusb0-dll:err [_usb_reap_async] timeout error\n'
```

jonasmalacofilho, 2020-12-12:
Have you tried a significantly larger timeout value, say 5 seconds (`timeout=5000`)?

theDweeb, 2020-12-14:
Yes, I have tried timeout=10000. There doesn't seem to be any activity on my USB sniffer showing data going out.

jonasmalacofilho, 2020-12-15:
Can you switch to libusb1 and run your test case with `PYUSB_DEBUG=debug LIBUSB_DEBUG=4` on the program's environment?

mcuee, 2020-12-24:
@theDweeb This usually means that the device can not understand the command sent to it. You have to understand the device communication protocol.

jonasmalacofilho, 2021-01-19:
I'm closing this issue due to inactivity, but please feel free to request it to be reopened.

closed 2021-01-19 by jonasmalacofilho