pyusb/pyusb #343: USBError ('libusb0-dll:err [control_msg] sending control message failed, win error: A device attached to the system is not functioning.')

oldnapalm, 2020-12-20:
Hello. Sorry if this issue is not related to PyUSB.

I'm using PyUSB to communicate to an ANT device (using the [python-ant](https://github.com/oldnapalm/python-ant) library). All works fine for me using a [Garmin](https://buy.garmin.com/en-US/US/p/10997) device, but some people are having problem using a [Cycplus](https://www.cycplus.com/products/ant-usb-stick-u1) device (although they have the same vendor and device IDs, 0fcf:1008) `Exception: USBError(None, b'libusb0-dll:err [control_msg] sending control message failed, win error: A device attached to the system is not functioning.\r\n\n')`

I also tried using libusb 1.0, it works fine with the Garmin device too, but with the Cycplus device they get the error `Exception: USBError(32, 'Pipe error')`

The project source is at https://github.com/oldnapalm/vpower

Any idea what can be causing the issue?

Thank you.

jonasmalacofilho, 2020-12-20:
Opa!

A few things could be happening, but maybe we should start by checking if both devices expose the same interfaces and endpoints?  Additionally, which drivers do they use on Windows?

Can you also have your testers run your program (with libusb1) with `PYUSB_DEBUG=debug` and `LIBUSB_DEBUG=4` on their environments?

mcuee, 2020-12-21:
The support USB devices (Garmin USB ANT dongles) are using libusb-win32 device driver (libusb0.sys) by default. Just wondering if things get better if you use the WinUSB driver instead (using Zadig to replace the driver). pyusb supports both libusb0.sys (libusb0/libusb-win32 backend) and winusb (libusb1 backend).
ANTUSB2 Stick (0fcf:1008: Dynastream Innovations, Inc.)
ANTUSB-m Stick (0fcf:1009: Dynastream Innovations, Inc.)

mcuee, 2020-12-21:
On the other hand, the fundamental issue here is probably that the "clone" Cycplus USB ANT+ stick is not compatible with the official Garmin ANTUSB2 stick. It should not use the same VID/PID in the first place. You may have to carry out some reverse engineering to see the differences.
https://www.cycplus.com/products/ant-usb-stick-u1

oldnapalm, 2020-12-21:
We tested with both libusb0 (using libusb-win32 v1.0.24.11584 from Windows ANT drivers and v1.2.6.0 from Zadig) and libusb1 backend (using WinUSB v6.1.7600.16385 from Zadig).
Using libusb-win32 we get the error message in the title. Using WinUSB the error is `USBError(32, 'Pipe error')`.
Also tried with UsbDk and the error is `USBError(5, 'Input/Output Error')`.

Devices information extracted using usbview

<details><summary>Cycplus</summary>

```
[Port1]  :  ANT USBStick2

Is Port User Connectable:         yes
Is Port Debug Capable:            no
Companion Port Number:            17
Companion Hub Symbolic Link Name: USB#ROOT_HUB30#4&121c824c&0&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}
Protocols Supported:
 USB 1.1:                         yes
 USB 2.0:                         yes
 USB 3.0:                         no

Device Power State:               PowerDeviceD0

       ---===>Device Information<===---
English product name: "ANT USBStick2"

ConnectionStatus:                  
Current Config Value:              0x01  -> Device Bus Speed: Full (is not SuperSpeed or higher capable)
Device Address:                    0x26
Open Pipes:                           2

          ===>Device Descriptor<===
bLength:                           0x12
bDescriptorType:                   0x01
bcdUSB:                          0x0200
bDeviceClass:                      0x00  -> This is an Interface Class Defined Device
bDeviceSubClass:                   0x00
bDeviceProtocol:                   0x00
bMaxPacketSize0:                   0x40 = (64) Bytes
idVendor:                        0x0FCF = Dynastream Innovations Inc.
idProduct:                       0x1008
bcdDevice:                       0x0100
iManufacturer:                     0x01
     English (United States)  "CYCPLUS"
iProduct:                          0x02
     English (United States)  "ANT USBStick2"
iSerialNumber:                     0x03
*!*ERROR:  No String Descriptor for index 3!
bNumConfigurations:                0x01

          ---===>Open Pipes<===---

          ===>Endpoint Descriptor<===
bLength:                           0x07
bDescriptorType:                   0x05
bEndpointAddress:                  0x81  -> Direction: IN - EndpointID: 1
bmAttributes:                      0x02  -> Bulk Transfer Type
wMaxPacketSize:                  0x0040 = 0x40 bytes
bInterval:                         0x01

          ===>Endpoint Descriptor<===
bLength:                           0x07
bDescriptorType:                   0x05
bEndpointAddress:                  0x01  -> Direction: OUT - EndpointID: 1
bmAttributes:                      0x02  -> Bulk Transfer Type
wMaxPacketSize:                  0x0040 = 0x40 bytes
bInterval:                         0x01

       ---===>Full Configuration Descriptor<===---

          ===>Configuration Descriptor<===
[... 559 lines cut ...]
[ 0.305708] [00001468] libusb: debug [usbi_handle_transfer_completion] transfer 000001BDBFFE8688 has callback 00007FF8E56F9390
[ 0.306097] [00001468] libusb: debug [sync_transfer_cb] actual_length=5
2020-12-21 09:12:02,030 DEBUG:usb.backend.libusb1:_LibUSB.get_configuration_descriptor(<usb.backend.libusb1._Device object at 0x000001BDC0194820>, 0)
[ 0.306223] [00001468] libusb: debug [libusb_free_transfer] transfer 000001BDBFFE8688
[ 0.306451] [00001c2c] libusb: debug [libusb_get_config_descriptor] index 0
2020-12-21 09:12:02,030 DEBUG:usb.backend.libusb1:_LibUSB.bulk_read(<usb.backend.libusb1._DeviceHandle object at 0x000001BDC0173EB0>, 129, 0, array('B', [0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]), 1000)
[ 0.307429] [00001c2c] libusb: debug [libusb_alloc_transfer] transfer 000001BDBFFE8688
[ 0.307580] [00001c2c] libusb: debug [libusb_submit_transfer] transfer 000001BDBFFE8688
[ 0.307735] [00001c2c] libusb: debug [add_to_flying_list] arm timer for timeout in 1000ms (first in line)
[ 0.307913] [00001c2c] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 0.308090] [00001c2c] libusb: debug [usbi_wait_for_events] WaitForMultipleObjects() for 2 HANDLEs with timeout in 60000ms
[ 0.307955] [00002190] libusb: debug [windows_iocp_thread] transfer 000001BDBFFE8688 completed, length 16
[ 0.308442] [00001c2c] libusb: debug [usbi_wait_for_events] WaitForMultipleObjects() returned 0
[ 0.308577] [00001c2c] libusb: debug [handle_event_trigger] event triggered
[ 0.308704] [00001c2c] libusb: debug [windows_handle_transfer_completion] handling transfer 000001BDBFFE8688 completion with errcode 0, length 16
[ 0.308873] [00001c2c] libusb: debug [arm_timer_for_next_timeout] no timeouts, disarming timer
[ 0.308985] [00001c2c] libusb: debug [usbi_handle_transfer_completion] transfer 000001BDBFFE8688 has callback 00007FF8E56F9390
[ 0.309148] [00001c2c] libusb: debug [sync_transfer_cb] actual_length=5
[ 0.309261] [00001c2c] libusb: debug [libusb_free_transfer] transfer 000001BDBFFE8688
2020-12-21 09:12:02,039 DEBUG:usb.backend.libusb1:_LibUSB.bulk_read(<usb.backend.libusb1._DeviceHandle object at 0x000001BDC0173EB0>, 129, 0, array('B', [0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]), 1000)
```

</details>

I understand that the Cycplus clone may not be compatible with the official Garmin stick, but the thing is, it seems to work using the Windows ANT drivers (libusb-win32 from Dynastream Innovations).

jonasmalacofilho, 2020-12-21:
The issue seems to lie in a lower level than PyUSB.  But see if forcing PyUSB to use its libusb0/libusb-win32 backend helps (with the libusb0/Windows ANT drivers):

```py
import usb.core
from usb.backend import libusb0

backend = libusb0.get_backend()

dev = usb.core.find(backend=backend)
```

oldnapalm, 2020-12-21:
I will try that (I don't have the Cycplus device so I need to ask someone else to test), but using libusb0 was our first try. Using libusb1 was a later test.

Could it be due to the Cycplus device not having a serial number?

```
iSerialNumber:                     0x03
*!*ERROR:  No String Descriptor for index 3!
```

jonasmalacofilho, 2020-12-21:
No, that's pretty standard.

oldnapalm, 2020-12-21:
Forcing the libusb0 backend as you suggested gives the same result as the first try (not forcing libusb1)

```
2020-12-21 14:17:40,150 DEBUG:usb.backend.libusb0:_LibUSB.enumerate_devices()
2020-12-21 14:17:40,150 DEBUG:usb.backend.libusb0:_LibUSB.get_device_descriptor(<usb.backend.libusb0._usb_device object at 0x000002F480D2FF40>)
Starting ANT node
2020-12-21 14:17:40,152 ERROR:usb.backend.libusb1:Error loading libusb 1.0 backend
Traceback (most recent call last):
  File "usb\backend\libusb1.py", line 961, in get_backend
  File "Lib\site-packages\pyinstaller_hooks_contrib-2020.8-py3.8.egg\_pyinstaller_hooks_contrib\hooks\rthooks\pyi_rth_usb.py", line 57, in _load_library
OSError: USB library could not be found
2020-12-21 14:17:40,153 ERROR:usb.backend.openusb:Error loading OpenUSB backend
Traceback (most recent call last):
  File "usb\backend\openusb.py", line 745, in get_backend
  File "Lib\site-packages\pyinstaller_hooks_contrib-2020.8-py3.8.egg\_pyinstaller_hooks_contrib\hooks\rthooks\pyi_rth_usb.py", line 57, in _load_library
OSError: USB library could not be found
2020-12-21 14:17:40,154 INFO:usb.core:find(): using backend "usb.backend.libusb0"
2020-12-21 14:17:40,154 DEBUG:usb.backend.libusb0:_LibUSB.enumerate_devices()
2020-12-21 14:17:40,156 DEBUG:usb.backend.libusb0:_LibUSB.get_device_descriptor(<usb.backend.libusb0._usb_device object at 0x000002F480E0D1C0>)
2020-12-21 14:17:40,157 DEBUG:usb.backend.libusb0:_LibUSB.open_device(<usb.backend.libusb0._usb_device object at 0x000002F480E0D1C0>)
2020-12-21 14:17:40,158 DEBUG:usb.backend.libusb0:_LibUSB.is_kernel_driver_active(3251265297696, 0)
2020-12-21 14:17:40,162 DEBUG:usb.backend.libusb0:_LibUSB.get_configuration_descriptor(<usb.backend.libusb0._usb_device object at 0x000002F480E0D1C0>, 0)
2020-12-21 14:17:40,163 DEBUG:usb.backend.libusb0:_LibUSB.set_configuration(3251265297696, 1)
2020-12-21 14:17:40,165 DEBUG:usb.backend.libusb0:_LibUSB.get_configuration_descriptor(<usb.backend.libusb0._usb_device object at 0x000002F480E0D1C0>, 0)
2020-12-21 14:17:40,166 DEBUG:usb.backend.libusb0:_LibUSB.get_interface_descriptor(<usb.backend.libusb0._usb_device object at 0x000002F480E0D1C0>, 0, 0, 0)
2020-12-21 14:17:40,166 DEBUG:usb.backend.libusb0:_LibUSB.get_configuration_descriptor(<usb.backend.libusb0._usb_device object at 0x000002F480E0D1C0>, 0)
2020-12-21 14:17:40,167 DEBUG:usb.backend.libusb0:_LibUSB.claim_interface(3251265297696, 0)
2020-12-21 14:17:40,168 DEBUG:usb.backend.libusb0:_LibUSB.ctrl_transfer(3251265297696, 129, 10, 0, 0, array('B', [0]), 1000)
Exception: USBError(None, b'libusb0-dll:err [control_msg] sending control message failed, win error: A device attached to the system is not functioning.\r\n\n')
```

mcuee, 2020-12-23:
So the issue is lower level than pyusb. Basically this means you have to reverse engineering the Cycplus device as it does not understnd the following control transfer, which presumably works with the Garmin USB ANT dongles.
```
ctrl_transfer(3251265297696, 129, 10, 0, 0, array('B', [0]), 1000).
```

oldnapalm, 2020-12-23:
I tried to change the way python-ant calls `find_descriptor` to be more like how it's done in the PyUSB tutorial.

Previously it did this
```
        dev.set_configuration()
        cfg = dev.get_active_configuration()
        interfaceNumber = cfg[(0, 0)].bInterfaceNumber

        intf = find_descriptor(cfg,
                               bInterfaceNumber=interfaceNumber,
                               bAlternateSetting=get_interface(dev, interfaceNumber))

        claim_interface(dev, interfaceNumber)

        endpoint_out_matcher = \
            lambda e: endpoint_direction(e.bEndpointAddress) == ENDPOINT_OUT
        epOut = find_descriptor(intf, custom_match=endpoint_out_matcher)
        assert epOut is not None

        endpoint_in_matcher = \
            lambda e: endpoint_direction(e.bEndpointAddress) == ENDPOINT_IN
        ep_in = find_descriptor(intf, custom_match=endpoint_in_matcher)
        assert ep_in is not None
```

I changed it to
```
        dev.set_configuration()
        cfg = dev.get_active_configuration()
        intf = cfg[(0,0)]
        interfaceNumber = intf.bInterfaceNumber

        claim_interface(dev, interfaceNumber)

        epOut = find_descriptor(intf,
            custom_match=lambda e: endpoint_direction(e.bEndpointAddress) == ENDPOINT_OUT)
        assert epOut is not None

        epIn = find_descriptor(intf,
            custom_match=lambda e: endpoint_direction(e.bEndpointAddress) == ENDPOINT_IN)
        assert epIn is not None
```

Now `ctrl_transfer` is never called. With the Garmin stick it still works normally and with the Cycplus it goes a little further, but times out reading in the main loop (previous reads seem to work)
```
Using LinearInterpolationPowerCalculator
2020-12-23 09:04:42,598 ERROR:usb.backend.libusb1:Error loading libusb 1.0 backend
Traceback (most recent call last):
  File "usb\backend\libusb1.py", line 961, in get_backend
  File "Lib\site-packages\pyinstaller_hooks_contrib-2020.8-py3.8.egg\_pyinstaller_hooks_contrib\hooks\rthooks\pyi_rth_usb.py", line 57, in _load_library
OSError: USB library could not be found
2020-12-23 09:04:42,599 ERROR:usb.backend.openusb:Error loading OpenUSB backend
Traceback (most recent call last):
  File "usb\backend\openusb.py", line 745, in get_backend
  File "Lib\site-packages\pyinstaller_hooks_contrib-2020.8-py3.8.egg\_pyinstaller_hooks_contrib\hooks\rthooks\pyi_rth_usb.py", line 57, in _load_library
OSError: USB library could not be found
2020-12-23 09:04:42,604 INFO:usb.core:find(): using backend "usb.backend.libusb0"
2020-12-23 09:04:42,604 DEBUG:usb.backend.libusb0:_LibUSB.enumerate_devices()
2020-12-23 09:04:42,606 DEBUG:usb.backend.libusb0:_LibUSB.get_device_descriptor(<usb.backend.libusb0._usb_device object at 0x0000023B27D291C0>)
2020-12-23 09:04:42,607 ERROR:usb.backend.libusb1:Error loading libusb 1.0 backend
Traceback (most recent call last):
  File "usb\backend\libusb1.py", line 961, in get_backend
  File "Lib\site-packages\pyinstaller_hooks_contrib-2020.8-py3.8.egg\_pyinstaller_hooks_contrib\hooks\rthooks\pyi_rth_usb.py", line 57, in _load_library
OSError: USB library could not be found
2020-12-23 09:04:42,608 ERROR:usb.backend.openusb:Error loading OpenUSB backend
Traceback (most recent call last):
  File "usb\backend\openusb.py", line 745, in get_backend
  File "Lib\site-packages\pyinstaller_hooks_contrib-2020.8-py3.8.egg\_pyinstaller_hooks_contrib\hooks\rthooks\pyi_rth_usb.py", line 57, in _load_library
OSError: USB library could not be found
2020-12-23 09:04:42,609 INFO:usb.core:find(): using backend "usb.backend.libusb0"
[... 102 lines cut ...]
  File "threading.py", line 870, in run
  File "ant\core\event.py", line 50, in EventPump
  File "ant\core\driver.py", line 79, in read
  File "ant\core\driver.py", line 245, in _read
  File "usb\core.py", line 421, in read
  File "usb\core.py", line 1017, in read
  File "usb\_debug.py", line 62, in do_trace
  File "usb\backend\libusb0.py", line 553, in bulk_read
  File "usb\backend\libusb0.py", line 665, in __read
  File "usb\backend\libusb0.py", line 447, in _check
usb.core.USBError: [Errno None] b'libusb0-dll:err [_usb_reap_async] timeout error\n'
```

I will try to increase the timeout but with the Garmin stick it keeps reading 4 times a second
```
Main wait loop
2020-12-23 07:25:21,214 DEBUG:usb.backend.libusb0:_LibUSB.bulk_read(2874219240176, 129, 0, array('B', [0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]), 1000)
2020-12-23 07:25:21,463 DEBUG:usb.backend.libusb0:_LibUSB.bulk_read(2874219240176, 129, 0, array('B', [0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]), 1000)
2020-12-23 07:25:21,713 DEBUG:usb.backend.libusb0:_LibUSB.bulk_read(2874219240176, 129, 0, array('B', [0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]), 1000)
2020-12-23 07:25:21,963 DEBUG:usb.backend.libusb0:_LibUSB.bulk_read(2874219240176, 129, 0, array('B', [0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]), 1000)
2020-12-23 07:25:22,212 DEBUG:usb.backend.libusb0:_LibUSB.bulk_read(2874219240176, 129, 0, array('B', [0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]), 1000)
2020-12-23 07:25:22,462 DEBUG:usb.backend.libusb0:_LibUSB.bulk_read(2874219240176, 129, 0, array('B', [0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]), 1000)
```

The Zwift app is not open source so it's a little harder to check how it's doing, but I will try eventually.

jonasmalacofilho, 2020-12-23:
Note that

```py
ctrl_transfer(dev_handle=3251265297696, bmRequestType=129, bRequest=10,
              wValue=0, wIndex=0, data=array('B', [0]), timeout=1000)
```

is equivalent to `GET_INTERFACE(interface=0, alternate_setting=0)`, which should be valid as long as the device is in a configured state (which it should be, in this case).

---

That said, resetting the configuration 

```py
dev.set_configuration()
cfg = dev.get_active_configuration()
```

can be troublesome with some devices, and it's generally preferable to only call `set_configuration` when necessary.

See [How (not) to call set_configuration() on a device already configured with the selected configuration?](https://github.com/pyusb/pyusb/blob/master/docs/faq.rst#how-not-to-call-set_configuration-on-a-device-already-configured-with-the-selected-configuration).

oldnapalm, 2020-12-23:
So `set_configuration` should not be called there? Anyway, tested without it and the issue persists.

I was reading this https://tacxfaqx.com/knowledge-base/cycplus-ant-stick/

For what I understand there's an issue with "continuous scan" in these devices. Maybe that's the cause of this issue?

I guess the apps that work with this stick don't use this "continuous scan", while python-ant does. Will look into it.

mcuee, 2020-12-24:
Close this for now as it is not a pyusb issue. 
@oldnapalm But you can still comment further if there are further questions.

closed 2020-12-24 by mcuee

rickwierenga, 2022-11-03:
For future readers, using libusbK instead of libusb-win32 fixes the problem for me.