pyusb/pyusb #344: TypeError: cannot use a str to initialize an array with typecode 'B'

weijunjun-git, 2020-12-21:
Traceback (most recent call last):
  File "C:\Users\wwx565944\AppData\Local\Programs\Python\Python39\lib\site-packages\usb\_interop.py", line 94, in as_array
    return array.array('B', data)
TypeError: cannot use a str to initialize an array with typecode 'B'

weijunjun-git, 2020-12-21:
1>.  a.frombytes(data) # usb/_interop.py line 99
File "C:\Users\wwx565944\AppData\Local\Programs\Python\Python39\lib\site-packages\usb\_interop.py", line 99, in as_array
    a.frombytes(data) # deprecated since 3.2
TypeError: a bytes-like object is required, not 'str'

weijunjun-git, 2020-12-21:
2>. a.fromstring(data)# usb/_interop.py line 99
File "C:\Users\wwx565944\AppData\Local\Programs\Python\Python39\lib\site-packages\usb\_interop.py", line 99, in as_array
    a.fromstring(data) # deprecated since 3.2
AttributeError: 'array.array' object has no attribute 'fromstring'

jonasmalacofilho, 2020-12-21:
Already fixed in the current git head (see: 4f3322e07fb651da35695f13aca27754ddc0d7b3).

closed 2020-12-21 by jonasmalacofilho

eblanshey, 2020-12-26:
FYI even using the latest fix I still get this error when writing e.g. `ep.write('t')`:

```
Traceback (most recent call last):
  File "C:\Users\Personal\Repositories\usb-python\.venv\lib\site-packages\usb\_interop.py", line 94, in as_array
    return array.array('B', data)
TypeError: cannot use a str to initialize an array with typecode 'B'
```

```
  File "C:\Users\Personal\Repositories\usb-python\.venv\lib\site-packages\usb\_interop.py", line 99, in as_array
    a.frombytes(data)
TypeError: a bytes-like object is required, not 'str'
```

If I change it to `ep.write(b't')` it works.

jonasmalacofilho, 2020-12-26:
Python's documentation was wrong, and 4f3322e07fb651da35695f13aca27754ddc0d7b3 broke that use case.

We'll probably bring it back, but you should generally handle string encodings yourself.  PyUSB was (indirectly) encoding them as UTF-8, and I imagine that was only made sense for you because of the overlap with 7-bit ASCII.

P.S. IIRC unicode strings in the USB spec are generally encoded in UTF-16, so what PyUSB used to do really didn't make a lot of sense.

eblanshey, 2020-12-26:
No worries, just wanted to let you know! I'm new to Python and I was following the tutorial pretty closely and got stuck here for a bit. Might be a good idea to update them to follow best encoding practices.

jonasmalacofilho, 2020-12-27:
Thank you for letting me know, particularly that the tutorial has some examples of writing strings.