pyusb/pyusb #345: How to find the DFU functional descriptor?

JerryVVan, 2020-12-22:
Hi, I want to find the DFU functional descriptor of the USB devices, but it seems that this descriptor wasn't defined in usb.util
```
# descriptor type
DESC_TYPE_DEVICE = 0x01
DESC_TYPE_CONFIG = 0x02
DESC_TYPE_STRING = 0x03
DESC_TYPE_INTERFACE = 0x04
DESC_TYPE_ENDPOINT = 0x05
```

mcuee, 2020-12-23:
You can always use pyusb to retrieve the descriptor. 
What you need to do is to read the DFU Specification from USBIF.
https://usb.org/sites/default/files/DFU_1.1.pdf

Is this a question or an enhancement request?

JerryVVan, 2020-12-23:
Thanks for response.
But I don't know how to use pyusb to get this descriptor. 
Could you show me a demo code?

jonasmalacofilho, 2020-12-23:
Hi,

I'm sorry, but guiding you step by step is outside the scope of an issue/not something I can take on right now.

Start by reading the DFU spec (and specific parts of the USB spec that may come up) and try to code something based on our general [Tutorial](https://github.com/pyusb/pyusb/blob/master/docs/tutorial.rst).  Also take a look at the [FAQ](https://github.com/pyusb/pyusb/blob/master/docs/faq.rst) for a couple of common problems.

After that, you should be able to find most answers to questions by searching our issues or code that depends on PyUSB.  In fact, there are many DFU tools out there, and one of them could already use PyUSB.

jonasmalacofilho, 2020-12-23:
@mcuee 

Ops, I assumed you wouldn't be able to help him in Python either... it that's not the case, please feel free to reopen.

closed 2020-12-23 by jonasmalacofilho

mcuee, 2020-12-23:
No problem. I agree to close this issue, as it is a question which requires the OP to study first.