pyusb/pyusb #347: Cannot claim interface using libusb1

eblanshey, 2020-12-27:
**Problem:** I am getting "unsupported" error when it tries to claim an interface using libusb1 on Windows 10. It DOES work using libusb0.

```
Traceback (most recent call last):
  File "c:\Users\Personal\Repositories\usb-python\host.py", line 36, in <module>
    ep_out.write('1'.encode('UTF-8'))
  File "C:\Users\Personal\Repositories\usb-python\.venv\lib\site-packages\usb\core.py", line 406, in write
    return self.device.write(self, data, timeout)
  File "C:\Users\Personal\Repositories\usb-python\.venv\lib\site-packages\usb\core.py", line 974, in write
    intf, ep = self._ctx.setup_request(self, endpoint)
  File "C:\Users\Personal\Repositories\usb-python\.venv\lib\site-packages\usb\core.py", line 113, in wrapper
    return f(self, *args, **kwargs)
  File "C:\Users\Personal\Repositories\usb-python\.venv\lib\site-packages\usb\core.py", line 227, in setup_request
    self.managed_claim_interface(device, intf)
  File "C:\Users\Personal\Repositories\usb-python\.venv\lib\site-packages\usb\core.py", line 113, in wrapper
    return f(self, *args, **kwargs)
  File "C:\Users\Personal\Repositories\usb-python\.venv\lib\site-packages\usb\core.py", line 178, in managed_claim_interface
    self.backend.claim_interface(self.handle, i)
  File "C:\Users\Personal\Repositories\usb-python\.venv\lib\site-packages\usb\_debug.py", line 62, in do_trace
    return f(*args, **named_args)
  File "C:\Users\Personal\Repositories\usb-python\.venv\lib\site-packages\usb\backend\libusb1.py", line 829, in claim_interface
    _check(self.lib.libusb_claim_interface(dev_handle.handle, intf))
  File "C:\Users\Personal\Repositories\usb-python\.venv\lib\site-packages\usb\backend\libusb1.py", line 600, in _check
    raise NotImplementedError(_strerror(ret))
NotImplementedError: Operation not supported or unimplemented on this platform
```

**Context:** Windows 10, Python 3.9.1, 64bit python, pyusb v 1.1.0. I didn't have libusb1 installed, so I downloaded the [latest release v1.0.24](https://github.com/libusb/libusb/releases) and copied `VS2019\MS64\dll\libusb-1.0.dll` to `C:\Windows\System32`. pyusb successfully finds this dll.

**What I tried:** 

- Using the dll from the `MinGW64\dll\` directory instead.
- Using the previous libusb1 version released in 2019, `v1.0.23`, and copied `MS64\dll\libusb-1.0.dll` to the System32 folder.
- All 3 drivers via Zadig: WinUSB, libusbk, and libusb-win32
- Latest pyusb version on `master`
- With libusb0, claiming the interface works, and reading from the device works.

**Debug logs using libusb1:**

```
2020-12-27 13:26:39,172 DEBUG:usb.backend.libusb1:_LibUSB.__init__(<WinDLL 'C:\Windows\System32\libusb-1.0.dll', handle 7ffba93a0000 at 0x243b4136eb0>)
2020-12-27 13:26:39,172 INFO:usb.core:find(): using backend "usb.backend.libusb1"
2020-12-27 13:26:39,172 DEBUG:usb.backend.libusb1:_LibUSB.enumerate_devices()
2020-12-27 13:26:39,203 DEBUG:usb.backend.libusb1:_LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x00000243B5F522E0>)
2020-12-27 13:26:39,203 DEBUG:usb.backend.libusb1:_LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x00000243B5F52490>)
2020-12-27 13:26:39,203 DEBUG:usb.backend.libusb1:_LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x00000243B5F52340>)
2020-12-27 13:26:39,203 DEBUG:usb.backend.libusb1:_LibUSB.open_device(<usb.backend.libusb1._Device object at 0x00000243B5F52340>)
2020-12-27 13:26:39,203 DEBUG:usb.backend.libusb1:_LibUSB.get_configuration(<usb.backend.libusb1._DeviceHandle object at 0x00000243B4136FA0>)
2020-12-27 13:26:39,203 DEBUG:usb.backend.libusb1:_LibUSB.get_configuration_descriptor(<usb.backend.libusb1._Device object at 0x00000243B5F52340>, 0)
2020-12-27 13:26:39,203 DEBUG:usb.backend.libusb1:_LibUSB.get_interface_descriptor(<usb.backend.libusb1._Device object at 0x00000243B5F52340>, 1, 0, 0)
2020-12-27 13:26:39,203 DEBUG:usb.backend.libusb1:_LibUSB.get_configuration_descriptor(<usb.backend.libusb1._Device object at 0x00000243B5F52340>, 0)
2020-12-27 13:26:39,203 DEBUG:usb.backend.libusb1:_LibUSB.get_endpoint_descriptor(<usb.backend.libusb1._Device object at 0x00000243B5F52340>, 0, 1, 0, 0)
2020-12-27 13:26:39,203 DEBUG:usb.backend.libusb1:_LibUSB.get_interface_descriptor(<usb.backend.libusb1._Device object at 0x00000243B5F52340>, 1, 0, 0)
2020-12-27 13:26:39,203 DEBUG:usb.backend.libusb1:_LibUSB.get_configuration_descriptor(<usb.backend.libusb1._Device object at 0x00000243B5F52340>, 0)
2020-12-27 13:26:39,203 DEBUG:usb.backend.libusb1:_LibUSB.get_endpoint_descriptor(<usb.backend.libusb1._Device object at 0x00000243B5F52340>, 0, 1, 0, 0)
2020-12-27 13:26:39,203 DEBUG:usb.backend.libusb1:_LibUSB.get_interface_descriptor(<usb.backend.libusb1._Device object at 0x00000243B5F52340>, 1, 0, 0)
2020-12-27 13:26:39,203 DEBUG:usb.backend.libusb1:_LibUSB.get_configuration_descriptor(<usb.backend.libusb1._Device object at 0x00000243B5F52340>, 0)
2020-12-27 13:26:39,203 DEBUG:usb.backend.libusb1:_LibUSB.get_endpoint_descriptor(<usb.backend.libusb1._Device object at 0x00000243B5F52340>, 1, 1, 0, 0)
2020-12-27 13:26:39,203 DEBUG:usb.backend.libusb1:_LibUSB.get_interface_descriptor(<usb.backend.libusb1._Device object at 0x00000243B5F52340>, 1, 0, 0)
2020-12-27 13:26:39,203 DEBUG:usb.backend.libusb1:_LibUSB.get_configuration_descriptor(<usb.backend.libusb1._Device object at 0x00000243B5F52340>, 0)
2020-12-27 13:26:39,203 DEBUG:usb.backend.libusb1:_LibUSB.get_configuration_descriptor(<usb.backend.libusb1._Device object at 0x00000243B5F52340>, 0)
2020-12-27 13:26:39,206 DEBUG:usb.backend.libusb1:_LibUSB.get_interface_descriptor(<usb.backend.libusb1._Device object at 0x00000243B5F52340>, 0, 0, 0)
2020-12-27 13:26:39,206 DEBUG:usb.backend.libusb1:_LibUSB.get_configuration_descriptor(<usb.backend.libusb1._Device object at 0x00000243B5F52340>, 0)
2020-12-27 13:26:39,206 DEBUG:usb.backend.libusb1:_LibUSB.get_endpoint_descriptor(<usb.backend.libusb1._Device object at 0x00000243B5F52340>, 0, 0, 0, 0)
2020-12-27 13:26:39,206 DEBUG:usb.backend.libusb1:_LibUSB.get_interface_descriptor(<usb.backend.libusb1._Device object at 0x00000243B5F52340>, 0, 0, 0)
2020-12-27 13:26:39,206 DEBUG:usb.backend.libusb1:_LibUSB.get_configuration_descriptor(<usb.backend.libusb1._Device object at 0x00000243B5F52340>, 0)
2020-12-27 13:26:39,206 DEBUG:usb.backend.libusb1:_LibUSB.get_interface_descriptor(<usb.backend.libusb1._Device object at 0x00000243B5F52340>, 0, 1, 0)
2020-12-27 13:26:39,206 DEBUG:usb.backend.libusb1:_LibUSB.get_configuration_descriptor(<usb.backend.libusb1._Device object at 0x00000243B5F52340>, 0)
2020-12-27 13:26:39,206 DEBUG:usb.backend.libusb1:_LibUSB.get_interface_descriptor(<usb.backend.libusb1._Device object at 0x00000243B5F52340>, 1, 0, 0)
2020-12-27 13:26:39,206 DEBUG:usb.backend.libusb1:_LibUSB.get_configuration_descriptor(<usb.backend.libusb1._Device object at 0x00000243B5F52340>, 0)
2020-12-27 13:26:39,206 DEBUG:usb.backend.libusb1:_LibUSB.get_endpoint_descriptor(<usb.backend.libusb1._Device object at 0x00000243B5F52340>, 0, 1, 0, 0)
2020-12-27 13:26:39,206 DEBUG:usb.backend.libusb1:_LibUSB.get_interface_descriptor(<usb.backend.libusb1._Device object at 0x00000243B5F52340>, 1, 0, 0)
2020-12-27 13:26:39,206 DEBUG:usb.backend.libusb1:_LibUSB.get_configuration_descriptor(<usb.backend.libusb1._Device object at 0x00000243B5F52340>, 0)
2020-12-27 13:26:39,207 DEBUG:usb.backend.libusb1:_LibUSB.claim_interface(<usb.backend.libusb1._DeviceHandle object at 0x00000243B4136FA0>, 1)
2020-12-27 13:26:39,239 DEBUG:usb.backend.libusb1:_LibUSB.close_device(<usb.backend.libusb1._DeviceHandle object at 0x00000243B4136FA0>)
2020-12-27 13:26:39,239 DEBUG:usb.backend.libusb1:_LibUSB._finalize_object()
```

**Why I am doing this if libusb0 works:** A) libusb0 doesn't detect one of the config's interfaces that libusb1 does, B) the device is not detecting a "write" command via usblib0 on the available interface, which may be explained by A), not connecting to the right interface.

Any help would be appreciated!

eblanshey, 2020-12-28:
Okay, it seems this is either a driver or libusb issue, not a pyusb issue. libusb shows that there are 3 interfaces, and I have been installing WinUSB drivers for only the first one. The second interface, the one with "CDC Data USB Device Interface Class" has bulk in/out endpoints that I need to communicate with, but this interface is not being listed in Zadig (even with List All) so I can't install a driver for it. This device is also not being listed in the Windows Device Manager. The `xusb.exe` utility cannot connect (claim) to the second interface either, due to "unrecognized device driver."

I was confused because when I installed libusbk/libusb-win32 on interface 0, I was able to claim and partially communicate (reads only) with interface 1 (when using libusb0.1 only). If I reverted interface 0 to the default "USB Serial (CDC)" via Zadig, I was no longer able to communicate with interface 1. Interesting thing is that I know the bulk in/out is working because I am able to communicate with it using a terminal via serial port, when the driver for interface 0 (which technically has an Interrupt Transfer Type)  is "USB Serial (CDC)". Not sure how that works, but it works. So I am at a loss as to how to do the same thing programmatically via libusb.

I understand this may be off topic so please close if needed. Perhaps I will try to post to the libusb mailing list.

eblanshey, 2020-12-29:
I still do not know why I couldn't connect to the interface. I ended up using a much simpler method to communicate with the bulk in/out endpoint on the USB device by using pySerial, using Window's built-in drivers.

closed 2020-12-29 by eblanshey

jonasmalacofilho, 2020-12-29:
Thanks for the detailed info nonetheless!

mcuee, 2021-01-21:
https://github.com/libusb/libusb/issues/85
Most likely you hit the limitation of libusb Windows when it comes to USB devices with IAD.