pyusb/pyusb #348: usb.core.USBError: [Errno 19] No such device (it may have been disconnected)

WanManito, 2020-12-29:
pyusb==1.1.0

When a HID device has two interfaces, one is the analog keyboard output, the other is the normal communication channel, this API can't work normally. When I switch to the normal interface, I will prompt the title error. This problem can be perfectly solved in Java, Android and C #. I don't know if it can be solved in this API, thank you

WanManito, 2020-12-29:
2020-12-29 10:57:32,516 DEBUG:usb.backend.libusb1:_LibUSB.__init__(<WinDLL 'C:\Windows\system32\libusb-1.0.dll', handle 7ffccc380000 at 0x1b45aeff7c8>)
[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.000000] [000046a0] libusb: debug [libusb_init] created default context
[ 0.000000] [000046a0] libusb: debug [libusb_init] libusb v1.0.23.11397
[ 0.000000] [000046a0] libusb: debug [windows_init_dlls] Will use CancelIoEx for I/O cancellation
[ 0.000998] [000046a0] libusb: debug [get_windows_version] Windows 10 64-bit
[ 0.000998] [000046a0] libusb: debug [windows_init_clock] hires timer available (Frequency: 10000000 Hz)
[ 0.000998] [000046a0] libusb: debug [windows_init_clock] timer thread will run on core #0
[ 0.001994] [000046a0] libusb: debug [htab_create] using 1021 entries hash table
[ 0.011969] [000046a0] libusb: info [winusbx_init] libusbK DLL is not available, will use native WinUSB
[ 0.013964] [000046a0] libusb: debug [winusbx_init] initalized sub API libusbK
[ 0.014968] [000046a0] libusb: debug [winusbx_init] initalized sub API libusb0
[ 0.014968] [000046a0] libusb: debug [winusbx_init] initalized sub API WinUSB
[ 0.016954] [000046a0] libusb: info [windows_init] UsbDk backend is not available
[ 0.016954] [000046a0] libusb: debug [usbi_add_pollfd] add fd 0 events 1
2020-12-29 10:57:32,534 INFO:usb.core:find(): using backend "usb.backend.libusb1"
2020-12-29 10:57:32,535 DEBUG:usb.backend.libusb1:_LibUSB.enumerate_devices()
[ 0.017952] [000046a0] libusb: debug [libusb_get_device_list]
[ 0.035904] [000046a0] libusb: debug [winusb_get_device_list] allocating new device for session [5F]
[ 0.035904] [000046a0] libusb: debug [winusb_get_device_list] allocating new device for session [60]
[ 0.036900] [000046a0] libusb: debug [winusb_get_device_list] allocating new device for session [61]
[ 0.036900] [000046a0] libusb: debug [get_api_type] driver(s): plser
[ 0.036900] [000046a0] libusb: debug [get_api_type] upper filter driver(s): serenum
[ 0.036900] [000046a0] libusb: debug [winusb_get_device_list] allocating new device for session [25]
[ 0.037897] [000046a0] libusb: debug [get_api_type] driver(s): BTHUSB
[ 0.037897] [000046a0] libusb: debug [get_api_type] lower filter driver(s): ibtusb
[ 0.037897] [000046a0] libusb: debug [winusb_get_device_list] allocating new device for session [3F]
[ 0.038895] [000046a0] libusb: debug [get_api_type] driver(s): AX88772
[ 0.038895] [000046a0] libusb: debug [winusb_get_device_list] allocating new device for session [45]
[ 0.039893] [000046a0] libusb: debug [get_api_type] driver(s): usbccgp
[ 0.039893] [000046a0] libusb: debug [get_api_type] matched driver name against Composite API
[ 0.039893] [000046a0] libusb: debug [winusb_get_device_list] allocating new device for session [46]
[ 0.039893] [000046a0] libusb: debug [get_api_type] driver(s): usbccgp
[ 0.040892] [000046a0] libusb: debug [get_api_type] matched driver name against Composite API
[ 0.040892] [000046a0] libusb: debug [winusb_get_device_list] allocating new device for session [47]
[ 0.040892] [000046a0] libusb: debug [get_api_type] driver(s): WUDFRd
[ 0.040892] [000046a0] libusb: debug [get_api_type] lower filter driver(s): WinUsb
[ 0.041889] [000046a0] libusb: debug [get_api_type] matched lower filter driver name against WinUSB
[ 0.041889] [000046a0] libusb: debug [winusb_get_device_list] allocating new device for session [48]
[ 0.042885] [000046a0] libusb: debug [get_api_type] driver(s): HidUsb
[ 0.042885] [000046a0] libusb: debug [get_api_type] matched driver name against HID API
[ 0.043885] [000046a0] libusb: debug [winusb_get_device_list] allocating new device for session [49]
[ 0.043885] [000046a0] libusb: debug [enumerate_hcd_root_hub] assigning HCD 'PCI\VEN_8086&DEV_9D2F&SUBSYS_16041D72&REV_21\3&11583659&0&A0' bus number 1
[ 0.047872] [000046a0] libusb: debug [winusb_get_device_list] found existing device for session [5F]
[ 0.047872] [000046a0] libusb: debug [init_device] found 1 configurations (active conf: 1)
[ 0.047872] [000046a0] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 41 bytes)
[ 0.048868] [000046a0] libusb: debug [init_device] (bus: 1, addr: 13, depth: 2, port: 4): 'USB\VID_0BDA&PID_5411\6&191677B0&1&4'
[ 0.049869] [000046a0] libusb: debug [winusb_get_device_list] found existing device for session [25]
[ 0.049869] [000046a0] libusb: debug [init_device] found 1 configurations (active conf: 1)
[ 0.049869] [000046a0] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 39 bytes)
[ 0.049869] [000046a0] libusb: debug [init_device] (bus: 1, addr: 14, depth: 3, port: 1): 'USB\VID_067B&PID_23C3\DLAZB19C116'
[ 0.050863] [000046a0] libusb: debug [winusb_get_device_list] found existing device for session [60]
[ 0.050863] [000046a0] libusb: debug [init_device] found 1 configurations (active conf: 1)
[ 0.051862] [000046a0] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 25 bytes)
[ 0.051862] [000046a0] libusb: debug [init_device] (bus: 1, addr: 1, depth: 1, port: 1): 'USB\VID_05E3&PID_0608\5&E4A4C6&0&1'
[ 0.052862] [000046a0] libusb: debug [winusb_get_device_list] found existing device for session [45]
[ 0.052862] [000046a0] libusb: debug [init_device] found 1 configurations (active conf: 1)
[... cut ...]
[ 0.370012] [000046a0] libusb: debug [libusb_unref_device] destroy device 1.1
[ 0.370012] [000046a0] libusb: debug [libusb_unref_device] destroy device 1.0
2020-12-29 10:57:32,889 DEBUG:usb.backend.libusb1:_LibUSB._finalize_object()
[ 0.372007] [000046a0] libusb: debug [libusb_exit]
[ 0.373005] [000046a0] libusb: debug [libusb_exit] destroying default context
[ 0.373005] [000046a0] libusb: debug [usbi_remove_pollfd] remove fd 0
[ 0.375996] [00003e64] libusb: debug [windows_clock_gettime_threaded] timer thread quitting

WanManito, 2020-12-29:
[image]
interface[0] It is available, but how do I get it

WanManito, 2020-12-29:
import usb.core
import usb.util
import time

import platform

# find our device
dev = usb.core.find(idVendor=0x03EB, idProduct=0x2421)  

# was it found?
if dev is None:
    raise ValueError('Device not found')

if platform.system().lower() == 'linux':
    if dev.is_kernel_driver_active(0):
        dev.detach_kernel_driver(0)  

# set the active configuration. With no arguments, the first
# configuration will be the active one
# dev.set_configuration()

# get an endpoint instance
# configuration = dev.get_active_configuration()
configuration = dev[0]

cfg_ = configuration[(0, 0)]
# alt = usb.util.find_descriptor(cfg_, find_all=True, bInterfaceNumber=1)
# print(alt)

# 0x2 
outPoint = usb.util.find_descriptor(
    cfg_,
    # match the first OUT endpoint
    custom_match=lambda e: usb.util.endpoint_direction(e.bEndpointAddress) == usb.util.ENDPOINT_OUT)

# 0x81 
inPoint = usb.util.find_descriptor(
    cfg_,
    # match the first IN endpoint
    custom_match=lambda e: usb.util.endpoint_direction(e.bEndpointAddress) == usb.util.ENDPOINT_IN)

# assert outPoint, inPoint is not None

outPoint.write("msg")  

read = inPoint.read(64)

jonasmalacofilho, 2020-12-29:
Your mileage will vary with HIDs: [Does libusb support USB HID devices?](https://github.com/libusb/libusb/wiki/FAQ#does-libusb-support-usb-hid-devices).

WanManito, 2020-12-29:
It should be supported, because libusb is also used in Java and C #, and now it is "debug [set]_ composite_ interface] interface[0] = \\.\HID#VID_ 03EB&amp;PID_ 2421&amp;MI_ 00#8&amp;6AEAA9B&amp;0&amp;0000#{4D1E55B2-F16F-11CF-88CB-001111000030} [ 0.096741] [000046a0] libusb: debug [winusb_ get_ device_ list] setting composite interface for [46]: [ 0.112698] [000046a0] libusb: debug [set_ composite_ interface] interface[1] = \\.\HID#VID_ 03EB&amp;PID_ 2421&amp;MI_ 01#8&amp;1E7236E0&amp;0&amp;0000#{4D1E55B2-F16F-11CF-88CB-001111000030}#KBD"，
"Interface [0]" is what I want @jonasmalacofilho

WanManito, 2020-12-29:
In other platforms, it only distinguishes between keyboard mode and communication mode, that is, which channel can be used in the interface. It can be seen in the debug log that it can distinguish between the two channels, which can be realized by logic

jonasmalacofilho, 2020-12-29:
> "Interface [0]" is what I want @jonasmalacofilho

@WanManito this is your seventh not-really-an-issue issue here.  I'm sorry, but I'm not consulting for you.  Please pinpoint a specific and, preferably, reproduceable (with common devices) problem; if a reproduceable example is not possible, please show real/practical C code using libusb that works and that shows what exactly is PyUSB doing wrong.

---

P.S.  And you don't even appear to be using PyUSB for a free/open source software project!

WanManito, 2021-01-05:
We use Cpython hidapi instead

WanManito, 2021-01-05:
This is the implementation of "Cpython hidapi", which can distinguish different channels of the same device. I've tried pyusb for a long time, but I haven't found a way
[image]

WanManito, 2021-01-05:
[image]
[image]
It can only find one

jonasmalacofilho, 2021-01-05:
As I mentioned...

Your mileage will vary with HIDs: see libusb's ["Does libusb support USB HID devices?"](https://github.com/libusb/libusb/wiki/FAQ#does-libusb-support-usb-hid-devices) FAQ entry.  And your libusb debug output shows:

```
[ 0.255317] [000046a0] libusb: warning [hid_open] could not open HID device in R/W mode (keyboard or mouse?) - trying without
[ 0.256320] [000046a0] libusb: error [hid_open] could not open device \\.\USB#VID_03EB&PID_2421#123123123123#{A5DCBF10-6530-11D2-901F-00C04FB951ED} (interface 1): [2] 系统找不到指定的文件。
[ 0.256320] [000046a0] libusb: debug [libusb_open] open 1.19 returns -4
Traceback (most recent call last):
```

The fact that "cpthon hidapi" (it's cython-hidapi, right?) works as you expect does not change the sad truth that you simply may not be able to open some HID interfaces on Windows, with PyUSB or any other libusb wrapper.  You could try different drivers... but it's probably easier to stick to HIDAPI if you can.

---

> [...]
> It can only find one

Working as intended, there is only one USB device (USB device != USB interface).

Look, it seems that you want (or at least need) help.  If there was anything terribly wrong with your code, I would have told immediately... but there isn't.  There also doesn't appear to be anything wrong with PyUSB or libusb (apart from documented limitations).  It just doesn't seem to be a way to use libusb with that interface.[1]

If you don't mind, I suggest you open source your project, then ask for help in the right venues (like forums related to the specific domain you're working on, but also our [not so active] mailing list).

[1] Although I admit our communication has to been as clear as I normally like, so I might be missing something.

WanManito, 2021-01-05:
@jonasmalacofilho Thank you for the interruption

WanManito, 2021-01-05:
@jonasmalacofilho Thank you for your answer. Thank you

closed 2021-01-05 by WanManito