pyusb/pyusb #355: Mac M1 support

2231puppy, 2021-01-21:
The Mac M1 series seems to be unsupported. Here is my debug output:

`2021-01-21 16:49:22,365 ERROR:usb.libloader:Libusb 1 (/opt/homebrew/Cellar/libusb/1.0.24/lib/libusb-1.0.dylib) could not be loaded
Traceback (most recent call last):
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/usb/libloader.py", line 118, in load_library
    return ctypes.CDLL(lib)
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/ctypes/__init__.py", line 374, in __init__
    self._handle = _dlopen(self._name, mode)
OSError: dlopen(/opt/homebrew/Cellar/libusb/1.0.24/lib/libusb-1.0.dylib, 6): no suitable image found.  Did find:
	/opt/homebrew/Cellar/libusb/1.0.24/lib/libusb-1.0.dylib: mach-o, but wrong architecture
	/opt/homebrew/Cellar/libusb/1.0.24/lib/libusb-1.0.0.dylib: mach-o, but wrong architecture
	/opt/homebrew/Cellar/libusb/1.0.24/lib/libusb-1.0.dylib: mach-o, but wrong architecture
	/opt/homebrew/Cellar/libusb/1.0.24/lib/libusb-1.0.0.dylib: mach-o, but wrong architecture
2021-01-21 16:49:22,367 ERROR:usb.backend.libusb1:Error loading libusb 1.0 backend
2021-01-21 16:49:22,367 ERROR:usb.libloader:'OpenUSB library' could not be found
2021-01-21 16:49:22,367 ERROR:usb.backend.openusb:Error loading OpenUSB backend
2021-01-21 16:49:22,369 ERROR:usb.libloader:'Libusb 0' could not be found
2021-01-21 16:49:22,369 ERROR:usb.backend.libusb0:Error loading libusb 0.1 backend
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/usb/core.py", line 1299, in find
    raise NoBackendError('No backend available')
usb.core.NoBackendError: No backend available`

Please add this support.

jonasmalacofilho, 2021-01-21:
The correct architecture of LibUSB must be installed:

```
libusb-1.0.dylib: mach-o, but wrong architecture
```

There's nothing for PyUSB to do here.  **Maybe** the libusb homebrew formula needs to be adjusted?

closed 2021-01-21 by jonasmalacofilho

mcuee, 2021-01-28:
FYI libusb and pyusb work fine under my Mac Mini M1.

You need to install homebrew correctly. Recommendation is to run native M1 version in /opt/homebrew and Intel version under /usr/local/homebrew.

Ref:
https://github.com/Homebrew/brew/issues/7857

mvanaltvorst, 2021-02-12:
This issue is not resolved.

```
>>> import usb.core
>>> dev = usb.core.find()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/Library/Python/3.8/site-packages/usb/core.py", line 1299, in find
    raise NoBackendError('No backend available')
usb.core.NoBackendError: No backend available
```

My environment:
```
> brew info libusb
libusb: stable 1.0.24 (bottled), HEAD
Library for USB device access
https://libusb.info/
/opt/homebrew/Cellar/libusb/1.0.24 (22 files, 560.3KB) *
  Poured from bottle on 2021-02-12 at 21:50:38
From: https://github.com/Homebrew/homebrew-core/blob/HEAD/Formula/libusb.rb
License: LGPL-2.1-or-later
==> Options
--HEAD
	Install HEAD version
==> Analytics
install: 91,011 (30 days), 270,440 (90 days), 755,150 (365 days)
install-on-request: 8,181 (30 days), 26,816 (90 days), 74,764 (365 days)
build-error: 0 (30 days)

> brew list libusb
/opt/homebrew/Cellar/libusb/1.0.24/include/libusb-1.0/libusb.h
/opt/homebrew/Cellar/libusb/1.0.24/lib/libusb-1.0.0.dylib
/opt/homebrew/Cellar/libusb/1.0.24/lib/pkgconfig/libusb-1.0.pc
/opt/homebrew/Cellar/libusb/1.0.24/lib/ (2 other files)
/opt/homebrew/Cellar/libusb/1.0.24/share/libusb/ (9 files)

> otool -L /opt/homebrew/Cellar/libusb/1.0.24/lib/libusb-1.0.0.dylib
/opt/homebrew/Cellar/libusb/1.0.24/lib/libusb-1.0.0.dylib:
	/opt/homebrew/opt/libusb/lib/libusb-1.0.0.dylib (compatibility version 4.0.0, current version 4.0.0)
	/usr/lib/libobjc.A.dylib (compatibility version 1.0.0, current version 228.0.0)
	/System/Library/Frameworks/IOKit.framework/Versions/A/IOKit (compatibility version 1.0.0, current version 275.0.0)
	/System/Library/Frameworks/CoreFoundation.framework/Versions/A/CoreFoundation (compatibility version 150.0.0, current version 1770.255.0)
	/usr/lib/libSystem.B.dylib (compatibility version 1.0.0, current version 1292.60.1)
```
I think @mcuee is using the x86_64 version of libusb via Rosetta 2. This is not a viable solution in the long term. Libusb supports a native ARM variant of their library, and it would be great if pyusb is able to take advantage of it as well.

jonasmalacofilho, 2021-02-12:
@mvanaltvorst and @mcuee 

Can you both execute

```
$ python -c "import ctypes.util; print(ctypes.util.find_library('usb'))"
```

on a M1 Mac, trying to pick the suitable (ARM) Python installation?

mvanaltvorst, 2021-02-12:
```
Python 3.8.2 (default, Dec 21 2020, 15:06:03) 
[Clang 12.0.0 (clang-1200.0.32.29)] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> import ctypes.util
>>> print(ctypes.util.find_library('usb'))
None
```

jonasmalacofilho, 2021-02-12:
@mvanaltvorst this suggests a problem with Python 3.8.2 `ctypes` on M1 Macs.

A quick search led to me to [Issue41100 ](https://bugs.python.org/issue41100), where the last couple of comments are:

> Is there an ETA for having Big Sur support backported to the 3.8 branch, or is it already there and I'm just not seeing it here?
> —*John Reese, Feb. 8*

> There is no ETA. I'm spending more time away from the computer and on other projects when I'm not at work. I also have no personal need for 3.8 support.
> If anyone else is up to the job: the back port to 3.9 should be a good starting point, but I don't trust the output of the cherry_picker tool to 3.8.
> —*Ronald Oussoren, Feb. 9*

It seems that you would have better luck with Python 3.9.1.

mcuee, 2021-02-13:
I am using M1 version of homebrew and indeed Python 3.9.1.
```
mcuee@mcuees-Mac-mini ~ % which brew  
/opt/homebrew/bin/brew

mcuee@mcuees-Mac-mini ~ % python3 --version
Python 3.9.1

mcuee@mcuees-Mac-mini ~ % python3 -c "import ctypes.util; print(ctypes.util.find_library('usb'))"
/opt/homebrew/lib/libusb.dylib

mcuee@mcuees-Mac-mini ~ % file /opt/homebrew/Cellar/libusb/1.0.24/lib/libusb-1.0.0.dylib
/opt/homebrew/Cellar/libusb/1.0.24/lib/libusb-1.0.0.dylib: Mach-O 64-bit dynamically linked shared library arm64

mcuee, 2021-02-13:
Anyway, if you install Python3.8 from homebrew, pyusb works as well.

```
brew search python
brew info python@3.8
brew install python@3.8
/opt/homebrew/opt/python@3.8/bin/pip3 install pyusb
...
mcuee@mcuees-Mac-mini ~ % /opt/homebrew/opt/python@3.8/bin/python3 --version
Python 3.8.7
mcuee@mcuees-Mac-mini ~ % /opt/homebrew/opt/python@3.8/bin/python3 -c "import ctypes.util; print(ctypes.util.find_library('usb'))"
/opt/homebrew/lib/libusb.dylib

With the above commands, pyusb works as well.

```

mcuee, 2021-02-13:
BTW, if you look at the formula Python 3.8, patched version is used by homebrew as of now.
https://github.com/Homebrew/homebrew-core/blob/master/Formula/python@3.8.rb

```
# Patch for MACOSX_DEPLOYMENT_TARGET on Big Sur, not yet backported to the 3.8 branch
  # https://bugs.python.org/issue42504
  # https://github.com/python/cpython/pull/23556
  patch do
    url "https://github.com/python/cpython/commit/09a698b4.patch?full_index=1"
    sha256 "a297c429ca83847d1d3ae84bdbb66e8a035731fd17955205de37c902d13300bf"
  end

  # Remove this block when upstream backports arm64 compatibility to the 3.8 branch
  if Hardware::CPU.arm?
    # Upstream PRs #20171, #21114, #21224 and #21249
    patch do
      url "https://raw.githubusercontent.com/Homebrew/formula-patches/cef51fde/python/3.8.7.patch"
      sha256 "01ebca9fed99fd0b34abfd198e7d6137e3f851d9d1a07ded837b28c7998e810c"
    end
  end

```

mvanaltvorst, 2021-02-13:
> @mvanaltvorst this suggests a problem with Python 3.8.2 `ctypes` on M1 Macs.
>
> A quick search led to me to [Issue41100 ](https://bugs.python.org/issue41100), where the last couple of comments are: [...]

You're right, it seems like the version of Python Xcode uses doesn't work correctly. Installing the latest version of Python3 with home-brew seems to have fixed my problems!

mcuee, 2021-02-14:
Just want to share a tip if you have both Apple Silicon Homebrew and x86_64 Homebrew (Rosetta 2) installed. You can automatically select the right brew depending whether you are using Rosetta or not.

Adapted from the following URL.
https://cutecoder.org/software/run-command-line-apple-silicon/
```
mcuee@mcuees-Mac-mini ~ % cat .zprofile 
if [ "$(sysctl -n sysctl.proc_translated)" = "1" ]; then
    eval $(/usr/local/Homebrew/bin/brew shellenv)
else
    eval $(/opt/homebrew/bin/brew shellenv)
fi
```

jonasmalacofilho, 2021-04-14:
For reference, a temporary workaround is to construct the backend with a custom `find_library()` function.  More information can be found in [#360 (comment)](https://github.com/pyusb/pyusb/issues/360#issuecomment-812802678).

mvanaltvorst, 2021-05-07:
> For reference, a temporary workaround is to construct the backend with a custom `find_library()` function. More information can be found in [#360 (comment)](https://github.com/pyusb/pyusb/issues/360#issuecomment-812802678).

Do you have more information on this? Seems like pyusb still cannot find libusb with the latest 3.9.2, nearly three months later.

jonasmalacofilho, 2021-05-07:
> I'm using a miniforge Python installation, which does not have the patch that homebrew 3.8 supposedly contains.

You should probably report the problem to miniforge.  Since both CPython and Homebrew have been able to backport the fix, I'm hopeful miniforge will be able to do the same.

Either way, I'm sorry, but backporting Python patches to custom Python distributions is well outside the scope of PyUSB.

mvanaltvorst, 2021-05-07:
Thanks for your quick response, will try that!

mellow-hype, 2021-11-20:
For anyone else still having this issue, I manage python installations with pyenv and installed libusb with homebrew and was having this issue as well. I was finally able to resolve the issue by creating a new symlink in `/opt/homebrew/lib/` named `libusb.dylib` linked to the existing link created by homebrew.

```
ln -s /opt/homebrew/lib/libusb-1.0.0.dylib /opt/homebrew/lib/libusb.dylib
```

It seems the expected file name is `libusb.dylib`.

7ombie, 2022-01-16:
Same issue (M1, Monterey). I already had Python 3.9 installed locally, so used `brew` to install `python3` (in its own directories), then used this to install `pyusb` into the Homebrew Python libraries:

``` sh
/opt/homebrew/bin/pip3 install pyusb
```

Running the same version of Python lets me use `libusb`:

``` sh
/opt/homebrew/bin/python3
```

I'll tidy up my OS later, but this works at least. Thanks for the pointers, @everyone.

7ombie, 2022-01-17:
Probably unrelated, by PyUSB was super unreliable. It would randomly find the device about one in five attempts. I just have to keep running the script. Sometimes it would send transfers, sometimes not. And when it read data (from an interrupt pipe), it would often drop byte. It was unworkable, and the whole point of using Python was to make prototyping simple.

jonasmalacofilho, 2022-01-17:
@mcuee

I find it unlikely that PyUSB or LibUSB would be so unrealiable on their own; maybe it's a problem with that user's specific device?  (But I'm not going to engage with a vague "unworkable" comment).

---

Anyway, can you let help answer a few questions about how are we doing on M1 Macs these days?

- can PyUSB find libusb-1.0?  what needs to be installed?
- is the symlink workaround [mentioned before](https://github.com/pyusb/pyusb/issues/355#issuecomment-974726078) necessary?
- can PyUSB+libusb reliably find and read/write from/to a device?

7ombie, 2022-01-18:
The device (Novation Twitch) works with a macOS driver. As well, things that don't work with PyUSB work as documented with WebUSB. The device itself does seem a bit flakey to be honest (Novation took a cheap and cheerful approach to USB back when the Twitch was designed), so that may be a factor, but PyUSB was remarkably useless (on my machine).

I would open an issue, but there were so many things wrong, it'd take too long to break it down into anything useful.

I understand that this information is not directly useful (and I did say it's probably unrelated), but I figured I should still make a note of it here, so the developers are at least aware that one M1 user had some serious stability issues on their machine, just in case it turns out to be relevant later.

mcuee, 2022-01-19:
> @mcuee
> 
> I find it unlikely that PyUSB or LibUSB would be so unrealiable on their own; maybe it's a problem with that user's specific device? (But I'm not going to engage with a vague "unworkable" comment).
> 
> Anyway, can you help answer a few questions about how we are doing on M1 Macs these days?
> 
> * can PyUSB find libusb-1.0?  what needs to be installed?

Yes, just install Homebrew.

> * is the symlink workaround [mentioned before](https://github.com/pyusb/pyusb/issues/355#issuecomment-974726078) necessary?

Not necessary if using homebrew.

> * can PyUSB+libusb reliably find and read/write from/to a device?

Yes if the device is behaving well.

mcuee, 2022-01-19:
The whole problem of libusb under macOS is usually because there is an existing macOS kernel driver with exclusive access to the device. Then there is no easy way to detach the kernel driver. In that case, it is not recommended to use libusb/pyusb.

Ref: 
https://github.com/libusb/libusb/issues/1014
https://github.com/libusb/libusb/issues/972

mellow-hype, 2022-01-24:
hey @jonasmalacofilho -- in case it helps, I recently installed libusb via homebrew on a new M1 mac and ended up running into either the same or similar issue and resolving it by using `export DYLD_LIBRARY_PATH=/opt/homebrew/lib` prior to running the package that was leveraging libusb. I had forgotten about the fix I had previously noted above. Looking at both "fixes", I think the issue may be in the custom directory homebrew uses on M1s (`/opt/homebrew`) and it not being in the default dyld search path (at least not for non-homebrew Python installations).

mcuee, 2022-01-24:
I think the main difference is that I am using the Homebrew python 3 and not Apple Python 3.

jonasmalacofilho, 2022-01-25:
Thank you all for the extra information.

mcuee, 2022-03-09:
Just an example run log to show that `export DYLD_LIBRARY_PATH=/opt/homebrew/lib` is the right fix if you are using non-Homebrew python installations, for example, Apple Python3, or Python.org Python3, or others.

I am using a virtual env from Python.org Python3.10 installation in the following example.

```
(py310venv_universal) mcuee@mcuees-Mac-mini python % python
Python 3.10.2 (v3.10.2:a58ebcc701, Jan 13 2022, 14:50:16) [Clang 13.0.0 (clang-1300.0.29.30)] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> import usb.core
>>> dev=usb.core.find()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/Users/mcuee/build/python/py310venv_universal/lib/python3.10/site-packages/usb/core.py", line 1309, in find
    raise NoBackendError('No backend available')
usb.core.NoBackendError: No backend available
>>> exit()

(py310venv_universal) mcuee@mcuees-Mac-mini python % export DYLD_LIBRARY_PATH=/opt/homebrew/lib
(py310venv_universal) mcuee@mcuees-Mac-mini python % python
Python 3.10.2 (v3.10.2:a58ebcc701, Jan 13 2022, 14:50:16) [Clang 13.0.0 (clang-1300.0.29.30)] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> import usb.core
>>> dev=usb.core.find()
>>> dev
<DEVICE ID 1915:1025 on Bus 002 Address 009>
>>> 
```

jonasmalacofilho, 2022-07-15:
@jveitchmichaelis, at this point (and for a significant part of the thread up to this point), failures to search the DLLs in the usual places are most likely issues with these interpreters.

That said, one thing you could check before raising an issue to Conda is whether your running it natively on Arm.

jveitchmichaelis, 2022-07-17:
@jonasmalacofilho good call with checking to see if it's running a native version. I found [this](https://stackoverflow.com/questions/65415996/how-to-specify-the-architecture-or-platform-for-a-new-conda-environment-apple) SO post which boils down to creating a new env with:

```
CONDA_SUBDIR=osx-arm64 conda create -n <env_name> python=3.10 -c conda-forge --override-channels
```

and then installing:

```
conda install pyusb libusb -c conda-forge --override-channels
```

(I also ran the suggested `conda env config vars set CONDA_SUBDIR=osx-arm64` inside the env, I guess because it's not set permanently above).

enterdoki, 2022-08-14:
> For anyone else still having this issue, I manage python installations with pyenv and installed libusb with homebrew and was having this issue as well. I was finally able to resolve the issue by creating a new symlink in `/usr/local/lib` named `libusb.dylib` linked to the existing link created by homebrew.
> 
> ```
> ln -s /opt/homebrew/lib/libusb-1.0.0.dylib //usr/local/lib/libusb.dylib
> ```
> 
> It seems the expected file name is `libusb.dylib`.

This worked for me. Thanks.

d3lg, 2023-10-19:
> > > For anyone else still having this issue, I manage python installations with pyenv and installed libusb with homebrew and was having this issue as well. I was finally able to resolve the issue by creating a new symlink in `/usr/local/lib` named `libusb.dylib` linked to the existing link created by homebrew.
> > > ```
> > > ln -s /opt/homebrew/lib/libusb-1.0.0.dylib //usr/local/lib/libusb.dylib
> > > ```
> > > 
> > > 
> > > It seems the expected file name is `libusb.dylib`.
> > 
> > 
> > This worked for me. Thanks. (M1 Pro)
> 
> Worked for me, too!

Worked for me, too!!!!!!!!!! (Y)