pyusb/pyusb #356: device.set_configuration() raises error with None errno

ryohajika, 2021-02-10:
Hi there :wave:

I'm trying to update one of [a legacy library](https://github.com/ryohajika/Adafruit-Trinket-USB) for the Trinket series Arduino compatible boards from Adafruit.
Basically it is all about realizing USB-serial communication on the trinket boards which don't have any USB-Serial conversion chip by a firmware-based implementation technique: https://github.com/obdev/v-usb .

I'm trying to update the python client code to realize data pass-through between two virtual serial ports made using `socat` command, but the error in below, when `device.set_configuration()` is executed.

```
File "TrinketFakeUsbSerialHostSW.py", line 136, in findTrinket
    device.set_configuration()
  File "build/bdist.macosx-11.2-x86_64/egg/usb/core.py", line 905, in set_configuration
  File "build/bdist.macosx-11.2-x86_64/egg/usb/core.py", line 113, in wrapper
  File "build/bdist.macosx-11.2-x86_64/egg/usb/core.py", line 159, in managed_set_configuration
  File "build/bdist.macosx-11.2-x86_64/egg/usb/backend/libusb1.py", line 812, in set_configuration
  File "build/bdist.macosx-11.2-x86_64/egg/usb/backend/libusb1.py", line 604, in _check
usb.core.USBError: [Errno None] Other error
```

Apparently both of the vendor ID and the product ID are modified based on the information I see in the system from the original implementation done by Adafruit, and I don't see any error to check and open the device using pyUSB. Does anybody have any idea what is the cause of this error?

Environment: MacBook 15inch (2018), macOS 11.2 (Intel, x86_64), Python 2.7.16 (bundled to the system), pyUSB 1.1.1 / Adafruit Trinket Pro (3V/12MHz/ATmega328P) which is connected to the computer via a satechi USB-C to USB3 adapter.

jonasmalacofilho, 2021-02-10:
This "Other error" (`Errno None`) is quite odd.  Can you run the code again under `PYUSB_DEBUG=debug` and `LIBUSB_DEBUG=4` and show me the complete output?  And what version of the libusb library do you have?

On another front, `set_configuration` can cause some issues (or, more accurately, unexpected behavior) if the device is already in the desired configuration state:

 - [PyUSB FAQ: How (not) to call set_configuration() on a device already configured with the selected configuration?](https://github.com/pyusb/pyusb/blob/master/docs/faq.rst#how-not-to-call-set_configuration-on-a-device-already-configured-with-the-selected-configuration)
 - [libusb: Caveats: Configuration selection and handling](http://libusb.sourceforge.net/api-1.0/libusb_caveats.html#configsel)

Maybe the approach described in those documents will help.

---

P.S. Please send the debug output for the "Other error" regardless of whether the above fixes your issue, as there might be a problem in our `_check` function.

P.P.S. Python 2.7 is no longer supported in PyUSB 1.1.1.  If you absolutely must stay with Python 2.7 (please note that it has reached end-of-life more than a year ago), the last PyUSB version to support it was 1.1.0.

ryohajika, 2021-02-10:
Thank you for your quick response!
I just quickly got log with those debug options... for me it seems like either libusb or pyUSB is not grabbing the device info...

<details><summary>Console log (macOS11.2, Python 2.7.16, libusb 1.0, pyUSB 1.1.0)</summary><p>

```

WARNING: Python 2.7 is not recommended.
This version is included in macOS for compatibility with legacy software.
Future versions of macOS will not include Python 2.7.
Instead, it is recommended that you transition to using 'python3' from within Terminal.

Python 2.7.16 (default, Dec 21 2020, 23:00:36)
[GCC Apple LLVM 12.0.0 (clang-1200.0.30.4) [+internal-os, ptrauth-isa=sign+stri on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> import usb.core
>>> usb.__version__
u'1.1.0'
>>> device = usb.core.find(idVendor = 0x1781, idProduct = 0x0c9f)
2021-02-11 01:00:58,683 DEBUG:usb.backend.libusb1:_LibUSB.__init__(<CDLL '/usr/local/lib/libusb-1.0.dylib', handle 7fc76a132ea0 at 10cae6710>)
[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.000008] [00049057] libusb: debug [libusb_init] created default context
[ 0.000014] [00049057] libusb: debug [libusb_init] libusb v1.0.24.11584
[ 0.000201] [00049057] libusb: debug [darwin_get_cached_device] could not get connected port number
[ 0.000207] [00049057] libusb: debug [darwin_get_cached_device] finding cached device for sessionID 0x0
[ 0.000226] [00049057] libusb: debug [darwin_get_cached_device] caching new device with sessionID 0x0
[ 0.004196] [00049057] libusb: debug [darwin_cache_device_descriptor] cached device descriptor:
[ 0.004207] [00049057] libusb: debug [darwin_cache_device_descriptor]   bDescriptorType:    0x01
[ 0.004209] [00049057] libusb: debug [darwin_cache_device_descriptor]   bcdUSB:             0x0100
[ 0.004210] [00049057] libusb: debug [darwin_cache_device_descriptor]   bDeviceClass:       0x09
[ 0.004212] [00049057] libusb: debug [darwin_cache_device_descriptor]   bDeviceSubClass:    0xff
[ 0.004213] [00049057] libusb: debug [darwin_cache_device_descriptor]   bDeviceProtocol:    0x03
[ 0.004215] [00049057] libusb: debug [darwin_cache_device_descriptor]   bMaxPacketSize0:    0x08
[ 0.004216] [00049057] libusb: debug [darwin_cache_device_descriptor]   idVendor:           0x05ac
[ 0.004217] [00049057] libusb: debug [darwin_cache_device_descriptor]   idProduct:          0x8007
[ 0.004219] [00049057] libusb: debug [darwin_cache_device_descriptor]   bcdDevice:          0x0198
[ 0.004220] [00049057] libusb: debug [darwin_cache_device_descriptor]   iManufacturer:      0x00
[ 0.004278] [00049057] libusb: debug [darwin_cache_device_descriptor]   iProduct:           0x00
[ 0.004286] [00049057] libusb: debug [darwin_cache_device_descriptor]   iSerialNumber:      0x00
[ 0.004302] [00049057] libusb: debug [darwin_cache_device_descriptor]   bNumConfigurations: 0x01
[ 0.004412] [00049057] libusb: debug [darwin_get_cached_device] could not get connected port number
[ 0.004420] [00049057] libusb: debug [darwin_get_cached_device] finding cached device for sessionID 0x0
[ 0.004443] [00049057] libusb: debug [darwin_get_cached_device] matching sessionID/locationID 0x0/0x0 against cached device with sessionID/locationID 0x0/0x80000000
[ 0.004448] [00049057] libusb: debug [darwin_get_cached_device] using cached device for device
[ 0.004506] [00049057] libusb: debug [darwin_get_cached_device] could not get connected port number
[ 0.004510] [00049057] libusb: debug [darwin_get_cached_device] finding cached device for sessionID 0x0
[ 0.004526] [00049057] libusb: debug [darwin_get_cached_device] matching sessionID/locationID 0x0/0x14000000 against cached device with sessionID/locationID 0x0/0x80000000
[ 0.004529] [00049057] libusb: debug [darwin_get_cached_device] using cached device for device
[ 0.004577] [00049057] libusb: debug [darwin_get_cached_device] could not get connected port number
[ 0.004580] [00049057] libusb: debug [darwin_get_cached_device] finding cached device for sessionID 0x0
[ 0.004598] [00049057] libusb: debug [darwin_get_cached_device] matching sessionID/locationID 0x0/0x1000000 against cached device with sessionID/locationID 0x0/0x80000000
[ 0.004601] [00049057] libusb: debug [darwin_get_cached_device] using cached device for device
[ 0.004674] [00049057] libusb: debug [darwin_get_cached_device] finding cached device for sessionID 0x381886aa
[ 0.006153] [00049057] libusb: debug [darwin_get_cached_device] matching sessionID/locationID 0x381886aa/0x80300000 against cached device with sessionID/locationID 0x0/0x80000000
[ 0.006165] [00049057] libusb: debug [darwin_get_cached_device] caching new device with sessionID 0x381886aa
[ 0.008339] [00049057] libusb: debug [darwin_device_from_service] set up plugin for service retry: out of resources
[ 0.008666] [00049057] libusb: debug [darwin_device_from_service] set up plugin for service retry: out of resources
[ 0.008936] [00049057] libusb: debug [darwin_device_from_service] set up plugin for service retry: out of resources
[ 0.009167] [00049057] libusb: debug [darwin_device_from_service] set up plugin for service retry: out of resources
[ 0.009407] [00049057] libusb: debug [darwin_device_from_service] set up plugin for service retry: out of resources
[ 0.009441] [00049057] libusb: debug [darwin_device_from_service] could not set up plugin for service: out of resources
[... cut ...]
  File "build/bdist.macosx-11.2-x86_64/egg/usb/core.py", line 113, in wrapper
  File "build/bdist.macosx-11.2-x86_64/egg/usb/core.py", line 159, in managed_set_configuration
  File "build/bdist.macosx-11.2-x86_64/egg/usb/_debug.py", line 62, in do_trace
  File "build/bdist.macosx-11.2-x86_64/egg/usb/backend/libusb1.py", line 812, in set_configuration
  File "build/bdist.macosx-11.2-x86_64/egg/usb/backend/libusb1.py", line 604, in _check
usb.core.USBError: [Errno None] Other error

```

</p></details>

ryohajika, 2021-02-10:
I keep seeing the `_check` before the error message `usb.core.USBError: [Errno None] Other error` so it could be related to `_check` I guess 🤔

jonasmalacofilho, 2021-02-10:
> for me it seems like either libusb or pyUSB is not grabbing the device info...

Can you try `print(device)` instead of just `device`?

> I keep seeing the _check before the error message usb.core.USBError: [Errno None] Other error so it could be related to _check I guess

Our `_check` function is a generic handler for libusb errors.  It probably isn't the underlying problem, but we generally expect more from it than simply `[Errno None] Other error`.

I wish I could see the raw return value from `libusb_set_configuration`.  Are you familiar with the [Python Debugger](https://docs.python.org/3/library/pdb.html)?

mcuee, 2021-02-11:
@ryohajika 
Two inputs from my side.
1) V-USB device (firmware based USB) may not be totally compatible with the USB specification. So I am not so sure if that is causing the problem here.
2) There are a few critical fixed in libusb after 1.0.24 release, not so sure if that helps if you can try latest git version of libusb.

mcuee, 2021-02-11:
@ryohajika Please post the device USB descriptors as well. Sometimes macOS can be more stringent.

ryohajika, 2021-02-16:
Hi @jonasmalacofilho and @mcuee 
Thank you for your feedbacks.

I still stick with the same environment, but I used the latest libusb from the master brunch. Also tried to use pdb but was not fully sure how to use it...
I understand the problem is derived from V-USB firmware, but I'm just curious if it is a problem that can be solved by python side :)
Please see the log in below:

<details>
<summary>Console log (macOS 11.0.1, Python 2.7.16, libusb 1.0.24.11606, pyUSB 1.1.0)</summary>

```

Python 2.7.16 (default, Oct 30 2020, 02:15:49)
[GCC Apple LLVM 12.0.0 (clang-1200.0.30.4) [+internal-os, ptrauth-isa=sign+stri on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> import usb.core, usb.util
>>> import serial
>>> usb.__version__
u'1.1.0'
>>> device = usb.core.find(idVendor = 0x1781, idProduct = 0x0c9f)
2021-02-16 21:47:12,677 DEBUG:usb.backend.libusb1:_LibUSB.__init__(<CDLL '/usr/local/lib/libusb-1.0.dylib', handle 14bf09db0 at 1047f2e80>)
[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.000023] [00108f9c] libusb: debug [libusb_init] created default context
[ 0.000049] [00108f9c] libusb: debug [libusb_init] libusb v1.0.24.11606
[ 0.000292] [00108f9c] libusb: debug [darwin_get_cached_device] finding cached device for sessionID 0xbc32304
[ 0.000366] [00108f9c] libusb: debug [darwin_get_cached_device] caching new device with sessionID 0xbc32304
[ 0.007816] [00108f9c] libusb: debug [darwin_cache_device_descriptor] cached device descriptor:
[ 0.008343] [00108f9c] libusb: debug [darwin_cache_device_descriptor]   bDescriptorType:    0x01
[ 0.008353] [00108f9c] libusb: debug [darwin_cache_device_descriptor]   bcdUSB:             0x0310
[ 0.008355] [00108f9c] libusb: debug [darwin_cache_device_descriptor]   bDeviceClass:       0x00
[ 0.008357] [00108f9c] libusb: debug [darwin_cache_device_descriptor]   bDeviceSubClass:    0x00
[ 0.008359] [00108f9c] libusb: debug [darwin_cache_device_descriptor]   bDeviceProtocol:    0x00
[ 0.008360] [00108f9c] libusb: debug [darwin_cache_device_descriptor]   bMaxPacketSize0:    0x09
[ 0.008362] [00108f9c] libusb: debug [darwin_cache_device_descriptor]   idVendor:           0x059f
[ 0.008364] [00108f9c] libusb: debug [darwin_cache_device_descriptor]   idProduct:          0x1093
[ 0.008365] [00108f9c] libusb: debug [darwin_cache_device_descriptor]   bcdDevice:          0x0100
[ 0.008367] [00108f9c] libusb: debug [darwin_cache_device_descriptor]   iManufacturer:      0x02
[ 0.008369] [00108f9c] libusb: debug [darwin_cache_device_descriptor]   iProduct:           0x03
[ 0.008410] [00108f9c] libusb: debug [darwin_cache_device_descriptor]   iSerialNumber:      0x01
[ 0.008415] [00108f9c] libusb: debug [darwin_cache_device_descriptor]   bNumConfigurations: 0x01
[ 0.008449] [00108f9c] libusb: debug [darwin_check_configuration] active config: 1, first config: 1
[ 0.008451] [00108f9c] libusb: debug [process_new_device] allocating new device in context 0x14d0079c0 for with session 0xbc32304
[ 0.008455] [00108f9c] libusb: debug [process_new_device] found device with address 1 port = 6 parent = 0x0 at 0x14d10641a
[ 0.008624] [00108f9c] libusb: debug [darwin_get_cached_device] finding cached device for sessionID 0xc5292e5
[ 0.008654] [00108f9c] libusb: debug [darwin_get_cached_device] matching sessionID/locationID 0xc5292e5/0x200000 against cached device with sessionID/locationID 0xbc32304/0x2600000
[ 0.008657] [00108f9c] libusb: debug [darwin_get_cached_device] caching new device with sessionID 0xc5292e5
[ 0.008950] [00108f9c] libusb: debug [darwin_cache_device_descriptor] cached device descriptor:
[ 0.008955] [00108f9c] libusb: debug [darwin_cache_device_descriptor]   bDescriptorType:    0x01
[ 0.008956] [00108f9c] libusb: debug [darwin_cache_device_descriptor]   bcdUSB:             0x0310
[ 0.008957] [00108f9c] libusb: debug [darwin_cache_device_descriptor]   bDeviceClass:       0x09
[ 0.008958] [00108f9c] libusb: debug [darwin_cache_device_descriptor]   bDeviceSubClass:    0x00
[ 0.008959] [00108f9c] libusb: debug [darwin_cache_device_descriptor]   bDeviceProtocol:    0x03
[ 0.008960] [00108f9c] libusb: debug [darwin_cache_device_descriptor]   bMaxPacketSize0:    0x09
[ 0.008962] [00108f9c] libusb: debug [darwin_cache_device_descriptor]   idVendor:           0x2109
[ 0.008963] [00108f9c] libusb: debug [darwin_cache_device_descriptor]   idProduct:          0x0817
[ 0.008964] [00108f9c] libusb: debug [darwin_cache_device_descriptor]   bcdDevice:          0x0383
[ 0.008965] [00108f9c] libusb: debug [darwin_cache_device_descriptor]   iManufacturer:      0x01
[ 0.008994] [00108f9c] libusb: debug [darwin_cache_device_descriptor]   iProduct:           0x02
[ 0.008998] [00108f9c] libusb: debug [darwin_cache_device_descriptor]   iSerialNumber:      0x00
[ 0.008999] [00108f9c] libusb: debug [darwin_cache_device_descriptor]   bNumConfigurations: 0x01
[ 0.009016] [00108f9c] libusb: debug [darwin_check_configuration] active config: 1, first config: 1
[ 0.009017] [00108f9c] libusb: debug [process_new_device] allocating new device in context 0x14d0079c0 for with session 0xc5292e5
[ 0.009020] [00108f9c] libusb: debug [process_new_device] found device with address 1 port = 2 parent = 0x0 at 0x12be041aa
[ 0.009067] [00108f9c] libusb: debug [darwin_get_cached_device] finding cached device for sessionID 0xc8ef02b
[... cut ...]
  File "/Users/ryohajika/Library/Python/2.7/lib/python/site-packages/usb/backend/libusb1.py", line 604, in _check
    raise USBError(_strerror(ret), ret, _libusb_errno[ret])
usb.core.USBError: [Errno None] Other error
>>> exit()
2021-02-16 21:56:27,169 DEBUG:usb.backend.libusb1:_LibUSB.close_device(<usb.backend.libusb1._DeviceHandle object at 0x103374128>)
[135.251666] [0010a0c4] libusb: debug [libusb_close]
Exception TypeError: "'NoneType' object is not callable" in <bound method _LibUSB.__del__ of <usb.backend.libusb1._LibUSB object at 0x1034caeb8>> ignored

```

</details>

jonasmalacofilho, 2021-02-23:
> Also tried to use pdb but was not fully sure how to use it...

That's ok.  The error is probably [coming from the kernel](https://github.com/libusb/libusb/blob/bda2344f5e7d460decd3968492faa7a5fb2797cd/libusb/os/darwin_usb.c#L149-L155).

> I'm just curious if it is a problem that can be solved by python side

I can't think of anything else other than checking if the device already is the desired configuration and, if that's the case, skipping `set_configuration` call altogether.

---

Closing this as there isn't anything PyUSB can do here.

closed 2021-02-23 by jonasmalacofilho

ryohajika, 2021-02-24:
Okay I'll look around to see if something can be fixed in v-usb or the adafruit library impl. Thank you so much for you all who get involved to this thing! :pray: