pyusb/pyusb #357: Resource busy during set_configuration or write

cassc, 2021-02-10:
With the following script, I got `Resource busy` error.

```python
import usb.core
import usb.util

print('r311 usb testing start')
dev = usb.core.find(idVendor=0x2109, idProduct=0x7638)
pkt_pwd_check = b'\xef\x01\xff\xff\xff\xff\x01\x00\x07\x13\x00\x00\x00\x00\x00\x1b'

dev.set_configuration()
cfg = dev.get_active_configuration()
print(f'config: {cfg}')

# get an endpoint instance
cfg = dev.get_active_configuration()
intf = cfg[(0,0)]

ep = usb.util.find_descriptor(
    intf,
    # match the first OUT endpoint
    custom_match = \
    lambda e: \
        usb.util.endpoint_direction(e.bEndpointAddress) == \
        usb.util.ENDPOINT_OUT)

assert ep is not None

# write the data
ep.write(pkt_pwd_check)
```

```
LIBUSB_DEBUG=4 PYUSB_DEBUG=debug python r311-reader-usb.py
r311 usb testing start
2021-02-10 11:55:13,196 DEBUG:usb.backend.libusb1:_LibUSB.__init__(<CDLL 'libusb-1.0.so.0', handle 557ad14f2e10 at 0x7f139d8e36d0>)
[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.000013] [00003411] libusb: debug [libusb_init] created default context
[ 0.000028] [00003411] libusb: debug [libusb_init] libusb v1.0.24.11584
[ 0.000036] [00003411] libusb: debug [get_kernel_version] reported kernel version is 5.4.91-1-lts
[ 0.000059] [00003411] libusb: debug [op_init] found usbfs at /dev/bus/usb
[ 0.000064] [00003411] libusb: debug [op_init] max iso packet length is (likely) 98304 bytes
[ 0.000080] [00003411] libusb: debug [op_init] sysfs is available
[ 0.000643] [00003413] libusb: debug [linux_udev_event_thread_main] udev event thread entering
[ 0.004368] [00003411] libusb: debug [linux_get_device_address] getting address for device: usb1 detached: 0
[ 0.004382] [00003411] libusb: debug [linux_get_device_address] scan usb1
[ 0.004412] [00003411] libusb: debug [linux_get_device_address] bus=1 dev=1
[ 0.004417] [00003411] libusb: debug [linux_enumerate_device] busnum 1 devaddr 1 session_id 257
[ 0.004422] [00003411] libusb: debug [linux_enumerate_device] allocating new device for 1/1 (session 257)
[ 0.004529] [00003411] libusb: debug [linux_get_device_address] getting address for device: 1-1 detached: 0
[ 0.004536] [00003411] libusb: debug [linux_get_device_address] scan 1-1
[ 0.004563] [00003411] libusb: debug [linux_get_device_address] bus=1 dev=16
[ 0.004568] [00003411] libusb: debug [linux_enumerate_device] busnum 1 devaddr 16 session_id 272
[ 0.004573] [00003411] libusb: debug [linux_enumerate_device] allocating new device for 1/16 (session 272)
[ 0.004601] [00003411] libusb: debug [linux_get_parent_info] dev 0x557ad1512460 (1-1) has parent 0x557ad1511cd0 (usb1) port 1
[ 0.004685] [00003411] libusb: debug [linux_get_device_address] getting address for device: 1-2 detached: 0
[ 0.004691] [00003411] libusb: debug [linux_get_device_address] scan 1-2
[ 0.004717] [00003411] libusb: debug [linux_get_device_address] bus=1 dev=2
[ 0.004722] [00003411] libusb: debug [linux_enumerate_device] busnum 1 devaddr 2 session_id 258
[ 0.004727] [00003411] libusb: debug [linux_enumerate_device] allocating new device for 1/2 (session 258)
[ 0.004754] [00003411] libusb: debug [linux_get_parent_info] dev 0x557ad15125d0 (1-2) has parent 0x557ad1511cd0 (usb1) port 2
[ 0.004842] [00003411] libusb: debug [linux_get_device_address] getting address for device: 1-2.3 detached: 0
[ 0.004848] [00003411] libusb: debug [linux_get_device_address] scan 1-2.3
[ 0.004876] [00003411] libusb: debug [linux_get_device_address] bus=1 dev=4
[ 0.004881] [00003411] libusb: debug [linux_enumerate_device] busnum 1 devaddr 4 session_id 260
[ 0.004886] [00003411] libusb: debug [linux_enumerate_device] allocating new device for 1/4 (session 260)
[ 0.004913] [00003411] libusb: debug [linux_get_parent_info] dev 0x557ad14f7610 (1-2.3) has parent 0x557ad15125d0 (1-2) port 3
[ 0.004997] [00003411] libusb: debug [linux_get_device_address] getting address for device: 1-2.4 detached: 0
[ 0.005003] [00003411] libusb: debug [linux_get_device_address] scan 1-2.4
[... 71 lines cut ...]
    raise USBError(_strerror(ret), ret, _libusb_errno[ret])
usb.core.USBError: [Errno 16] Resource busy
2021-02-10 11:55:13,211 DEBUG:usb.backend.libusb1:_LibUSB.close_device(<usb.backend.libusb1._DeviceHandle object at 0x7f139da4e5e0>)
[ 0.014534] [00003411] libusb: debug [libusb_close]  
[ 0.014545] [00003411] libusb: debug [usbi_remove_event_source] remove fd 7
2021-02-10 11:55:13,211 DEBUG:usb.backend.libusb1:_LibUSB._finalize_object()
[ 0.014676] [00003411] libusb: debug [libusb_exit]  
[ 0.014682] [00003411] libusb: debug [libusb_exit] destroying default context
[ 0.014688] [00003411] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 0.014693] [00003411] libusb: debug [handle_events] event sources modified, reallocating event data
[ 0.014701] [00003411] libusb: debug [usbi_wait_for_events] poll() 2 fds with timeout in 0ms
[ 0.014708] [00003411] libusb: debug [usbi_wait_for_events] poll() returned 0
[ 0.014714] [00003411] libusb: debug [libusb_unref_device] destroy device 2.2
[ 0.014719] [00003411] libusb: debug [libusb_unref_device] destroy device 2.1
[ 0.014724] [00003411] libusb: debug [libusb_unref_device] destroy device 1.5
[ 0.014729] [00003411] libusb: debug [libusb_unref_device] destroy device 1.3
[ 0.014734] [00003411] libusb: debug [libusb_unref_device] destroy device 1.9
[ 0.014738] [00003411] libusb: debug [libusb_unref_device] destroy device 1.4
[ 0.014743] [00003411] libusb: debug [libusb_unref_device] destroy device 1.2
[ 0.014748] [00003411] libusb: debug [libusb_unref_device] destroy device 1.16
[ 0.014753] [00003411] libusb: debug [libusb_unref_device] destroy device 1.1
[ 0.014758] [00003411] libusb: debug [usbi_remove_event_source] remove fd 6
[ 0.014768] [00003411] libusb: debug [usbi_remove_event_source] remove fd 5
[ 0.014790] [00003413] libusb: debug [linux_udev_event_thread_main] udev event thread exiting
```

jonasmalacofilho, 2021-02-10:
According to the libusb documentation for [`libusb_set_configuration`](http://libusb.sourceforge.net/api-1.0/group__libusb__dev.html#ga785ddea63a2b9bcb879a614ca4867bed):

> **Returns**
> - [...]
> - LIBUSB_ERROR_BUSY if interfaces are currently claimed 
> - [...]

See also:

 - [PyUSB FAQ: How (not) to call set_configuration() on a device already configured with the selected configuration?](https://github.com/pyusb/pyusb/blob/master/docs/faq.rst#how-not-to-call-set_configuration-on-a-device-already-configured-with-the-selected-configuration)
 - [libusb: Caveats: Configuration selection and handling](http://libusb.sourceforge.net/api-1.0/libusb_caveats.html#configsel)

closed 2021-02-10 by jonasmalacofilho

cassc, 2021-02-10:
If I just skip `set_configuration`, 

```python
import usb.core
import usb.util

print('r311 usb testing start')
dev = usb.core.find(idVendor=0x2109, idProduct=0x7638)
pkt_pwd_check = b'\xef\x01\xff\xff\xff\xff\x01\x00\x07\x13\x00\x00\x00\x00\x00\x1b'

cfg = dev.get_active_configuration()

# get an endpoint instance
cfg = dev.get_active_configuration()

print(cfg)

intf = cfg[(0,0)]

ep = usb.util.find_descriptor(
    intf,
    # match the first OUT endpoint
    custom_match = \
    lambda e: \
        usb.util.endpoint_direction(e.bEndpointAddress) == \
        usb.util.ENDPOINT_OUT)

assert ep is not None

# write the data
ep.write(pkt_pwd_check)
```

but I still get `Resource busy` error when writing data:

```
r311 usb testing start
2021-02-10 17:05:49,439 DEBUG:usb.backend.libusb1:_LibUSB.__init__(<CDLL 'libusb-1.0.so.0', handle 55cb48d47760 at 0x7efcc03e0670>)
[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.000014] [00000f4c] libusb: debug [libusb_init] created default context
[ 0.000027] [00000f4c] libusb: debug [libusb_init] libusb v1.0.24.11584
[ 0.000036] [00000f4c] libusb: debug [get_kernel_version] reported kernel version is 5.4.91-1-lts
[ 0.000063] [00000f4c] libusb: debug [op_init] found usbfs at /dev/bus/usb
[ 0.000068] [00000f4c] libusb: debug [op_init] max iso packet length is (likely) 98304 bytes
[ 0.000077] [00000f4c] libusb: debug [op_init] sysfs is available
[ 0.000309] [00000f4e] libusb: debug [linux_udev_event_thread_main] udev event thread entering
[ 0.003763] [00000f4c] libusb: debug [linux_get_device_address] getting address for device: usb1 detached: 0
[ 0.003787] [00000f4c] libusb: debug [linux_get_device_address] scan usb1
[ 0.003836] [00000f4c] libusb: debug [linux_get_device_address] bus=1 dev=1
[ 0.003846] [00000f4c] libusb: debug [linux_enumerate_device] busnum 1 devaddr 1 session_id 257
[ 0.003855] [00000f4c] libusb: debug [linux_enumerate_device] allocating new device for 1/1 (session 257)
[ 0.003990] [00000f4c] libusb: debug [linux_get_device_address] getting address for device: 1-1 detached: 0
[ 0.003996] [00000f4c] libusb: debug [linux_get_device_address] scan 1-1
[ 0.004025] [00000f4c] libusb: debug [linux_get_device_address] bus=1 dev=7
[ 0.004030] [00000f4c] libusb: debug [linux_enumerate_device] busnum 1 devaddr 7 session_id 263
[ 0.004037] [00000f4c] libusb: debug [linux_enumerate_device] allocating new device for 1/7 (session 263)
[ 0.004067] [00000f4c] libusb: debug [linux_get_parent_info] dev 0x55cb48d66c60 (1-1) has parent 0x55cb48d664d0 (usb1) port 1
[ 0.004140] [00000f4c] libusb: debug [linux_get_device_address] getting address for device: 1-2 detached: 0
[ 0.004146] [00000f4c] libusb: debug [linux_get_device_address] scan 1-2
[ 0.004170] [00000f4c] libusb: debug [linux_get_device_address] bus=1 dev=2
[ 0.004174] [00000f4c] libusb: debug [linux_enumerate_device] busnum 1 devaddr 2 session_id 258
[ 0.004178] [00000f4c] libusb: debug [linux_enumerate_device] allocating new device for 1/2 (session 258)
[ 0.004201] [00000f4c] libusb: debug [linux_get_parent_info] dev 0x55cb48d66dd0 (1-2) has parent 0x55cb48d664d0 (usb1) port 2
[ 0.004271] [00000f4c] libusb: debug [linux_get_device_address] getting address for device: 1-2.3 detached: 0
[ 0.004276] [00000f4c] libusb: debug [linux_get_device_address] scan 1-2.3
[ 0.004299] [00000f4c] libusb: debug [linux_get_device_address] bus=1 dev=4
[ 0.004302] [00000f4c] libusb: debug [linux_enumerate_device] busnum 1 devaddr 4 session_id 260
[ 0.004305] [00000f4c] libusb: debug [linux_enumerate_device] allocating new device for 1/4 (session 260)
[ 0.004327] [00000f4c] libusb: debug [linux_get_parent_info] dev 0x55cb48d4bf60 (1-2.3) has parent 0x55cb48d66dd0 (1-2) port 3
[... 155 lines cut ...]
    raise USBError(_strerror(ret), ret, _libusb_errno[ret])
usb.core.USBError: [Errno 16] Resource busy
2021-02-10 17:05:49,450 DEBUG:usb.backend.libusb1:_LibUSB.close_device(<usb.backend.libusb1._DeviceHandle object at 0x7efcc0543460>)
[ 0.011360] [00000f4c] libusb: debug [libusb_close]  
[ 0.011366] [00000f4c] libusb: debug [usbi_remove_event_source] remove fd 7
2021-02-10 17:05:49,450 DEBUG:usb.backend.libusb1:_LibUSB._finalize_object()
[ 0.011452] [00000f4c] libusb: debug [libusb_exit]  
[ 0.011458] [00000f4c] libusb: debug [libusb_exit] destroying default context
[ 0.011469] [00000f4c] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 0.011472] [00000f4c] libusb: debug [handle_events] event sources modified, reallocating event data
[ 0.011479] [00000f4c] libusb: debug [usbi_wait_for_events] poll() 2 fds with timeout in 0ms
[ 0.011484] [00000f4c] libusb: debug [usbi_wait_for_events] poll() returned 0
[ 0.011487] [00000f4c] libusb: debug [libusb_unref_device] destroy device 2.2
[ 0.011492] [00000f4c] libusb: debug [libusb_unref_device] destroy device 2.1
[ 0.011496] [00000f4c] libusb: debug [libusb_unref_device] destroy device 1.5
[ 0.011499] [00000f4c] libusb: debug [libusb_unref_device] destroy device 1.3
[ 0.011503] [00000f4c] libusb: debug [libusb_unref_device] destroy device 1.6
[ 0.011507] [00000f4c] libusb: debug [libusb_unref_device] destroy device 1.4
[ 0.011510] [00000f4c] libusb: debug [libusb_unref_device] destroy device 1.2
[ 0.011515] [00000f4c] libusb: debug [libusb_unref_device] destroy device 1.7
[ 0.011519] [00000f4c] libusb: debug [libusb_unref_device] destroy device 1.1
[ 0.011522] [00000f4c] libusb: debug [usbi_remove_event_source] remove fd 6
[ 0.011530] [00000f4c] libusb: debug [usbi_remove_event_source] remove fd 5
[ 0.011548] [00000f4e] libusb: debug [linux_udev_event_thread_main] udev event thread exiting
```

jonasmalacofilho, 2021-02-10:
Again, the interface is claimed by some other program or the kernel itself.

In the latter case, you may need to:

- detect that condition with `<device>.is_kernel_driver_active(interface)`;
- detach the kernel driver with `<device>.detach_kernel_driver(interface)`;
- and, once you're done, reattach it with `<device>.attach_kernel_driver(interface)`.

Unfortunately it seems that the need for detaching the kernel driver isn't very obviously documented neither here nor in libusb.
If you can, please contribute these improvements.

_(The only mention I could find right now is in the documentation for [`libusb_kernel_driver_active`](http://libusb.sourceforge.net/api-1.0/group__libusb__dev.html#ga1cabd4660a274f715eeb82de112e0779))._

mcuee, 2021-02-11:
libusb github wiki FAQ page updated. Hopefully this helps.
https://github.com/libusb/libusb/wiki/FAQ#Can_I_run_libusb_applications_on_Linux_when_there_is_already_a_kernel_driver_attached_to_it

jonasmalacofilho, 2021-02-11:
Thanks, @mcuee!