pyusb/pyusb #359: Pyusb with new version of libusb 1.0 on HID mangling one byte of data on Windows 10

EnochMakenp, 2021-03-30:
Using the latest PyUSB, in conjunction with the latest libusb1.0 on Windows 10 with an access card reader that behave as a HID device is missing the last byte of data in the ctrl_transfer communication. I'm asking for 8 bytes of data and only 7 are read, which is annoying as I want to read the card UID, coded on 8 bytes...

The code I use in PyUSB is:
    device = usb.core.find(idVendor=vendorID, idProduct=productID)
    ret = device.ctrl_transfer(0x21, 0x09, 0x0300, 0, (0x8f,0x00,0x00,0x00,0x00,0x00,0x00,0x00),8) #makes the request
    response = device.ctrl_transfer(0xa1, 1, 0x0300, 0, 8) # read the response, ask for 8 bytes
and after this the content of response is :
array('B', [186, 137, 24, 128, 0, 0, 229])
Note that I have only 7 bytes instead of the 8 requested 
(if I ask for 7 bytes, the command fail, but if I request 9 or more bytes , I still get the 7 bytes...)

I used Wireshark (with USBPcap) to spy on the USB transaction and the 8 bytes of data are available on the USB packet, as can be see in the following transcript:

Frame 8: 36 bytes on wire (288 bits), 36 bytes captured (288 bits) on interface wireshark_extcap2040, id 0
USB URB
    [Source: 5.11.0]
    [Destination: host]
    USBPcap pseudoheader length: 28
    IRP ID: 0xffff960ef084e060
    IRP USBD_STATUS: USBD_STATUS_SUCCESS (0x00000000)
    URB Function: URB_FUNCTION_CONTROL_TRANSFER (0x0008)
    IRP information: 0x01, Direction: PDO -> FDO
    URB bus id: 5
    Device address: 11
    Endpoint: 0x80, Direction: IN
    URB transfer type: URB_CONTROL (0x02)
    Packet Data Length: 8
    [Request in: 7]
    [Time from request: 0.000164000 seconds]
    Control transfer stage: Complete (3)
    [bInterfaceClass: Unknown (0xffff)]
    CONTROL response data: ba8918800000e5c9

The control data (ba8918800000e5c9) has 8 bytes and in the response above I'm missing the last byte "c9", and the collected bytes in python stop at "e5" (=229).

This Wireshark test is the reason why I suspect PyUSB or libusb1.0 messing around - with a preference for the second one, as there is a new version that is only 3 months old.

Note that I can not revert to the previous version of libusb1.0 (previous to 2020-12-11) as it does not work at all anymore ending systematically with "[Errno 19] No such device (it may have been disconnected). " when I try to use ctrl_transfer.

jonasmalacofilho, 2021-03-30:
I couldn't find an obvious source for this issue.

Can you show me the output from running that code with `PYUSB_DEBUG=debug` and `LIBUSB_DEBUG=4`?

jonasmalacofilho, 2021-03-30:
I just tested it and the following should work on a standard Windows Command Prompt shell:

```
set PYUSB_DEBUG=debug
set LIBUSB_DEBUG=4
<your program>
```

I'm not sure what's going on on your system.

EnochMakenp, 2021-03-30:
Apparently it does not work from Powershell - but it works from the standard command interface... go figure... anyway, here it is:
```
[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.002219] [000029b4] libusb: debug [libusb_init] created default context
[ 0.012362] [000029b4] libusb: debug [libusb_init] libusb v1.0.24.11584
[ 0.013836] [000029b4] libusb: debug [get_windows_version] Windows 10 64-bit
[ 0.014984] [000029b4] libusb: debug [htab_create] using 1021 entries hash table
[ 0.020298] [000029b4] libusb: info [winusbx_init] WinUSB DLL available (with isoch support)
[ 0.020695] [000029b4] libusb: info [winusbx_init] libusbK DLL is not available
[ 0.023508] [000029b4] libusb: info [windows_init] UsbDk backend is not available
[ 0.023680] [000029b4] libusb: debug [usbi_add_event_source] add HANDLE 0000000000000280 events 0
[ 0.028529] [000029b4] libusb: debug [usbi_io_init] using timer for timeouts
[ 0.036121] [000029b4] libusb: debug [usbi_add_event_source] add HANDLE 00000000000002E8 events 0
[ 0.024898] [00001118] libusb: debug [windows_iocp_thread] I/O completion thread started
[ 0.038436] [000029b4] libusb: debug [libusb_get_device_list]
[ 0.091154] [000029b4] libusb: debug [winusb_get_device_list] allocating new device for session [6A]
[ 0.091374] [000029b4] libusb: debug [winusb_get_device_list] allocating new device for session [6B]
[ 0.105151] [000029b4] libusb: debug [winusb_get_device_list] allocating new device for session [6C]
[ 0.105691] [000029b4] libusb: debug [winusb_get_device_list] allocating new device for session [6D]
[ 0.109410] [000029b4] libusb: debug [get_api_type] driver(s): BTHUSB
[ 0.111940] [000029b4] libusb: debug [get_api_type] upper filter driver(s): BtFilter
[ 0.120522] [000029b4] libusb: debug [get_api_type] lower filter driver(s): BtFilter
[ 0.121169] [000029b4] libusb: debug [winusb_get_device_list] allocating new device for session [47]
[ 0.123513] [000029b4] libusb: debug [get_api_type] driver(s): HidUsb
[ 0.123672] [000029b4] libusb: debug [get_api_type] matched driver name against HID API
[ 0.124210] [000029b4] libusb: debug [winusb_get_device_list] allocating new device for session [51]
[ 0.128655] [000029b4] libusb: debug [get_api_type] driver(s): usbccgp
[ 0.140143] [000029b4] libusb: debug [get_api_type] matched driver name against Composite API
[ 0.142465] [000029b4] libusb: debug [winusb_get_device_list] allocating new device for session [52]
[ 0.155678] [000029b4] libusb: debug [get_api_type] driver(s): usbccgp
[ 0.157856] [000029b4] libusb: debug [get_api_type] matched driver name against Composite API
[ 0.158273] [000029b4] libusb: debug [winusb_get_device_list] allocating new device for session [53]
[ 0.174705] [000029b4] libusb: debug [get_api_type] driver(s): usbccgp
[ 0.176768] [000029b4] libusb: debug [get_api_type] matched driver name against Composite API
[ 0.188142] [000029b4] libusb: debug [winusb_get_device_list] allocating new device for session [54]
[ 0.190839] [000029b4] libusb: debug [get_api_type] driver(s): rtux64w10
[ 0.191125] [000029b4] libusb: debug [winusb_get_device_list] allocating new device for session [55]
[ 0.206284] [000029b4] libusb: debug [enumerate_hcd_root_hub] assigning HCD 'PCI\VEN_8086&DEV_9D2F&SUBSYS_07E61028&REV_21\3&11583659&0&A0' bus number 1
[ 0.206431] [000029b4] libusb: debug [enumerate_hcd_root_hub] assigning HCD 'PCI\VEN_8086&DEV_15D4&SUBSYS_11112222&REV_02\6&25BD8816&0&001000E4' bus number 2
[ 0.208369] [000029b4] libusb: debug [winusb_get_device_list] found existing device for session [54]
[ 0.208611] [000029b4] libusb: debug [init_device] found 1 configurations (current config: 1)
[ 0.209174] [000029b4] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 59 bytes)
[ 0.216204] [000029b4] libusb: debug [init_device] (bus: 2, addr: 5, depth: 2, port: 6): 'USB\VID_1038&PID_1729\9&88AD693&0&6'
[ 0.216888] [000029b4] libusb: debug [winusb_get_device_list] found existing device for session [55]
[ 0.224425] [000029b4] libusb: debug [init_device] found 2 configurations (current config: 1)
[ 0.228799] [000029b4] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 57 bytes)
[ 0.238387] [000029b4] libusb: debug [cache_config_descriptors] cached config descriptor 1 (bConfigurationValue=2, 98 bytes)
[ 0.238673] [000029b4] libusb: debug [init_device] (bus: 2, addr: 7, depth: 2, port: 2): 'USB\VID_0BDA&PID_8153\000002000000'
[ 0.239505] [000029b4] libusb: debug [winusb_get_device_list] found existing device for session [53]
[ 0.239882] [000029b4] libusb: debug [init_device] found 1 configurations (current config: 1)
[ 0.240360] [000029b4] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 311 bytes)
[ 0.256138] [000029b4] libusb: debug [init_device] (bus: 2, addr: 4, depth: 2, port: 5): 'USB\VID_0BDA&PID_4014\200901010001'
[ 0.258939] [000029b4] libusb: debug [winusb_get_device_list] found existing device for session [47]
[ 0.268626] [000029b4] libusb: debug [init_device] found 1 configurations (current config: 1)
[ 0.269308] [000029b4] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 177 bytes)
[ 0.271932] [000029b4] libusb: debug [init_device] (bus: 1, addr: 2, depth: 1, port: 7): 'USB\VID_0489&PID_E0A2\5&364997A6&0&7'
[... cut ...]
[ 0.432586] [000029b4] libusb: debug [libusb_unref_device] destroy device 2.1
[ 0.433107] [000029b4] libusb: debug [libusb_unref_device] destroy device 2.0
[ 0.433658] [000029b4] libusb: debug [libusb_exit]
[ 0.434145] [000029b4] libusb: debug [libusb_exit] destroying default context
[ 0.434652] [000029b4] libusb: debug [usbi_remove_event_source] remove HANDLE 00000000000002E8
[ 0.435107] [000029b4] libusb: debug [usbi_remove_event_source] remove HANDLE 0000000000000280
[ 0.435678] [00001118] libusb: debug [windows_iocp_thread] I/O completion thread exiting
```

EnochMakenp, 2021-03-31:
The core problem is tackled in the related issue opened on libusb: libusb/libusb#902
The issue here will be closed.

closed 2021-03-31 by EnochMakenp