pyusb/pyusb #360: Mac: how to package libusb in a way that allows pyusb to find it?

morinted, 2021-04-03:
I am trying to package an `app` bundle in Mac with pyusb.

We already package our own python, taking advantage of macholib to include any necessary shared libraries.

I'm looking for a way to package `libusb-1.0.0.dynalib` in such a way that pyusb will be able to find it.

I noticed that [libusb1](https://pypi.org/project/libusb1/) will include its own package directory in its search and was wondering if something similar exists for pyusb, or if there's a better alternative.

jonasmalacofilho, 2021-04-03:
By default we use `ctypes.util.find_library` (see [`usb/libloader.py`](https://github.com/pyusb/pyusb/blob/v1.1.1/usb/libloader.py)).

But it's also possible to supply a custom `find_library` function to [`usb.backend.libusb1.get_backend()`](https://github.com/pyusb/pyusb/blob/a54ebc75a95fd9d6e1b1501f10d8fc0025ca8c0b/usb/backend/libusb1.py#L957), and use that backend in [`usb.core.find()`](https://github.com/pyusb/pyusb/blob/a54ebc75a95fd9d6e1b1501f10d8fc0025ca8c0b/usb/core.py#L1213).

morinted, 2021-04-03:
That's fantastic, thank you!

closed 2021-04-03 by morinted