pyusb/pyusb #362: FreeBSD: No USB backend found

cederom, 2021-05-11:
Hello world :-)

I am using https://github.com/pyocd/pyOCD that depends on `pyusb`. It stopped working with my probe and other folks on Linux also find this problem (reported here https://github.com/pyocd/pyOCD/issues/1144) :-(
```
(..)
  File "/home/x/usr/local/venv37zephyr/lib/python3.7/site-packages/pyocd/probe/pydapaccess/interface/__init__.py", line 62, in <module>
    raise DAPAccessIntf.DeviceError("No USB backend found")
pyocd.probe.pydapaccess.dap_access_api.DeviceError: No USB backend found
```

Reinstalling `pyusb` from latest head sources does not fix the problem. FreeBSD has its own LibUSB implementation. All other `libusb` applications work. Clearly `pyusb` has problem with backend detection :-)

```
(venv37zephyr) ls -al /usr/lib/libusb*
-r--r--r--  1 root  wheel  582956 28 paź  2020 /usr/lib/libusb_p.a
-r--r--r--  1 root  wheel  554926 28 paź  2020 /usr/lib/libusb.a
lrwxr-xr-x  1 root  wheel      11 19 lis  2019 /usr/lib/libusb.so -> libusb.so.3
-r--r--r--  1 root  wheel   90616 28 paź  2020 /usr/lib/libusb.so.3
-r--r--r--  1 root  wheel   72622 28 paź  2020 /usr/lib/libusbhid_p.a
-r--r--r--  1 root  wheel   70878 28 paź  2020 /usr/lib/libusbhid.a
lrwxr-xr-x  1 root  wheel      14 19 lis  2019 /usr/lib/libusbhid.so -> libusbhid.so.4
-r--r--r--  1 root  wheel   15912 28 paź  2020 /usr/lib/libusbhid.so.4

(venv37zephyr) uname -a
FreeBSD 0xCFMX4 12.2-RELEASE-p3 FreeBSD 12.2-RELEASE-p3 #0 r358209M: Sat Jan 30 15:39:23 CET 2021     root@0xCFMX4:/usr/obj/usr/src/amd64.amd64/sys/GENERIC  amd64
```

Any hints welcome :-)

cederom, 2021-05-12:
Thanks @jonasmalacofilho :-) I can confirm this is not the `pyusb` issue but `pyOCD` problem! It only selects `pyusb` backend for a selected list of platforms.

Last questions:
* Can we assume `pyusbd` a good "failsafe" backend that works on all platforms where `libusb` is present (except Windows where `hidapiusb` or `pywinusb` is used and Darwin where `hidapiusb` is used)?
* Is `pyusb` confirmed to work on all BSD operating systems? I would like to use `"BSD" in platform.system()` rather than `platform.system()=="FreeBSD"` kind of check to match all BSD platforms?

jonasmalacofilho, 2021-05-12:
Thanks!

> Is pyusb confirmed to work on all BSD operating systems? I would like to use "BSD" in platform.system() rather than platform.system()=="FreeBSD" kind of check to match all BSD platforms?

As PyUSB itself is a pure-Python library, it _should_ work on any platform with Python (2.7 or 3.5+), a working [`ctypes.util.find_library()`](https://docs.python.org/3/library/ctypes.html#finding-shared-libraries) and sensible naming of libraries like libusb.

But I can't confirm that this is indeed the case on all BSDs, or even all Python distributions.  For example, `find_library()` is currently  broken for M1 Macs on some Python distributions.

> Can we assume pyusbd a good "failsafe" backend that works on all platforms where libusb is present (except Windows where hidapiusb or pywinusb is used and Darwin where hidapiusb is used)?

It depends on the goals/requirements of the project and on the device type.  The latter is probably why PyOCD doesn't use PyUSB on Mac or Windows.

Anyway, at a glance PyUSB seems like a reasonable fallback to me, but it's possible that the PyOCD developers really prefer the conservative approach of just errorring out early instead of using a potentially untested (on that platform) backend.

mcuee, 2021-05-21:
libusb under FreeBSD is not under libusb project but under FreeBSD USB developers so it is under good care. 

BUT, libusb itself is considered as EXPERIMENTAL for OpenBSD and NetBSD, due to lack of participation from the OpenBSD/NetBSD community.
https://github.com/libusb/libusb/wiki