pyusb/pyusb #363: Incompatible version is released to Python 2

pazeshun, 2021-05-12:
I am not sure when pyusb dropped Python 2 support (https://github.com/pyusb/pyusb/commit/eaf690840b3b041646fc862a0f77e6b0bed24dc1?), but pyusb v1.1.1 and v1.1.0 are currently released to Python 2 and cause setuptools failure in installtion on Ubuntu 18.04 with `python-pip` and `python-setuptools` apt packages:
```
$ python --version
Python 2.7.17
$ pip --version
pip 9.0.1 from /usr/lib/python2.7/dist-packages (python 2.7)
$ pip show setuptools
Name: setuptools
Version: 39.0.1
Summary: Easily download, build, install, upgrade, and uninstall Python packages
Home-page: https://github.com/pypa/setuptools
Author: Python Packaging Authority
Author-email: distutils-sig@python.org
License: UNKNOWN
Location: /usr/lib/python2.7/dist-packages
Requires: 
$ pip install pyusb --user --no-cache-dir
Collecting pyusb
  Downloading https://files.pythonhosted.org/packages/b9/8d/25c4e446a07e918eb39b5af25c4a83a89db95ae44e4ed5a46c3c53b0a4d6/pyusb-1.1.1.tar.gz (74kB)
    100% |████████████████████████████████| 81kB 16.6MB/s 
    Complete output from command python setup.py egg_info:
    /tmp/easy_install-xlwoLp/setuptools_scm-6.0.1/src
    <pkg_resources.WorkingSet object at 0x7f78235f1250>
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-build-cg7SkJ/pyusb/setup.py", line 117, in <module>
        'Topic :: System :: Hardware :: Hardware Drivers'
      File "/usr/lib/python2.7/dist-packages/setuptools/__init__.py", line 128, in setup
        _install_setup_requires(attrs)
      File "/usr/lib/python2.7/dist-packages/setuptools/__init__.py", line 123, in _install_setup_requires
        dist.fetch_build_eggs(dist.setup_requires)
      File "/usr/lib/python2.7/dist-packages/setuptools/dist.py", line 513, in fetch_build_eggs
        replace_conflicting=True,
      File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 774, in resolve
        replace_conflicting=replace_conflicting
      File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 1057, in best_match
        return self.obtain(req, installer)
      File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 1069, in obtain
        return installer(requirement)
      File "/usr/lib/python2.7/dist-packages/setuptools/dist.py", line 580, in fetch_build_egg
        return cmd.easy_install(req)
      File "/usr/lib/python2.7/dist-packages/setuptools/command/easy_install.py", line 698, in easy_install
        return self.install_item(spec, dist.location, tmpdir, deps)
      File "/usr/lib/python2.7/dist-packages/setuptools/command/easy_install.py", line 724, in install_item
        dists = self.install_eggs(spec, download, tmpdir)
      File "/usr/lib/python2.7/dist-packages/setuptools/command/easy_install.py", line 909, in install_eggs
        return self.build_and_install(setup_script, setup_base)
      File "/usr/lib/python2.7/dist-packages/setuptools/command/easy_install.py", line 1177, in build_and_install
        self.run_setup(setup_script, setup_base, args)
      File "/usr/lib/python2.7/dist-packages/setuptools/command/easy_install.py", line 1163, in run_setup
        run_setup(setup_script, args)
      File "/usr/lib/python2.7/dist-packages/setuptools/sandbox.py", line 253, in run_setup
        raise
      File "/usr/lib/python2.7/contextlib.py", line 35, in __exit__
        self.gen.throw(type, value, traceback)
      File "/usr/lib/python2.7/dist-packages/setuptools/sandbox.py", line 195, in setup_context
        yield
      File "/usr/lib/python2.7/contextlib.py", line 35, in __exit__
        self.gen.throw(type, value, traceback)
      File "/usr/lib/python2.7/dist-packages/setuptools/sandbox.py", line 166, in save_modules
        saved_exc.resume()
      File "/usr/lib/python2.7/dist-packages/setuptools/sandbox.py", line 141, in resume
        six.reraise(type, exc, self._tb)
      File "/usr/lib/python2.7/dist-packages/setuptools/sandbox.py", line 154, in save_modules
        yield saved
      File "/usr/lib/python2.7/dist-packages/setuptools/sandbox.py", line 195, in setup_context
        yield
      File "/usr/lib/python2.7/dist-packages/setuptools/sandbox.py", line 250, in run_setup
        _execfile(setup_script, ns)
      File "/usr/lib/python2.7/dist-packages/setuptools/sandbox.py", line 45, in _execfile
[... 88 lines cut ...]
Installing collected packages: pyusb
  Running setup.py install for pyusb ... done
Successfully installed pyusb-1.0.2
```

After investigating, I found that this problem can be solved by a newer setuptools:
```
$ pip install setuptools --user --no-cache-dir
Collecting setuptools
  Downloading https://files.pythonhosted.org/packages/e1/b7/182161210a13158cd3ccc41ee19aadef54496b74f2817cc147006ec932b4/setuptools-44.1.1-py2.py3-none-any.whl (583kB)
    100% |████████████████████████████████| 583kB 35.9MB/s 
Installing collected packages: setuptools
Successfully installed setuptools-44.1.1
$ pip install pyusb --user --no-cache-dir
Collecting pyusb
  Downloading https://files.pythonhosted.org/packages/b9/8d/25c4e446a07e918eb39b5af25c4a83a89db95ae44e4ed5a46c3c53b0a4d6/pyusb-1.1.1.tar.gz (74kB)
    100% |████████████████████████████████| 81kB 16.1MB/s 
Installing collected packages: pyusb
  Running setup.py install for pyusb ... done
Successfully installed pyusb-1.1.1
```

Is this OK for pyusb project?
I am not familiar with Python, but I think this dependency should be automatically resolved when `pip install pyusb` is run.
On the other hand, if pyusb has already dropped Python 2 support, a newer release to Python 2 seems strange, so I am not sure if a bug fix release to Python 2 is correct.

jonasmalacofilho, 2021-05-12:
PyUSB 1.1.0 and 1.1.1 are compatible with Python 2.7, but do indeed introduce a dependency to `setuptools_scm`, which could _potentially_ be problematic on distros without very outdated `setuptools`.

However, we [explicitly tested Ubuntu 18.04](https://github.com/pyusb/pyusb/pull/307#issuecomment-645667247) at the time of the change, and found no issues with its `setuptools==39.0.1`.

From your logs, it seems that you're using a much newer version of `setuptools_scm` that what was in the Ubuntu repository at the time of our testing, and _it_ appears to be incompatible with the old `setuptools`.  Where does it (`setuptools_scm==6.0.1`) comes from?

pazeshun, 2021-05-13:
@jonasmalacofilho Thank you for your reply.

On my environment, `setuptools_scm` is not installed even after I updated `setuptools` and installed the latest pyusb:
```
$ pip install setuptools --user --no-cache-dir
Collecting setuptools
  Downloading https://files.pythonhosted.org/packages/e1/b7/182161210a13158cd3ccc41ee19aadef54496b74f2817cc147006ec932b4/setuptools-44.1.1-py2.py3-none-any.whl (583kB)
    100% |████████████████████████████████| 583kB 41.7MB/s 
Installing collected packages: setuptools
Successfully installed setuptools-44.1.1
$ pip install pyusb --user --no-cache-dir
Collecting pyusb
  Downloading https://files.pythonhosted.org/packages/b9/8d/25c4e446a07e918eb39b5af25c4a83a89db95ae44e4ed5a46c3c53b0a4d6/pyusb-1.1.1.tar.gz (74kB)
    100% |████████████████████████████████| 81kB 11.4MB/s 
Installing collected packages: pyusb
  Running setup.py install for pyusb ... done
Successfully installed pyusb-1.1.1
$ pip show setuptools_scm  # No output
$ pip show setuptools-scm  # No output
```

I found `setuptools_scm` can be explicitly installed by `sudo apt install python-setuptools-scm`.
After that, the latest pyusb can be installed with no problem:
```
$ sudo apt install python-setuptools-scm
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following packages were automatically installed and are no longer required:
  linux-hwe-5.4-headers-5.4.0-42 python3-click python3-colorama
Use 'sudo apt autoremove' to remove them.
The following NEW packages will be installed:
  python-setuptools-scm
0 upgraded, 1 newly installed, 0 to remove and 15 not upgraded.
Need to get 13.7 kB of archives.
After this operation, 62.5 kB of additional disk space will be used.
Get:1 http://jp.archive.ubuntu.com/ubuntu bionic/universe amd64 python-setuptools-scm all 1.15.6-1 [13.7 kB]
Fetched 13.7 kB in 0s (586 kB/s)                 
Selecting previously unselected package python-setuptools-scm.
(Reading database ... 322882 files and directories currently installed.)
Preparing to unpack .../python-setuptools-scm_1.15.6-1_all.deb ...
Unpacking python-setuptools-scm (1.15.6-1) ...
Setting up python-setuptools-scm (1.15.6-1) ...
$ pip show setuptools
Name: setuptools
Version: 39.0.1
Summary: Easily download, build, install, upgrade, and uninstall Python packages
Home-page: https://github.com/pypa/setuptools
Author: Python Packaging Authority
Author-email: distutils-sig@python.org
License: UNKNOWN
Location: /usr/lib/python2.7/dist-packages
Requires: 
$ pip show setuptools-scm
Name: setuptools-scm
Version: 1.15.6
Summary: the blessed package to manage pytyour versions by scm tags
Home-page: https://github.com/pypa/setuptools_scm/
Author: Ronny Pfannschmidt
Author-email: opensource@ronnypfannschmidt.de
License: MIT
Location: /usr/lib/python2.7/dist-packages
Requires: 
$ pip install pyusb --user --no-cache-dir
Collecting pyusb
  Downloading https://files.pythonhosted.org/packages/b9/8d/25c4e446a07e918eb39b5af25c4a83a89db95ae44e4ed5a46c3c53b0a4d6/pyusb-1.1.1.tar.gz (74kB)
    100% |################################| 81kB 21.3MB/s 
Installing collected packages: pyusb
  Running setup.py install for pyusb ... done
Successfully installed pyusb-1.1.1
```

I think the problem is that `python-setuptools-scm` is not automatically installed in the minimum process to install pyusb (`sudo apt install python-pip`->`sudo pip install pyusb`) because `python-setuptools-scm` does not required by `python-pip`:
```
$ apt-cache rdepends python-setuptools-scm
python-setuptools-scm
Reverse Depends:
  sagemath
```

In this case, I think `setuptools_scm` should be automatically installed at `sudo pip install pyusb`.
On my environment, installing only `setuptools_scm` by `pip` solves the problem, so pyusb can have `pip` dependency on `setuptools_scm`:
```
$ pip install setuptools-scm --user --no-cache-dir --no-deps
Collecting setuptools-scm
  Downloading https://files.pythonhosted.org/packages/6a/18/23ad8654c5c8d91d1238b2d52882e50152473f2bd2db0da60215b51f401b/setuptools_scm-5.0.2-py2.py3-none-any.whl
Installing collected packages: setuptools-scm
Successfully installed setuptools-scm-5.0.2
$ pip show setuptools
Name: setuptools
Version: 39.0.1
Summary: Easily download, build, install, upgrade, and uninstall Python packages
Home-page: https://github.com/pypa/setuptools
Author: Python Packaging Authority
Author-email: distutils-sig@python.org
License: UNKNOWN
Location: /usr/lib/python2.7/dist-packages
Requires: 
$ pip show setuptools-scm
Name: setuptools-scm
Version: 5.0.2
Summary: the blessed package to manage your versions by scm tags
Home-page: https://github.com/pypa/setuptools_scm/
Author: Ronny Pfannschmidt
Author-email: opensource@ronnypfannschmidt.de
License: MIT
Location: /home/hiro/.local/lib/python2.7/site-packages
Requires: setuptools
$ pip install pyusb --
user --no-cache-dir
Collecting pyusb
  Downloading https://files.pythonhosted.org/packages/b9/8d/25c4e446a07e918eb39b5af25c4a83a89db95ae44e4ed5a46c3c53b0a4d6/pyusb-1.1.1.tar.gz (74kB)
    100% |################################| 81kB 15.0MB/s 
Installing collected packages: pyusb
  Running setup.py install for pyusb ... done
Successfully installed pyusb-1.1.1
```

However, I am not sure it is a common solution for all platforms listed in https://github.com/pyusb/pyusb/pull/307#issuecomment-645667247.

pazeshun, 2021-05-13:
I am also not sure how to release a new version to Python 2.
If the master branch before #364 is compatible with Python 2 (I don't think so because CI doesn't check Python 2, though), reverting #364 -> adding fixing commits -> releasing the last version to Python 2 -> re-applying #364 .

jonasmalacofilho, 2021-05-13:
I was going to say it must have been automatically and temporarily installed by pip, but the versions doesn't match:

Your bug was triggered with 6.0.1,

```
[...]
      File "/tmp/easy_install-xlwoLp/setuptools_scm-6.0.1/src/setuptools_scm/utils.py", line 41
        print(*k)
              ^
    SyntaxError: invalid syntax
```

but when you manually/explicitly installed it (with pip) you got version 5.0.2.

Still, your problem is that setuptools_scm > 5 no longer supports Python 2.7, and yet it somehow got used before.

I noticed your `pip` version is also fairly old: it's possible it may not yet fully/always honor the `python_requires>3.6` directive for setuptools_scm==6.0.1 (in [setup.cfg](https://github.com/pypa/setuptools_scm/blob/v6.0.1/setup.cfg#L31)).

---

Small correction to a previous comment I made: Python 2.7/3.5 support ended with PyUSB 1.1.0, per the release notes for PyUSB 1.1.1.

jonasmalacofilho, 2021-05-13:
> I am also not sure how to release a new version to Python 2.

It appears to be a problem _in pip._  And while maybe we could work around it in our setup.py (we do something for very old versions of setuptools), maintaining Python 2 support is a burden I can no longer carry.

Everyone kept it alive as long as possible.  But it's time to call it.

---

P.S. As a practical note, support for Python 2.7 was dropped on a patch release; this was chosen because a dependency's EOL is not (IMO) a breaking change.  But it also means that a Python 2 patch for PyUSB 1.1.x would first need to bring back Python 2.7 support, and only then fix it.  Which is totally opposite to finally letting Python 2 go...

pazeshun, 2021-05-13:
> I was going to say it must have been automatically and temporarily installed by pip, but the versions doesn't match

Oh, I didn't know about temporary installation. Now I understand my investigation was wrong. Sorry for your trouble...

> I noticed your `pip` version is also fairly old: it's possible it may not yet fully/always honor the `python_requires>3.6` directive for setuptools_scm==6.0.1 (in [setup.cfg](https://github.com/pypa/setuptools_scm/blob/v6.0.1/setup.cfg#L31)).

> It appears to be a problem _in pip._

Your assumption seems right.
When I updated `pip`, the error in installing pyusb disappeared:
```
$ pip install pip -U --user
Collecting pip
  Downloading https://files.pythonhosted.org/packages/27/79/8a850fe3496446ff0d584327ae44e7500daf6764ca1a382d2d02789accf7/pip-20.3.4-py2.py3-none-any.whl (1.5MB)
    100% |████████████████████████████████| 1.5MB 890kB/s 
Installing collected packages: pip
Successfully installed pip-20.3.4
$ pip show pip
WARNING: pip is being invoked by an old script wrapper. This will fail in a future version of pip.
Please see https://github.com/pypa/pip/issues/5599 for advice on fixing the underlying issue.
To avoid this problem you can invoke Python with '-m pip' instead of running pip directly.
DEPRECATION: Python 2.7 reached the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 is no longer maintained. pip 21.0 will drop support for Python 2.7 in January 2021. More details about Python 2 support in pip can be found at https://pip.pypa.io/en/latest/development/release-process/#python-2-support pip 21.0 will remove support for this functionality.
Name: pip
Version: 20.3.4
Summary: The PyPA recommended tool for installing Python packages.
Home-page: https://pip.pypa.io/
Author: The pip developers
Author-email: distutils-sig@python.org
License: MIT
Location: /home/hiro/.local/lib/python2.7/site-packages
Requires: 
Required-by: 
$ pip install pyusb -U --user --no-cache-dir
WARNING: pip is being invoked by an old script wrapper. This will fail in a future version of pip.
Please see https://github.com/pypa/pip/issues/5599 for advice on fixing the underlying issue.
To avoid this problem you can invoke Python with '-m pip' instead of running pip directly.
DEPRECATION: Python 2.7 reached the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 is no longer maintained. pip 21.0 will drop support for Python 2.7 in January 2021. More details about Python 2 support in pip can be found at https://pip.pypa.io/en/latest/development/release-process/#python-2-support pip 21.0 will remove support for this functionality.
Collecting pyusb
  Downloading pyusb-1.1.1.tar.gz (74 kB)
     |████████████████████████████████| 74 kB 13.2 MB/s 
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
    Preparing wheel metadata ... done
Building wheels for collected packages: pyusb
  Building wheel for pyusb (PEP 517) ... done
  Created wheel for pyusb: filename=pyusb-1.1.1-py2-none-any.whl size=58052 sha256=29cc581eda3d25c60c0e6f184d3fa5320c89637a70d105ea7dbdfeb6b9fb9e98
  Stored in directory: /tmp/pip-ephem-wheel-cache-an3DtY/wheels/3a/8f/92/fbf127fcc393d9ae11cdc66adbef53b5b525b4e553a244257c
Successfully built pyusb
Installing collected packages: pyusb
Successfully installed pyusb-1.1.1
```
When I checked the verbose log of the above pyusb installation, I confirmed that setuptools_scm 5.0.2 was temporarily installed.

Now my understanding is that...
- PyUSB 1.1.0 is compatible with Python 2, but in order to install it, `pip` should be upgraded from the version of `python-pip` on Ubuntu 18.04
- PyUSB 1.1.1 is not compatible with Python 2, but is released to Python 2

> And while maybe we could work around it in our setup.py (we do something for very old versions of setuptools), maintaining Python 2 support is a burden I can no longer carry.

> But it also means that a Python 2 patch for PyUSB 1.1.x would first need to bring back Python 2.7 support, and only then fix it. Which is totally opposite to finally letting Python 2 go...

OK.
If you have time, please consider "yank"ing current PyUSB 1.1.1 and creating a post-release including #364 .
https://pypi.org/help/#yanked
https://www.python.org/dev/peps/pep-0440/#post-releases

jonasmalacofilho, 2021-05-14:
> ("Yank" can be recognized pip 20.3+.)

Ouch, that means it probably won't be recognized by most users anyway.  Given that, I don't think a special 1.1.2 release is worth the effort.

But I'll consider yanking 1.1.1 after the next normal release (1.2.0).

jonasmalacofilho, 2021-07-05:
PyUSB 1.2.0 has been released, and PyUSB 1.1.1 has been yanked.

closed 2021-07-05 by jonasmalacofilho

pazeshun, 2021-07-05:
I confirmed `pip` 20.3.4 on Python 2.7.17 installs PyUSB 1.1.0.
@jonasmalacofilho Thank you very much for your support!