pyusb/pyusb #365: FreeBSD: ImportError: cannot import name 'ENODATA' from 'errno' (unknown location)

cederom, 2021-05-13:
Hello world :-)

I have encountered this unhandled exception on FreeBSD using `pyocd`. This time it looks like `pyusb` issue. Exists in master:
```
(venv37zephyr) pyocd flash build/zephyr/zephyr.hex
0001742:CRITICAL:__main__:uncaught exception: cannot import name 'ENODATA' from 'errno' (unknown location)
Traceback (most recent call last):
  File "/home/x/usr/local/venv37zephyr/lib/python3.7/site-packages/pyocd-0.30.4.dev31-py3.7.egg/pyocd/__main__.py", line 150, in run
    status = cmd.invoke()
  File "/home/x/usr/local/venv37zephyr/lib/python3.7/site-packages/pyocd-0.30.4.dev31-py3.7.egg/pyocd/subcommands/flash_cmd.py", line 86, in invoke
    with session:
  File "/home/x/usr/local/venv37zephyr/lib/python3.7/site-packages/pyocd-0.30.4.dev31-py3.7.egg/pyocd/core/session.py", line 343, in __enter__
    self.open()
  File "/home/x/usr/local/venv37zephyr/lib/python3.7/site-packages/pyocd-0.30.4.dev31-py3.7.egg/pyocd/core/session.py", line 457, in open
    self._probe.open()
  File "/home/x/usr/local/venv37zephyr/lib/python3.7/site-packages/pyocd-0.30.4.dev31-py3.7.egg/pyocd/probe/cmsis_dap_probe.py", line 170, in open
    self._link.open()
  File "/home/x/usr/local/venv37zephyr/lib/python3.7/site-packages/pyocd-0.30.4.dev31-py3.7.egg/pyocd/utility/concurrency.py", line 28, in _locking
    return func(self, *args, **kwargs)
  File "/home/x/usr/local/venv37zephyr/lib/python3.7/site-packages/pyocd-0.30.4.dev31-py3.7.egg/pyocd/probe/pydapaccess/dap_access_cmsis_dap.py", line 635, in open
    self._interface.open()
  File "/home/x/usr/local/venv37zephyr/lib/python3.7/site-packages/pyocd-0.30.4.dev31-py3.7.egg/pyocd/probe/pydapaccess/interface/pyusb_backend.py", line 102, in open
    if dev.is_kernel_driver_active(interface_number):
  File "/home/x/usr/local/venv37zephyr/lib/python3.7/site-packages/pyusb-1.1.1.post7+g90d3e05-py3.7.egg/usb/core.py", line 1107, in is_kernel_driver_active
    interface)
  File "/home/x/usr/local/venv37zephyr/lib/python3.7/site-packages/pyusb-1.1.1.post7+g90d3e05-py3.7.egg/usb/backend/libusb0.py", line 624, in is_kernel_driver_active
    from errno import ENODATA
ImportError: cannot import name 'ENODATA' from 'errno' (unknown location)

(venv37zephyr) uname -a
FreeBSD hexagon 12.2-RELEASE-p6 FreeBSD 12.2-RELEASE-p6 GENERIC  amd64

(venv37zephyr) which python
/home/x/usr/local/venv37zephyr/bin/python

(venv37zephyr) python --version
Python 3.7.10
```

Do you think this is our local LibUSB implementation issue @hselasky ?

cederom, 2021-05-13:
Okay this has nothing to do with LibUSB, just invalid use of import, please verify fix proposed here https://github.com/pyusb/pyusb/pull/366 - I am just not sure if we should return negative value or the errno value :-)

cederom, 2021-05-13:
As noted by @jonasmalacofilho it is not the import issue. As noted by @eblot before https://github.com/pyusb/pyusb/issues/299 this part may cause troubles - and it is - code is tied to Linux implementation, but FreeBSD has this function implemented too, so presence check is passed, and then we encounter unhandled exception on Linux-only `errno.ENODATA` import. See discussion in the PR: https://github.com/pyusb/pyusb/pull/366.

mcuee, 2021-05-21:
As mentioned by Jonas in #366, #299 is not really the issue. The issue is more because of the libusb FreeBSD implementation and the way pyusb deals with it.

cederom, 2021-06-23:
As described in https://github.com/pyusb/pyusb/pull/366 the fix [1] proposed by @jonasmalacofilho seems better than mine and also solves the issue, please merge as solution :-) Thank you!! :-)

[1] https://github.com/jonasmalacofilho/pyusb/tree/fix-libusb0-is-kernel-driver-active

commit 582226aa (2021-06-23): libusb0: differentiate when usb_get_driver_np is missing

Fixes: #365
Closes: #366
closed 2021-06-23 by jonasmalacofilho