pyusb/pyusb #367: Data sent to stdout is invalid according to the receiving piped program, ffplay

c0debreaker, 2021-05-14:
The code below works great. The raw.bin file is properly formatted and can be parsed by ffmpeg.
```
with open("raw.bin", "wb") as myFile:
    RxData = None
    while True:
        try:
            data = dev.read(3, 100)
            myFile.write(data)
```

However, I also need to output  the value of data variable to stdout. I tried this code but the resulting text is invalid according to ffplay. I guess the line that contains `join` is the culprit.
```
while True:
    try:
        data = dev.read(3, 100)
        sret = ''.join([chr(x) for x in data])
        sys.stdout.write(sret)
```

An example value of data variable is below. I get the value below when I use `print(data)` right after `dev.read`.
```
array('B', [88, 115, 135, 243, 123, 51, 73, 67, 251, 75, 104, 11, 245, 181, 183, 221, 108, 152, 4, 194, 216, 124, 243, 13, 57, 254, 248, 173, 45, 8, 100, 181, 222, 165, 135, 45, 218, 182, 226, 137, 47, 91, 144, 208, 37, 64, 156, 70, 89, 169, 164, 171, 166, 59, 207, 200, 123, 70, 37, 224, 127, 58, 98, 155, 86, 167, 214, 239, 93, 205, 191, 217, 117, 162, 141, 152, 147, 117, 98, 220, 138, 88, 21, 82, 126, 27, 121, 178, 142, 73, 225, 130, 253, 217, 71, 116, 142, 137, 252, 29, 157, 102, 120, 190, 123, 54, 184, 254, 29, 56, 179, 245, 97, 200, 227, 172, 52, 226, 154, 210, 244, 0, 133, 31, 127, 12, 176, 62, 222, 59, 114, 231, 105, 25, 3, 111, 147, 196, 167, 148, 241, 219, 128, 100, 104, 247, 138, 51, 89, 168, 96, 230, 232, 186, 5, 145, 20, 136, 28, 73, 232, 63, 22, 65, 185, 211, 254, 224, 1, 183, 127, 182, 204, 126, 95, 21, 56, 157, 233, 243, 23, 193, 101, 183, 167, 243, 137, 120, 176, 118, 44, 95, 213, 110, 225, 76, 146, 92, 75, 46, 241, 202, 160, 123, 40, 204, 32, 91, 215, 160, 83, 122, 227, 95, 162, 243, 98, 243, 2, 187, 125, 38, 250, 215, 13, 82, 50, 52, 229, 74, 119, 82, 240, 249, 171, 139, 83, 226, 124, 100, 184, 1, 145, 110, 56, 166, 194, 234, 122, 139, 205, 150, 87, 112, 8, 248, 15, 81, 177, 87, 162, 43, 164, 22, 172, 189, 180, 84, 77, 68, 68, 119, 248, 65, 43, 232, 31, 129, 153, 17, 115, 161, 106, 193, 216, 216, 29, 164, 95, 220, 147, 66, 180, 84, 229, 94, 84, 2, 65, 21, 94, 21, 48, 254, 147, 198, 96, 118, 86, 174, 0, 153, 187, 1, 213, 90, 65, 93, 83, 152, 7, 43, 108, 142, 124, 26, 73, 159, 86, 58, 83, 235, 40, 211, 215, 220, 107, 191, 43, 17, 236, 234, 232, 157, 128, 85, 225, 180, 248, 24, 13, 7, 201, 58, 37, 235, 206, 172, 139, 251, 160, 207, 241, 253, 48, 62, 48, 103, 83, 174, 50, 252, 134, 47, 12, 50, 20, 77, 21, 50, 103, 251, 145, 84, 151, 156, 64, 199, 240, 85, 76, 55, 228, 184, 154, 128, 133, 82, 4, 247, 40, 89, 106, 177, 51, 169, 181, 244, 16, 204, 160, 168, 197, 147, 68, 147, 185, 197, 226, 136, 35, 135, 160, 21, 112, 71, 202, 228, 127, 127, 196, 85, 158, 195, 154, 52, 152, 122, 104, 64, 69, 220, 147, 254, 67, 93, 3, 250, 191, 111, 85, 195, 14, 135, 82, 67, 148, 89, 110, 72, 105, 24, 201, 251, 103, 223, 231, 128, 16, 35, 46, 94, 190, 4, 54, 153, 31, 224, 196, 2, 82, 200, 127, 3, 33, 253, 255, 209, 82, 249, 150, 219, 1, 223, 120, 62, 247, 39, 143, 230, 56, 11, 164, 106, 215, 5, 83, 101, 226, 110, 129, 76])
```

Any help would be greatly appreciated.

jonasmalacofilho, 2021-05-14:
Your problem is that in

```py
sret = ''.join([chr(x) for x in data])
```

you interpret each byte read as a Unicode Scalar Value.  And then you write the resulting string to stdout, using UTF-8 by default.

These are not equivalent: for example, a byte value of 0xdb (219 decimal) becomes two bytes 0xc39b after these steps.

---

Closing the issue as it's not related to USB or PyUSB.

closed 2021-05-14 by jonasmalacofilho

c0debreaker, 2021-05-14:
Thank you for explaining the issue. What is the proper way to do it?

c0debreaker, 2021-05-14:
I'm good. It's working now. :)