pyusb/pyusb #368: Unable to write on isochronous OUT

FrontMage, 2021-05-20:
Hi, I've been trying to run the code on tutorial docs but found no luck.

```python
import usb.core
import usb.util

dev = usb.core.find(idVendor=0x2F9D, idProduct=0x3101)
reattach = False
if dev.is_kernel_driver_active(1):
    reattach = True
    dev.detach_kernel_driver(1)

if dev is None:
    raise ValueError("Device not found")

cfg = dev.get_active_configuration()
print(cfg)
dev.write(0x2, [0])

if reattach:
    dev.attach_kernel_driver(1)

```

outputs
```bash
2021-05-20 14:05:46,094 DEBUG:usb.backend.libusb1:_LibUSB.__init__(<CDLL 'libusb-1.0.so.0', handle 55eeac719720 at 0x7f0133210460>)
[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.000012] [00027103] libusb: debug [libusb_init] created default context
[ 0.000021] [00027103] libusb: debug [libusb_init] libusb v1.0.24.11584
[ 0.000028] [00027103] libusb: debug [get_kernel_version] reported kernel version is 5.10.34-1-MANJARO
[ 0.000046] [00027103] libusb: debug [op_init] found usbfs at /dev/bus/usb
[ 0.000049] [00027103] libusb: debug [op_init] max iso packet length is (likely) 98304 bytes
[ 0.000055] [00027103] libusb: debug [op_init] sysfs is available
[ 0.000399] [0002715e] libusb: debug [linux_udev_event_thread_main] udev event thread entering
[ 0.004676] [00027103] libusb: debug [linux_get_device_address] getting address for device: usb1 detached: 0
[ 0.004700] [00027103] libusb: debug [linux_get_device_address] scan usb1
[ 0.004725] [00027103] libusb: debug [linux_get_device_address] bus=1 dev=1
[ 0.004729] [00027103] libusb: debug [linux_enumerate_device] busnum 1 devaddr 1 session_id 257
[ 0.004733] [00027103] libusb: debug [linux_enumerate_device] allocating new device for 1/1 (session 257)
[ 0.004799] [00027103] libusb: debug [linux_get_device_address] getting address for device: 1-5 detached: 0
[ 0.004803] [00027103] libusb: debug [linux_get_device_address] scan 1-5
[ 0.004819] [00027103] libusb: debug [linux_get_device_address] bus=1 dev=6
[ 0.004822] [00027103] libusb: debug [linux_enumerate_device] busnum 1 devaddr 6 session_id 262
[ 0.004824] [00027103] libusb: debug [linux_enumerate_device] allocating new device for 1/6 (session 262)
[ 0.004840] [00027103] libusb: debug [linux_get_parent_info] dev 0x55eeac73cdf0 (1-5) has parent 0x55eeac73c540 (usb1) port 5
[ 0.004891] [00027103] libusb: debug [linux_get_device_address] getting address for device: 1-6 detached: 0
[ 0.004895] [00027103] libusb: debug [linux_get_device_address] scan 1-6
[ 0.004909] [00027103] libusb: debug [linux_get_device_address] bus=1 dev=3
[ 0.004911] [00027103] libusb: debug [linux_enumerate_device] busnum 1 devaddr 3 session_id 259
[ 0.004914] [00027103] libusb: debug [linux_enumerate_device] allocating new device for 1/3 (session 259)
[ 0.004930] [00027103] libusb: debug [linux_get_parent_info] dev 0x55eeac73caa0 (1-6) has parent 0x55eeac73c540 (usb1) port 6
[ 0.004978] [00027103] libusb: debug [linux_get_device_address] getting address for device: usb2 detached: 0
[ 0.004983] [00027103] libusb: debug [linux_get_device_address] scan usb2
[ 0.004997] [00027103] libusb: debug [linux_get_device_address] bus=2 dev=1
[ 0.005000] [00027103] libusb: debug [linux_enumerate_device] busnum 2 devaddr 1 session_id 513
[ 0.005002] [00027103] libusb: debug [linux_enumerate_device] allocating new device for 2/1 (session 513)
[ 0.005062] [00027103] libusb: debug [linux_get_device_address] getting address for device: 2-2 detached: 0
[ 0.005066] [00027103] libusb: debug [linux_get_device_address] scan 2-2
[ 0.005081] [00027103] libusb: debug [linux_get_device_address] bus=2 dev=2
[ 0.005083] [00027103] libusb: debug [linux_enumerate_device] busnum 2 devaddr 2 session_id 514
[ 0.005086] [00027103] libusb: debug [linux_enumerate_device] allocating new device for 2/2 (session 514)
[ 0.005106] [00027103] libusb: debug [linux_get_parent_info] dev 0x55eeac71f570 (2-2) has parent 0x55eeac73d0d0 (usb2) port 2
[ 0.005174] [00027103] libusb: debug [linux_get_device_address] getting address for device: usb3 detached: 0
[ 0.005182] [00027103] libusb: debug [linux_get_device_address] scan usb3
[ 0.005206] [00027103] libusb: debug [linux_get_device_address] bus=3 dev=1
[ 0.005210] [00027103] libusb: debug [linux_enumerate_device] busnum 3 devaddr 1 session_id 769
[ 0.005214] [00027103] libusb: debug [linux_enumerate_device] allocating new device for 3/1 (session 769)
[ 0.005292] [00027103] libusb: debug [linux_get_device_address] getting address for device: 3-2 detached: 0
[... 292 lines cut ...]
[ 0.012022] [00027103] libusb: debug [libusb_unref_device] destroy device 6.1
[ 0.012025] [00027103] libusb: debug [libusb_unref_device] destroy device 5.1
[ 0.012028] [00027103] libusb: debug [libusb_unref_device] destroy device 4.1
[ 0.012030] [00027103] libusb: debug [libusb_unref_device] destroy device 3.3
[ 0.012032] [00027103] libusb: debug [libusb_unref_device] destroy device 3.2
[ 0.012035] [00027103] libusb: debug [libusb_unref_device] destroy device 3.1
[ 0.012037] [00027103] libusb: debug [libusb_unref_device] destroy device 2.2
[ 0.012039] [00027103] libusb: debug [libusb_unref_device] destroy device 2.1
[ 0.012042] [00027103] libusb: debug [libusb_unref_device] destroy device 1.3
[ 0.012044] [00027103] libusb: debug [libusb_unref_device] destroy device 1.6
[ 0.012046] [00027103] libusb: debug [libusb_unref_device] destroy device 1.1
[ 0.012049] [00027103] libusb: debug [usbi_remove_event_source] remove fd 6
[ 0.012055] [00027103] libusb: debug [usbi_remove_event_source] remove fd 5
[ 0.012074] [0002715e] libusb: debug [linux_udev_event_thread_main] udev event thread exiting

```

The problem seems to be something wrong with submitting the transfer.
```bash
[ 0.009953] [00026f4e] libusb: error [submit_iso_transfer] submiturb failed, errno=2
[ 0.009955] [00026f4e] libusb: debug [submit_iso_transfer] first URB failed, easy peasy  
```

I search through libusb's docs, but there is no errno=2 found, only negative error numbers.
I'm confused here, can someone help?

jonasmalacofilho, 2021-05-20:
> I search through libusb's docs, but there is no errno=2 found, only negative error numbers.

My interpretation is that libusb functions return 0 on success or a negative error code, where `error_code == -error_number`.

Assuming that's correct, you're getting `LIBUSB_ERROR_INVALID_PARAM`, which is mentioned in the docs for [`libusb_submit_transfer()`](https://libusb.sourceforge.io/api-1.0/group__libusb__asyncio.html#gabb0932601f2c7dad2fee4b27962848ce):

> LIBUSB_ERROR_INVALID_PARAM if the transfer size is larger than the operating system and/or hardware can support (see [Transfer length limitations](https://libusb.sourceforge.io/api-1.0/group__libusb__asyncio.html#asynclimits)) 

Keep in mind that the [implementation](https://github.com/libusb/libusb/blob/v1.0.24/libusb/os/linux_usbfs.c#L2052) appears to return that error from a few different places.  For example, that error is also returned for `transfer->num_iso_packets < 1`, which is calculated [here](https://github.com/pyusb/pyusb/blob/v1.1.1/usb/backend/libusb1.py#L659).

FrontMage, 2021-05-20:
@jonasmalacofilho 
Thanks for the insight!
A bit weird though, since my data is just a single `0`, which does not seem to be larger than anything other than an empty array.
Perhaps there is an isochronous write example I can refer to?

mcuee, 2021-05-20:
You are sending this to an audio device which may have some requirements on setting up the pipe. You can not just any how send the data.

mcuee, 2021-05-20:
If you are learning USB, libusb and pyusb, do not use isochronous transfer as a start, it is tough to get it right.

The best is to start with a USB MCU demo board with proper generic USB FW examples, and start with control transfer, interrupt transfer and bulk transfer. 

DO NOT start with a mouse, keyboard or USB mass storage device either.

FrontMage, 2021-05-20:
@mcuee Totally agree, isochronous is insanely complicated, the example code on the `libusb` repo is not working either.
I can't even find any other examples showing how to stream data using isochronous.
Thanks for the help, I will continue looking for solutions.

mcuee, 2021-05-21:
So I will close this issue.

closed 2021-05-21 by mcuee

jonasmalacofilho, 2021-05-21:
I made a mistake before: the error number that appears in `submiturb failed, errno=2` [comes](https://github.com/libusb/libusb/blob/v1.0.24/libusb/os/linux_usbfs.c#L2159) from the `SUBMITURB` ioctl, and has nothing to do with the libusb error the function returns.

And, if it helps, in that context `errno=2` [means](https://elixir.bootlin.com/linux/v5.12.5/source/include/uapi/asm-generic/errno-base.h#L6) `ENOENT`, which in turn is [documented](https://www.kernel.org/doc/html/latest/driver-api/usb/error-codes.html) to mean:

> specified interface or endpoint does not exist or is not enabled

mcuee, 2021-05-21:
Yes, that is right. The device is a bit strange. The default interface 0 has no Endpoints associated with it which cause the abover errno=2.

Interface 1 Alternative Setting 0 has no valid end points either. Interface 1 Alternative Setting 1 has valid EP OUT 2 (0x02).

FrontMage, 2021-05-22:
@jonasmalacofilho @mcuee 
Thanks for the help!
There are 3 endpoints, 0x2 iso out, 0x81 iso in, 0x82 interrupt in.
Write to 0x2 will give the above error, while write on 0x81 and 0x82 will give a `Resource busy` error which I think is caused by not claiming the device.
From what I've learnt, endpoint IN means device transfer data into the computer OUT means the other way.
So these errors are not making any sense, since the endpoint clearly exits. About the enable part, I believe they are enabled, but might need to be claimed to be used.
I tried to claim the device, detach and reattach kernel drivers, but I always get `Resource busy`.

mcuee, 2021-05-22:
Good progress. 

I recommend you to read the basic of USB here.
https://www.usbmadesimple.co.uk/

mcuee, 2021-05-22:
And read pyusb tutorial in more detail about dev.set_interface_altsetting().
https://github.com/pyusb/pyusb/blob/master/docs/tutorial.rst