pyusb/pyusb #369: Read less bytes than 'size_or_buffer' parameter when 'ZZZZZZC' is larger than 2097672

vhicipos2, 2021-05-20:
I used dev.read(0x86,2097672,10000) to read 2097672 bytes from a 68013A demo board, but it only returned 2097664 bytes.
I had to use another dev.read(0x86,8,10000) to get the last 8 bytes. Below is my code :
`length = 262209
dr = dev.read(0x86,length*8,10000)
if int(len(dr)/8) != length:
    dr = dr + dev.read(0x86,(length-int(len(dr)/8))*8,10000)`
[image]
I  have checked all 'length' less than 262209，the read function returned the correct number of bytes.
2097672 = 1024*2048+512+8, is this a clue?

vhicipos2, 2021-05-20:
The read speed is ~40MB/s，so 10000ms timeout is sufficient.

jonasmalacofilho, 2021-05-20:
This behavior depends on the device, specifically what its USB interface supports and what the protocol it uses actually mandates.  Thus, no `read()` is guaranteed to return exactly as many bytes as you requested.

> 2097672 = 1024*2048+512+8, is this a clue?

Not really, since isn't a simple limit+spill pattern.

vhicipos2, 2021-05-20:
Hi Jonas,
 I tried to write and read back 7200000 bytes. Below is the record of USB monitor:
[image]
The max bytes for a single read/write request seemed to be 2097152, so a 7200000 bytes write was split into 3x2097152+908544, that's fine.
However, a 7200000 bytes read was split into 3x2097152+908288, the last 256 bytes were lost. It looks to be the host's fault rather than the device's, because the host just requested 908288 bytes. May there be some bugs in pyusb or libusb driver?
[image]

jonasmalacofilho, 2021-05-20:
I'm not sure what you're using to capture the traffic, but couldn't that screenshot be interpreted to mean that the host asked (IN) for 512 bytes, but the device only returned (OUT/STATUS_SUCCESS) 256?

This is not to say that there can't be bugs in PyUSB, libusb or even WinUSB.  However, what proof do you have that the device is actually trying to completely fulfill your reads?

vhicipos2, 2021-05-20:
Hi Jonas,
  I use USB monitor Pro, a software based tool. In fact, read() was called twice, the first read(7200000) returned only 7200000-256 bytes, so I had to send the second read(256) to get the lost data.
'the host asked (IN) for 512 bytes, but the device only returned (OUT/STATUS_SUCCESS) 256' this behavior was OK for me. I just wonder why the first read(7200000) was split into 3x2097152+**908288** rather than 3x2097152+**908544**.
For now I have no solid proof the device works correctly, but simulation of FPGA code showed that 7200000 bytes were send to 68013A's FIFO interface, and two sequential read() got exactly 7200000 bytes.

jonasmalacofilho, 2021-05-20:
> [...] was split into 3x2097152+908288 rather than 3x2097152+908544.

Well, actually, `3x2097152 + 908544 + 256`, which seems fine to me.  As long as you can read all the data with a loop over `read()`, that's within spec.

I have not worked with the 68013A, but some things could be explain this behavior:

 - not all data is available yet at the USB interface when the host read request reaches it;
 - the FIFO you mentioned is probably bounded, and at some point not all data can fit within it, and only when a read clears the queue can more data be placed into it;
 - placing data into the queue may not be atomic and, if that case, another host read request could reach the interface before it's completely full, further chunking the data into multiple reads.

mcuee, 2021-05-21:
@vhicipos2 
I will suggest you ask the question in libusb mailing list. Tim Roberts is an Windows driver expert and he also knows USB/libusb well.

libusb mailing list: you need to subscribe in order to post.
https://sourceforge.net/projects/libusb/lists/libusb-devel

vhicipos2, 2021-05-21:
Hi Jonas and mcuee,
    Thanks for your suggestions! Besides asking help from libusb, I will get a hardware USB monitor and logic analyzer to check on the device's side. It may cost some time and for now I can use read() twice as a walkaround.

mcuee, 2021-05-21:
No problem. I do not think this is a pyusb problem so I will close the issue here. However, you can always reopen the issue if your further testing shows there is a problem with pyusb.

closed 2021-05-21 by mcuee