pyusb/pyusb #371: Milestone discussion

mcuee, 2021-06-15:
https://github.com/pyusb/pyusb/milestones

Just wondering what should be included in 1.2.0 release. I think the main missing feature like hotplug and async support are not ready.

jonasmalacofilho, 2021-06-15:
I agree that hotplug and async aren't ready.

In another project I've switched to time-based releases a year ago, and the experiment has been successful in ensuring timely, stable releases with less maintainer stress.  There are some differences – that project has to rely on contributors with access to the various supported devices, and it's also more active – but I think [the model](https://github.com/liquidctl/liquidctl/blob/main/docs/developer/process.md#release-cycle-and-pre-release-freeze-periods) is suitable here, with minor adjustments.

Specifically, my idea (before you opened this issue) was to release 1.2.0 in early July and to instill a 6-month release cycle, which would mean a 1.3.0 release in early January 2022 (or 1.2.x if no new features were added by then).

Additionally, the two freeze periods from liquidctl's model should probably be combined into a single 1-month (or longer) freeze, to compensate for lower release frequency and, thus, a higher assurance that the release is "good".

What do you think about this idea?

_(I also invite other interested parties to comment)._

mcuee, 2021-06-15:
It will be great to have a 6 month release cycle. I would be happy with 1 year release cycle myself based on the projects I am involved in. :-)

jonasmalacofilho, 2021-06-16:
I think 6 months is long enough to not overburden us, and keeping the time between releases shorter avoids pilling up too much code that hasn't seen comprehensive adoption and that, because of it, could be subtly broken. : )

jonasmalacofilho, 2021-06-16:
Updated the milestones and added a new document on our development process (`docs/internal/process.md`).  Closing...

closed 2021-06-16 by jonasmalacofilho