pyusb/pyusb #373: Potential issues with the test codes

mcuee, 2021-06-26:
I understand that the HW based test has been disabled. However, it seems to me that I can still run it and there are a few error messages which may point to the errors of the test codes (probably they are not matching with the latest pyusb codes).

I just built the bmfw from Wander (https://github.com/walac/bmfw) and here is the run log. I am using Python 3.9.5 within a virtual environment under Windows 10 64bit. The two failed cases are kind of expected because my FW built configuration may not match the test code (for example no isochronous endpoint). But the two errors may be a problem.

```
(py39venv) C:\work\libusb\pyusb\tests [master ≡ +1 ~0 -0 !]> python
Python 3.9.5 (tags/v3.9.5:0a7dcbd, May  3 2021, 17:27:52) [MSC v.1928 64 bit (AMD64)] on win32
Type "help", "copyright", "credits" or "license" for more information.

(py39venv) C:\work\libusb\pyusb\tests [master ≡ +1 ~0 -0 !]> python .\testall.py
FE.F.........E......
======================================================================
ERROR: runTest (test_control.ControlTest)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "C:\work\libusb\pyusb\tests\test_control.py", line 56, in runTest
    self.test_getset_descriptor()
  File "C:\work\libusb\pyusb\tests\test_control.py", line 107, in test_getset_descriptor
    self.assertEqual(struct.unpack(dev_fmt, ret.tostring()), dev_descr)
AttributeError: 'array.array' object has no attribute 'tostring'

======================================================================
ERROR: runTest (test_legacy.LegacyTest)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "C:\work\libusb\pyusb\tests\test_legacy.py", line 63, in runTest
    self.test_get_descriptor()
  File "C:\work\libusb\pyusb\tests\test_legacy.py", line 241, in test_get_descriptor
    self.assertEqual(struct.unpack(dev_fmt, ret.tostring()), dev_descr)
AttributeError: 'array.array' object has no attribute 'tostring'

======================================================================
FAIL: runTest (test_backend.BackendTest)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "C:\work\libusb\pyusb\tests\test_backend.py", line 64, in runTest
    self.test_iso_write_read()
  File "C:\work\libusb\pyusb\tests\test_backend.py", line 205, in test_iso_write_read
    self.__write_read(
  File "C:\work\libusb\pyusb\tests\test_backend.py", line 275, in __write_read
    self.assertEqual(ret, length, str(ret) + ' != ' + str(length))
AssertionError: 0 != 64 : 0 != 64

======================================================================
FAIL: runTest (test_integration.DeviceTest)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "C:\work\libusb\pyusb\tests\test_integration.py", line 68, in runTest
    self.test_write_read()
  File "C:\work\libusb\pyusb\tests\test_integration.py", line 165, in test_write_read
    self.assertTrue(
AssertionError: False is not true : array('B', [0, 1, 5, 4, 3, 2, 1, 0, 55, 54, 53, 52, 51, 50, 49, 48, 47, 46, 45, 44, 43, 42, 41, 40, 39, 38, 37, 36, 35, 34, 33, 32, 0, 1, 5, 4, 3, 2, 1, 0, 23, 22, 21, 20, 19, 18, 17, 16, 15, 14, 13, 12, 11, 10, 9, 8, 7, 6, 5, 4, 3, 2, 1, 0]) != array('B', [0, 1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 26, 27, 28, 29, 30, 31, 32, 33, 34, 35, 36, 37, 38, 39, 40, 41, 42, 43, 44, 45, 46, 47, 48, 49, 50, 51, 52, 53, 54, 55, 56, 57, 58, 59, 60, 61, 62, 63]), in interface = 2

----------------------------------------------------------------------
Ran 20 tests in 2.694s

FAILED (failures=2, errors=2)

```

mcuee, 2021-06-26:
Since we do not support legacy Python versions, I think the following simple fixes will do.

```
(py39venv) C:\work\libusb\pyusb\tests [master ≡ +1 ~2 -0 !]> git diff
diff --git a/tests/test_control.py b/tests/test_control.py
index cfeaa87..4bdd121 100644
--- a/tests/test_control.py
+++ b/tests/test_control.py
@@ -104,7 +104,7 @@ class ControlTest(unittest.TestCase):
                     self.dev.bDescriptorType,
                     0
                 )
-        self.assertEqual(struct.unpack(dev_fmt, ret.tostring()), dev_descr)
+        self.assertEqual(struct.unpack(dev_fmt, ret.tobytes()), dev_descr)

     @methodtrace(utils.logger)
     def test_getset_configuration(self):
diff --git a/tests/test_legacy.py b/tests/test_legacy.py
index f5d4978..7edceee 100644
--- a/tests/test_legacy.py
+++ b/tests/test_legacy.py
@@ -238,7 +238,7 @@ class LegacyTest(unittest.TestCase):
                     0,
                     struct.calcsize(dev_fmt))

-        self.assertEqual(struct.unpack(dev_fmt, ret.tostring()), dev_descr)
+        self.assertEqual(struct.unpack(dev_fmt, ret.tobytes()), dev_descr)

     def __write_read(self, write_fn, read_fn, ep):
         for data in (utils.get_array_data1(), utils.get_array_data2()):

```

mcuee, 2021-06-26:
Once I applied the above simple patch, it is now running better.

```

(py39venv) C:\work\libusb\pyusb\tests [master ≡ +1 ~2 -0 !]> python .\testall.py
F..F................
======================================================================
FAIL: runTest (test_backend.BackendTest)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "C:\work\libusb\pyusb\tests\test_backend.py", line 64, in runTest
    self.test_iso_write_read()
  File "C:\work\libusb\pyusb\tests\test_backend.py", line 205, in test_iso_write_read
    self.__write_read(
  File "C:\work\libusb\pyusb\tests\test_backend.py", line 275, in __write_read
    self.assertEqual(ret, length, str(ret) + ' != ' + str(length))
AssertionError: 0 != 64 : 0 != 64

======================================================================
FAIL: runTest (test_integration.DeviceTest)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "C:\work\libusb\pyusb\tests\test_integration.py", line 68, in runTest
    self.test_write_read()
  File "C:\work\libusb\pyusb\tests\test_integration.py", line 165, in test_write_read
    self.assertTrue(
AssertionError: False is not true : array('B', [0, 1, 5, 4, 3, 2, 1, 0, 55, 54, 53, 52, 51, 50, 49, 48, 47, 46, 45, 44, 43, 42, 41, 40, 39, 38, 37, 36, 35, 34, 33, 32, 0, 1, 5, 4, 3, 2, 1, 0, 23, 22, 21, 20, 19, 18, 17, 16, 15, 14, 13, 12, 11, 10, 9, 8, 7, 6, 5, 4, 3, 2, 1, 0]) != array('B', [0, 1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 26, 27, 28, 29, 30, 31, 32, 33, 34, 35, 36, 37, 38, 39, 40, 41, 42, 43, 44, 45, 46, 47, 48, 49, 50, 51, 52, 53, 54, 55, 56, 57, 58, 59, 60, 61, 62, 63]), in interface = 2

----------------------------------------------------------------------
Ran 20 tests in 2.708s

FAILED (failures=2)

```

mcuee, 2021-06-26:
@jonasmalacofilho 
BTW, I think the HW based tests are not really disabled if I run testall.py, right? Is this the intention of the patch? I'd like to keep it this way since I can still run the HW based tests.
https://github.com/pyusb/pyusb/commit/a16251f3d62de1e0b50cdfb431482d08a34355b4

jonasmalacofilho, 2021-06-26:
> Since we do not support legacy Python versions, I think the following simple fixes will do.

That makes sense, `array.tostring()` was removed in Python 3.9.  Applied, thanks.

> The two failed cases are kind of expected because my FW built configuration may not match the test code (for example no isochronous endpoint).

> Once I applied the above simple patch, it is now running better and the two failed cases are expected.

What firmware configuration did you use?

> BTW, I think the HW based tests are not really disabled if I run testall.py, right? Is this the intention of the patch? I'd like to keep it this way since I can still run the HW based tests.

I no longer remember what I was thinking when I merged that PR, but it certainly makes sense to keep the hardware tests enabled in `testall.py`.  (Otherwise it would be better to remove them).

mcuee, 2021-06-26:
Thanks for merging the patch.

As for the firmware configuration, I am building Wander's FW as is. I was wrong to say that there were no isoc endpoints.

```
~ ❯ lsusb1 -vvv -d 04d8:fa2e

Bus 002 Device 009: ID 04d8:fa2e  
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0         8
  idVendor           0x04d8 
  idProduct          0xfa2e 
  bcdDevice            0.01
  iManufacturer           1 
  iProduct                2 
  iSerial                 3 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x004e
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xc0
      Self Powered
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass         0 
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x01  EP 1 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
        bInterval               0
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       1
      bNumEndpoints           2
      bInterfaceClass         0 
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
        bInterval               1
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       2
      bNumEndpoints           2
      bInterfaceClass         0 
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x03  EP 3 OUT
        bmAttributes            5
          Transfer Type            Isochronous
          Synch Type               Asynchronous
          Usage Type               Data
        wMaxPacketSize     0x0020  1x 32 bytes
        bInterval               1
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            5
          Transfer Type            Isochronous
          Synch Type               Asynchronous
          Usage Type               Data
        wMaxPacketSize     0x0020  1x 32 bytes
        bInterval               1
can't get device qualifier: No such file or directory
can't get debug descriptor: No such file or directory
Device Status:     0x0001
  Self Powered
```

mcuee, 2021-06-26:
Running log under macOS Big Sur (Mac Mini M1):

```
pyusb/tests on  master [?] via 🐍 v2.7.16 ❯ python3 testall.py 
E..............EE..E
======================================================================
ERROR: runTest (test_backend.BackendTest)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/Users/mcuee/build/pyusb/pyusb/tests/test_backend.py", line 63, in runTest
    self.test_intr_write_read()
  File "/Users/mcuee/build/pyusb/pyusb/tests/test_backend.py", line 191, in test_intr_write_read
    self.__write_read(
  File "/Users/mcuee/build/pyusb/pyusb/tests/test_backend.py", line 257, in __write_read
    ret = write_fn(self.handle, ep, intf, data, 1000)
  File "/Users/mcuee/build/pyusb/pyusb/usb/backend/libusb1.py", line 855, in intr_write
    return self.__write(self.lib.libusb_interrupt_transfer,
  File "/Users/mcuee/build/pyusb/pyusb/usb/backend/libusb1.py", line 938, in __write
    _check(retval)
  File "/Users/mcuee/build/pyusb/pyusb/usb/backend/libusb1.py", line 604, in _check
    raise USBError(_strerror(ret), ret, _libusb_errno[ret])
usb.core.USBError: [Errno 2] Entity not found

======================================================================
ERROR: runTest (test_legacy.LegacyTest)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/Users/mcuee/build/pyusb/pyusb/tests/test_legacy.py", line 59, in runTest
    self.test_bulk_write_read()
  File "/Users/mcuee/build/pyusb/pyusb/tests/test_legacy.py", line 158, in test_bulk_write_read
    self.__write_read(
  File "/Users/mcuee/build/pyusb/pyusb/tests/test_legacy.py", line 260, in __write_read
    ret = read_fn(ep | usb.util.ENDPOINT_IN, length, 1000)
  File "/Users/mcuee/build/pyusb/pyusb/usb/legacy.py", line 167, in bulkRead
    return self.dev.read(endpoint, size, timeout)
  File "/Users/mcuee/build/pyusb/pyusb/usb/core.py", line 1026, in read
    ret = fn(
  File "/Users/mcuee/build/pyusb/pyusb/usb/backend/libusb1.py", line 846, in bulk_read
    return self.__read(self.lib.libusb_bulk_transfer,
  File "/Users/mcuee/build/pyusb/pyusb/usb/backend/libusb1.py", line 954, in __read
    _check(retval)
  File "/Users/mcuee/build/pyusb/pyusb/usb/backend/libusb1.py", line 604, in _check
    raise USBError(_strerror(ret), ret, _libusb_errno[ret])
usb.core.USBError: [Errno 84] Overflow

======================================================================
ERROR: runTest (test_integration.DeviceTest)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/Users/mcuee/build/pyusb/pyusb/tests/test_integration.py", line 68, in runTest
    self.test_write_read()
  File "/Users/mcuee/build/pyusb/pyusb/tests/test_integration.py", line 161, in test_write_read
    ret = self.dev.read(eps[alt] | usb.util.ENDPOINT_IN, length)
  File "/Users/mcuee/build/pyusb/pyusb/usb/core.py", line 1026, in read
    ret = fn(
  File "/Users/mcuee/build/pyusb/pyusb/usb/backend/libusb1.py", line 846, in bulk_read
    return self.__read(self.lib.libusb_bulk_transfer,
  File "/Users/mcuee/build/pyusb/pyusb/usb/backend/libusb1.py", line 954, in __read
    _check(retval)
  File "/Users/mcuee/build/pyusb/pyusb/usb/backend/libusb1.py", line 602, in _check
    raise USBTimeoutError(_strerror(ret), ret, _libusb_errno[ret])
usb.core.USBTimeoutError: [Errno 60] Operation timed out

======================================================================
ERROR: runTest (test_integration.EndpointTest)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/Users/mcuee/build/pyusb/pyusb/tests/test_integration.py", line 349, in runTest
    self.test_write_read()
  File "/Users/mcuee/build/pyusb/pyusb/tests/test_integration.py", line 372, in test_write_read
[... 77 lines cut ...]
======================================================================
ERROR: runTest (test_integration.EndpointTest)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/Users/mcuee/build/pyusb/pyusb/tests/test_integration.py", line 349, in runTest
    self.test_write_read()
  File "/Users/mcuee/build/pyusb/pyusb/tests/test_integration.py", line 372, in test_write_read
    ret = self.ep_in.read(length)
  File "/Users/mcuee/build/pyusb/pyusb/usb/core.py", line 423, in read
    return self.device.read(self, size_or_buffer, timeout)
  File "/Users/mcuee/build/pyusb/pyusb/usb/core.py", line 1026, in read
    ret = fn(
  File "/Users/mcuee/build/pyusb/pyusb/usb/backend/libusb1.py", line 846, in bulk_read
    return self.__read(self.lib.libusb_bulk_transfer,
  File "/Users/mcuee/build/pyusb/pyusb/usb/backend/libusb1.py", line 954, in __read
    _check(retval)
  File "/Users/mcuee/build/pyusb/pyusb/usb/backend/libusb1.py", line 602, in _check
    raise USBTimeoutError(_strerror(ret), ret, _libusb_errno[ret])
usb.core.USBTimeoutError: [Errno 60] Operation timed out

----------------------------------------------------------------------
Ran 20 tests in 2.308s

FAILED (errors=4)
```

mcuee, 2021-06-26:
Tests seem to be okay under Linux. This was probably the main platform Wander tested.

```
pyusb/tests on  master [?] via 🐍 v3.9.5 ❯ sudo python3 testall.py
..........................
----------------------------------------------------------------------
Ran 26 tests in 0.501s

OK

```

mcuee, 2021-06-26:
On the other hand, I think USB PICs are not that popular. ARM based is probably more popular. 

I will explore Linux Gadget based solution as well. Unfortunately my Raspberry Pi 2, 3B+ and Raspberry Pi 400 are not supporting USB OTG, only USB hosts. I will need to dig out my other NanoPi and Orange Pi boards.

mcuee, 2021-06-26:
Close this issue for now. I will talk about tests related topics in #235.

closed 2021-06-26 by mcuee