pyusb/pyusb #374: libusb0 backend kernel driver detach for macOS with latest git libusb

mcuee, 2021-06-27:
One of the major change for the upcoming libusb 1.0.25 release is to add kernel driver detaching for macOS. This change should not affect libusb1 backend of pyusb. However, the current libusb0 has hard coded codes saying the feature is only available for Linux. This may need to be fixed.

https://github.com/libusb/libusb/pull/911
https://github.com/libusb/libusb/blob/master/ChangeLog
2021-05-16: v1.0.25-pre
* Darwin (macOS): Add support for detaching kernel drivers with authorization.

mcuee, 2021-06-27:
Take note macOS users will use libusb-compat-0.1 which is just a wrapper of libusb-1.0.
https://github.com/libusb/libusb-compat-0.1/blob/master/libusb/core.c

jonasmalacofilho, 2021-06-27:
We don't restrict kernel detachment on Darwin, even on libusb0: we only check that `usb_detach_kernel_driver_np()` is present and delegate everything to it.

https://github.com/pyusb/pyusb/blob/5d7cb25f6628fa14b1206e2e46492a160aaedd0f/usb/backend/libusb0.py#L648-L652

What I think will be an issue is our `is_kernel_driver_active()` backend method, because there we do have some custom logic, to handle all the different semantics `usb_get_driver_np()` may have depending on the implementation.

https://github.com/pyusb/pyusb/blob/5d7cb25f6628fa14b1206e2e46492a160aaedd0f/usb/backend/libusb0.py#L620-L646

Can we assume _all_ Darwin users will be running libusb-compat-0.1?  Because if that's the case, the required change is trivial (`'dummy' => True, -ENODATA => False`).

mcuee, 2021-06-27:
Yes we can safe to assume all Darwin users to use libusb-compat-0.1 now (or even since many years ago). Nathan has been with the libusb project since the 0.1 days.

commit f0c513c9 (2021-06-30): libusb0: implement is_kernel_driver_active() for Mac OS

Assumes the libusb0 implementation in use is libusb-compat, and depends
on libusb_kernel_driver_active() support for that platform (expected in
libusb 1.0.25).

Fixes: #374
closed 2021-06-30 by jonasmalacofilho