pyusb/pyusb #375: NotImplementedError: Operation not supported or unimplemented on this platform

TitouanLabourdette, 2021-06-28:
Hi.

I'm in an internship and I want to read the USB port to see the type of data transiting. Ideally, I want to be able to both send and receive USB packets.
I'm trying to read on a port with the .read() method but I have this error popping up : 
```
NotImplementedError: Operation not supported or unimplemented on this platform
```

I don't understand what could be the problem. Everything is up to date, I use the latest version of Python and Spyder, on Windows 10.

The code is as follow : 
```Python
import usb.core
import usb.util

# find our device
dev = usb.core.find(idVendor = 0x152D, idProduct = 0x0576)
# was it found?
if dev is None:
    raise ValueError('Device not found')
else:
    print(dev)
    dev.read(0x81,100)
```

jonasmalacofilho, 2021-06-28:
> I'm in an internship and I want to read the USB port to see the type of data transiting.

Using a USB traffic analyzer is probably more appropriate for that job.  There are hardware options, but so far I've been able to get by with software-only tools like Wireshark.

(Technically Wireshark doesn't actually capture the data, only displays it, but searching for "Wireshark + USB + [operating system]" should help you get started).

> I'm trying to read on a port with the .read() method but I have this error popping up :

Most likely, the driver that is installed for that device isn't compatible with libusb.  Their FAQ has [more information](https://github.com/libusb/libusb/wiki/FAQ#How_to_use_libusb_under_Windows) on using it with Windows.

TitouanLabourdette, 2021-06-28:
> Using a USB traffic analyzer is probably more appropriate for that job. There are hardware options, but so far I've been able to get by with software-only tools like Wireshark.
> (Technically [...]

I know of Wireshark and it really helped me to understand some things but I want to automatically read USB port to respond to certain data with a script now. I will look on Internet with the keywords you said.

> Most likely, the driver that is installed for that device isn't compatible with libusb. Their FAQ has more information on using it with Windows.

I will check this then.

Thank you very much for your quick answer.

jonasmalacofilho, 2021-06-30:
> I want to automatically read USB port to respond to certain data with a script now.

In that case PyUSB is a suitable tool for the job after all.

TitouanLabourdette, 2021-06-30:
> In that case PyUSB is a suitable tool for the job after all.

Well, yes, that was what I was trying to use in the snippet of code I wrote in my first post. But the problem is that once I have my device plugged in and found with the `usb.core.find()`'s method, I can't do `dev.read()` because it says `Operation not supported or unimplemented on this platform` and I don't understand why.

I followed the tutorial provided here : https://github.com/pyusb/pyusb/blob/master/docs/tutorial.rst

Maybe I have overlooked something ?

jonasmalacofilho, 2021-07-01:
I still think the most likely cause is that the installed driver isn't compatible with libusb.

> Most likely, the driver that is installed for that device isn't compatible with libusb. Their FAQ has [more information](https://github.com/libusb/libusb/wiki/FAQ#How_to_use_libusb_under_Windows) on using it with Windows.

TitouanLabourdette, 2021-07-01:
You were right, the port used were indeed not the same depending of the type of devices connected. I couldn't find any COM port because I used a hard drive. With another peripheral I finally got an open COM port.

Thank you very much for your help.

closed 2021-07-01 by jonasmalacofilho