pyusb/pyusb #378: 1,2,1: pytest is failing in project unit

kloczek, 2021-07-09:
Just normal build, install and test cycle used on building package from non-root account:
- "setup.py build"
- "setup.py install --root </install/prefix>"
- "pytest with PYTHONPATH pointing to setearch and sitelib inside </install/prefix>
```console
+ PYTHONPATH=/home/tkloczko/rpmbuild/BUILDROOT/python-pyusb-1.2.1-2.fc35.x86_64/usr/lib64/python3.8/site-packages:/home/tkloczko/rpmbuild/BUILDROOT/python-pyusb-1.2.1-2.fc35.x86_64/usr/lib/python3.8/site-packages
+ /usr/bin/pytest -ra -q
=========================================================================== test session starts ============================================================================
platform linux -- Python 3.8.11, pytest-6.2.4, py-1.10.0, pluggy-0.13.1
benchmark: 3.4.1 (defaults: timer=time.perf_counter disable_gc=False min_rounds=5 min_time=0.000005 max_time=1.0 calibration_precision=10 warmup=False warmup_iterations=100000)
rootdir: /home/tkloczko/rpmbuild/BUILD/pyusb-1.2.1, configfile: setup.cfg
plugins: forked-1.3.0, shutil-1.7.0, virtualenv-1.7.0, expect-1.1.0, httpbin-1.0.0, flake8-1.0.7, timeout-1.4.2, betamax-0.8.1, freezegun-0.4.2, case-1.5.3, isort-1.3.0, aspectlib-1.5.2, asyncio-0.15.1, toolbox-0.5, xprocess-0.17.1, aiohttp-0.3.0, checkdocs-2.7.0, mock-3.6.1, rerunfailures-9.1.1, requests-mock-1.9.3, cov-2.12.1, pyfakefs-4.5.0, flaky-3.7.0, hypothesis-6.14.0, benchmark-3.4.1, xdist-2.3.0, pylama-7.7.1, datadir-1.3.1, regressions-2.2.0, cases-3.6.3, Faker-8.10.0
collected 16 items

. F                                                                                                                                                                  [  7%]
tests/test_find.py .                                                                                                                                                 [ 14%]
tests/test_interop.py ......                                                                                                                                         [ 57%]
tests/test_util.py ......                                                                                                                                            [100%]

================================================================================= FAILURES =================================================================================
_______________________________________________________________________________ test session _______________________________________________________________________________

cls = <class '_pytest.runner.CallInfo'>, func = <function call_runtest_hook.<locals>.<lambda> at 0x7f47bd8f6940>, when = 'call'
reraise = (<class '_pytest.outcomes.Exit'>, <class 'KeyboardInterrupt'>)

    @classmethod
    def from_call(
        cls,
        func: "Callable[[], TResult]",
        when: "Literal['collect', 'setup', 'call', 'teardown']",
        reraise: Optional[
            Union[Type[BaseException], Tuple[Type[BaseException], ...]]
        ] = None,
    ) -> "CallInfo[TResult]":
        excinfo = None
        start = timing.time()
        precise_start = timing.perf_counter()
        try:
>           result: Optional[TResult] = func()

/usr/lib/python3.8/site-packages/_pytest/runner.py:311:
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _

>       lambda: ihook(item=item, **kwds), when=when, reraise=reraise
    )

/usr/lib/python3.8/site-packages/_pytest/runner.py:255:
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _

self = <_HookCaller 'pytest_runtest_call'>, args = (), kwargs = {'item': <CheckdocsItem project>}, notincall = set()

    def __call__(self, *args, **kwargs):
        if args:
            raise TypeError("hook calling supports only keyword arguments")
        assert not self.is_historic()
        if self.spec and self.spec.argnames:
            notincall = (
                set(self.spec.argnames) - set(["__multicall__"]) - set(kwargs.keys())
            )
            if notincall:
                warnings.warn(
                    "Argument(s) {} which are declared in the hookspec "
                    "can not be found in this hook call".format(tuple(notincall)),
                    stacklevel=2,
                )
>       return self._hookexec(self, self.get_hookimpls(), kwargs)

/usr/lib/python3.8/site-packages/pluggy/hooks.py:286:
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _

[... 446 lines cut ...]
                    # read() call run on child threads, with the timeout
                    # being done in a join() on those threads.  communicate()
                    # _after_ kill() is required to collect that and add it
                    # to the exception.
                    exc.stdout, exc.stderr = process.communicate()
                else:
                    # POSIX _communicate already populated the output so
                    # far into the TimeoutExpired exception.
                    process.wait()
                raise
            except:  # Including KeyboardInterrupt, communicate handled that.
                process.kill()
                # We don't call process.wait() as .__exit__ does that for us.
                raise
            retcode = process.poll()
            if check and retcode:
>               raise CalledProcessError(retcode, process.args,
                                         output=stdout, stderr=stderr)
E               subprocess.CalledProcessError: Command '['/usr/bin/python3', '/usr/lib/python3.8/site-packages/pep517/in_process/_in_process.py', 'prepare_metadata_for_build_wheel', '/tmp/tmpvhork5cr']' returned non-zero exit status 1.

/usr/lib64/python3.8/subprocess.py:516: CalledProcessError
========================================================================= short test summary info ==========================================================================
FAILED ::project - subprocess.CalledProcessError: Command '['/usr/bin/python3', '/usr/lib/python3.8/site-packages/pep517/in_process/_in_process.py', 'prepare_metadata_fo...
====================================================================== 1 failed, 13 passed in 19.33s =======================================================================
```

jonasmalacofilho, 2021-07-09:
The way you're running pytest appears to break it for some reason.  But I don't see any relation with PyUSB; in fact, even with pytest somewhat broken, our tests ran and passed.

jonasmalacofilho, 2021-07-09:
At a glance, it's the exactly same issue you reported to another project 5 days ago: [jamielennox/requests-mock#175 (comment)](https://github.com/jamielennox/requests-mock/issues/175#issuecomment-873661531).

It's also possibly loosely related to pypa/pep517#116, specifically to `pep517.meta` [(apparently) being deprecated](https://github.com/pypa/pep517/issues/116#issuecomment-864399353).

You may want to report this to pytest and/or pep517, since it's clearly impact what appears to be a fairly reasonable use case.

closed 2021-07-09 by jonasmalacofilho

kloczek, 2021-07-09:
> The way you're running pytest appears to break it for some reason. But I don't see any relation with PyUSB; in fact, even with pytest somewhat broken, our tests ran and passed.

OK. So how you propose to test what has been installed from non-root account on packaging module into rpm package?
Issue is that this methodology only ocassioonaly is not working (10 cases on total +530 already packaged python modukes).
```console
[tkloczko@barrel SPECS]$ grep BUG python-*|grep pytest | grep error
python-cffi.spec:# BUG: pytest errors https://foss.heptapod.net/pypy/cffi/-/issues/500
python-logutils.spec:# BUG: test suite is failing https://bitbucket.org/vinay.sajip/logutils/issues/4/035-pytest-test-suite-warnings-and-errors
python-path.spec:# BUG: pytest errors https://github.com/jaraco/path/issues/202
python-pbr.spec:# BUG: pytest errors https://bugs.launchpad.net/pbr/+bug/1930619
python-pip.spec:# BUG: pytest errors https://github.com/pypa/pip/issues/10123
python-pyfakefs.spec:# BUG: pytest is failing https://github.com/jmcgeheeiv/pyfakefs/issues/608 flaky cause part of the errors
python-pyrsistent.spec:# BUG: pytest errors when pytest-randomly is used https://github.com/tobgu/pyrsistent/issues/228
python-pytest-rerunfailures.spec:# BUG: tox warnings and pytest errors https://github.com/pytest-dev/pytest-rerunfailures/issues/159
python-urllib3.spec:# BUG: pytest errors and warnings https://github.com/urllib3/urllib3/issues/2264
python-whoosh.spec:# BUG: pytest error https://github.com/whoosh-community/whoosh/issues/571
```
```console
[tkloczko@barrel SPECS]$ ls python-* -1|wc -l
534
[tkloczko@barrel SPECS]$ grep %pytest python-* |wc -l
426
```

kloczek, 2021-07-09:
Correction:
```console
[tkloczko@barrel SPECS]$ grep "BUG: pytest" python-*| grep error
python-cffi.spec:# BUG: pytest errors https://foss.heptapod.net/pypy/cffi/-/issues/500
python-path.spec:# BUG: pytest errors https://github.com/jaraco/path/issues/202
python-pbr.spec:# BUG: pytest errors https://bugs.launchpad.net/pbr/+bug/1930619
python-pip.spec:# BUG: pytest errors https://github.com/pypa/pip/issues/10123
python-pyfakefs.spec:# BUG: pytest is failing https://github.com/jmcgeheeiv/pyfakefs/issues/608 flaky cause part of the errors
python-pyrsistent.spec:# BUG: pytest errors when pytest-randomly is used https://github.com/tobgu/pyrsistent/issues/228
python-urllib3.spec:# BUG: pytest errors and warnings https://github.com/urllib3/urllib3/issues/2264
python-whoosh.spec:# BUG: pytest error https://github.com/whoosh-community/whoosh/issues/571
[tkloczko@barrel SPECS]$ grep "BUG: pytest" python-*| grep error|wc -l
8
```

jonasmalacofilho, 2021-07-09:
> OK. So how you propose to test what has been installed from non-root account on packaging module into rpm package?
> Issue is that this methodology only ocassioonaly is not working (10 cases on total +530 already packaged python modukes).

I'm not sure.  I think pytest isn't really an integration testing tool, at least not to my knowlegde.

While in general that may be a significant limitation, our particular tests are just unit tests, with no use of real devices or lower level or system APIs.  So what I can propose for the packing of PyUSB is simple to run pytest on the sources, just using the appropriate `python` executable and standard library.

Regarding the occasional nature of these failures in general, it does seem that there's something that some projects do that triggers the problem.  But I'm not familiar with pytest's code...

If you report all the affected projects to pytest, as well as your setup and your motivation, they may be able to spot what's the issue.  And if it turns out to be something we can mitigate on our side (within reason), please let me know.

kloczek, 2021-07-12:
> The way you're running pytest appears to break it for some reason. But I don't see any relation with PyUSB; in fact, even with pytest somewhat broken, our tests ran and passed.

Do you have any suggestion how should I run pytest on building rpm package from non-root account?

jonasmalacofilho, 2021-07-13:
Instead of trying to test the installed modules, have you tried running `pytest` on the sources' root (but still within your isolated build environment).

kloczek, 2021-07-13:
OK. Issue is that typical package build procedure intention is  to test not source tree but what will be psackaged.
In other words if source tree tests will pass with no errors it may not mean that it is true about what will be packagrd.
Just kind of SQ (Stupid Question): can you tell what in this case does not allow use that kind of testing procedure? (from higher point of view)
Looking on the output really I cannot figure what is violated when that exact procedure is used ..
(I'm asking because I would like to understand corretctly that context)

jonasmalacofilho, 2021-07-13:
> OK. Issue is that typical package build procedure intention is to test not source tree but what will be psackaged.

While what you claim is true for integration tests, I don't think it's not the norm for unit tests.  In compiled languages the unit tests are not even part of the final executable.

> Just kind of SQ (Stupid Question): can you tell what in this case does not allow use that kind of testing procedure? (from higher point of view)

No idea, sorry.  I don't think we're doing anything too special here, so I'm really surprised that the issue only affects some projects [using pytest] and not others.

Have you tried opening an issue on the pytest repository?

kloczek, 2021-07-13:
> Have you tried opening an issue on the pytest repository?

I must to spend a bit more time to look on what exactly is failing pyusb using described above methodology to be able describe potentiall pyetst issue :/
Will let you know when will open the ticket.

jonasmalacofilho, 2021-07-13:
Well, it's up to you.  But IMO you can rule a PyUSB problem that's unrelated to pytest.  So, at this point I would be comfortable opening an issue or at least asking for assistance from the pytest maintainers: even if we need to change something here, it will be related to pytest usage, or some weird conflict between pytest and something else that we're using.

jonasmalacofilho, 2021-07-13:
Also, try running `pytest` with `--verbose`.