pyusb/pyusb #379: Please read the Tutorial and FAQ before asking questions

mcuee, 2021-08-01:
As mentioned in the README file, there is a tutorial and FAQ document. Please read it before raising questions. Thanks.

Tutorial: https://github.com/pyusb/pyusb/blob/master/docs/tutorial.rst
FAQ: https://github.com/pyusb/pyusb/blob/master/docs/faq.rst

jonasmalacofilho, 2021-08-01:
Do you think issue templates could help?

Or (/and) maybe we just need to make the documentation easier to find.

mcuee, 2021-08-01:
Personally I think issue template just deters the users from submitting issues. So I am against the idea of issue templates. 

So I think documentation will help. Examples will help too. Right now we do not really have full examples other than the tutorials.

mcuee, 2021-08-02:
I spent a bit of time labeling all the github issues. It seems to me that better documentation and better examples will help to sort out many issues.

jonasmalacofilho, 2021-08-24:
> Personally I think issue template just deters the users from submitting issues. 

I partially agree: they can indeed act as a deterrent.  But I think that can be managed/tuned.

What I suggest is something fairly light/relaxed, where there's even still an option opt out of the templates.  Something like what I have in [liquidctl](https://github.com/liquidctl/liquidctl/issues/new?assignees=&labels=bug&template=bug_report.md), but obviously adapted to our needs (mention the faq/tutorial; `lsusb -v` output; etc.).

---

Either way, this issue being pinned doesn't appear to have helped.

mcuee, 2021-08-24:
Okay, un-pin this one.

Your suggestion is good. I have no objections with a light template.

mcuee, 2021-08-24:
In this case, I will close this issue.

closed 2021-08-24 by mcuee

mcuee, 2021-08-24:
BTW, I have added the following simple patch to the FAQ. I find this to be a common issue for Windows users.
https://github.com/pyusb/pyusb/commit/67550ed625e4252c0d586dd8ed5de681f4ad4edc