pyusb/pyusb #380: pyusb isoc transfer under macOS

mcuee, 2021-08-02:
So far I have not been able to get pyusb isoc transfer to work under macOS and Windows. On Windows I think libusb-1.0 itself has some issues. So I will test out macOS and Linux first.

I am using Cyusb FX3 known good Firmware isoloop and isosrcsink.
https://github.com/hantianjz/cy_fx3_sdk/tree/master/cyusb_linux_1.0.5/fx3_images

Issues:
```
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/opt/homebrew/lib/python3.9/site-packages/usb/core.py", line 408, in write
    return self.device.write(self, data, timeout)
  File "/opt/homebrew/lib/python3.9/site-packages/usb/core.py", line 979, in write
    return fn(
  File "/opt/homebrew/lib/python3.9/site-packages/usb/backend/libusb1.py", line 874, in iso_write
    return handler.submit(self.ctx)
  File "/opt/homebrew/lib/python3.9/site-packages/usb/backend/libusb1.py", line 680, in submit
    _check(_lib.libusb_submit_transfer(self.transfer))
  File "/opt/homebrew/lib/python3.9/site-packages/usb/backend/libusb1.py", line 604, in _check
    raise USBError(_strerror(ret), ret, _libusb_errno[ret])
usb.core.USBError: [Errno 2] Entity not found
```

mcuee, 2021-08-02:
Run log following the turorial (no issues with the bulkloop and bulksrcsink example):
```
libusb on  master [?] took 11m41s ❯ python3
Python 3.9.6 (default, Jun 28 2021, 19:24:41) 
[Clang 12.0.5 (clang-1205.0.22.9)] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> import usb.core
>>> import usb.util
>>> dev = usb.core.find(idVendor=0x04b4, idProduct=0x00f0)
>>> dev
<DEVICE ID 04b4:00f0 on Bus 002 Address 013>
>>> dev.set_configuration()
>>> dev
<DEVICE ID 04b4:00f0 on Bus 002 Address 013>
>>> cfg = dev.get_active_configuration()
>>> intf = cfg[(0,1)]
>>> cfg
<CONFIGURATION 1: 100 mA>
>>> intf
<INTERFACE 0, 1: Vendor Specific>
>>> ep = usb.util.find_descriptor(
...     intf,
...     # match the first OUT endpoint
...     custom_match = \
...     lambda e: \
...         usb.util.endpoint_direction(e.bEndpointAddress) == \
...         usb.util.ENDPOINT_OUT)
>>> ep
<ENDPOINT 0x3: Isochronous OUT>
>>> dat="A"*1024
>>> ep.write(dat)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/opt/homebrew/lib/python3.9/site-packages/usb/core.py", line 408, in write
    return self.device.write(self, data, timeout)
  File "/opt/homebrew/lib/python3.9/site-packages/usb/core.py", line 979, in write
    return fn(
  File "/opt/homebrew/lib/python3.9/site-packages/usb/backend/libusb1.py", line 874, in iso_write
    return handler.submit(self.ctx)
  File "/opt/homebrew/lib/python3.9/site-packages/usb/backend/libusb1.py", line 680, in submit
    _check(_lib.libusb_submit_transfer(self.transfer))
  File "/opt/homebrew/lib/python3.9/site-packages/usb/backend/libusb1.py", line 604, in _check
    raise USBError(_strerror(ret), ret, _libusb_errno[ret])
usb.core.USBError: [Errno 2] Entity not found
```

mcuee, 2021-08-02:
"lsusb -vvv" output

```

libusb on  master [?] ❯ lsusb1 -vvv -d 04b4:00f0

Bus 002 Device 013: ID 04b4:00f0  
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               3.10
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0         9
  idVendor           0x04b4 
  idProduct          0x00f0 
  bcdDevice            0.00
  iManufacturer           1 
  iProduct                2 
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0035
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              400mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           0
      bInterfaceClass       255 
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       1
      bNumEndpoints           2
      bInterfaceClass       255 
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x03  EP 3 OUT
        bmAttributes            1
          Transfer Type            Isochronous
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0400  1x 1024 bytes
        bInterval               1
        bMaxBurst               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            1
          Transfer Type            Isochronous
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0400  1x 1024 bytes
        bInterval               1
        bMaxBurst               0
Binary Object Store Descriptor:
  bLength                 5
  bDescriptorType        15
  wTotalLength       0x0016
  bNumDeviceCaps          2
  USB 2.0 Extension Device Capability:
    bLength                 7
    bDescriptorType        16
    bDevCapabilityType      2
    bmAttributes   0x0000641e
      BESL Link Power Management (LPM) Supported
    BESL value     1024 us 
    Deep BESL value    24576 us 
  SuperSpeed USB Device Capability:
    bLength                10
    bDescriptorType        16
    bDevCapabilityType      3
    bmAttributes         0x00
    wSpeedsSupported   0x000e
      Device can operate at Full Speed (12Mbps)
      Device can operate at High Speed (480Mbps)
      Device can operate at SuperSpeed (5Gbps)
    bFunctionalitySupport   3
      Lowest fully-functional device speed is SuperSpeed (5Gbps)
    bU1DevExitLat           0 micro seconds
    bU2DevExitLat           0 micro seconds
can't get debug descriptor: No such file or directory
Device Status:     0x0000
  (Bus Powered)
```

mcuee, 2021-08-02:
Error messages under Linux (Raspberry Pi 400 running Ubuntu Linux 64bit)

```
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python3.9/dist-packages/pyusb-1.1.1.post14+g5d7cb25-py3.9.egg/usb/core.py", line 408, in write
  File "/usr/local/lib/python3.9/dist-packages/pyusb-1.1.1.post14+g5d7cb25-py3.9.egg/usb/core.py", line 986, in write
  File "/usr/local/lib/python3.9/dist-packages/pyusb-1.1.1.post14+g5d7cb25-py3.9.egg/usb/backend/libusb1.py", line 874, in iso_write
  File "/usr/local/lib/python3.9/dist-packages/pyusb-1.1.1.post14+g5d7cb25-py3.9.egg/usb/backend/libusb1.py", line 680, in submit
  File "/usr/local/lib/python3.9/dist-packages/pyusb-1.1.1.post14+g5d7cb25-py3.9.egg/usb/backend/libusb1.py", line 604, in _check
usb.core.USBError: [Errno 5] Input/Output Error

libusb on  master took 3m2s ❯ uname -a
Linux rpi400Ubuntu 5.11.0-1015-raspi #16-Ubuntu SMP PREEMPT Wed Jul 14 11:04:58 UTC 2021 aarch64 aarch64 aarch64 GNU/Linux

```

mcuee, 2021-08-03:
Same results under Windows as macOS, I am using libusbk driver, as libusb WinUSB does not work well with isoc transfer.

```
#!/usr/bin/python

### quick mod from https://github.com/danielkucera/cyusb3014-breakout/blob/master/software/loopback-test.py

import usb.core
import usb.util
import time

# find our device
dev = usb.core.find(idVendor=0x04b4, idProduct=0x00f0 )

# was it found?
if dev is None:
    raise ValueError('Device not found')

# set the active configuration. With no arguments, the first
# configuration will be the active one
dev.set_configuration()

# get an endpoint instance
cfg = dev.get_active_configuration()

intf = cfg[(0,1)]

ep = usb.util.find_descriptor(
    intf,
    # match the first OUT endpoint
    custom_match = \
    lambda e: \
        e.bEndpointAddress == 3)

ep2 = usb.util.find_descriptor(
    intf,
    # match the first IN endpoint
    custom_match = \
    lambda e: \
        e.bEndpointAddress == 0x83)

assert ep is not None
assert ep2 is not None

print ('ep:',ep)
print ('ep2:',ep2)

chunkw = 2**16

print ("Starting CYUSB3014 python loopback test")

def write_loop():
    start1 = time.time()
    trans1 = 0
    dat = "A"*chunkw
    while True:
        ep.write(dat)
        trans1 = trans1 + len(dat)
        if time.time() > start1 + 1:
            bps1 = trans1/(time.time() - start1)
            print ("Write: transfered %d kB/s" % (bps1/(1024)) )
            trans1 = 0
            start1 = time.time()

write_loop()

(py39venv) C:\work\libusb\python_usb [master ≡ +4 ~0 -0 !]> python .\cyusb_fx3_benchmark_isoloop_python3_write.py
ep:       ENDPOINT 0x3: Isochronous OUT ========================
       bLength          :    0x7 (7 bytes)
       bDescriptorType  :    0x5 Endpoint
       bEndpointAddress :    0x3 OUT
       bmAttributes     :    0x1 Isochronous
       wMaxPacketSize   :  0x400 (1024 bytes)
       bInterval        :    0x1
ep2:       ENDPOINT 0x83: Isochronous IN ========================
       bLength          :    0x7 (7 bytes)
       bDescriptorType  :    0x5 Endpoint
       bEndpointAddress :   0x83 IN
       bmAttributes     :    0x1 Isochronous
       wMaxPacketSize   :  0x400 (1024 bytes)
       bInterval        :    0x1
Starting CYUSB3014 python loopback test
Traceback (most recent call last):
  File "C:\work\libusb\python_usb\cyusb_fx3_benchmark_isoloop_python3_write.py", line 62, in <module>
    write_loop()
  File "C:\work\libusb\python_usb\cyusb_fx3_benchmark_isoloop_python3_write.py", line 54, in write_loop
    ep.write(dat)
  File "C:\work\python\py39venv\lib\site-packages\usb\core.py", line 408, in write
    return self.device.write(self, data, timeout)
  File "C:\work\python\py39venv\lib\site-packages\usb\core.py", line 979, in write
    return fn(
  File "C:\work\python\py39venv\lib\site-packages\usb\backend\libusb1.py", line 874, in iso_write
    return handler.submit(self.ctx)
  File "C:\work\python\py39venv\lib\site-packages\usb\backend\libusb1.py", line 680, in submit
    _check(_lib.libusb_submit_transfer(self.transfer))
  File "C:\work\python\py39venv\lib\site-packages\usb\backend\libusb1.py", line 604, in _check
    raise USBError(_strerror(ret), ret, _libusb_errno[ret])
usb.core.USBError: [Errno 2] Entity not found
```

mcuee, 2021-08-03:
But let me try to get the C based program to work first under any of the OS, before confirming that pyusb is the issue.
https://github.com/walac/isochronous

mcuee, 2021-08-03:
Looks like it does work under macOS, tested with the pyusb benchmark firmware, I need to explicitly set the alt_setting 2 in this case.

```
python_usb on  master [?] via 🐍 v3.9.6 (py396venv) ❯ python
Python 3.9.6 (default, Jun 28 2021, 19:24:41) 
[Clang 12.0.5 (clang-1205.0.22.9)] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> import usb.core
>>> import usb.util
>>> dev = usb.core.find(idVendor=0x04d8, idProduct=0xfa2e)
>>> dev.set_configuration()
>>> dev
<DEVICE ID 04d8:fa2e on Bus 002 Address 008>
>>> cfg = dev.get_active_configuration()
>>> intf = cfg[(0,2)]
>>> ep = usb.util.find_descriptor(
...     intf,
...     # match the first OUT endpoint
...     custom_match = \
...     lambda e: \
...         usb.util.endpoint_direction(e.bEndpointAddress) == \
...         usb.util.ENDPOINT_OUT)
>>> ep
<ENDPOINT 0x3: Isochronous OUT>
>>> dat = "A" * 64
>>> dev.set_interface_altsetting(intf,2)
>>> ep.write(dat)
64
>>> ret = dev.read(0x83, 64, 1000)
>>> sret = ''.join([chr(x) for x in ret])
>>> dat
'AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA'
>>> sret
'AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA'
```

mcuee, 2021-08-03:
Same thing that it actually works with the pyusb benchmark firmware, pyusb results match the C based code.

```
~/build/python via 🐍 v3.9.5 (mypy39venv) took 2m37s ❯ python
Python 3.9.5 (default, May 11 2021, 08:20:37) 
[GCC 10.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import usb.core
>>> import usb.util
>>> dev = usb.core.find(idVendor=0x04d8, idProduct=0xfa2e)
>>> dev.set_configuration()
>>> cfg = dev.get_active_configuration()
>>> intf = cfg[(0,2)]
>>> dev.set_interface_altsetting(intf,2)
>>> dat = "A" * 96
>>> ep = usb.util.find_descriptor(
...     intf,
...     # match the first OUT endpoint
...     custom_match = \
...     lambda e: \
...         usb.util.endpoint_direction(e.bEndpointAddress) == \
...         usb.util.ENDPOINT_OUT)
>>> ep
<ENDPOINT 0x3: Isochronous OUT>
>>> ep.write(dat)
96
>>> ret = dev.read(0x83, 96, 1000)
>>> sret = ''.join([chr(x) for x in ret])
>>> dat
'AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA'
>>> sret
'AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA'
>>> dat = "A" * 128
>>> ep.write(dat)
128
>>> ret = dev.read(0x83, 128, 1000)
>>> sret = ''.join([chr(x) for x in ret])
>>> dat
'AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA'
>>> sret
'AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA'
>>> exit()
```

mcuee, 2021-08-03:
C based code: 2 packets loopback are okay, more packets will cause problems, looks like FW bug to me. So pyusb is okay with full speed isoc transfer under Linux and macOS. I will try high speed and superspeed later.

```
Packet 0 requested to transfer 32 bytes, transfered 32 bytes.
Packet 1 requested to transfer 32 bytes, transfered 32 bytes.
Packet 2 requested to transfer 32 bytes, transfered 32 bytes.
Packet 3 requested to transfer 32 bytes, transfered 32 bytes.
Packet 0 requested to transfer 32 bytes, transfered 32 bytes.
Packet 1 requested to transfer 32 bytes, transfered 32 bytes.
Packet 2 requested to transfer 32 bytes, transfered 0 bytes.
Packet 3 requested to transfer 32 bytes, transfered 0 bytes.

```

mcuee, 2021-08-03:
The benchmark firmware does not really work under Windows with the C based code, so it is also not working under pyusb under Windows. Therefore I will test with the FX3 known good isoloop and isosrcsink FW which is known to work under Windows with libusbK (but may not work well with libusb-1.0 Windows, need to test it out).

mcuee, 2021-08-03:
I am seeing some funny results from the FX3 isoloop test (the readback data is funny), but it may not be related to pyusb, but rather libusb-1.0 Windows.

```
#!/usr/bin/python

### quick mod from https://github.com/danielkucera/cyusb3014-breakout/blob/master/software/loopback-test.py

import usb.core
import usb.util
import time
import _thread

# find our device
dev = usb.core.find(idVendor=0x04b4, idProduct=0x00f0 )

# was it found?
if dev is None:
    raise ValueError('Device not found')

# set the active configuration. With no arguments, the first
# configuration will be the active one
dev.set_configuration()

# get an endpoint instance
cfg = dev.get_active_configuration()

#intf = cfg[(0,1)]
#dev.set_interface_altsetting(interface = 0, alternate_setting = 1)
intf = cfg[(0,1)]
dev.set_interface_altsetting(intf,1)

epout = usb.util.find_descriptor(
    intf,
    # match the first OUT endpoint
    custom_match = \
    lambda e: \
        e.bEndpointAddress == 3)

epin = usb.util.find_descriptor(
    intf,
    # match the first IN endpoint
    custom_match = \
    lambda e: \
        e.bEndpointAddress == 0x83)

assert epout is not None
assert epin is not None

print ('epout:',epout)
print ('epin:',epin)

chunkr = 2**16
chunkw = 2**16
dat = "A"*chunkw
print ("Starting CYUSB3014 python loopback test")

def read_loop():
    start = time.time()
    trans = 0
    while True:
        data = dev.read(epin, chunkr, 1000)
        #datareadback = ''.join([chr(x) for x in data])
        #print(len(datareadback))
        #assert datareadback == dat
        trans = trans + len(data)
        if time.time() > start + 1:
            bps = trans/(time.time() - start)
            print ("Read: transferred %d kB/s" % (bps/(1024)) )
            trans = 0
            start = time.time()

def write_loop():
    start1 = time.time()
    trans1 = 0
    while True:
        epout.write(dat)
        trans1 = trans1 + len(dat)
        if time.time() > start1 + 1:
            bps1 = trans1/(time.time() - start1)
            print ("Write: transfered %d kB/s" % (bps1/(1024)) )
            trans1 = 0
            start1 = time.time()

_thread.start_new_thread( write_loop, () )

read_loop()

```

```
(py39venv) C:\work\libusb\python_usb [master ≡ +5 ~0 -0 !]> python .\cyusb_fx3_benchmark_isoloop_python3.py
epout:       ENDPOINT 0x3: Isochronous OUT ========================
       bLength          :    0x7 (7 bytes)
       bDescriptorType  :    0x5 Endpoint
       bEndpointAddress :    0x3 OUT
       bmAttributes     :    0x1 Isochronous
       wMaxPacketSize   :  0x400 (1024 bytes)
       bInterval        :    0x1
epin:       ENDPOINT 0x83: Isochronous IN ========================
       bLength          :    0x7 (7 bytes)
       bDescriptorType  :    0x5 Endpoint
       bEndpointAddress :   0x83 IN
       bmAttributes     :    0x1 Isochronous
       wMaxPacketSize   :  0x400 (1024 bytes)
       bInterval        :    0x1
Starting CYUSB3014 python loopback test
Read: transferred 6819 kB/s
Write: transfered 7960 kB/s
Read: transferred 6807 kB/s
Write: transfered 8000 kB/s
Read: transferred 6841 kB/s
Write: transfered 8004 kB/s
Read: transferred 6858 kB/s
Write: transfered 8006 kB/s
Read: transferred 6876 kB/s
Write: transfered 8002 kB/s
Read: transferred 6868 kB/s
Write: transfered 8000 kB/s
Read: transferred 6868 kB/s
Write: transfered 8001 kB/s
Read: transferred 6795 kB/s
Write: transfered 8004 kB/s
Read: transferred 6858 kB/s
Write: transfered 8007 kB/s
Read: transferred 6880 kB/s
Write: transfered 7998 kB/s
Write: transfered 7995 kB/s
Read: transferred 6871 kB/s
Write: transfered 8000 kB/s
Read: transferred 6852 kB/s
Write: transfered 7999 kB/s
Read: transferred 6874 kB/s
Read: transferred 6797 kB/s
Write: transfered 8006 kB/s
```

mcuee, 2021-08-03:
No issues under macOS, tested with my Mac Mini M1, I also checked returned data and they seem to be okay.
```
   
python_usb on  master [?] via 🐍 v2.7.16 took 31s ❯ python3 cyusb_fx3_isoloop_python3.py
epout:       ENDPOINT 0x3: Isochronous OUT ========================
       bLength          :    0x7 (7 bytes)
       bDescriptorType  :    0x5 Endpoint
       bEndpointAddress :    0x3 OUT
       bmAttributes     :    0x1 Isochronous
       wMaxPacketSize   :  0x400 (1024 bytes)
       bInterval        :    0x1
epin:       ENDPOINT 0x83: Isochronous IN ========================
       bLength          :    0x7 (7 bytes)
       bDescriptorType  :    0x5 Endpoint
       bEndpointAddress :   0x83 IN
       bmAttributes     :    0x1 Isochronous
       wMaxPacketSize   :  0x400 (1024 bytes)
       bInterval        :    0x1
Starting CYUSB3014 python loopback test
Read: transferred 5298 kB/s
Write: transfered 5323 kB/s
Read: transferred 5250 kB/s
Write: transfered 5333 kB/s
Read: transferred 4567 kB/s
Write: transfered 5322 kB/s
Read: transferred 3916 kB/s
Write: transfered 5333 kB/s
Read: transferred 3916 kB/s
Write: transfered 5333 kB/s
Read: transferred 3916 kB/s
Write: transfered 5333 kB/s
Read: transferred 4095 kB/s
Write: transfered 5332 kB/s
Read: transferred 4583 kB/s
Write: transfered 5333 kB/s
Read: transferred 4583 kB/s
Write: transfered 5333 kB/s
```

mcuee, 2021-08-03:
Same results under my Raspberry Pi 400 running Ubuntu 64bit Linux.
```
python_usb on  master [?] via 🐍 v3.9.5 took 9s ❯ python3 cyusb_fx3_isoloop_python3.py
epout:       ENDPOINT 0x3: Isochronous OUT ========================
       bLength          :    0x7 (7 bytes)
       bDescriptorType  :    0x5 Endpoint
       bEndpointAddress :    0x3 OUT
       bmAttributes     :    0x1 Isochronous
       wMaxPacketSize   :  0x400 (1024 bytes)
       bInterval        :    0x1
epin:       ENDPOINT 0x83: Isochronous IN ========================
       bLength          :    0x7 (7 bytes)
       bDescriptorType  :    0x5 Endpoint
       bEndpointAddress :   0x83 IN
       bmAttributes     :    0x1 Isochronous
       wMaxPacketSize   :  0x400 (1024 bytes)
       bInterval        :    0x1
Starting CYUSB3014 python loopback test
Read: transferred 5729 kB/s
Write: transfered 6315 kB/s
Write: transfered 6334 kB/s
Read: transferred 6208 kB/s
Read: transferred 6210 kB/s
Write: transfered 6320 kB/s
Write: transfered 6308 kB/s
Read: transferred 6182 kB/s
Write: transfered 6141 kB/s
Read: transferred 6037 kB/s
Read: transferred 5659 kB/s
Write: transfered 5740 kB/s
Read: transferred 5585 kB/s
Write: transfered 5677 kB/s
Read: transferred 5702 kB/s
Write: transfered 5751 kB/s
Read: transferred 5728 kB/s
Write: transfered 5824 kB/s
Read: transferred 5668 kB/s
Write: transfered 5747 kB/s

```

mcuee, 2021-08-03:
In summary, pyusb isoc codes seem to work under Linux and macOS. It does not seem to work well under Windows but the issue may well be with libusb-1.0 itself and not pyusb. So I will close this issue.

mcuee, 2021-08-03:
On the other hand, real isoc API should really be async, so the improvement for async API should still be applicable.

jonasmalacofilho, 2021-08-03:
Thanks for testing this, and I'm glad to hear that it seems to be mostly working (barring FW or libusb1 limitations).

> On the other hand, real isoc API should really be async, so the improvement for async API should still be applicable.

Can you clarify?  I'm not familiar with the use cases for isoc transfers...

Implementation wise, as long as it makes sense to have blocking isoc read and write, their current implementations for libusb1 seem to be correct (repeatedly calling `libusb_handle_events()` in a loop, [per the libusb1 documentation](https://libusb.sourceforge.io/api-1.0/group__libusb__asyncio.html#asyncevent)).

mcuee, 2021-08-03:
Yes if there are real use case of block isoc read and write, then the implementation seems to be correct. But majority of the use case of isoc transfer is really for video/audio and other types real time data, blocking transfer may not be good enough. 

What I mean is that the WIP #303 is not complete and it needs to implement for control transfer and isoc transfer. As for high level API, I am not an expert for Python so I am not so sure what will be good there.

For isoc transfer, you can see that libusb-1.0 API is more complicated than interrupt/bulk transfer. So I am not so sure if the approach of #303 can cover isoc transfer well enough or not.

And Travis actually creates quite some high level APIs for isoc transfer. Even though libusbk is Windows only, somehow I feel the API is nicer for async transfer and isoc transfer than libusb-1.0.
http://libusbk.sourceforge.net/UsbK3/modules.html

mcuee, 2022-03-29:
Some testing results under Windows with libusb-1.0.26rc1. Isoc transfer seems to work now under Windows with libusbk driver.
https://github.com/libusb/libusb/issues/1099

tormodvolden, 2022-03-29:
I see in the log in the libusb ticket that it fails in bulk_read. Should that be iso_read?

mcuee, 2022-03-29:
I will generate the debug log. 0x82 is an isoc endpoint. The FW is from Cypress and is a known-good firmware (tested with Cypress Windows CySteeam app and libusbk kbench). I believe the problem comes from libusb Windows (isoc problem with WinUSB driver) and not pyusb.
https://github.com/libusb/libusb/issues/1099#issuecomment-1081330942

tormodvolden, 2022-03-29:
Great. Yes, since it works with libusbk, the problem is probably in the libusb winusb code. But in any case your debug log might shed some light on what goes wrong.

tormodvolden, 2022-03-29:
I am starting to wonder if it is a pyusb bug, setting the transfer type wrongly to bulk when submitting the transfers, and for some reason this still works fine on libusbk.

jonasmalacofilho, 2022-03-29:
@mcuee can you also enable `PYUSB_DEBUG=debug` when generating that debug log?

tormodvolden, 2022-03-29:
See the order: bulk = 0, intr=1, iso=2 in usb/core.py
```
        fn_map = {
                    util.ENDPOINT_TYPE_BULK:backend.bulk_write,
                    util.ENDPOINT_TYPE_INTR:backend.intr_write,
                    util.ENDPOINT_TYPE_ISO:backend.iso_write
                }

        intf, ep = self._ctx.setup_request(self, endpoint)
        fn = fn_map[util.endpoint_type(ep.bmAttributes)]

```
versus e.g. from libusb.h```

        LIBUSB_ENDPOINT_TRANSFER_TYPE_CONTROL = 0x0,
        LIBUSB_ENDPOINT_TRANSFER_TYPE_ISOCHRONOUS = 0x1,
        LIBUSB_ENDPOINT_TRANSFER_TYPE_BULK = 0x2,
        LIBUSB_ENDPOINT_TRANSFER_TYPE_INTERRUPT = 0x3
```

tormodvolden, 2022-03-29:
I tested isochronous transfers with WinUSB with libfreenect/XBox (all C/C++) and it works well. I can see `winusbx_submit_iso_transfer` in the log. So I am even more convinced there is a pyusb issue.

jonasmalacofilho, 2022-03-29:
Hey, thank you both for looking into this some more.

While waiting for mcuee's log, I looked at https://github.com/libusb/libusb/issues/1099#issuecomment-1081865420 and there does appear to be a problem in PyUSB: the traceback clearly shows that we're issuing a `libusb_bulk_transfer()` call, which is completely unexpected for a isochronous endpoint.

There isn't anything immediately/obviously wrong with `<Device>.read()`.[^1]  However, it still depends on

```py
intf, ep = self._ctx.setup_request(self, endpoint)
```

returning the correct endpoint, which I'm no longer convinced it does _when altsettings are involved._  And that appears to be case (https://github.com/libusb/libusb/issues/1099#issuecomment-1081330942).

@mcuee, besides the LibUSB+PyUSB debug log, is it also possible to try with altsetting 0?

[^1]: Like @tormodvolden already spotted, `fn_map` is a dictionary/hash map, so the order of the entries is not important.  The definitions of the `util.ENDPOINT_TYPE_*` constants also appear to be correct.

tormodvolden, 2022-03-29:
There will always be altsettings with isochronous transfers, because any interface must have a (default?) altsetting without isochronous endpoints.

mcuee, 2022-03-30:
Okay, I will reopen this ticket. It seems to me pyusb does not work with isoc transfer for high speed and superspeed device (not yet checking out full speed device).

Here is the debug under Linux: Ubuntu 20.04 x64.

The following debug is with PYUSB_DEBUG=debug and LIBUSB_DEBUG=3 (it is too much with 4).

```
2022-03-30 08:36:44,878 DEBUG:usb.backend.libusb1:_LibUSB.__init__(<CDLL 'libusb-1.0.so.0', handle d1b3b0 at 0x7fb98590e8e0>)
2022-03-30 08:36:44,880 INFO:usb.core:find(): using backend "usb.backend.libusb1"
2022-03-30 08:36:44,880 DEBUG:usb.backend.libusb1:_LibUSB.enumerate_devices()
2022-03-30 08:36:44,880 DEBUG:usb.backend.libusb1:_LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x7fb985b1a730>)
2022-03-30 08:36:44,880 DEBUG:usb.backend.libusb1:_LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x7fb985b1a910>)
2022-03-30 08:36:44,880 DEBUG:usb.backend.libusb1:_Device._finalize_object()
2022-03-30 08:36:44,880 DEBUG:usb.backend.libusb1:_LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x7fb985b1a790>)
2022-03-30 08:36:44,880 DEBUG:usb.backend.libusb1:_Device._finalize_object()
2022-03-30 08:36:44,880 DEBUG:usb.backend.libusb1:_LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x7fb985b1a970>)
2022-03-30 08:36:44,880 DEBUG:usb.backend.libusb1:_Device._finalize_object()
2022-03-30 08:36:44,880 DEBUG:usb.backend.libusb1:_LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x7fb985b1a820>)
2022-03-30 08:36:44,880 DEBUG:usb.backend.libusb1:_Device._finalize_object()
2022-03-30 08:36:44,880 DEBUG:usb.backend.libusb1:_LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x7fb985b1aa00>)
2022-03-30 08:36:44,880 DEBUG:usb.backend.libusb1:_Device._finalize_object()
2022-03-30 08:36:44,880 DEBUG:usb.backend.libusb1:_LibUSB.get_configuration_descriptor(<usb.backend.libusb1._Device object at 0x7fb985b1aa00>, 0)
2022-03-30 08:36:44,880 DEBUG:usb.backend.libusb1:_LibUSB.open_device(<usb.backend.libusb1._Device object at 0x7fb985b1aa00>)
2022-03-30 08:36:44,880 DEBUG:usb.backend.libusb1:_LibUSB.set_configuration(<usb.backend.libusb1._DeviceHandle object at 0x7fb985dd8d00>, 1)
2022-03-30 08:36:44,882 DEBUG:usb.backend.libusb1:_LibUSB.get_configuration_descriptor(<usb.backend.libusb1._Device object at 0x7fb985b1aa00>, 0)
2022-03-30 08:36:44,882 DEBUG:usb.backend.libusb1:_LibUSB.get_configuration_descriptor(<usb.backend.libusb1._Device object at 0x7fb985b1aa00>, 0)
2022-03-30 08:36:44,882 DEBUG:usb.backend.libusb1:_LibUSB.get_interface_descriptor(<usb.backend.libusb1._Device object at 0x7fb985b1aa00>, 0, 0, 0)
2022-03-30 08:36:44,882 DEBUG:usb.backend.libusb1:_LibUSB.get_configuration_descriptor(<usb.backend.libusb1._Device object at 0x7fb985b1aa00>, 0)
2022-03-30 08:36:44,882 DEBUG:usb.backend.libusb1:_LibUSB.get_interface_descriptor(<usb.backend.libusb1._Device object at 0x7fb985b1aa00>, 0, 1, 0)
2022-03-30 08:36:44,882 DEBUG:usb.backend.libusb1:_LibUSB.get_configuration_descriptor(<usb.backend.libusb1._Device object at 0x7fb985b1aa00>, 0)
2022-03-30 08:36:44,882 DEBUG:usb.backend.libusb1:_LibUSB.get_interface_descriptor(<usb.backend.libusb1._Device object at 0x7fb985b1aa00>, 0, 2, 0)
2022-03-30 08:36:44,882 DEBUG:usb.backend.libusb1:_LibUSB.get_configuration_descriptor(<usb.backend.libusb1._Device object at 0x7fb985b1aa00>, 0)
2022-03-30 08:36:44,882 DEBUG:usb.backend.libusb1:_LibUSB.get_interface_descriptor(<usb.backend.libusb1._Device object at 0x7fb985b1aa00>, 0, 3, 0)
2022-03-30 08:36:44,882 DEBUG:usb.backend.libusb1:_LibUSB.get_configuration_descriptor(<usb.backend.libusb1._Device object at 0x7fb985b1aa00>, 0)
2022-03-30 08:36:44,882 DEBUG:usb.backend.libusb1:_LibUSB.get_interface_descriptor(<usb.backend.libusb1._Device object at 0x7fb985b1aa00>, 0, 4, 0)
2022-03-30 08:36:44,882 DEBUG:usb.backend.libusb1:_LibUSB.get_configuration_descriptor(<usb.backend.libusb1._Device object at 0x7fb985b1aa00>, 0)
2022-03-30 08:36:44,882 DEBUG:usb.backend.libusb1:_LibUSB.get_interface_descriptor(<usb.backend.libusb1._Device object at 0x7fb985b1aa00>, 0, 5, 0)
2022-03-30 08:36:44,882 DEBUG:usb.backend.libusb1:_LibUSB.get_configuration_descriptor(<usb.backend.libusb1._Device object at 0x7fb985b1aa00>, 0)
2022-03-30 08:36:44,882 DEBUG:usb.backend.libusb1:_LibUSB.get_interface_descriptor(<usb.backend.libusb1._Device object at 0x7fb985b1aa00>, 0, 6, 0)
2022-03-30 08:36:44,882 DEBUG:usb.backend.libusb1:_LibUSB.get_configuration_descriptor(<usb.backend.libusb1._Device object at 0x7fb985b1aa00>, 0)
2022-03-30 08:36:44,882 DEBUG:usb.backend.libusb1:_LibUSB.claim_interface(<usb.backend.libusb1._DeviceHandle object at 0x7fb985dd8d00>, 0)
2022-03-30 08:36:44,882 DEBUG:usb.backend.libusb1:_LibUSB.set_interface_altsetting(<usb.backend.libusb1._DeviceHandle object at 0x7fb985dd8d00>, 0, 6)
2022-03-30 08:36:44,883 DEBUG:usb.backend.libusb1:_LibUSB.get_interface_descriptor(<usb.backend.libusb1._Device object at 0x7fb985b1aa00>, 0, 6, 0)
2022-03-30 08:36:44,883 DEBUG:usb.backend.libusb1:_LibUSB.get_configuration_descriptor(<usb.backend.libusb1._Device object at 0x7fb985b1aa00>, 0)
2022-03-30 08:36:44,883 DEBUG:usb.backend.libusb1:_LibUSB.get_endpoint_descriptor(<usb.backend.libusb1._Device object at 0x7fb985b1aa00>, 0, 0, 6, 0)
2022-03-30 08:36:44,883 DEBUG:usb.backend.libusb1:_LibUSB.get_interface_descriptor(<usb.backend.libusb1._Device object at 0x7fb985b1aa00>, 0, 6, 0)
[... cut ...]  65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65]), 1000)
libusb: error [submit_bulk_transfer] submiturb failed, errno=22
...
```

mcuee, 2022-03-30:
Under Linux it seems to me the other simpler device with only one alternative setting does work. This is a SuperSpeed FX3 device running isoc loop back test. I will try Windows later.

```
2022-03-30 08:56:41,987 DEBUG:usb.backend.libusb1:_LibUSB.__init__(<CDLL 'libusb-1.0.so.0', handle 26bd3b0 at 0x7fab6e27c8e0>)
libusb: warning [libusb_init] installing new context as implicit default
2022-03-30 08:56:41,989 INFO:usb.core:find(): using backend "usb.backend.libusb1"
2022-03-30 08:56:41,989 DEBUG:usb.backend.libusb1:_LibUSB.enumerate_devices()
2022-03-30 08:56:41,989 DEBUG:usb.backend.libusb1:_LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x7fab6e448730>)
2022-03-30 08:56:41,989 DEBUG:usb.backend.libusb1:_LibUSB.get_configuration_descriptor(<usb.backend.libusb1._Device object at 0x7fab6e448730>, 0)
2022-03-30 08:56:41,989 DEBUG:usb.backend.libusb1:_LibUSB.open_device(<usb.backend.libusb1._Device object at 0x7fab6e448730>)
2022-03-30 08:56:41,989 DEBUG:usb.backend.libusb1:_LibUSB.set_configuration(<usb.backend.libusb1._DeviceHandle object at 0x7fab6e747d90>, 1)
2022-03-30 08:56:41,989 DEBUG:usb.backend.libusb1:_LibUSB.get_configuration_descriptor(<usb.backend.libusb1._Device object at 0x7fab6e448730>, 0)
2022-03-30 08:56:41,989 DEBUG:usb.backend.libusb1:_LibUSB.get_configuration_descriptor(<usb.backend.libusb1._Device object at 0x7fab6e448730>, 0)
2022-03-30 08:56:41,989 DEBUG:usb.backend.libusb1:_LibUSB.get_interface_descriptor(<usb.backend.libusb1._Device object at 0x7fab6e448730>, 0, 0, 0)
2022-03-30 08:56:41,990 DEBUG:usb.backend.libusb1:_LibUSB.get_configuration_descriptor(<usb.backend.libusb1._Device object at 0x7fab6e448730>, 0)
2022-03-30 08:56:41,990 DEBUG:usb.backend.libusb1:_LibUSB.get_interface_descriptor(<usb.backend.libusb1._Device object at 0x7fab6e448730>, 0, 1, 0)
2022-03-30 08:56:41,990 DEBUG:usb.backend.libusb1:_LibUSB.get_configuration_descriptor(<usb.backend.libusb1._Device object at 0x7fab6e448730>, 0)
2022-03-30 08:56:41,990 DEBUG:usb.backend.libusb1:_LibUSB.claim_interface(<usb.backend.libusb1._DeviceHandle object at 0x7fab6e747d90>, 0)
2022-03-30 08:56:41,990 DEBUG:usb.backend.libusb1:_LibUSB.set_interface_altsetting(<usb.backend.libusb1._DeviceHandle object at 0x7fab6e747d90>, 0, 1)
2022-03-30 08:56:41,991 DEBUG:usb.backend.libusb1:_LibUSB.get_interface_descriptor(<usb.backend.libusb1._Device object at 0x7fab6e448730>, 0, 1, 0)
2022-03-30 08:56:41,991 DEBUG:usb.backend.libusb1:_LibUSB.get_configuration_descriptor(<usb.backend.libusb1._Device object at 0x7fab6e448730>, 0)
2022-03-30 08:56:41,991 DEBUG:usb.backend.libusb1:_LibUSB.get_endpoint_descriptor(<usb.backend.libusb1._Device object at 0x7fab6e448730>, 0, 0, 1, 0)
2022-03-30 08:56:41,991 DEBUG:usb.backend.libusb1:_LibUSB.get_interface_descriptor(<usb.backend.libusb1._Device object at 0x7fab6e448730>, 0, 1, 0)
2022-03-30 08:56:41,991 DEBUG:usb.backend.libusb1:_LibUSB.get_configuration_descriptor(<usb.backend.libusb1._Device object at 0x7fab6e448730>, 0)
2022-03-30 08:56:41,991 DEBUG:usb.backend.libusb1:_LibUSB.get_endpoint_descriptor(<usb.backend.libusb1._Device object at 0x7fab6e448730>, 1, 0, 1, 0)
2022-03-30 08:56:41,991 DEBUG:usb.backend.libusb1:_LibUSB.get_interface_descriptor(<usb.backend.libusb1._Device object at 0x7fab6e448730>, 0, 1, 0)
2022-03-30 08:56:41,991 DEBUG:usb.backend.libusb1:_LibUSB.get_configuration_descriptor(<usb.backend.libusb1._Device object at 0x7fab6e448730>, 0)
2022-03-30 08:56:41,991 DEBUG:usb.backend.libusb1:_LibUSB.get_endpoint_descriptor(<usb.backend.libusb1._Device object at 0x7fab6e448730>, 0, 0, 1, 0)
2022-03-30 08:56:41,991 DEBUG:usb.backend.libusb1:_LibUSB.get_interface_descriptor(<usb.backend.libusb1._Device object at 0x7fab6e448730>, 0, 1, 0)
2022-03-30 08:56:41,991 DEBUG:usb.backend.libusb1:_LibUSB.get_configuration_descriptor(<usb.backend.libusb1._Device object at 0x7fab6e448730>, 0)
2022-03-30 08:56:41,991 DEBUG:usb.backend.libusb1:_LibUSB.get_endpoint_descriptor(<usb.backend.libusb1._Device object at 0x7fab6e448730>, 0, 0, 1, 0)
2022-03-30 08:56:41,991 DEBUG:usb.backend.libusb1:_LibUSB.get_interface_descriptor(<usb.backend.libusb1._Device object at 0x7fab6e448730>, 0, 1, 0)
2022-03-30 08:56:41,991 DEBUG:usb.backend.libusb1:_LibUSB.get_configuration_descriptor(<usb.backend.libusb1._Device object at 0x7fab6e448730>, 0)
2022-03-30 08:56:41,991 DEBUG:usb.backend.libusb1:_LibUSB.get_endpoint_descriptor(<usb.backend.libusb1._Device object at 0x7fab6e448730>, 1, 0, 1, 0)
2022-03-30 08:56:41,991 DEBUG:usb.backend.libusb1:_LibUSB.get_interface_descriptor(<usb.backend.libusb1._Device object at 0x7fab6e448730>, 0, 1, 0)
2022-03-30 08:56:41,991 DEBUG:usb.backend.libusb1:_LibUSB.get_configuration_descriptor(<usb.backend.libusb1._Device object at 0x7fab6e448730>, 0)
2022-03-30 08:56:41,991 DEBUG:usb.backend.libusb1:_LibUSB.get_configuration_descriptor(<usb.backend.libusb1._Device object at 0x7fab6e448730>, 0)
2022-03-30 08:56:41,991 DEBUG:usb.backend.libusb1:_LibUSB.get_interface_descriptor(<usb.backend.libusb1._Device object at 0x7fab6e448730>, 0, 0, 0)
2022-03-30 08:56:41,991 DEBUG:usb.backend.libusb1:_LibUSB.get_configuration_descriptor(<usb.backend.libusb1._Device object at 0x7fab6e448730>, 0)
2022-03-30 08:56:41,991 DEBUG:usb.backend.libusb1:_LibUSB.get_interface_descriptor(<usb.backend.libusb1._Device object at 0x7fab6e448730>, 0, 1, 0)
[... cut ...]  65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65]), 1000)
```

mcuee, 2022-03-30:
Under Windows, it seems to work as well for libusbk driver, but got problem with WinUSB, still both are using isoc transfer.

Debug log for libusbk.
```

```

mcuee, 2022-03-30:
Debug log for libusbk:
```
2022-03-30 10:41:22,912 DEBUG:usb.backend.libusb1:_LibUSB.__init__(<WinDLL 'C:\work\python\py39venv\Scripts\libusb-1.0.dll', handle 7fff891a0000 at 0x1a5472c9f70>)
libusb: info [winusbx_init] WinUSB DLL available (with isoch support)
libusb: error [load_usbdk_helper_dll] Failed to load UsbDkHelper.dll: [126] The specified module could not be found.
libusb: info [windows_init] UsbDk backend is not available
libusb: warning [libusb_init] installing new context as implicit default
2022-03-30 10:41:22,921 INFO:usb.core:find(): using backend "usb.backend.libusb1"
2022-03-30 10:41:22,922 DEBUG:usb.backend.libusb1:_LibUSB.enumerate_devices()
2022-03-30 10:41:22,960 DEBUG:usb.backend.libusb1:_LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x000001A547445B80>)
2022-03-30 10:41:22,961 DEBUG:usb.backend.libusb1:_LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x000001A547445D30>)
2022-03-30 10:41:22,961 DEBUG:usb.backend.libusb1:_LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x000001A547445BE0>)
2022-03-30 10:41:22,961 DEBUG:usb.backend.libusb1:_LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x000001A547445D90>)
2022-03-30 10:41:22,961 DEBUG:usb.backend.libusb1:_LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x000001A547445C70>)
2022-03-30 10:41:22,962 DEBUG:usb.backend.libusb1:_LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x000001A547445DF0>)
2022-03-30 10:41:22,962 DEBUG:usb.backend.libusb1:_LibUSB.get_configuration_descriptor(<usb.backend.libusb1._Device object at 0x000001A547445DF0>, 0)
2022-03-30 10:41:22,962 DEBUG:usb.backend.libusb1:_LibUSB.open_device(<usb.backend.libusb1._Device object at 0x000001A547445DF0>)
2022-03-30 10:41:22,962 DEBUG:usb.backend.libusb1:_LibUSB.get_configuration_descriptor(<usb.backend.libusb1._Device object at 0x000001A547445DF0>, 0)
2022-03-30 10:41:22,963 DEBUG:usb.backend.libusb1:_LibUSB.get_configuration_descriptor(<usb.backend.libusb1._Device object at 0x000001A547445DF0>, 0)
2022-03-30 10:41:22,963 DEBUG:usb.backend.libusb1:_LibUSB.get_interface_descriptor(<usb.backend.libusb1._Device object at 0x000001A547445DF0>, 0, 0, 0)
2022-03-30 10:41:22,963 DEBUG:usb.backend.libusb1:_LibUSB.get_configuration_descriptor(<usb.backend.libusb1._Device object at 0x000001A547445DF0>, 0)
2022-03-30 10:41:22,963 DEBUG:usb.backend.libusb1:_LibUSB.get_interface_descriptor(<usb.backend.libusb1._Device object at 0x000001A547445DF0>, 0, 1, 0)
2022-03-30 10:41:22,963 DEBUG:usb.backend.libusb1:_LibUSB.get_configuration_descriptor(<usb.backend.libusb1._Device object at 0x000001A547445DF0>, 0)
2022-03-30 10:41:22,964 DEBUG:usb.backend.libusb1:_LibUSB.claim_interface(<usb.backend.libusb1._DeviceHandle object at 0x000001A547387D00>, 0)
2022-03-30 10:41:22,964 DEBUG:usb.backend.libusb1:_LibUSB.set_interface_altsetting(<usb.backend.libusb1._DeviceHandle object at 0x000001A547387D00>, 0, 1)
2022-03-30 10:41:22,965 DEBUG:usb.backend.libusb1:_LibUSB.get_interface_descriptor(<usb.backend.libusb1._Device object at 0x000001A547445DF0>, 0, 1, 0)
2022-03-30 10:41:22,965 DEBUG:usb.backend.libusb1:_LibUSB.get_configuration_descriptor(<usb.backend.libusb1._Device object at 0x000001A547445DF0>, 0)
2022-03-30 10:41:22,965 DEBUG:usb.backend.libusb1:_LibUSB.get_endpoint_descriptor(<usb.backend.libusb1._Device object at 0x000001A547445DF0>, 0, 0, 1, 0)
2022-03-30 10:41:22,965 DEBUG:usb.backend.libusb1:_LibUSB.get_interface_descriptor(<usb.backend.libusb1._Device object at 0x000001A547445DF0>, 0, 1, 0)
2022-03-30 10:41:22,966 DEBUG:usb.backend.libusb1:_LibUSB.get_configuration_descriptor(<usb.backend.libusb1._Device object at 0x000001A547445DF0>, 0)
2022-03-30 10:41:22,966 DEBUG:usb.backend.libusb1:_LibUSB.get_endpoint_descriptor(<usb.backend.libusb1._Device object at 0x000001A547445DF0>, 1, 0, 1, 0)
2022-03-30 10:41:22,966 DEBUG:usb.backend.libusb1:_LibUSB.get_interface_descriptor(<usb.backend.libusb1._Device object at 0x000001A547445DF0>, 0, 1, 0)
2022-03-30 10:41:22,966 DEBUG:usb.backend.libusb1:_LibUSB.get_configuration_descriptor(<usb.backend.libusb1._Device object at 0x000001A547445DF0>, 0)
2022-03-30 10:41:22,966 DEBUG:usb.backend.libusb1:_LibUSB.get_endpoint_descriptor(<usb.backend.libusb1._Device object at 0x000001A547445DF0>, 0, 0, 1, 0)
2022-03-30 10:41:22,967 DEBUG:usb.backend.libusb1:_LibUSB.get_interface_descriptor(<usb.backend.libusb1._Device object at 0x000001A547445DF0>, 0, 1, 0)
2022-03-30 10:41:22,967 DEBUG:usb.backend.libusb1:_LibUSB.get_configuration_descriptor(<usb.backend.libusb1._Device object at 0x000001A547445DF0>, 0)
2022-03-30 10:41:22,967 DEBUG:usb.backend.libusb1:_LibUSB.get_endpoint_descriptor(<usb.backend.libusb1._Device object at 0x000001A547445DF0>, 0, 0, 1, 0)
2022-03-30 10:41:22,967 DEBUG:usb.backend.libusb1:_LibUSB.get_interface_descriptor(<usb.backend.libusb1._Device object at 0x000001A547445DF0>, 0, 1, 0)
2022-03-30 10:41:22,967 DEBUG:usb.backend.libusb1:_LibUSB.get_configuration_descriptor(<usb.backend.libusb1._Device object at 0x000001A547445DF0>, 0)
2022-03-30 10:41:22,967 DEBUG:usb.backend.libusb1:_LibUSB.get_endpoint_descriptor(<usb.backend.libusb1._Device object at 0x000001A547445DF0>, 1, 0, 1, 0)
2022-03-30 10:41:22,968 DEBUG:usb.backend.libusb1:_LibUSB.get_interface_descriptor(<usb.backend.libusb1._Device object at 0x000001A547445DF0>, 0, 1, 0)
[... cut ...]  65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65]), 1000)
...
```

jonasmalacofilho, 2022-03-30:
@tormodvolden ,

> Something with self._ep_info going stale? Maybe do ._ep_info.clear() in managed_set_interface() ?

I think that can also happen, but it's probably not the case here, since no reads or writes were done before the altsetting change.

---

@mcuee,

Thanks, I'll take a look.

mcuee, 2022-03-30:
Windows libusbk debug log for the FX2LP device running Cystream FW.

```
2022-03-30 11:53:58,201 DEBUG:usb.backend.libusb1:_LibUSB.__init__(<WinDLL 'C:\work\python\py39venv\Scripts\libusb-1.0.dll', handle 7fff86bb0000 at 0x19a2b4e9f70>)
libusb: info [winusbx_init] WinUSB DLL available (with isoch support)
libusb: error [load_usbdk_helper_dll] Failed to load UsbDkHelper.dll: [126] The specified module could not be found.
libusb: info [windows_init] UsbDk backend is not available
libusb: warning [libusb_init] installing new context as implicit default
2022-03-30 11:53:58,209 INFO:usb.core:find(): using backend "usb.backend.libusb1"
2022-03-30 11:53:58,210 DEBUG:usb.backend.libusb1:_LibUSB.enumerate_devices()
2022-03-30 11:53:58,250 DEBUG:usb.backend.libusb1:_LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x0000019A2B664BB0>)
2022-03-30 11:53:58,250 DEBUG:usb.backend.libusb1:_LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x0000019A2B664D60>)
2022-03-30 11:53:58,251 DEBUG:usb.backend.libusb1:_LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x0000019A2B664C10>)
2022-03-30 11:53:58,251 DEBUG:usb.backend.libusb1:_LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x0000019A2B664DC0>)
2022-03-30 11:53:58,251 DEBUG:usb.backend.libusb1:_LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x0000019A2B664CA0>)
2022-03-30 11:53:58,251 DEBUG:usb.backend.libusb1:_LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x0000019A2B664E20>)
2022-03-30 11:53:58,252 DEBUG:usb.backend.libusb1:_LibUSB.get_configuration_descriptor(<usb.backend.libusb1._Device object at 0x0000019A2B664E20>, 0)
2022-03-30 11:53:58,252 DEBUG:usb.backend.libusb1:_LibUSB.open_device(<usb.backend.libusb1._Device object at 0x0000019A2B664E20>)
2022-03-30 11:53:58,253 DEBUG:usb.backend.libusb1:_LibUSB.get_configuration_descriptor(<usb.backend.libusb1._Device object at 0x0000019A2B664E20>, 0)
2022-03-30 11:53:58,253 DEBUG:usb.backend.libusb1:_LibUSB.get_configuration_descriptor(<usb.backend.libusb1._Device object at 0x0000019A2B664E20>, 0)
2022-03-30 11:53:58,253 DEBUG:usb.backend.libusb1:_LibUSB.get_interface_descriptor(<usb.backend.libusb1._Device object at 0x0000019A2B664E20>, 0, 0, 0)
2022-03-30 11:53:58,254 DEBUG:usb.backend.libusb1:_LibUSB.get_configuration_descriptor(<usb.backend.libusb1._Device object at 0x0000019A2B664E20>, 0)
2022-03-30 11:53:58,254 DEBUG:usb.backend.libusb1:_LibUSB.get_interface_descriptor(<usb.backend.libusb1._Device object at 0x0000019A2B664E20>, 0, 1, 0)
2022-03-30 11:53:58,254 DEBUG:usb.backend.libusb1:_LibUSB.get_configuration_descriptor(<usb.backend.libusb1._Device object at 0x0000019A2B664E20>, 0)
2022-03-30 11:53:58,255 DEBUG:usb.backend.libusb1:_LibUSB.get_interface_descriptor(<usb.backend.libusb1._Device object at 0x0000019A2B664E20>, 0, 2, 0)
2022-03-30 11:53:58,255 DEBUG:usb.backend.libusb1:_LibUSB.get_configuration_descriptor(<usb.backend.libusb1._Device object at 0x0000019A2B664E20>, 0)
2022-03-30 11:53:58,255 DEBUG:usb.backend.libusb1:_LibUSB.get_interface_descriptor(<usb.backend.libusb1._Device object at 0x0000019A2B664E20>, 0, 3, 0)
2022-03-30 11:53:58,256 DEBUG:usb.backend.libusb1:_LibUSB.get_configuration_descriptor(<usb.backend.libusb1._Device object at 0x0000019A2B664E20>, 0)
2022-03-30 11:53:58,256 DEBUG:usb.backend.libusb1:_LibUSB.get_interface_descriptor(<usb.backend.libusb1._Device object at 0x0000019A2B664E20>, 0, 4, 0)
2022-03-30 11:53:58,256 DEBUG:usb.backend.libusb1:_LibUSB.get_configuration_descriptor(<usb.backend.libusb1._Device object at 0x0000019A2B664E20>, 0)
2022-03-30 11:53:58,257 DEBUG:usb.backend.libusb1:_LibUSB.get_interface_descriptor(<usb.backend.libusb1._Device object at 0x0000019A2B664E20>, 0, 5, 0)
2022-03-30 11:53:58,257 DEBUG:usb.backend.libusb1:_LibUSB.get_configuration_descriptor(<usb.backend.libusb1._Device object at 0x0000019A2B664E20>, 0)
2022-03-30 11:53:58,257 DEBUG:usb.backend.libusb1:_LibUSB.get_interface_descriptor(<usb.backend.libusb1._Device object at 0x0000019A2B664E20>, 0, 6, 0)
2022-03-30 11:53:58,257 DEBUG:usb.backend.libusb1:_LibUSB.get_configuration_descriptor(<usb.backend.libusb1._Device object at 0x0000019A2B664E20>, 0)
2022-03-30 11:53:58,258 DEBUG:usb.backend.libusb1:_LibUSB.claim_interface(<usb.backend.libusb1._DeviceHandle object at 0x0000019A2B5A6D30>, 0)
2022-03-30 11:53:58,258 DEBUG:usb.backend.libusb1:_LibUSB.set_interface_altsetting(<usb.backend.libusb1._DeviceHandle object at 0x0000019A2B5A6D30>, 0, 6)
2022-03-30 11:53:58,259 DEBUG:usb.backend.libusb1:_LibUSB.get_interface_descriptor(<usb.backend.libusb1._Device object at 0x0000019A2B664E20>, 0, 6, 0)
2022-03-30 11:53:58,259 DEBUG:usb.backend.libusb1:_LibUSB.get_configuration_descriptor(<usb.backend.libusb1._Device object at 0x0000019A2B664E20>, 0)
2022-03-30 11:53:58,259 DEBUG:usb.backend.libusb1:_LibUSB.get_endpoint_descriptor(<usb.backend.libusb1._Device object at 0x0000019A2B664E20>, 0, 0, 6, 0)
2022-03-30 11:53:58,259 DEBUG:usb.backend.libusb1:_LibUSB.get_interface_descriptor(<usb.backend.libusb1._Device object at 0x0000019A2B664E20>, 0, 6, 0)
2022-03-30 11:53:58,259 DEBUG:usb.backend.libusb1:_LibUSB.get_configuration_descriptor(<usb.backend.libusb1._Device object at 0x0000019A2B664E20>, 0)
2022-03-30 11:53:58,259 DEBUG:usb.backend.libusb1:_LibUSB.get_endpoint_descriptor(<usb.backend.libusb1._Device object at 0x0000019A2B664E20>, 1, 0, 6, 0)
[... cut ...] 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65]), 1000)

```

mcuee, 2022-03-30:
WinUSB driver debug log.

```
2022-03-30 11:56:29,222 DEBUG:usb.backend.libusb1:_LibUSB.__init__(<WinDLL 'C:\work\python\py39venv\Scripts\libusb-1.0.dll', handle 7fff71b60000 at 0x1ceaea79f70>)
libusb: info [winusbx_init] WinUSB DLL available (with isoch support)
libusb: error [load_usbdk_helper_dll] Failed to load UsbDkHelper.dll: [126] The specified module could not be found.
libusb: info [windows_init] UsbDk backend is not available
libusb: warning [libusb_init] installing new context as implicit default
2022-03-30 11:56:29,232 INFO:usb.core:find(): using backend "usb.backend.libusb1"
2022-03-30 11:56:29,232 DEBUG:usb.backend.libusb1:_LibUSB.enumerate_devices()
2022-03-30 11:56:29,270 DEBUG:usb.backend.libusb1:_LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x000001CEAEC04BB0>)
2022-03-30 11:56:29,270 DEBUG:usb.backend.libusb1:_LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x000001CEAEC04D60>)
2022-03-30 11:56:29,271 DEBUG:usb.backend.libusb1:_LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x000001CEAEC04C10>)
2022-03-30 11:56:29,271 DEBUG:usb.backend.libusb1:_LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x000001CEAEC04DC0>)
2022-03-30 11:56:29,271 DEBUG:usb.backend.libusb1:_LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x000001CEAEC04CA0>)
2022-03-30 11:56:29,271 DEBUG:usb.backend.libusb1:_LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x000001CEAEC04E20>)
2022-03-30 11:56:29,271 DEBUG:usb.backend.libusb1:_LibUSB.get_configuration_descriptor(<usb.backend.libusb1._Device object at 0x000001CEAEC04E20>, 0)
2022-03-30 11:56:29,272 DEBUG:usb.backend.libusb1:_LibUSB.open_device(<usb.backend.libusb1._Device object at 0x000001CEAEC04E20>)
2022-03-30 11:56:29,272 DEBUG:usb.backend.libusb1:_LibUSB.get_configuration_descriptor(<usb.backend.libusb1._Device object at 0x000001CEAEC04E20>, 0)
2022-03-30 11:56:29,272 DEBUG:usb.backend.libusb1:_LibUSB.get_configuration_descriptor(<usb.backend.libusb1._Device object at 0x000001CEAEC04E20>, 0)
2022-03-30 11:56:29,272 DEBUG:usb.backend.libusb1:_LibUSB.get_interface_descriptor(<usb.backend.libusb1._Device object at 0x000001CEAEC04E20>, 0, 0, 0)
2022-03-30 11:56:29,273 DEBUG:usb.backend.libusb1:_LibUSB.get_configuration_descriptor(<usb.backend.libusb1._Device object at 0x000001CEAEC04E20>, 0)
2022-03-30 11:56:29,273 DEBUG:usb.backend.libusb1:_LibUSB.get_interface_descriptor(<usb.backend.libusb1._Device object at 0x000001CEAEC04E20>, 0, 1, 0)
2022-03-30 11:56:29,273 DEBUG:usb.backend.libusb1:_LibUSB.get_configuration_descriptor(<usb.backend.libusb1._Device object at 0x000001CEAEC04E20>, 0)
2022-03-30 11:56:29,273 DEBUG:usb.backend.libusb1:_LibUSB.get_interface_descriptor(<usb.backend.libusb1._Device object at 0x000001CEAEC04E20>, 0, 2, 0)
2022-03-30 11:56:29,273 DEBUG:usb.backend.libusb1:_LibUSB.get_configuration_descriptor(<usb.backend.libusb1._Device object at 0x000001CEAEC04E20>, 0)
2022-03-30 11:56:29,274 DEBUG:usb.backend.libusb1:_LibUSB.get_interface_descriptor(<usb.backend.libusb1._Device object at 0x000001CEAEC04E20>, 0, 3, 0)
2022-03-30 11:56:29,274 DEBUG:usb.backend.libusb1:_LibUSB.get_configuration_descriptor(<usb.backend.libusb1._Device object at 0x000001CEAEC04E20>, 0)
2022-03-30 11:56:29,274 DEBUG:usb.backend.libusb1:_LibUSB.get_interface_descriptor(<usb.backend.libusb1._Device object at 0x000001CEAEC04E20>, 0, 4, 0)
2022-03-30 11:56:29,274 DEBUG:usb.backend.libusb1:_LibUSB.get_configuration_descriptor(<usb.backend.libusb1._Device object at 0x000001CEAEC04E20>, 0)
2022-03-30 11:56:29,274 DEBUG:usb.backend.libusb1:_LibUSB.get_interface_descriptor(<usb.backend.libusb1._Device object at 0x000001CEAEC04E20>, 0, 5, 0)
2022-03-30 11:56:29,275 DEBUG:usb.backend.libusb1:_LibUSB.get_configuration_descriptor(<usb.backend.libusb1._Device object at 0x000001CEAEC04E20>, 0)
2022-03-30 11:56:29,275 DEBUG:usb.backend.libusb1:_LibUSB.get_interface_descriptor(<usb.backend.libusb1._Device object at 0x000001CEAEC04E20>, 0, 6, 0)
2022-03-30 11:56:29,275 DEBUG:usb.backend.libusb1:_LibUSB.get_configuration_descriptor(<usb.backend.libusb1._Device object at 0x000001CEAEC04E20>, 0)
2022-03-30 11:56:29,275 DEBUG:usb.backend.libusb1:_LibUSB.claim_interface(<usb.backend.libusb1._DeviceHandle object at 0x000001CEAEB37D30>, 0)
2022-03-30 11:56:29,276 DEBUG:usb.backend.libusb1:_LibUSB.set_interface_altsetting(<usb.backend.libusb1._DeviceHandle object at 0x000001CEAEB37D30>, 0, 6)
2022-03-30 11:56:29,276 DEBUG:usb.backend.libusb1:_LibUSB.get_interface_descriptor(<usb.backend.libusb1._Device object at 0x000001CEAEC04E20>, 0, 6, 0)
2022-03-30 11:56:29,277 DEBUG:usb.backend.libusb1:_LibUSB.get_configuration_descriptor(<usb.backend.libusb1._Device object at 0x000001CEAEC04E20>, 0)
2022-03-30 11:56:29,277 DEBUG:usb.backend.libusb1:_LibUSB.get_endpoint_descriptor(<usb.backend.libusb1._Device object at 0x000001CEAEC04E20>, 0, 0, 6, 0)
2022-03-30 11:56:29,277 DEBUG:usb.backend.libusb1:_LibUSB.get_interface_descriptor(<usb.backend.libusb1._Device object at 0x000001CEAEC04E20>, 0, 6, 0)
2022-03-30 11:56:29,277 DEBUG:usb.backend.libusb1:_LibUSB.get_configuration_descriptor(<usb.backend.libusb1._Device object at 0x000001CEAEC04E20>, 0)
2022-03-30 11:56:29,277 DEBUG:usb.backend.libusb1:_LibUSB.get_endpoint_descriptor(<usb.backend.libusb1._Device object at 0x000001CEAEC04E20>, 1, 0, 6, 0)
[... cut ...]
       bEndpointAddress :   0x82 IN
       bmAttributes     :    0x1 Isochronous
       wMaxPacketSize   :  0x400 (1024 bytes)
       bInterval        :    0x1
      ENDPOINT 0x6: Isochronous OUT ========================
       bLength          :    0x7 (7 bytes)
       bDescriptorType  :    0x5 Endpoint
       bEndpointAddress :    0x6 OUT
       bmAttributes     :    0x1 Isochronous
       wMaxPacketSize   :  0x400 (1024 bytes)
       bInterval        :    0x1
Starting CYUSB FX3 ISOC Loopbacktest

```

mcuee, 2022-03-30:
Device Descriptor of the Cystream FW for the Cypress EZUSB FX2LP.

```
<DEVICE>
	FriendlyName="Cypress FX2LP StreamerExample Device"
	Manufacturer="Cypress"
	Product="CY-Stream"
	SerialNumber=""
	Configurations="1"
	MaxPacketSize="64"
	VendorID="04 B4"
	ProductID="10 03"
	Class="00h"
	SubClass="00h"
	Protocol="00h"
	BcdDevice="00 00"
	BcdUSB="02 00"
	<CONFIGURATION>
		Configuration="0"
		ConfigurationValue="1"
		Attributes="A0h"
		Interfaces="1"
		DescriptorType="2"
		DescriptorLength="9"
		TotalLength="135"
		MaxPower="50"
		<INTERFACE>
			Interface="0"
			InterfaceNumber="0"
			AltSetting="0"
			Class="FFh"
			Subclass="00h"
			Protocol="0"
			Endpoints="1"
			DescriptorType="4"
			DescriptorLength="9"
			<ENDPOINT>
				Type="BULK"
				Direction="IN"
				Address="82h"
				Attributes="02h"
				MaxPktSize="512"
				DescriptorType="5"
				DescriptorLength="7"
				Interval="0"
			</ENDPOINT>
		</INTERFACE>
		<INTERFACE>
			Interface="0"
			InterfaceNumber="0"
			AltSetting="1"
			Class="FFh"
			Subclass="00h"
			Protocol="0"
			Endpoints="1"
			DescriptorType="4"
			DescriptorLength="9"
			<ENDPOINT>
				Type="BULK"
				Direction="OUT"
				Address="02h"
				Attributes="02h"
				MaxPktSize="512"
				DescriptorType="5"
				DescriptorLength="7"
				Interval="0"
			</ENDPOINT>
		</INTERFACE>
		<INTERFACE>
			Interface="0"
			InterfaceNumber="0"
			AltSetting="2"
			Class="FFh"
			Subclass="00h"
			Protocol="0"
			Endpoints="2"
			DescriptorType="4"
			DescriptorLength="9"
			<ENDPOINT>
				Type="BULK"
				Direction="IN"
				Address="82h"
				Attributes="02h"
				MaxPktSize="512"
				DescriptorType="5"
				DescriptorLength="7"
				Interval="0"
			</ENDPOINT>
			<ENDPOINT>
				Type="BULK"
				Direction="OUT"
				Address="06h"
				Attributes="02h"
				MaxPktSize="512"
				DescriptorType="5"
				DescriptorLength="7"
				Interval="0"
			</ENDPOINT>
		</INTERFACE>
		<INTERFACE>
			Interface="0"
			InterfaceNumber="0"
			AltSetting="3"
			Class="FFh"
			Subclass="00h"
			Protocol="0"
			Endpoints="1"
			DescriptorType="4"
			DescriptorLength="9"
			<ENDPOINT>
				Type="ISOC"
				Direction="IN"
				Address="82h"
				Attributes="01h"
				MaxPktSize="3072"
				DescriptorType="5"
				DescriptorLength="7"
				Interval="1"
			</ENDPOINT>
		</INTERFACE>
		<INTERFACE>
			Interface="0"
			InterfaceNumber="0"
			AltSetting="4"
			Class="FFh"
			Subclass="00h"
			Protocol="0"
			Endpoints="1"
			DescriptorType="4"
			DescriptorLength="9"
			<ENDPOINT>
				Type="ISOC"
				Direction="OUT"
				Address="02h"
				Attributes="01h"
				MaxPktSize="3072"
				DescriptorType="5"
				DescriptorLength="7"
				Interval="1"
			</ENDPOINT>
		</INTERFACE>
		<INTERFACE>
			Interface="0"
			InterfaceNumber="0"
			AltSetting="5"
			Class="FFh"
			Subclass="00h"
			Protocol="0"
			Endpoints="1"
			DescriptorType="4"
			DescriptorLength="9"
			<ENDPOINT>
				Type="ISOC"
				Direction="IN"
				Address="82h"
				Attributes="01h"
				MaxPktSize="1024"
				DescriptorType="5"
				DescriptorLength="7"
				Interval="1"
			</ENDPOINT>
		</INTERFACE>
		<INTERFACE>
			Interface="0"
			InterfaceNumber="0"
			AltSetting="6"
			Class="FFh"
			Subclass="00h"
			Protocol="0"
			Endpoints="2"
			DescriptorType="4"
			DescriptorLength="9"
			<ENDPOINT>
				Type="ISOC"
				Direction="IN"
				Address="82h"
				Attributes="01h"
				MaxPktSize="1024"
				DescriptorType="5"
				DescriptorLength="7"
				Interval="1"
			</ENDPOINT>
			<ENDPOINT>
				Type="ISOC"
				Direction="OUT"
				Address="06h"
				Attributes="01h"
				MaxPktSize="1024"
				DescriptorType="5"
				DescriptorLength="7"
				Interval="1"
			</ENDPOINT>
		</INTERFACE>
	</CONFIGURATION>
</DEVICE>

```

mcuee, 2022-03-30:
There seems to be no issue with WinUSB or libusbk if I test alt interface 2 (bulk endpoints).
```
2022-03-30 12:04:01,266 DEBUG:usb.backend.libusb1:_LibUSB.__init__(<WinDLL 'C:\work\python\py39venv\Scripts\libusb-1.0.dll', handle 7fff891d0000 at 0x1d64cc39f70>)
libusb: info [winusbx_init] WinUSB DLL available (with isoch support)
libusb: error [load_usbdk_helper_dll] Failed to load UsbDkHelper.dll: [126] The specified module could not be found.
libusb: info [windows_init] UsbDk backend is not available
libusb: warning [libusb_init] installing new context as implicit default
2022-03-30 12:04:01,273 INFO:usb.core:find(): using backend "usb.backend.libusb1"
2022-03-30 12:04:01,273 DEBUG:usb.backend.libusb1:_LibUSB.enumerate_devices()
2022-03-30 12:04:01,299 DEBUG:usb.backend.libusb1:_LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x000001D64E765B80>)
2022-03-30 12:04:01,300 DEBUG:usb.backend.libusb1:_LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x000001D64E765D30>)
2022-03-30 12:04:01,300 DEBUG:usb.backend.libusb1:_LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x000001D64E765BE0>)
2022-03-30 12:04:01,300 DEBUG:usb.backend.libusb1:_LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x000001D64E765D90>)
2022-03-30 12:04:01,300 DEBUG:usb.backend.libusb1:_LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x000001D64E765C70>)
2022-03-30 12:04:01,300 DEBUG:usb.backend.libusb1:_LibUSB.get_device_descriptor(<usb.backend.libusb1._Device object at 0x000001D64E765DF0>)
2022-03-30 12:04:01,300 DEBUG:usb.backend.libusb1:_LibUSB.get_configuration_descriptor(<usb.backend.libusb1._Device object at 0x000001D64E765DF0>, 0)
2022-03-30 12:04:01,301 DEBUG:usb.backend.libusb1:_LibUSB.open_device(<usb.backend.libusb1._Device object at 0x000001D64E765DF0>)
2022-03-30 12:04:01,301 DEBUG:usb.backend.libusb1:_LibUSB.get_configuration_descriptor(<usb.backend.libusb1._Device object at 0x000001D64E765DF0>, 0)
2022-03-30 12:04:01,301 DEBUG:usb.backend.libusb1:_LibUSB.get_configuration_descriptor(<usb.backend.libusb1._Device object at 0x000001D64E765DF0>, 0)
2022-03-30 12:04:01,301 DEBUG:usb.backend.libusb1:_LibUSB.get_interface_descriptor(<usb.backend.libusb1._Device object at 0x000001D64E765DF0>, 0, 0, 0)
2022-03-30 12:04:01,301 DEBUG:usb.backend.libusb1:_LibUSB.get_configuration_descriptor(<usb.backend.libusb1._Device object at 0x000001D64E765DF0>, 0)
2022-03-30 12:04:01,301 DEBUG:usb.backend.libusb1:_LibUSB.get_interface_descriptor(<usb.backend.libusb1._Device object at 0x000001D64E765DF0>, 0, 1, 0)
2022-03-30 12:04:01,301 DEBUG:usb.backend.libusb1:_LibUSB.get_configuration_descriptor(<usb.backend.libusb1._Device object at 0x000001D64E765DF0>, 0)
2022-03-30 12:04:01,301 DEBUG:usb.backend.libusb1:_LibUSB.get_interface_descriptor(<usb.backend.libusb1._Device object at 0x000001D64E765DF0>, 0, 2, 0)
2022-03-30 12:04:01,302 DEBUG:usb.backend.libusb1:_LibUSB.get_configuration_descriptor(<usb.backend.libusb1._Device object at 0x000001D64E765DF0>, 0)
2022-03-30 12:04:01,302 DEBUG:usb.backend.libusb1:_LibUSB.claim_interface(<usb.backend.libusb1._DeviceHandle object at 0x000001D64CCF7D00>, 0)
2022-03-30 12:04:01,302 DEBUG:usb.backend.libusb1:_LibUSB.set_interface_altsetting(<usb.backend.libusb1._DeviceHandle object at 0x000001D64CCF7D00>, 0, 2)
2022-03-30 12:04:01,303 DEBUG:usb.backend.libusb1:_LibUSB.get_interface_descriptor(<usb.backend.libusb1._Device object at 0x000001D64E765DF0>, 0, 2, 0)
2022-03-30 12:04:01,303 DEBUG:usb.backend.libusb1:_LibUSB.get_configuration_descriptor(<usb.backend.libusb1._Device object at 0x000001D64E765DF0>, 0)
2022-03-30 12:04:01,303 DEBUG:usb.backend.libusb1:_LibUSB.get_endpoint_descriptor(<usb.backend.libusb1._Device object at 0x000001D64E765DF0>, 0, 0, 2, 0)
2022-03-30 12:04:01,303 DEBUG:usb.backend.libusb1:_LibUSB.get_interface_descriptor(<usb.backend.libusb1._Device object at 0x000001D64E765DF0>, 0, 2, 0)
2022-03-30 12:04:01,303 DEBUG:usb.backend.libusb1:_LibUSB.get_configuration_descriptor(<usb.backend.libusb1._Device object at 0x000001D64E765DF0>, 0)
2022-03-30 12:04:01,303 DEBUG:usb.backend.libusb1:_LibUSB.get_endpoint_descriptor(<usb.backend.libusb1._Device object at 0x000001D64E765DF0>, 1, 0, 2, 0)
2022-03-30 12:04:01,303 DEBUG:usb.backend.libusb1:_LibUSB.get_interface_descriptor(<usb.backend.libusb1._Device object at 0x000001D64E765DF0>, 0, 2, 0)
2022-03-30 12:04:01,303 DEBUG:usb.backend.libusb1:_LibUSB.get_configuration_descriptor(<usb.backend.libusb1._Device object at 0x000001D64E765DF0>, 0)
2022-03-30 12:04:01,303 DEBUG:usb.backend.libusb1:_LibUSB.get_endpoint_descriptor(<usb.backend.libusb1._Device object at 0x000001D64E765DF0>, 0, 0, 2, 0)
2022-03-30 12:04:01,303 DEBUG:usb.backend.libusb1:_LibUSB.get_interface_descriptor(<usb.backend.libusb1._Device object at 0x000001D64E765DF0>, 0, 2, 0)
2022-03-30 12:04:01,304 DEBUG:usb.backend.libusb1:_LibUSB.get_configuration_descriptor(<usb.backend.libusb1._Device object at 0x000001D64E765DF0>, 0)
2022-03-30 12:04:01,304 DEBUG:usb.backend.libusb1:_LibUSB.get_endpoint_descriptor(<usb.backend.libusb1._Device object at 0x000001D64E765DF0>, 1, 0, 2, 0)
2022-03-30 12:04:01,304 DEBUG:usb.backend.libusb1:_LibUSB.get_interface_descriptor(<usb.backend.libusb1._Device object at 0x000001D64E765DF0>, 0, 2, 0)
2022-03-30 12:04:01,304 DEBUG:usb.backend.libusb1:_LibUSB.get_configuration_descriptor(<usb.backend.libusb1._Device object at 0x000001D64E765DF0>, 0)
[... cut ...] 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65, 65]), 1000)

```

mcuee, 2022-03-30:
I have deleted the SuperSpeed debug log which points to libusb WinUSB issue as mentioned in libusb/libusb#1104. 

The pyusb issue seems to be with the CyStream FW for FX2LP.

jonasmalacofilho, 2022-03-30:
@mcuee, I haven't finished going over all of your logs, but can you test the code in the [`staging`](https://github.com/pyusb/pyusb/tree/staging) branch (currently at 302eb7e9950c6726abcb6bb7e1f1bfbfcb9560aa)?  It has a couple of fixes related to this.

mcuee, 2022-03-30:
@jonasmalacofilho Yes it now submits isoc transfer. 
For libusbk it always crashes. For WinUSB the log is different but also got problems.
[winusb_staging.zip](https://github.com/pyusb/pyusb/files/8377769/winusb_staging.zip)
[libusbk_staging.zip](https://github.com/pyusb/pyusb/files/8377771/libusbk_staging.zip)

jonasmalacofilho, 2022-03-31:
Now there seem to be a few different errors:

- the reads fail somewhere deep in libusb with errcode 87 (WinUSB) or 31 (libusbk); might it be caused by your transfer size (2**14) being to too large for Windows (and Linux)?[^1]
- the writes with WinUSB seem to also fail, again deep in libusb, due to packets not being "contiguous in memory", but PyUSB literally sends an array, so I'm not sure what else that error might mean; could it be that we need to pad the buffer to match the number of iso packets _exactly_ before passing it to `libusb_fill_iso_transfer()`?[^2]
- we also seem to have a new finalizer bug, at least when something else goes wrong.

Can you take a look at the first two?

In the meantime, I'll look into the new finalizer bug.

[^1]: https://libusb.sourceforge.io/api-1.0/group__libusb__asyncio.html#asynclimits
[^2]: https://github.com/pyusb/pyusb/blob/1caac015d1d62d4097eac00aa3ce244a01ebe5c3/usb/backend/libusb1.py#L656-L674

mcuee, 2022-03-31:
The reason to use 2**14 (16384) is that WinUSB backend somehow requires the size to be at least 8 times wht MaxPacketSize of the isoc endpoint, so it will be at least 8192 (2**13). I will try the staging git branch under Linux first to see how it goes under Linux. I can also play with the transfer size to see if that is the problem.

mcuee, 2022-03-31:
The staging branch seems to work fine under Ubuntu 20.04 Linux.
```
(mypy38venv) mcuee@UbuntuSwift3:~/build/python/python_usb$ python cyusb_fx2_isoc_loopback_python3.py 
    INTERFACE 0, 6: Vendor Specific ========================
     bLength            :    0x9 (9 bytes)
     bDescriptorType    :    0x4 Interface
     bInterfaceNumber   :    0x0
     bAlternateSetting  :    0x6
     bNumEndpoints      :    0x2
     bInterfaceClass    :   0xff Vendor Specific
     bInterfaceSubClass :    0x0
     bInterfaceProtocol :    0x0
     iInterface         :    0x0 
      ENDPOINT 0x82: Isochronous IN ========================
       bLength          :    0x7 (7 bytes)
       bDescriptorType  :    0x5 Endpoint
       bEndpointAddress :   0x82 IN
       bmAttributes     :    0x1 Isochronous
       wMaxPacketSize   :  0x400 (1024 bytes)
       bInterval        :    0x1
      ENDPOINT 0x6: Isochronous OUT ========================
       bLength          :    0x7 (7 bytes)
       bDescriptorType  :    0x5 Endpoint
       bEndpointAddress :    0x6 OUT
       bmAttributes     :    0x1 Isochronous
       wMaxPacketSize   :  0x400 (1024 bytes)
       bInterval        :    0x1
ep:       ENDPOINT 0x6: Isochronous OUT ========================
       bLength          :    0x7 (7 bytes)
       bDescriptorType  :    0x5 Endpoint
       bEndpointAddress :    0x6 OUT
       bmAttributes     :    0x1 Isochronous
       wMaxPacketSize   :  0x400 (1024 bytes)
       bInterval        :    0x1
ep2:       ENDPOINT 0x82: Isochronous IN ========================
       bLength          :    0x7 (7 bytes)
       bDescriptorType  :    0x5 Endpoint
       bEndpointAddress :   0x82 IN
       bmAttributes     :    0x1 Isochronous
       wMaxPacketSize   :  0x400 (1024 bytes)
       bInterval        :    0x1
Starting CYUSB FX2LP CyStream python isoc loopback test
Transfered 3623 kB/s
Transfered 3380 kB/s
Transfered 3679 kB/s
Transfered 3558 kB/s
Transfered 3394 kB/s
Transfered 3556 kB/s
^CTraceback (most recent call last):
  File "cyusb_fx2_isoc_loopback_python3.py", line 74, in <module>
    read_loop()
  File "cyusb_fx2_isoc_loopback_python3.py", line 56, in read_loop
    data = dev.read(0x82, chunkr, 1000)
  File "/home/mcuee/build/python/mypy38venv/lib/python3.8/site-packages/usb/core.py", line 1037, in read
    ret = fn(
  File "/home/mcuee/build/python/mypy38venv/lib/python3.8/site-packages/usb/_debug.py", line 62, in do_trace
    return f(*args, **named_args)
  File "/home/mcuee/build/python/mypy38venv/lib/python3.8/site-packages/usb/backend/libusb1.py", line 879, in iso_read
    return handler.submit(self.ctx)
  File "/home/mcuee/build/python/mypy38venv/lib/python3.8/site-packages/usb/backend/libusb1.py", line 684, in submit
    _check(_lib.libusb_handle_events(ctx))
KeyboardInterrupt
Traceback (most recent call last):
  File "/usr/lib/python3.8/weakref.py", line 642, in _exitfunc
    f()
  File "/usr/lib/python3.8/weakref.py", line 566, in __call__
    return info.func(*info.args, **(info.kwargs or {}))
  File "/home/mcuee/build/python/mypy38venv/lib/python3.8/site-packages/usb/_objfinalizer.py", line 102, in _do_finalize_object_ref
    obj._do_finalize_object()
  File "/home/mcuee/build/python/mypy38venv/lib/python3.8/site-packages/usb/_objfinalizer.py", line 71, in _do_finalize_object
    self._finalize_object()
  File "/home/mcuee/build/python/mypy38venv/lib/python3.8/site-packages/usb/backend/libusb1.py", line 677, in _finalize_object
    _lib.libusb_free_transfer(self.transfer)
AttributeError: '_IsoTransferHandler' object has no attribute 'transfer'

```

mcuee, 2022-03-31:
> the writes with WinUSB seem to also fail, again deep in libusb, due to packets not being "contiguous in memory", but PyUSB literally sends an array, so I'm not sure what else that error might mean; [...]

The relevant libusb codes are around here:
https://github.com/libusb/libusb/blob/master/libusb/os/windows_winusb.c#L2990
```
			// WinUSB only supports isoch transfers spanning a full USB frames. Later, we might be smarter about this
			// and allocate a temporary buffer. However, this is harder than it seems as its destruction would depend on overlapped
			// IO...
			if (transfer->dev_handle->dev->speed >= LIBUSB_SPEED_HIGH) // Microframes (125us)
				iso_transfer_size_multiple = (pipe_info_ex.MaximumBytesPerInterval * 8) / interval;
			else // Normal Frames (1ms)
				iso_transfer_size_multiple = pipe_info_ex.MaximumBytesPerInterval / interval;

			if (transfer->length % iso_transfer_size_multiple != 0) {
				usbi_err(TRANSFER_CTX(transfer), "length of isoch buffer must be a multiple of the MaximumBytesPerInterval * 8 / Interval");
				return LIBUSB_ERROR_INVALID_PARAM;
			}
		} else {
			// If this is an OUT transfer, we make sure the isoch packets are contiguous as this isn't supported otherwise.
			bool size_should_be_zero = false;

			for (idx = 0; idx < transfer->num_iso_packets; ++idx) {
				if ((size_should_be_zero && transfer->iso_packet_desc[idx].length != 0) ||
					(transfer->iso_packet_desc[idx].length != pipe_info_ex.MaximumBytesPerInterval && idx + 1 < transfer->num_iso_packets && transfer->iso_packet_desc[idx + 1].length > 0)) {
					usbi_err(TRANSFER_CTX(transfer), "isoch packets for OUT transfer with WinUSB must be contiguous in memory");
					return LIBUSB_ERROR_INVALID_PARAM;
				}

				size_should_be_zero = (transfer->iso_packet_desc[idx].length == 0);
				out_transfer_length += transfer->iso_packet_desc[idx].length;
			}
		}

```

mcuee, 2022-03-31:
I am not an expert here -- isochronous transfer is hard.
The relevant Microsoft documentation is here.
https://docs.microsoft.com/en-us/windows-hardware/drivers/usbcon/getting-set-up-to-use-windows-devices-usb

@tormodvolden -- please chime in here as well if you can. Thanks.

mcuee, 2022-03-31:
I think I can reproduce the issue with C based application, a quick hack of the libusb example.

```
/*
 * libusb example program to measure Atmel SAM3U isochronous performance
 * Copyright (C) 2012 Harald Welte <laforge@gnumonks.org>
 *
 * Copied with the author's permission under LGPL-2.1 from
 * http://git.gnumonks.org/cgi-bin/gitweb.cgi?p=sam3u-tests.git;a=blob;f=usb-benchmark-project/host/benchmark.c;h=74959f7ee88f1597286cd435f312a8ff52c56b7e
 *
 * An Atmel SAM3U test firmware is also available in the above repository.
 *
 * This library is free software; you can redistribute it and/or
 * modify it under the terms of the GNU Lesser General Public
 * License as published by the Free Software Foundation; either
 * version 2.1 of the License, or (at your option) any later version.
 *
 * This library is distributed in the hope that it will be useful,
 * but WITHOUT ANY WARRANTY; without even the implied warranty of
 * MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the GNU
 * Lesser General Public License for more details.
 *
 * You should have received a copy of the GNU Lesser General Public
 * License along with this library; if not, write to the Free Software
 * Foundation, Inc., 51 Franklin Street, Fifth Floor, Boston, MA 02110-1301 USA
 */

#include <config.h>

#include <errno.h>
#include <signal.h>
#include <stdio.h>
#include <stdlib.h>
#ifdef HAVE_SYS_TIME_H
#include <sys/time.h>
#endif
#include <time.h>

#include "libusb.h"

#define EP_ISO_IN	0x82

static volatile sig_atomic_t do_exit = 0;
static struct libusb_device_handle *devh = NULL;

static unsigned long num_bytes = 0, num_xfer = 0;
static struct timeval tv_start;

static void get_timestamp(struct timeval *tv)
{
#if defined(PLATFORM_WINDOWS)
	static LARGE_INTEGER frequency;
	LARGE_INTEGER counter;

	if (!frequency.QuadPart)
		QueryPerformanceFrequency(&frequency);

	QueryPerformanceCounter(&counter);
	counter.QuadPart *= 1000000;
	counter.QuadPart /= frequency.QuadPart;

	tv->tv_sec = (long)(counter.QuadPart / 1000000ULL);
	tv->tv_usec = (long)(counter.QuadPart % 1000000ULL);
#elif defined(HAVE_CLOCK_GETTIME)
	struct timespec ts;

	(void)clock_gettime(CLOCK_MONOTONIC, &ts);
	tv->tv_sec = ts.tv_sec;
	tv->tv_usec = (int)(ts.tv_nsec / 1000L);
#else
[... 139 lines cut ...]
	rc = libusb_set_interface_alt_setting(devh, 0, 6);
	if (rc < 0) {
		fprintf(stderr, "Error set alt interface: %s\n", libusb_error_name(rc));
		goto out;
	}

	benchmark_in(EP_ISO_IN);

	while (!do_exit) {
		rc = libusb_handle_events(NULL);
		if (rc != LIBUSB_SUCCESS)
			break;
	}

	/* Measurement has already been done by the signal handler. */

	libusb_release_interface(devh, 2);
out:
	if (devh)
		libusb_close(devh);
	libusb_exit(NULL);
	return rc;
}

```

mcuee, 2022-03-31:
@jonasmalacofilho I think the staging tree is good. As for the following, hopefully you can fix it.
> we also seem to have a new finalizer bug, at least when something else goes wrong.