pyusb/pyusb #383: pyusb finds my device with usb.core.find(find_all=1) but not with usb.core.find(idVendor=2341, idProduct=42)

artisticfox8, 2021-08-19:
Hello, I am using Python 3.9.6 on Linux Mint MATE 20.2 and I encountered this very weird bug, when trying out PyUSB with my Arduino 2560 MEGA R3 board. I want to tell you about it. Thanks for your attention.

python terminal:
```
>>> import usb
>>> d= usb.core.find(find_all=1)
>>> print(d)
<generator object find.<locals>.device_iter at 0x7f803ee00900>
>>> print(list(d))
[<DEVICE ID 1d6b:0003 on Bus 002 Address 001>, <DEVICE ID 17ef:608d on Bus 001 Address 004>, <DEVICE ID 0bda:0129 on Bus 001 Address 003>, <DEVICE ID 17ef:6018 on Bus 001 Address 002>, <DEVICE ID 2341:0042 on Bus 001 Address 008>, <DEVICE ID 1d6b:0002 on Bus 001 Address 001>]
>>> dev = usb.core.find(idVendor=2341, idProduct=42)
>>> print(dev)
None
>>> 
```
I used idProduct=42 because Python did not let me use idProduct= 0042

Here is how I found the idVendor=2341, idProduct=42 using dmesg:
```

[ 1168.955071] usb 1-3: new full-speed USB device number 5 using xhci_hcd
[ 1169.105966] usb 1-3: New USB device found, idVendor=2341, idProduct=0042, bcdDevice= 0.01
[ 1169.105975] usb 1-3: New USB device strings: Mfr=1, Product=2, SerialNumber=220
[ 1169.105980] usb 1-3: Manufacturer: Arduino (www.arduino.cc)
[ 1169.105985] usb 1-3: SerialNumber: 8573432333135151E091
[ 1169.256569] cdc_acm 1-3:1.0: ttyACM0: USB ACM device
[ 1169.258756] usbcore: registered new interface driver cdc_acm
[ 1169.258759] cdc_acm: USB Abstract Control Model driver for USB modems and ISDN adapters
[ 1280.956566] usb 1-3: USB disconnect, device number 5
[ 1337.539196] usb 1-3: new full-speed USB device number 6 using xhci_hcd
[ 1337.690347] usb 1-3: New USB device found, idVendor=2341, idProduct=0042, bcdDevice= 0.01
[ 1337.690353] usb 1-3: New USB device strings: Mfr=1, Product=2, SerialNumber=220
[ 1337.690356] usb 1-3: Manufacturer: Arduino (www.arduino.cc)
[ 1337.690359] usb 1-3: SerialNumber: 8573432333135151E091
```
Using lsusb confirms it:
```
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 004: ID 17ef:608d Lenovo Lenovo USB Optical Mouse
Bus 001 Device 007: ID 2341:0042 Arduino SA Mega 2560 R3 (CDC ACM)
Bus 001 Device 003: ID 0bda:0129 Realtek Semiconductor Corp. RTS5129 Card Reader Controller
Bus 001 Device 002: ID 17ef:6018 Lenovo Lenovo USB Keyboard
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```
Thanks for reading, hopefully you can resolve this

jonasmalacofilho, 2021-08-19:
USB vendor/product IDs are normally (and in your case) shown in hexadecimal.

In Python, you must use the `0x` prefix for hexadecimal literals:

```py
usb.core.find(idVendor=0x2341, idProduct=0x0042)
```

closed 2021-08-19 by jonasmalacofilho

artisticfox8, 2021-08-20:
> 
> 
> USB vendor/product IDs are normally (and in your case) shown in hexadecimal.
> 
> In Python, you must use the `0x` prefix for hexadecimal literals:
> 
> ```python
> usb.core.find(idVendor=0x2341, idProduct=0x0042)
> ```

Thanks for the reply. It would be nice and very user friendly of you to add a hex number check to the find function, and convert the integer to hex if it isn't already.  Or raise an ValueError: Integer is not hexadecimal or something.

jonasmalacofilho, 2021-08-24:
> It would be nice and very user friendly of you to add a hex number check to the find function, and convert the integer to hex if it isn't already. Or raise an ValueError: Integer is not hexadecimal or something.

The use of hexadecimal literals is just a convention used when dealing with USB IDs.  You are absolutely free to ignore it, as long as the integer value has the correct value, regardless of the base used for the literal.

Also, there's no way in Python to differentiate between `0x01` and `1`: both are simply integer literals with the value, in base 10, one.