pyusb/pyusb #384: The default example at tutorial.rst does not work: ep is None

artisticfox8, 2021-08-19:
Hello, I opened your tutorial to try it out and modified it only slightly, so my Arduino device is found without errors. It is 95% same, and there error is not mine, so I'd appreciate if you take a look at it :)

Here is the script:

```
import usb.core #pyusb
import usb.util
import usb.backend.libusb1

backend = usb.backend.libusb1.get_backend(find_library=lambda x: "C:/Python39/Lib/site-packages/usb1/libusb-1.0.dll") #didn't work otherwise, found no backend, later I downloaded the WinUSB driver using the Zadig program
d = usb.core.find(backend=backend, find_all=True)

# find our device Arduino MEGA 2560 idVendor=2341, idProduct=42
dev = usb.core.find(find_all=1) #only this way, if I put idVendor and idProduct, there are errors
dev2 = 0
for x in list(dev):
    if "2341:0042" in str(x):
        dev2 = x
        print("d",dev2) #works
# was it found?
if dev is None:
    raise ValueError('Device not found')

# set the active configuration. With no arguments, the first
# configuration will be the active one
dev2.set_configuration()

# get an endpoint instance
cfg = dev2.get_active_configuration()
intf = cfg[(0,0)]

ep = usb.util.find_descriptor( #works up until here
    intf,
    # match the first OUT endpoint
    custom_match = \
    lambda e: \
        usb.util.endpoint_direction(e.bEndpointAddress) == \
        usb.util.ENDPOINT_OUT)

assert ep is not None #it is

# write the data
ep.write('1')
```
Thanks for your patience!

mcuee, 2021-08-20:
What is the OS? Do you have access to the descriptor of the device? If this is Windows, did you follow the FAQ to install the supported driver by using Zadig?
https://github.com/pyusb/pyusb/blob/master/docs/faq.rst

Please also take note the following code will most likely not working. You have to understand the USB communication protocol in order to talk to your device.
```
# write the data
ep.write('1')
```

artisticfox8, 2021-08-20:
> What is the OS? Do you have access to the descriptor of the device? If this is Windows, did you follow the FAQ to install the supported driver by using Zadig? [...]

Thanks for the reply! The OS is Windows 10. I don't understand your second question, sorry. I know the numbers of productID and vendorID from the `lsusb` Linux command.  
I changed the for loop to `dev2 = usb.core.find(idVendor=0x2341, idProduct=0x0042)` (I learned numbers must be preceded with 0x here.
The only error still is 
```
assert ep is not None
AssertionError
```
I am sure the WInUSB driver is installed, here is the screenshot:
[image]

artisticfox8, 2021-08-20:
> ```
> # write the data
> ep.write('1')
> ```

As for talking to the device I only want to send strings to the serial, much like the classic Arduino IDE Serial. 

Here is my Arduino code btw:

```
int x;
void setup() {
 Serial.begin(115200);
 pinMode(LED_BUILTIN, OUTPUT);
 Serial.setTimeout(1);

}
void loop() {
 while (!Serial.available());
 x = Serial.readString().toInt();
 if(x==1){
   digitalWrite(LED_BUILTIN, LOW);
 }else{
   if(x==2){
     digitalWrite(LED_BUILTIN, HIGH);
   }
 }
 Serial.print(x + 1);
}

```

artisticfox8, 2021-08-20:
> What is the OS? Do you have access to the descriptor of the device? If this is Windows, did you follow the FAQ to install the supported driver by using Zadig? [...]

I tried to run it on Linux Mint 20.2 MATE edition, same error:
```

fox@fox-Lenovo-V320-15IAP-Desktop:~/Documents/code$ sudo python3 "testuju usb2.py"
Traceback (most recent call last):
  File "testuju usb2.py", line 36, in <module>
    assert ep is not None
AssertionError
```

mcuee, 2021-08-20:
For USB serial port, you should use pyserial instead and not pyusb.
https://github.com/pyserial/pyserial

artisticfox8, 2021-08-20:
> For USB serial port, you should use pyserial instead and not pyusb.
> https://github.com/pyserial/pyserial

I need to make that work on Android eventually. Pyserial does not work in Termux on Android. I've been told that Arduino + pyusb is possible. Could you maybe navigate me to a tutorial that could do this?

artisticfox8, 2021-08-20:
Even if simple `ep.write('1')` isn't gonna work, all the preceding parts shoudl. Examples should work without troubleshooting.

mcuee, 2021-08-20:
If you want to use pyusb, you have to reverse engineering the communication protocol.
https://slomkowski.eu/tutorials/eavesdropping-usb-and-writing-driver-in-python/

As for Android, sorry I can not help you. It is not easy to get libusb to work under Android.

But for the tutorial, you can try to debug in the following steps.

1) print out "lsusb -vvv" about your device under Linux

2) set environment variable PYUSB_DEBUG='debug' or use the following codes, then post the debug log.
https://github.com/pyusb/pyusb/blob/master/docs/faq.rst
```
import os
os.environ['PYUSB_DEBUG'] = 'debug'
```

3) if the above still does not help, set environment variable LIBUSB_DEBUG=4, then run you python codes and post the full debug log.

mcuee, 2021-08-20:
Then from there, hopefully you can find the problem. 

As for the USB communication protocol of your particular USB serial port implementation, you can either check out the Linux driver side, or use reverse engineering.

Example: https://slomkowski.eu/tutorials/eavesdropping-usb-and-writing-driver-in-python/

mcuee, 2021-08-20:
But if you are beginner, I will recommend you to try out something simpler. It is not an easy task to get libusb+pyusb working under Android (with or without Termux).

mcuee, 2021-08-20:
As for Serial under Android, there seems to be some solutions.
1) with python example: https://github.com/jacklinquan/usbserial4a
2) without python: https://github.com/mik3y/usb-serial-for-android

mcuee, 2021-08-21:
BTW, from what mentioned in the other issue, this is a CDC-ACM device. If you want to use libusb/pyusb, you have to understand the CDC-ACM protocol.

The example here may help you a bit.
https://github.com/tytouf/libusb-cdc-example/blob/master/cdc_example.c

mcuee, 2021-08-23:
Another discussion:
https://github.com/pyserial/pyserial/issues/603

mcuee, 2021-08-23:
BTW, there is a better way to find the device as mentioned in the other issue.
https://github.com/pyusb/pyusb/issues/383

mcuee, 2021-08-23:
I will close this issue for now. I believe you can sort out the issue under Linux/Windows. 

For Android/Termux, please try to discuss in existing issue #285.

artisticfox8, 2021-08-23:
> I will close this issue for now. I believe you can sort out the issue under Linux/Windows.
> 
> For Android/Termux, please try to discuss in existing issue #285 or in pyserial issue [pyserial/pyserial#428](https://github.com/pyserial/pyserial/issues/428) and [pyserial/pyserial#603](https://github.com/pyserial/pyserial/issues/603).

Big Thanks to you! You have been very helpful and provided  many useful links. However, you didn't adress the original problem, and that is that on both Linux Mint 20.2 and Windows 10, the default example did not work. It should work without debugging, it's an example after all.

mcuee, 2021-08-24:
> However, you didn't adress the original problem, and that is that on both Linux Mint 20.2 and Windows 10, the default example did not work. It should work without debugging, it's an example after all.

No it is not true that the default example will work without debugging when it comes to libusb and pyusb. You have to play your part as well. As mentioned before, please follow the following steps in debugging.

1) print out "lsusb -vvv" about your device under Linux

2) set environment variable PYUSB_DEBUG='debug' or use the following codes, then post the debug log.
https://github.com/pyusb/pyusb/blob/master/docs/faq.rst
```
import os
os.environ['PYUSB_DEBUG'] = 'debug'
```

3) if the above still does not help, set environment variable LIBUSB_DEBUG=4, then run you python codes and post the full debug log.

mcuee, 2021-08-24:
This is my guess but you need to debug by yourself.

1) for Windows, you install WinUSB driver. In that case, you have to use libusb-1.0 backend. In that case, you have to put libusb-1.0.dll in the right place.
https://github.com/pyusb/pyusb/blob/master/docs/faq.rst#how-do-i-install-libusb-on-windows

2) under Linux, the kernel cdc-acm driver is claiming the device, you have to detach the driver. You can use udev rule to unbind the driver, or use the kernel driver detaching function. Something like the following.
https://github.com/tytouf/libusb-cdc-example/blob/master/cdc_example.py

mcuee, 2021-08-24:
Actually once you print out the "lsusb -vvv" and you may find the issues. CDC-ACM devices have two interfaces. In the case of your device,  I believe the OUT endpoint will reside at interface 1 and not interface 0. So you will not able to find it.

mcuee, 2021-08-24:
I have an Arduino Leonardo (0x2341, 0x8036) and here is the test code and run log under Windows.

```
import usb.core #pyusb
import usb.util
import usb.backend.libusb1

be = usb.backend.libusb1.get_backend()

# find our device Arduino Leonardo idVendor=0x2341, idProduct=0x8036
dev = usb.core.find(idVendor=0x2341, idProduct=0x8036, backend=be)

if dev is None:
    raise ValueError('Device not found')

# set the active configuration. With no arguments, the first
# configuration will be the active one
dev.set_configuration()

# get an endpoint instance
cfg = dev.get_active_configuration()
#print(cfg)
intf0 = cfg[(0,0)]
print(intf0)
intf1 = cfg[(1,0)]
print(intf1)

ep = usb.util.find_descriptor(
    intf1,
    # match the first OUT endpoint
    custom_match = \
    lambda e: \
        usb.util.endpoint_direction(e.bEndpointAddress) == \
        usb.util.ENDPOINT_OUT)

assert ep is not None #it is

# write the data
ep.write('1')

(py39venv) C:\work\python> python .\test_cdcacm.py
    INTERFACE 0: CDC Communication =========================
     bLength            :    0x9 (9 bytes)
     bDescriptorType    :    0x4 Interface
     bInterfaceNumber   :    0x0
     bAlternateSetting  :    0x0
     bNumEndpoints      :    0x1
     bInterfaceClass    :    0x2 CDC Communication
     bInterfaceSubClass :    0x2
     bInterfaceProtocol :    0x0
     iInterface         :    0x0
      ENDPOINT 0x81: Interrupt IN ==========================
       bLength          :    0x7 (7 bytes)
       bDescriptorType  :    0x5 Endpoint
       bEndpointAddress :   0x81 IN
       bmAttributes     :    0x3 Interrupt
       wMaxPacketSize   :   0x10 (16 bytes)
       bInterval        :   0x40
    INTERFACE 1: CDC Data ==================================
     bLength            :    0x9 (9 bytes)
     bDescriptorType    :    0x4 Interface
     bInterfaceNumber   :    0x1
     bAlternateSetting  :    0x0
     bNumEndpoints      :    0x2
     bInterfaceClass    :    0xa CDC Data
     bInterfaceSubClass :    0x0
     bInterfaceProtocol :    0x0
     iInterface         :    0x0
      ENDPOINT 0x2: Bulk OUT ===============================
       bLength          :    0x7 (7 bytes)
       bDescriptorType  :    0x5 Endpoint
       bEndpointAddress :    0x2 OUT
       bmAttributes     :    0x2 Bulk
       wMaxPacketSize   :   0x40 (64 bytes)
       bInterval        :    0x0
      ENDPOINT 0x83: Bulk IN ===============================
       bLength          :    0x7 (7 bytes)
       bDescriptorType  :    0x5 Endpoint
       bEndpointAddress :   0x83 IN
       bmAttributes     :    0x2 Bulk
       wMaxPacketSize   :   0x40 (64 bytes)
       bInterval        :    0x0
Traceback (most recent call last):
  File "C:\work\python\test_cdcacm.py", line 36, in <module>
    ep.write('1')
  File "C:\work\python\py39venv\lib\site-packages\usb\core.py", line 408, in write
    return self.device.write(self, data, timeout)
  File "C:\work\python\py39venv\lib\site-packages\usb\core.py", line 989, in write
    return fn(
  File "C:\work\python\py39venv\lib\site-packages\usb\backend\libusb1.py", line 837, in bulk_write
    return self.__write(self.lib.libusb_bulk_transfer,
  File "C:\work\python\py39venv\lib\site-packages\usb\backend\libusb1.py", line 938, in __write
    _check(retval)
  File "C:\work\python\py39venv\lib\site-packages\usb\backend\libusb1.py", line 602, in _check
    raise USBTimeoutError(_strerror(ret), ret, _libusb_errno[ret])
usb.core.USBTimeoutError: [Errno 10060] Operation timed out
```

jonasmalacofilho, 2021-08-24:
@artisticfox8,

I can't see any issue with the [getting started example](https://github.com/pyusb/pyusb/blob/master/docs/tutorial.rst#lets-get-it-started).  It even suggests the use of hexadecimal literals when dealing with USB IDs, which was one of your concerns.

No matter what example we use, it can only really work with a handful of devices: knowing and following the protocol expected by the particular device is absolutely necessary to know what/when to read or write from/to USB devices.

Also, the getting started example is not the place to put all the possible platform and USB class specifics, like kernel driver replacement on Linux/Windows.

closed 2021-08-24 by jonasmalacofilho

mcuee, 2021-08-24:
@artisticfox8 Just using your Arduino program, then actually 'ep.write()' can work as well.

```
import time
import usb.core
import usb.util
import usb.backend.libusb1

be = usb.backend.libusb1.get_backend()

# find our device Arduino Leonardo idVendor=0x2341, idProduct=0x8036
dev = usb.core.find(idVendor=0x2341, idProduct=0x8036, backend=be)

if dev is None:
    raise ValueError('Device not found')

# set the active configuration. With no arguments, the first
# configuration will be the active one
dev.set_configuration()

# get an endpoint instance
cfg = dev.get_active_configuration()
#print(cfg)
intf0 = cfg[(0,0)]
print(intf0)
intf1 = cfg[(1,0)]
print(intf1)

epout = usb.util.find_descriptor(
    intf1,
    # match the first OUT endpoint
    custom_match = \
    lambda e: \
        usb.util.endpoint_direction(e.bEndpointAddress) == \
        usb.util.ENDPOINT_OUT)
      

assert epout is not None

# write the data
print("Turning the LED ON")
epout.write('2')
time.sleep(1)
print("Turning the LED OFF")
epout.write('1')
```

Rung log under Windows 10 (with WinUSB driver to replace the original USB Composite Device driver) with Arduino Leonardo.
```

(py39venv) C:\work\python> python .\test_cdcacm.py
    INTERFACE 0: CDC Communication =========================
     bLength            :    0x9 (9 bytes)
     bDescriptorType    :    0x4 Interface
     bInterfaceNumber   :    0x0
     bAlternateSetting  :    0x0
     bNumEndpoints      :    0x1
     bInterfaceClass    :    0x2 CDC Communication
     bInterfaceSubClass :    0x2
     bInterfaceProtocol :    0x0
     iInterface         :    0x0
      ENDPOINT 0x81: Interrupt IN ==========================
       bLength          :    0x7 (7 bytes)
       bDescriptorType  :    0x5 Endpoint
       bEndpointAddress :   0x81 IN
       bmAttributes     :    0x3 Interrupt
       wMaxPacketSize   :   0x10 (16 bytes)
       bInterval        :   0x40
    INTERFACE 1: CDC Data ==================================
     bLength            :    0x9 (9 bytes)
     bDescriptorType    :    0x4 Interface
     bInterfaceNumber   :    0x1
     bAlternateSetting  :    0x0
     bNumEndpoints      :    0x2
     bInterfaceClass    :    0xa CDC Data
     bInterfaceSubClass :    0x0
     bInterfaceProtocol :    0x0
     iInterface         :    0x0
      ENDPOINT 0x2: Bulk OUT ===============================
       bLength          :    0x7 (7 bytes)
       bDescriptorType  :    0x5 Endpoint
       bEndpointAddress :    0x2 OUT
       bmAttributes     :    0x2 Bulk
       wMaxPacketSize   :   0x40 (64 bytes)
       bInterval        :    0x0
      ENDPOINT 0x83: Bulk IN ===============================
       bLength          :    0x7 (7 bytes)
       bDescriptorType  :    0x5 Endpoint
       bEndpointAddress :   0x83 IN
       bmAttributes     :    0x2 Bulk
       wMaxPacketSize   :   0x40 (64 bytes)
       bInterval        :    0x0
Turning the LED ON
Turning the LED OFF
```

mcuee, 2021-08-24:
Solution for Android: https://github.com/pyserial/pyserial/issues/428

I just test this using my Android phone and Arduino Leonardo, it works fine with your Arduino example.
https://github.com/jacklinquan/usbserial4a

Follow the guide to install pydroid and then use pip to install usbserial4a. Then use the following example.
https://github.com/jacklinquan/usbserial4a/blob/master/example/usbserial4a_ui_example.py

artisticfox8, 2021-08-24:
>     1. lsusb -vvv"

Here is lsusb -vvv (after setting the environment variable):
```
Bus 001 Device 009: ID 2341:0042 Arduino SA Mega 2560 R3 (CDC ACM)
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               1.10
  bDeviceClass            2 Communications
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0         8
  idVendor           0x2341 Arduino SA
  idProduct          0x0042 Mega 2560 R3 (CDC ACM)
  bcdDevice            0.01
  iManufacturer           1 
  iProduct                2 
  iSerial               220 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x003e
    bNumInterfaces          2
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xc0
      Self Powered
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         2 Communications
      bInterfaceSubClass      2 Abstract (modem)
      bInterfaceProtocol      1 AT-commands (v.25ter)
      iInterface              0 
      CDC Header:
        bcdCDC               10.01
      CDC ACM:
        bmCapabilities       0x06
          sends break
          line coding and serial state
      CDC Union:
        bMasterInterface        0
        bSlaveInterface         1 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval             255
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass        10 CDC Data
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x04  EP 4 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1

```
Thanks for being friendly and patient with me!

artisticfox8, 2021-08-24:
usb.core.USBError: [Errno 16] Resource busy

> > However, you didn't adress the original problem, and that is that on both Linux Mint 20.2 and Windows 10, the default example did not work. It should work without debugging, it's an example after all.
> 
> No it is not true that the default example will work without debugging when it comes to libusb and pyusb. You have to play your part as well. As mentioned before, please follow the following steps in debugging.
> 
>     1. print out "lsusb -vvv" about your device under Linux
> 
>     2. set environment variable PYUSB_DEBUG='debug' or use the following codes, then post the debug log.
>        https://github.com/pyusb/pyusb/blob/master/docs/faq.rst
> 
> 
> ```
> import os
> os.environ['PYUSB_DEBUG'] = 'debug'
> ```
> 
>     1. if the above still does not help, set environment variable LIBUSB_DEBUG=4, then run you python codes and post the full debug log.

Here is the log:
```
d DEVICE ID 2341:0042 on Bus 001 Address 009 =================
 bLength                :   0x12 (18 bytes)
 bDescriptorType        :    0x1 Device
 bcdUSB                 :  0x110 USB 1.1
 bDeviceClass           :    0x2 Communications Device
 bDeviceSubClass        :    0x0
 bDeviceProtocol        :    0x0
 bMaxPacketSize0        :    0x8 (8 bytes)
 idVendor               : 0x2341
 idProduct              : 0x0042
 bcdDevice              :    0x1 Device 0.01
 iManufacturer          :    0x1 Arduino (www.arduino.cc)
 iProduct               :    0x2 Error Accessing String
 iSerialNumber          :   0xdc 8573432333135151E091
 bNumConfigurations     :    0x1
  CONFIGURATION 1: 100 mA ==================================
   bLength              :    0x9 (9 bytes)
   bDescriptorType      :    0x2 Configuration
   wTotalLength         :   0x3e (62 bytes)
   bNumInterfaces       :    0x2
   bConfigurationValue  :    0x1
   iConfiguration       :    0x0 
   bmAttributes         :   0xc0 Self Powered
   bMaxPower            :   0x32 (100 mA)
    INTERFACE 0: CDC Communication =========================
     bLength            :    0x9 (9 bytes)
     bDescriptorType    :    0x4 Interface
     bInterfaceNumber   :    0x0
     bAlternateSetting  :    0x0
     bNumEndpoints      :    0x1
     bInterfaceClass    :    0x2 CDC Communication
     bInterfaceSubClass :    0x2
     bInterfaceProtocol :    0x1
     iInterface         :    0x0 
      ENDPOINT 0x82: Interrupt IN ==========================
       bLength          :    0x7 (7 bytes)
       bDescriptorType  :    0x5 Endpoint
       bEndpointAddress :   0x82 IN
       bmAttributes     :    0x3 Interrupt
       wMaxPacketSize   :    0x8 (8 bytes)
       bInterval        :   0xff
    INTERFACE 1: CDC Data ==================================
     bLength            :    0x9 (9 bytes)
     bDescriptorType    :    0x4 Interface
     bInterfaceNumber   :    0x1
     bAlternateSetting  :    0x0
     bNumEndpoints      :    0x2
     bInterfaceClass    :    0xa CDC Data
     bInterfaceSubClass :    0x0
[... 88 lines cut ...]
       wMaxPacketSize   :   0x40 (64 bytes)
       bInterval        :    0x1
      ENDPOINT 0x83: Bulk IN ===============================
       bLength          :    0x7 (7 bytes)
       bDescriptorType  :    0x5 Endpoint
       bEndpointAddress :   0x83 IN
       bmAttributes     :    0x2 Bulk
       wMaxPacketSize   :   0x40 (64 bytes)
       bInterval        :    0x1
Traceback (most recent call last):
  File "arduino github default example test.py", line 21, in <module>
    dev2.set_configuration()
  File "/usr/local/lib/python3.8/dist-packages/usb/core.py", line 915, in set_configuration
    self._ctx.managed_set_configuration(self, configuration)
  File "/usr/local/lib/python3.8/dist-packages/usb/core.py", line 113, in wrapper
    return f(self, *args, **kwargs)
  File "/usr/local/lib/python3.8/dist-packages/usb/core.py", line 159, in managed_set_configuration
    self.backend.set_configuration(self.handle, cfg.bConfigurationValue)
  File "/usr/local/lib/python3.8/dist-packages/usb/backend/libusb1.py", line 812, in set_configuration
    _check(self.lib.libusb_set_configuration(dev_handle.handle, config_value))
  File "/usr/local/lib/python3.8/dist-packages/usb/backend/libusb1.py", line 604, in _check
    raise USBError(_strerror(ret), ret, _libusb_errno[ret])
usb.core.USBError: [Errno 16] Resource busy

```

artisticfox8, 2021-08-24:
Even after I created `/etc/modprobe.d/blacklist` and set the contents to `blacklist cdc-acm` (as suggested here https://stackoverflow.com/questions/48650539/prevent-cdc-acm-driver-from-loading-device). I get usb.core.USBError: [Errno 16] Resource busy when I run the file.

artisticfox8, 2021-08-24:
> usb.core.USBError: [Errno 16] Resource busy
> 
> > > However, you didn't adress the original problem, and that is that on both Linux Mint 20.2 and Windows 10, the default example did not work. It should work without debugging, it's an example after all.
> > 
> > 
> > No it is not true that the default example will work without debugging when it comes to libusb and pyusb. You have to play your part as well. As mentioned before, please follow the following steps in debugging.
> > ```
> > 1. print out "lsusb -vvv" about your device under Linux
> > 
> > 2. set environment variable PYUSB_DEBUG='debug' or use the following codes, then post the debug log.
> >    https://github.com/pyusb/pyusb/blob/master/docs/faq.rst
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > ```
> > import os
> > os.environ['PYUSB_DEBUG'] = 'debug'
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > ```
> > 1. if the above still does not help, set environment variable LIBUSB_DEBUG=4, then run you python codes and post the full debug log.
> > ```
> 
> Here is the log (under Linux):
> 
> ```
> d DEVICE ID 2341:0042 on Bus 001 Address 009 =================
>  bLength                :   0x12 (18 bytes)
>  bDescriptorType        :    0x1 Device
>  bcdUSB                 :  0x110 USB 1.1
>  bDeviceClass           :    0x2 Communications Device
>  bDeviceSubClass        :    0x0
>  bDeviceProtocol        :    0x0
>  bMaxPacketSize0        :    0x8 (8 bytes)
>  idVendor               : 0x2341
>  idProduct              : 0x0042
>  bcdDevice              :    0x1 Device 0.01
>  iManufacturer          :    0x1 Arduino (www.arduino.cc)
>  iProduct               :    0x2 Error Accessing String
>  iSerialNumber          :   0xdc 8573432333135151E091
>  bNumConfigurations     :    0x1
>   CONFIGURATION 1: 100 mA ==================================
>    bLength              :    0x9 (9 bytes)
>    bDescriptorType      :    0x2 Configuration
>    wTotalLength         :   0x3e (62 bytes)
>    bNumInterfaces       :    0x2
>    bConfigurationValue  :    0x1
>    iConfiguration       :    0x0 
>    bmAttributes         :   0xc0 Self Powered
>    bMaxPower            :   0x32 (100 mA)
>     INTERFACE 0: CDC Communication =========================
[... 113 lines cut ...]
>        bInterval        :    0x1
>       ENDPOINT 0x83: Bulk IN ===============================
>        bLength          :    0x7 (7 bytes)
>        bDescriptorType  :    0x5 Endpoint
>        bEndpointAddress :   0x83 IN
>        bmAttributes     :    0x2 Bulk
>        wMaxPacketSize   :   0x40 (64 bytes)
>        bInterval        :    0x1
> Traceback (most recent call last):
>   File "arduino github default example test.py", line 21, in <module>
>     dev2.set_configuration()
>   File "/usr/local/lib/python3.8/dist-packages/usb/core.py", line 915, in set_configuration
>     self._ctx.managed_set_configuration(self, configuration)
>   File "/usr/local/lib/python3.8/dist-packages/usb/core.py", line 113, in wrapper
>     return f(self, *args, **kwargs)
>   File "/usr/local/lib/python3.8/dist-packages/usb/core.py", line 159, in managed_set_configuration
>     self.backend.set_configuration(self.handle, cfg.bConfigurationValue)
>   File "/usr/local/lib/python3.8/dist-packages/usb/backend/libusb1.py", line 812, in set_configuration
>     _check(self.lib.libusb_set_configuration(dev_handle.handle, config_value))
>   File "/usr/local/lib/python3.8/dist-packages/usb/backend/libusb1.py", line 604, in _check
>     raise USBError(_strerror(ret), ret, _libusb_errno[ret])
> usb.core.USBError: [Errno 16] Resource busy
> ```

This is the log when I ran your python script (I modified the vendorID and deviceID so the device was found)

mcuee, 2021-08-24:
As mentioned, you have to detach the kernel driver under Linux and you may also need to set the right permission by using udev rules (or you can run as sudo). You have to carry out this debug by yourself.

mcuee, 2021-08-24:
Tip here: 
Manual driver binding and unbinding
https://lwn.net/Articles/143397/

mcuee, 2021-08-24:
Once you have got the kernel driver detached and have the right permission, you do not need to use sudo and it will work.

```

(py35venv) mcuee@orangepizero:~/libusb/python_usb$ uname -a
Linux orangepizero 4.14.70-sunxi #265 SMP Wed Sep 19 10:01:19 CEST 2018 armv7l GNU/Linux

(py35venv) mcuee@orangepizero:~/libusb/python_usb$ python test_cdcacm.py
    INTERFACE 0: CDC Communication =========================
     bLength            :    0x9 (9 bytes)
     bDescriptorType    :    0x4 Interface
     bInterfaceNumber   :    0x0
     bAlternateSetting  :    0x0
     bNumEndpoints      :    0x1
     bInterfaceClass    :    0x2 CDC Communication
     bInterfaceSubClass :    0x2
     bInterfaceProtocol :    0x0
     iInterface         :    0x0
      ENDPOINT 0x81: Interrupt IN ==========================
       bLength          :    0x7 (7 bytes)
       bDescriptorType  :    0x5 Endpoint
       bEndpointAddress :   0x81 IN
       bmAttributes     :    0x3 Interrupt
       wMaxPacketSize   :   0x10 (16 bytes)
       bInterval        :   0x40
    INTERFACE 1: CDC Data ==================================
     bLength            :    0x9 (9 bytes)
     bDescriptorType    :    0x4 Interface
     bInterfaceNumber   :    0x1
     bAlternateSetting  :    0x0
     bNumEndpoints      :    0x2
     bInterfaceClass    :    0xa CDC Data
     bInterfaceSubClass :    0x0
     bInterfaceProtocol :    0x0
     iInterface         :    0x0
      ENDPOINT 0x2: Bulk OUT ===============================
       bLength          :    0x7 (7 bytes)
       bDescriptorType  :    0x5 Endpoint
       bEndpointAddress :    0x2 OUT
       bmAttributes     :    0x2 Bulk
       wMaxPacketSize   :   0x40 (64 bytes)
       bInterval        :    0x0
      ENDPOINT 0x83: Bulk IN ===============================
       bLength          :    0x7 (7 bytes)
       bDescriptorType  :    0x5 Endpoint
       bEndpointAddress :   0x83 IN
       bmAttributes     :    0x2 Bulk
       wMaxPacketSize   :   0x40 (64 bytes)
       bInterval        :    0x0
Turning the LED ON
Turning the LED OFF
```

artisticfox8, 2021-08-24:
> Tips here: just take note that you are using cdc-acm driver here.
> 
>     1. Manual driver binding and unbinding. You can also use the tip to check if your driver blacklist is working or not if you want to use that method (you can even use brutal force "sudo rmmod cdc-acm" method if just for quick test). By right blacklist the driver will also work.
>        https://lwn.net/Articles/143397/
> 
>     2. Using udev rules to unbind kernel driver (recommended way from my side)
>        https://sourceforge.net/p/libusb/mailman/message/27708732/
>        http://vogelchr.blogspot.com/2016/02/linuxudev-unbinding-from-one-kernel.html

Ok so I've tried number 2, didn't work. Number 1 works. However I get:
```
Turning the LED ON
Turning the LED OFF
Traceback (most recent call last):
  File "testting gtihub2.py", line 42, in <module>
    epout.write('1')
  File "/usr/local/lib/python3.8/dist-packages/usb/core.py", line 408, in write
    return self.device.write(self, data, timeout)
  File "/usr/local/lib/python3.8/dist-packages/usb/core.py", line 989, in write
    return fn(
  File "/usr/local/lib/python3.8/dist-packages/usb/backend/libusb1.py", line 837, in bulk_write
    return self.__write(self.lib.libusb_bulk_transfer,
  File "/usr/local/lib/python3.8/dist-packages/usb/backend/libusb1.py", line 938, in __write
    _check(retval)
  File "/usr/local/lib/python3.8/dist-packages/usb/backend/libusb1.py", line 602, in _check
    raise USBTimeoutError(_strerror(ret), ret, _libusb_errno[ret])
usb.core.USBTimeoutError: [Errno 110] Operation timed out

```

mcuee, 2021-08-25:
That is normal. As mentioned in the early comments, 'ep.write()' call is not guaranteed to work and the device may not understand the command.

What you need to do is to understand the device communication protocol. In this case, there are two ways.

1) The FW is open Atmaga16U2 or Atmega8U2 is used with CDC-ACM FW is acting as the USB to serial chip in your  Arduino Mega 2560. So you can read the FW to see how to communicate with the device.
https://github.com/arduino/ArduinoCore-avr/tree/master/firmwares/atmegaxxu2/arduino-usbserial

2) You can also use reverse engineering method mentioned earlier. You can just use Linux in this case. Use back the original kernel cdc-acm driver. Then use Arduino serial monitor to talk to your device. Use Wireshark to capture the traffic (from the starting of USB serial connection to the end of transfer).
https://slomkowski.eu/tutorials/eavesdropping-usb-and-writing-driver-in-python/

You have to go through this exercise by yourself. 

Reading ABCs of USB may help you as well.
https://www.usbmadesimple.co.uk/

CDC-ACM spec is also good to read.
https://www.usb.org/document-library/class-definitions-communication-devices-12

artisticfox8, 2021-08-25:
> That is normal. As mentioned in the early comments, 'ep.write()' call is not guaranteed to work and the device may not understand the command. It happens to work on my particular board but not necessarily your board.
> 
> What you need to do is to understand the device communication protocol. In this case, there are a few ways.
> 
>     1. The FW is open Atmaga16U2 or Atmega8U2 is used with CDC-ACM FW is acting as the USB to serial chip in your  Arduino Mega 2560. So you can read the FW to see how to communicate with the device.
>        https://github.com/arduino/ArduinoCore-avr/tree/master/firmwares/atmegaxxu2/arduino-usbserial
> 
>     2. The device is supposed to be compliant to the CDC-ACM protocol. So you can read the spec here.
>        https://www.usb.org/document-library/class-definitions-communication-devices-12
> 
> 
> And here is a host side Python (not using pyusb, under Android) code which may help you.
> https://github.com/jacklinquan/usbserial4a/blob/master/usbserial4a/cdcacmserial4a.py
> 
>     1. You can also use reverse engineering method mentioned earlier. You can just use Linux in this case. Use back the original kernel cdc-acm driver. Then use Arduino serial monitor to talk to your device. Use Wireshark to capture the traffic (from the starting of USB serial connection to the end of transfer).
>        https://slomkowski.eu/tutorials/eavesdropping-usb-and-writing-driver-in-python/
> 
> 
> You have to go through this exercise by yourself. Unfortunately there is no easy shortcut when it comes to usb/libusb/pyusb. You have to understand the communication protocol. It is much more difficult than the serial. So it is much easier you talk to your Arduino board using serial protocol.
> 
> Reading ABCs of USB may help you as well.
> https://www.usbmadesimple.co.uk/

Thanks for your reply! I've tested the python android library you linked and it works nicely! I think I'll use that for now, but thanks for the other links you posted, I appreciate it. Thanks again! If I ever meet you in person, I'll buy you a beer!

mcuee, 2021-08-25:
Glad that you have got the Python Android library working with your device.