pyusb/pyusb #388: Detaching kernel in Windows 10 (using libusb0)

EduardoNigro, 2021-09-21:
Hello,

Will it ever be possible to disconnect my USB device using **_detach_kernel_driver()_**?
 **_pyusb_** is installed on a Windows 10 platform using libusb0 (installed with Zadig)

I can send and receive commands to the device. However, I cannot disconnect the device in a graceful way. Only by physically disconnecting the USB cable. 

Finally, I used to code for this particular device using MATLAB and the USB driver that was supplied with their installation (**_hidapi-1.1.jar_**) and I know it is possible to disconnect from the device via software.

Thanks,
Eduardo

Below is an example of the code that connects to the device:

> import usb.core
> 
> ID_VENDOR_LEGO = 0x0694
> ID_PRODUCT_EV3 = 0x0005
> device = usb.core.find(idVendor=ID_VENDOR_LEGO, idProduct=ID_PRODUCT_EV3)
> print(device)
> 

The output is:

> DEVICE ID 0694:0005 on Bus 000 Address 006 =================
>  bLength                :   0x12 (18 bytes)
>  bDescriptorType        :    0x1 Device
>  bcdUSB                 :  0x200 USB 2.0
>  bDeviceClass           :    0x0 Specified at interface
>  bDeviceSubClass        :    0x0
>  bDeviceProtocol        :    0x0
>  bMaxPacketSize0        :   0x40 (64 bytes)
>  idVendor               : 0x0694
>  idProduct              : 0x0005
>  bcdDevice              :  0x216 Device 2.16
>  iManufacturer          :    0x1 LEGO Group
>  iProduct               :    0x2 EV3
>  iSerialNumber          :    0x3 001653470e58
>  bNumConfigurations     :    0x1
>   CONFIGURATION 1: 2 mA ====================================
>    bLength              :    0x9 (9 bytes)
>    bDescriptorType      :    0x2 Configuration
>    wTotalLength         :   0x29 (41 bytes)
>    bNumInterfaces       :    0x1
>    bConfigurationValue  :    0x1
>    iConfiguration       :    0x1 LEGO Group
>    bmAttributes         :   0xc0 Self Powered
>    bMaxPower            :    0x1 (2 mA)
>     INTERFACE 0: Human Interface Device ====================
>      bLength            :    0x9 (9 bytes)
>      bDescriptorType    :    0x4 Interface
>      bInterfaceNumber   :    0x0
>      bAlternateSetting  :    0x0
>      bNumEndpoints      :    0x2
>      bInterfaceClass    :    0x3 Human Interface Device
>      bInterfaceSubClass :    0x0
>      bInterfaceProtocol :    0x0
>      iInterface         :    0x4 Xfer data to and from EV3 brick
>       ENDPOINT 0x81: Interrupt IN ==========================
>        bLength          :    0x7 (7 bytes)
>        bDescriptorType  :    0x5 Endpoint
>        bEndpointAddress :   0x81 IN
>        bmAttributes     :    0x3 Interrupt
>        wMaxPacketSize   :  0x400 (1024 bytes)
>        bInterval        :    0x4
>       ENDPOINT 0x1: Interrupt OUT ==========================
>        bLength          :    0x7 (7 bytes)
>        bDescriptorType  :    0x5 Endpoint
>        bEndpointAddress :    0x1 OUT
>        bmAttributes     :    0x3 Interrupt
>        wMaxPacketSize   :  0x400 (1024 bytes)
>        bInterval        :    0x4
>

mcuee, 2021-09-23:
No it is not possible to detach the kernel driver under Windows.

What do you mean by disconnect the device? 

And for HID device, it is recommended you use HIDAPI (and cython-hidapi Python binding) under Windows. It is not recommended to use pyusb. How do you disconnect the device with hidapi and its Java binding)?

EduardoNigro, 2021-09-23:
Bummer.

Okay, so let me answer you last question first. And please bear with me since I only know so much (or so little) about USB communication. After the MATLAB package is installed (in order to communicate with the LEGO EV3 device), two files seem to be used in the communication class: **_hidapi-1.1.jar_** and **_mwhidapi.class_**. 

The USB communication class constructor method looks something like this:

>         function obj = CommUSB
>             % constructor
> 
>             % Adding EV3 to java path
>             jPath = javaclasspath('-dynamic');
>             ev3path = fileparts(which('legoev3'));
>             if ~ismember(ev3path, jPath)
>                 javaaddpath(ev3path);
>             end
>             hidjarpath = fullfile(ev3path, 'hidapi-1.1.jar');
>             if ~ismember(hidjarpath, jPath)
>                 javaaddpath(hidjarpath);
>             end
>             
>             try
>                 obj.USBHandle = mwhidapi;
>             catch 
>                 error('Connect USB Failed ');
>             end
>             obj.Status = 1;
>             
>         end

Once a handle to an instance (I believe) of the **_mwhidapi_** class is assigned to the **_USBHandle_** property (or attribute), four methods are available:

- **_open()_**
- **_close()_**
- **_read()_**
- **_write()_**

So, answering your first question, what I mean by disconnecting a device is to have a **_close()_** method so the device can then be later opened by some other software.

I have been using this communication class for many years as the foundation of many robotics projects with the LEGO EV3 using MATLAB. I really want to port everything to Python.

As a side note, I did install the **_cython-hidapi_** package and, while the read and write methods seem to be working, the reply message never changes despite the fact I'm sending distinct write messages. With **_pyusb_**, on the other hand, it works just fine! I get the exact replies I'm expecting to get. The only thing that's missing is that good old **_close()_** method...

Finally, why using **_pyusb_** isn't recommended?

Thank you so much,
Eduardo

mcuee, 2021-09-24:
For HID devices, libusb project recommends using HDIAPI, therefore pyusb is not recommended and you should use cython-hidapi. You can raise the issue to cython-hidapi or HIDAPI project.
https://github.com/libusb/libusb/wiki/FAQ#Does_libusb_support_USB_HID_devices

mcuee, 2021-09-24:
The libusb0 backend is not really recommended and you should give libusb-1.0 backend a try. You need to download libusb Windows binary from here and put libusb-1.0.dll to get pyusb working with libusb-1.0 backend. 
https://github.com/libusb/libusb/releases

However, in that case, please do not use libusb0.sys device driver, rather you should try first with the default HID driver since  libusb-1.0 Windows backend support HID devices as well but there are some limitations.

mcuee, 2021-09-24:
Anyway, back to pyusb if you still want to use pyusb.

> I can send and receive commands to the device. However, I cannot disconnect the device in a graceful way. Only by physically disconnecting the USB cable.
> So, answering your first question, what I mean by disconnecting a device is to have a close() method so the device can then be later opened by some other software.

What do you mean by that? What is your pyusb code? It should work based on what you say.

mcuee, 2021-09-24:
I think I understand your issue already.

Your other software is using the default Windows HID driver. So when you switch to libusb0.sys driver, you will not be able to use the other dsoftware as that software relies on native Windows HID API.

When you use HIDAPI or its Java bindings, you will not need to switch driver (as it still uses native Windows HIDAPI). Therefore the other software will of course still work.

Therefore you have two possible solutions. I will recommend the HIDAPI/cython-hidapi method.
1) Use Windows Device manager to set the device to go back to use WIndows HID driver. Then use libusb-1.0 API and pyusb (you need libusb-1.0.dll from libusb project).
2) Use Windows Device manager to set the device to go back to use WIndows HID driver. The use Cython-HIDAPI and try to figure out the issue there.

EduardoNigro, 2021-09-24:
> Your other software is using the default Windows HID driver. So when you switch to libusb0.sys driver, you will not be able to use the other dsoftware as that software relies on native Windows HID API.

YES!

> When you use HIDAPI or its Java bindings, you will not need to switch driver (as it still uses native Windows HIDAPI). Therefore the other software will of course still work.

YES!

Okay.
**For solution 1:** When I download and extract the libusb-1.0 files, there are a few folders containing the **_libusb-1.0.dll_**. Which one I should get it from (I presume MinGW32?) and where should I place it (System32?). Do I need to use Zadig for the driver installation?
**For solution 2:** I'll collect some more info and raise the issue with the cython-hidapi folk.

But still, for the **_close()_** method thing... Shouldn't I have something similar in pyusb? Below is some test code using Cython-Hidapi:

> import hid
> 
> #### # Opening EV3 device
> ID_VENDOR_LEGO = 0x0694
> ID_PRODUCT_EV3 = 0x0005
> device = hid.device()
> device.open(ID_VENDOR_LEGO, ID_PRODUCT_EV3)
> 
> #### # Getting EV3 brick serial number
> device.write(b'\n\x00\x00\x00\x00\x10\x00\xd3\x0c\x02\x10`')
> reply = device.read(1025)[0:24]
> print(reply)
> data = "".join(map(chr,reply[5:reply.index(0,5,-1)]))
> print(data)
> 
> #### # Getting EV3 brick name
> device.write(b'\t\x00\x00\x00\x00\x08\x00\xd3\r\x08`')
> reply = device.read(1025)[0:24]
> print(reply)
> data = "".join(map(chr,reply[5:reply.index(0,5,-1)]))
> print(data)
> 
> #### # Closing EV3 device
> device.close()

The "equivalent" code using pyusb is:

> import usb.core
> 
> #### # Finding EV3 device
> ID_VENDOR_LEGO = 0x0694
> ID_PRODUCT_EV3 = 0x0005
> device = usb.core.find(idVendor=ID_VENDOR_LEGO, idProduct=ID_PRODUCT_EV3)
> print(device)
> 
> #### # Connecting to device
> EP_IN  = 0x81
> EP_OUT = 0x01
> device.set_configuration()
> reply = device.read(EP_IN, 1024, 100)[0:5]
> print(reply)
> 
> #### # Getting brick serial number
> device.write(EP_OUT, b'\n\x00\x00\x00\x00\x10\x00\xd3\x0c\x02\x10`', 100)
> reply = device.read(EP_IN, 1024, 100)[0:24]
> print(reply)
> data = "".join(map(chr,reply[5:reply.index(0,5,-1)]))
> print(data)
> 
> #### # Getting brick name
> device.write(EP_OUT, b'\t\x00\x00\x00\x00\x08\x00\xd3\r\x08`', 100)
> reply = device.read(EP_IN, 1024, 100)[0:24]
> print(reply)
> data = "".join(map(chr,reply[5:reply.index(0,5,-1)]))
> print(data)

The correct output is:

[17, 0, 0, 0, 2, 48, 48, 49, 54, 53, 51, 52, 55, 48, 101, 53, 56, 0, 0, 0, 0, 0, 0, 0]
001653470e58
[9, 0, 0, 0, 2, 69, 86, 51, 0, 126, 126, 126, 126, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]
EV3

Which I can only achieve using the other MATLAB-based software or pyusb. When I use cython-hidapi I get a repeat answer from the last time the other software send a command! I can run the two-command file for the cython-hidapi version ad-nauseum and the reply doesn't change. Unless I go back to the other MATLAB-based software (or pyusb) and send a new command, which then will get "stuck" when I run the hidapi version. But this I believe is an issue for them to help me with.

mcuee, 2021-09-24:
> For solution 1: When I download and extract the libusb-1.0 files, there are a few folders containing the libusb-1.0.dll. Which one I should get it from (I presume MinGW32?) and where should I place it (System32?). Do I need to use Zadig for the driver installation?

No need to use Zadig. But rather you have to revert the action by Zadig and go back to the original Windows HID driver. You can do that with Windows Device Manager.
https://github.com/pbatard/libwdi/wiki/FAQ#Help_Zadig_replaced_the_driver_for_the_wrong_device_How_do_I_restore_it

As for the libusb-1.0.dll file, for 64bit Python, please use the one from "libusb-1.0.24\MinGW64\dll" or "libusb-1.0.24_bin\VS2019\MS64\dll"(you may need redistribution files of VS2019). I like the one from VS2019 as it is smaller.

If for 32bit Python, then use the one from "libusb-1.0.24\MinGW32\dll" or "libusb-1.0.24_bin\VS2019\MS32\dll".

You can put the dll file in the same directory of your Python.exe file (eg: if using virtual environment, it will be under "venv\Scripts" directory.

mcuee, 2021-09-24:
As for the different between pyusb and cython-hidapi, I am not that familiar with cython-hidapi (I am however relatively familar with HIDAPI). I believe Windows HID API is more limited and may be more stringent with regard to your codes. 

HID API will talk in terms of reports. You need to use Input/Output/Feature report to talk to the device. You can say it is higher level API compare to libusb (talk in terms of Control/Interrupt transfer with regard to HID devices).

mcuee, 2021-09-24:
I will close this issue as there is no issue with pyusb in this case.

closed 2021-09-24 by mcuee

EduardoNigro, 2021-09-24:
Thank you for all the tips Sir!
Time for me to go pester the Cython-Hidapi guy ;)