pyusb/pyusb #390: Advice for debugging a communication issue when using pyusb

ap--, 2021-09-23:
Hi @jonasmalacofilho and @mcuee,

I'm using pyusb in my spectrometer library [python-seabreeze](https://github.com/ap--/python-seabreeze) and a user reported an issue where communication with a specific spectrometer model via pyusb works on Windows, but fails on Linux.
Another different type of spectrometer works on both of the user's operating systems, so we can rule out some library installation issue.

We have debug logs that were created with `LIBUSB_DEBUG=4` and `PYUSB_DEBUG=debug` from the Linux and the Windows machine. On both machines the pyusb backend is `libusb1` and the libusb library version is `v1.0.24.11584`

❌  &nbsp; **Here is the linux log:** https://github.com/ap--/python-seabreeze/issues/133#issuecomment-926065589
✅  &nbsp; **And here the windows log:** https://github.com/ap--/python-seabreeze/issues/133#issuecomment-926073213

The linux log shows that a timeout occurs when it's trying to read information back after the first command. But I'm out of ideas what the issue could be.

Could you give me some advice how to best proceed with the debugging here?

Thank you so much,
Cheers,
Andreas 😃

mcuee, 2021-09-24:
I will recommend using wireshark to capture the low level USB traffic. It uses usbmon capability of the Linux kernel. I suspect it to be a kernel USB host controller issue.
https://www.kernel.org/doc/html/latest/usb/usbmon.html

mcuee, 2021-09-24:
If possible, please ask the user to try a different Linux distribution to see if he can still reproduce the issue or not.

The first post seems to indicate a Raspberry Pi host. In that case, it should be not so difficult to use another Linux distro by popping in another MicroSD card. 
https://github.com/ap--/python-seabreeze/issues/133#issue-1001304371

ap--, 2021-09-26:
Thank you so much for the quick reply.
So we now have logs for:

❌  &nbsp; Raspberry Pi (I assume running Raspbian) [libusb v1.0.22.11312]: https://github.com/ap--/python-seabreeze/issues/133#issuecomment-926631864
❌  &nbsp; Virtual Machine with Ubuntu [libusb v1.0.24.11584]: https://github.com/ap--/python-seabreeze/issues/133#issuecomment-926065589
✅  &nbsp; Windows [libusb v1.0.24.11584]: https://github.com/ap--/python-seabreeze/issues/133#issuecomment-926073213

And we'll collect a packet capture soon. Is there anything specific we should look out for?
We will also try to increase the read timeout to see if that fixes the issue.

Cheers,
Andreas 😊

ap--, 2021-09-27:
We do have packet captures for windows and from the vm now: https://github.com/ap--/python-seabreeze/files/7237325/pcap_files.zip
But I'm not sure what I would need to be looking for specificly.

We also tried to increase the read timeout, but the issue persistet.

Thanks again for your help,
Cheers
Andreas :blush:

mcuee, 2021-09-28:
Unfortunately the capture does not tell much. The first bulk-in failed under Linux. I will suggest you ask in linux-usb mailing list to get better help. 
http://vger.kernel.org/vger-lists.html#linux-usb

```
Frame 115: 27 bytes on wire (216 bits), 27 bytes captured (216 bits)
USB URB
    [Source: host]
    [Destination: 1.3.1]
    USBPcap pseudoheader length: 27
    IRP ID: 0xffffb08a631744e0
    IRP USBD_STATUS: Unknown (0xffffffff)
    URB Function: URB_FUNCTION_BULK_OR_INTERRUPT_TRANSFER (0x0009)
    IRP information: 0x00, Direction: FDO -> PDO
        0000 000. = Reserved: 0x00
        .... ...0 = Direction: FDO -> PDO (0x0)
    URB bus id: 1
    Device address: 3
    Endpoint: 0x81, Direction: IN
        1... .... = Direction: IN (1)
        .... 0001 = Endpoint number: 1
    URB transfer type: URB_BULK (0x03)
    Packet Data Length: 0
    [bInterfaceClass: Vendor Specific (0xff)]

```

ap--, 2021-09-28:
Thank you so much for your help @mcuee
I'll go and ask the `linux-usb` mailing list then.
In the meantime I will close this issue, and report back with whatever I find out.

Thanks again,
Andreas :smiley:

closed 2021-09-28 by ap--