pyusb/pyusb #391: [Errno 16] Resource busy

tronchama, 2021-09-28:
Bear with me a second, I know there has been lots of "Resource busy" issues in this project and the Internet, namely Stack Overflow, is also full to the brim with them. Nevertheless, I'd still need some help here because I'm genuinely lost.

My code is as follows and its objective is simple: Check which configuration is this device configured on currently and change it to the second one. The specific device I hard-coded does indeed has that second configuration available.

```
import usb.core
import usb.util
import usb.control

# find our device
dev = usb.core.find(idVendor=0x04e8, idProduct=0x6860)

# was it found?
if dev is None:
    raise ValueError('Device not found')
   
#test change config
try:
    cfg = dev.get_active_configuration()
except usb.core.USBError:
    cfg = None
if cfg is None or cfg.bConfigurationValue != 2:
    for intf in cfg:
    	print(intf)
    	print(dev.is_kernel_driver_active(intf.bInterfaceNumber))
    	dev.detach_kernel_driver(intf.bInterfaceNumber)
    	usb.util.claim_interface(dev, intf.bInterfaceNumber)

dev.set_configuration(2)
```
Code right now is not made to be 100% functional but to test this possible bug/user error. Its current issues are:

1. `print(intf)` prints the current interface.
2. `is_kernel_driver_active()` prints `False`.
3. `detach_kernel_driver()` throws `[Errno 2] Entity not found`.
4. `claim_interface() `throws `[Errno 16] Resource busy`.

As you can see, the code works erratically and, to the best of my knowledge, even contradicts itself. To be extra sure, I swapped my code a couple times and implemented several solutions, for example this one stated in Issue #76, [here](https://github.com/pyusb/pyusb/issues/76#issuecomment-118460796). This specific solution made my code jump straight to `dev.set_configuration(2)` because the condition at the innermost `if` loop was always `False`, which throws yet another `[Errno 16] Resource busy`. Aside from this error, `dmesg` also shows me the following when running the previous solution or jumping straight to `dev.set_configuration(2)`:

`[ 5483.405249] usb 1-4: usbfs: interface 0 claimed by usbfs while 'python3' sets config #2`

I've used pyusb before and I'm well aquainted with how it works, specifically control transfers, but to be extra careful, I decided to read [libusb's FAQ](https://github.com/libusb/libusb/wiki/FAQ#Can_I_run_libusb_applications_on_Linux_when_there_is_already_a_kernel_driver_attached_to_it), [pyusb's FAQ](https://github.com/pyusb/pyusb/blob/master/docs/faq.rst#how-not-to-call-set_configuration-on-a-device-already-configured-with-the-selected-configuration) and [Configuration selection and handling](https://libusb.sourceforge.io/api-1.0/libusb_caveats.html#configsel).

After making sure I understood everything I was doing, I decided to debug the code myself using `LIBUSB_DEBUG=4` and `PYUSB_DEBUG=debug`. This is my code's stderr along with the program's stdout (minus the prints to avoid visual cluttering):
```
2021-09-28 12:29:00,302 DEBUG:usb.backend.libusb1:_LibUSB.__init__(<CDLL 'libusb-1.0.so.0', handle b808b0 at 0x7f7d06cc3ee0>)
[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.000005] [00001b3a] libusb: debug [libusb_init] created default context
[ 0.000023] [00001b3a] libusb: debug [libusb_init] libusb v1.0.23.11397
[ 0.000052] [00001b3a] libusb: debug [find_usbfs_path] found usbfs at /dev/bus/usb
[ 0.000069] [00001b3a] libusb: debug [get_kernel_version] reported kernel version is 5.11.0-36-generic
[ 0.000074] [00001b3a] libusb: debug [op_init] bulk continuation flag supported
[ 0.000077] [00001b3a] libusb: debug [op_init] zero length packet flag supported
[ 0.000081] [00001b3a] libusb: debug [op_init] max iso packet length is (likely) 49152 bytes
[ 0.000094] [00001b3a] libusb: debug [op_init] sysfs can relate devices
[ 0.000099] [00001b3a] libusb: debug [op_init] sysfs has complete descriptors
[ 0.000574] [00001b3c] libusb: debug [linux_udev_event_thread_main] udev event thread entering.
[ 0.004934] [00001b3a] libusb: debug [linux_get_device_address] getting address for device: usb1 detached: 0
[ 0.004955] [00001b3a] libusb: debug [linux_get_device_address] scan usb1
[ 0.004989] [00001b3a] libusb: debug [linux_get_device_address] bus=1 dev=1
[ 0.004995] [00001b3a] libusb: debug [linux_enumerate_device] busnum 1 devaddr 1 session_id 257
[ 0.004999] [00001b3a] libusb: debug [linux_enumerate_device] allocating new device for 1/1 (session 257)
[ 0.005106] [00001b3a] libusb: debug [linux_get_device_address] getting address for device: 1-12 detached: 0
[ 0.005116] [00001b3a] libusb: debug [linux_get_device_address] scan 1-12
[ 0.005141] [00001b3a] libusb: debug [linux_get_device_address] bus=1 dev=3
[ 0.005145] [00001b3a] libusb: debug [linux_enumerate_device] busnum 1 devaddr 3 session_id 259
[ 0.005150] [00001b3a] libusb: debug [linux_enumerate_device] allocating new device for 1/3 (session 259)
[ 0.005176] [00001b3a] libusb: debug [linux_get_parent_info] Dev 0xba4220 (1-12) has parent 0xba43d0 (usb1) port 12
[ 0.005250] [00001b3a] libusb: debug [linux_get_device_address] getting address for device: 1-4 detached: 0
[... 298 lines cut ...]
[ 0.074745] [00001b94] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 0.074747] [00001b94] libusb: debug [handle_events] poll fds modified, reallocating
[ 0.074753] [00001b94] libusb: debug [handle_events] poll() 2 fds with timeout in 0ms
[ 0.074756] [00001b94] libusb: debug [handle_events] poll() returned 0
[ 0.074759] [00001b94] libusb: debug [libusb_unref_device] destroy device 6.1
[ 0.074761] [00001b94] libusb: debug [libusb_unref_device] destroy device 5.2
[ 0.074762] [00001b94] libusb: debug [libusb_unref_device] destroy device 5.1
[ 0.074764] [00001b94] libusb: debug [libusb_unref_device] destroy device 4.1
[ 0.074765] [00001b94] libusb: debug [libusb_unref_device] destroy device 3.2
[ 0.074766] [00001b94] libusb: debug [libusb_unref_device] destroy device 3.1
[ 0.074768] [00001b94] libusb: debug [libusb_unref_device] destroy device 2.1
[ 0.074770] [00001b94] libusb: debug [libusb_unref_device] destroy device 1.6
[ 0.074771] [00001b94] libusb: debug [libusb_unref_device] destroy device 1.3
[ 0.074773] [00001b94] libusb: debug [libusb_unref_device] destroy device 1.1
[ 0.074774] [00001b94] libusb: debug [usbi_remove_pollfd] remove fd 6
[ 0.074790] [00001b94] libusb: debug [usbi_remove_pollfd] remove fd 8
[ 0.074850] [00001b96] libusb: debug [linux_udev_event_thread_main] udev event thread exiting
```
Personally I couldn't find anything of relative importance in these logs. I'd be glad if someone saw something I missed. Nevertheless, I also tried to [manually unbind](https://lwn.net/Articles/143397/) the `usbfs` module stated in the previous `dmesg` logs without luck. Unmounting the device or manually unbinding it from /`sys/bus/usb/drivers/usb/ ` was also futile. I blacklisted a bunch of kernel modules as well, but again, it didn't work.

Curiously enough, I tested a [similar](https://github.com/ud2/advisories/blob/master/android/samsung/nocve-2016-0004/usbswitcher.c) C script that worked flawlessly with no errors and switched configurations as desired. Nevertheless, I really want to keep using Python for my personal project.

If you have any idea about this I'd be glad to hear it.

`Linux v-HP-Desktop-M01-F1xxx 5.11.0-36-generic #40~20.04.1-Ubuntu SMP Sat Sep 18 02:14:19 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux`

jonasmalacofilho, 2021-09-28:
Hi!

This seems to be a tricky one...  so starting slow with the couple of things I noted:

> * `print(intf)` prints the current interface.
> * `is_kernel_driver_active()` prints `False`.
> * `detach_kernel_driver()` throws `[Errno 2] Entity not found`.

So far this makes sense: if there is no kernel driver active for that interface, `libusb_detach_kernel_driver()` will return `LIBUSB_ERROR_NOT_FOUND` ([aka](https://github.com/pyusb/pyusb/blob/67550ed625e4252c0d586dd8ed5de681f4ad4edc/usb/backend/libusb1.py#L108) "Entity not found").

> * `claim_interface() `throws `[Errno 16] Resource busy`.

Well, assuming a scenario where `libusb_detach_kernel_driver()` succeeded or didn't get to run (or this line wouldn't have been reached), this could also happen if the interface is claimed by another user-space program.

This would also explain:

> This specific solution made my code jump straight to `dev.set_configuration(2)` because the condition at the innermost `if` loop was always `False`, which throws yet another `[Errno 16] Resource [...]

---

One extra note: _unconditionally_ calling `dev.set_configuration(2)` (assuming it eventually succeeds) [could](https://github.com/pyusb/pyusb/blob/master/docs/faq.rst#how-not-to-call-set_configuration-on-a-device-already-configured-with-the-selected-configuration) cause the device to soft reset, if it happened to already be in that configuration.

mcuee, 2021-09-29:
@tronchama 
It is interesting that the C code works. It is using libusb 0.1 API. What is the OS you are testing with? Is it Ubuntu/Debian with legacy libusb-0.1? Or other distro with libusb-compat-0.1?

It has an interesting but strange usb_reset() before usb_set_configuration().
```
usb_reset(udev);
  r = usb_set_configuration(udev, 2);
```

tronchama, 2021-09-29:
@mcuee 
I'm using Ubuntu 20.04, it's my main OS. I installed libusb-0.1 via aptitude so *I guess* it's the legacy version. 
Funnily enough, this C doesn't work without that cheeky `usb_reset()`.
`dmesg` logs without that reset:
```
[82358.837594] usb 1-2: usbfs: interface 0 claimed by usbfs while 'a.out' sets config #2
```
`dmesg` logs with that reset:
```
[82416.935116] usb 1-2: reset high-speed USB device number 22 using xhci_hcd
[82417.145566] cdc_acm 1-2:2.1: ttyACM0: USB ACM device
```
Just for fun I tried adding that same reset to my Python code and the result is the same (busy error), although with a different `dmesg` output.
```
[82655.323508] usb 1-2: reset high-speed USB device number 23 using xhci_hcd
[82655.499946] usb 1-2: usbfs: process 35623 (events) did not claim interface 0 before use
[82655.500535] usb 1-2: usbfs: interface 0 claimed by usbfs while 'python3' sets config #2
```
It looks like it reseted fine but the end result doesn't change. I tried changing my Python's script backend to libusb-0.1 and it gives me the same resource busy error.
I also tried to use `usb_modeswitch -v dead -p beef -d` and `usb_modeswitch -v dead -p beef -u 2` to try and detach/swap configs respectively but it gives me, yet again, a resource busy error.

@jonasmalacofilho 

> this could also happen if the interface is claimed by another user-space program.

I don't know what else is using this resource, tho. I've tried blacklisting `usbfs` but it's not a module I can remove from memory and thus, my blacklisting does nothing.
On your extra note, I'm aware of that and I'll make sure everything is called correctly and in time *but* I'll focus on making this work without problems first :)

jonasmalacofilho, 2021-09-29:
Regarding `usbfs`, it is what libusb0/libusb1 use to access devices from user-space.

tronchama, 2021-09-29:
Just out of curiosity I followed both executions with `strace` and saw that C's `usb_reset(udev);` is actually a call to `ioctl(fd, USBDEVFS_RESET, 0)`, which makes sense. 
Since the C script is currently functional, I tried imitating it using a `ioctl` call in my Python script instead of the standard `dev.reset()`, plus adding the `dev.set_configuration(2)` right after it, _a la C_. This is the code I changed:
```
fd = os.open("/dev/bus/usb/001/009", os.O_WRONLY)
try:
	fcntl.ioctl(fd, USBDEVFS_RESET, 0)
	try:
		dev.set_configuration(2)
	except:
		#stuff
finally:
	os.close(fd)
```
And its  `strace` snippet:
```
ioctl(9, USBDEVFS_RESET, 0)             = 0
openat(AT_FDCWD, "/dev/bus/usb/001/009", O_RDWR|O_CLOEXEC) = 10
ioctl(10, USBDEVFS_GET_CAPABILITIES, 0x1f808bc) = 0
ioctl(10, USBDEVFS_SETCONFIGURATION, 0x7ffeca5dbf9c) = -1 EBUSY (Device or resource busy)
```
This is another `strace` snippet, this time from the C script:
```
ioctl(3, USBDEVFS_RESET, 0)             = 0
ioctl(3, USBDEVFS_SETCONFIGURATION, 0x7ffd1d99249c) = 0
write(2, "[*] Configuration switched!\n", 28[*] Configuration switched!
```
Honestly, after lots of hours sunk in and thousands of debug lines read I can't really see where the problem is. I ran the initial scripts and its variations on an Arch machine and all of them worked flawlessly. 

I think I'll drop the ball here. It looks like a weird bug in the library or kernel shenanigans. Honestly I don't have much time to try and fix this up, but I'm willing to post more logs if you wish to take a look into it, just in case. I'll just use the C script and I'll start working from there. Thanks a lot for getting back to me and trying to help. I'm not closing down the issue just in case you require more logs or snippets.

jonasmalacofilho, 2021-09-29:
> I also tried to use `usb_modeswitch -v dead -p beef -d` and `usb_modeswitch -v dead -p beef -u 2` to try and detach/swap configs respectively but it gives me, yet again, a resource busy error.

`usb_modeswitch` uses libusb as well (from C) so this problem doesn't appear to be specific to PyUSB.

> [...] on an Arch machine and all of them worked flawlessly.
> It looks like a weird bug in the library or kernel shenanigans.

Agreed.

> I'm not closing down the issue just in case you require more logs or snippets.

Since it doesn't appear to be specific to PyUSB, and we also don't have a good way to debug this, I think I'm going to close the issue.

closed 2021-09-29 by jonasmalacofilho

tronchama, 2021-09-29:
Thanks a lot for your time.

mcuee, 2021-09-30:
@tronchama 

As already mentioned by Jonas, it does not seem to be related to pyusb, but rather libusb-1.0 related, since usb_modeswitch also does not work, yet legacy libusb-0.1 works.

You may want to try out two things.
1) force using the legacy libusb-0.1 backend with pyusb. This may or may not work.
2) You are using libusb-1.0.23, maybe you want tp upgrade libusb to 1.0.24 or even latest git to see if that helps.

tronchama, 2021-09-30:
@mcuee 
> 1. Force using the legacy libusb-0.1 backend with pyusb. This may or may not work.

I tried and it didn't work, sadly.

> 2. You are using libusb-1.0.23, maybe you want tp upgrade libusb to 1.0.24 or even latest git to see if that helps.

Will definitely try this out, thanks a lot.

multiplemonomials, 2024-02-25:
I was getting a similar error message and had to do some digging.  Dug all the way into the kernel source.  As it [turns out](https://elixir.bootlin.com/linux/latest/source/drivers/usb/core/devio.c#L1558), the usbfs driver in Linux has a limitation that if *any* interface of the device is claimed by a kernel driver, *no* interface can be claimed by a libusb process.  So, it's not enough to just call detach_kernel_driver() on the interface you want to use!  You have to iterate through all of the interfaces and detach the kernel driver on *all* of them.

jonasmalacofilho, 2024-02-25:
Thanks, @multiplemonomials !

That appears worth mentioning somewhere in our docs. At the same time, I'm curious as to why this isn't documented (AFAIK) by LibUSB.

ProfessorLongBeard, 2024-08-16:
> Sorry for necro-post but thought this might be useful for others who end up here. I was getting a similar error message and had to do some digging. Dug all the way into the kernel source. As it [turns out](https://elixir.bootlin.com/linux/latest/source/drivers/usb/core/devio.c#L1558), the usbfs driver in Linux has a limitation that if _any_ interface of the device is claimed by a kernel driver, _no_ interface can be claimed by a libusb process. So, it's not enough to just call detach_kernel_driver() on the interface you want to use! You have to iterate through all of the interfaces and detach the kernel driver on _all_ of them.

I'm so glad I seen this comment when I did. I was porting an Qualcomm EDL tool written in Python that will be in C but kept running into "resource busy". After reading though all this and looking at the link provided in your comment the way is now clear! Thank you folks for keeping this issue open. it truly helped me a lot