pyusb/pyusb #396: Segmentation fault in WSL

q3w3e3, 2021-11-08:
while im aware that WSL has limited (or a complete lack of?) usb support I have been attempting to use pyVISA for a non USB device but am running into a segfault when pyUSB is run on WSL see [this issue](https://github.com/pyvisa/pyvisa/issues/634)

Versions:

win 10
Ubuntu 20.04.3 LTS (GNU/Linux 4.4.0-19041-Microsoft x86_64)
Python 3.8.10
pyusb==1.0.2

Steps to reproduce: 

see above linked issue for in situ

or for minimum segfault that is likely caused by the same issue:

1: enter python shell
2: `>>> import usb.core`
3: `>>> import dev = usb.core.find()`
4: segfault!

again, I understand that WSL doesn't have proper/complete usb support but a segfault isn't fantastic behavior when other packages that may otherwise work rely on pyUSB

jonasmalacofilho, 2021-11-08:
Well, let's take a look...

First, please use PyUSB 1.2.1 or git HEAD.

Second, try running with `PYUSB_DEBUG=debug LIBUSB_DEBUG=4` and let me take a look at the complete output.

q3w3e3, 2021-11-08:
1.2.1 

doesnt segfault! (my bad running an out of date version)

```Python 3.8.10 (default, Sep 28 2021, 16:10:42)
[GCC 9.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import usb.core
>>> dev = usb.core.find()
2021-11-08 17:08:11,074 DEBUG:usb.backend.libusb1:_LibUSB.__init__(<CDLL 'libusb-1.0.so.0', handle d932d0 at 0x7f850154c790>)
[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.000018] [00000e6a] libusb: debug [libusb_init] created default context
[ 0.001244] [00000e6a] libusb: debug [libusb_init] libusb v1.0.23.11397
[ 0.003130] [00000e6a] libusb: debug [find_usbfs_path] found usbfs at /dev/bus/usb
[ 0.003743] [00000e6a] libusb: debug [get_kernel_version] reported kernel version is 4.4.0-19041-Microsoft
[ 0.004130] [00000e6a] libusb: debug [op_init] bulk continuation flag supported
[ 0.004498] [00000e6a] libusb: debug [op_init] zero length packet flag supported
[ 0.004964] [00000e6a] libusb: debug [op_init] max iso packet length is (likely) 49152 bytes
[ 0.005528] [00000e6a] libusb: warning [op_init] sysfs not mounted
[ 0.007096] [00000e6a] libusb: error [linux_udev_start_event_monitor] failed to enable the udev monitor
[ 0.007697] [00000e6a] libusb: error [op_init] error starting hotplug event monitor
2021-11-08 17:08:11,083 ERROR:usb.backend.libusb1:Error loading libusb 1.0 backend
Traceback (most recent call last):
  File "/home/q3w3e3/.local/lib/python3.8/site-packages/usb/backend/libusb1.py", line 963, in get_backend
    _lib_object = _LibUSB(_lib)
  File "/home/q3w3e3/.local/lib/python3.8/site-packages/usb/_debug.py", line 62, in do_trace
    return f(*args, **named_args)
  File "/home/q3w3e3/.local/lib/python3.8/site-packages/usb/backend/libusb1.py", line 717, in __init__
    _check(self.lib.libusb_init(byref(self.ctx)))
  File "/home/q3w3e3/.local/lib/python3.8/site-packages/usb/backend/libusb1.py", line 604, in _check
    raise USBError(_strerror(ret), ret, _libusb_errno[ret])
usb.core.USBError: [Errno None] Other error
2021-11-08 17:08:11,093 DEBUG:usb.backend.libusb1:_LibUSB._finalize_object()
2021-11-08 17:08:11,291 ERROR:usb.libloader:'OpenUSB library' could not be found
2021-11-08 17:08:11,292 ERROR:usb.backend.openusb:Error loading OpenUSB backend
2021-11-08 17:08:11,876 ERROR:usb.libloader:'Libusb 0' could not be found
2021-11-08 17:08:11,877 ERROR:usb.backend.libusb0:Error loading libusb 0.1 backend
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/q3w3e3/.local/lib/python3.8/site-packages/usb/core.py", line 1309, in find
    raise NoBackendError('No backend available')
usb.core.NoBackendError: No backend available```

and the relevant packages

```$ pip freeze | grep -i usb
libusb==1.0.23b7
libusb1==1.9.3
pyusb==1.2.1```

q3w3e3, 2021-11-08:
and the debug output using the old version of pyusb:

```
Python 3.8.10 (default, Sep 28 2021, 16:10:42)
[GCC 9.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import libusb
>>> import usb.core
>>> dev = usb.core.find()
2021-11-08 17:12:15,163 DEBUG:usb.backend.libusb1:_LibUSB.__init__(<CDLL 'libusb-1.0.so.0', handle 1d87970 at 0x7fdbcbef90a0>)
[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.000008] [00000ea4] libusb: debug [libusb_init] created default context
[ 0.001606] [00000ea4] libusb: debug [libusb_init] libusb v1.0.22.11312
[ 0.003238] [00000ea4] libusb: debug [find_usbfs_path] found usbfs at /dev/bus/usb
[ 0.003810] [00000ea4] libusb: debug [get_kernel_version] reported kernel version is 4.4.0-19041-Microsoft
[ 0.004292] [00000ea4] libusb: debug [op_init] bulk continuation flag supported
[ 0.004771] [00000ea4] libusb: debug [op_init] zero length packet flag supported
[ 0.005239] [00000ea4] libusb: debug [op_init] max iso packet length is (likely) 49152 bytes
[ 0.005770] [00000ea4] libusb: warning [op_init] sysfs not mounted
[ 0.006804] [00000ea4] libusb: error [linux_udev_start_event_monitor] failed to enable the udev monitor
[ 0.007349] [00000ea4] libusb: error [op_init] error starting hotplug event monitor
2021-11-08 17:12:15,172 ERROR:usb.backend.libusb1:Error loading libusb 1.0 backend
Traceback (most recent call last):
  File "/home/q3w3e3/.local/lib/python3.8/site-packages/usb/backend/libusb1.py", line 945, in get_backend
    return _LibUSB(_lib)
  File "/home/q3w3e3/.local/lib/python3.8/site-packages/usb/_debug.py", line 60, in do_trace
    return f(*args, **named_args)
  File "/home/q3w3e3/.local/lib/python3.8/site-packages/usb/backend/libusb1.py", line 708, in __init__
    _check(self.lib.libusb_init(byref(self.ctx)))
  File "/home/q3w3e3/.local/lib/python3.8/site-packages/usb/backend/libusb1.py", line 595, in _check
    raise USBError(_strerror(ret), ret, _libusb_errno[ret])
usb.core.USBError: [Errno None] Other error
2021-11-08 17:12:15,183 DEBUG:usb.backend.libusb1:_LibUSB._finalize_object()
[ 0.020177] [00000ea4] libusb: debug [libusb_exit]
[ 0.020619] [00000ea4] libusb: debug [libusb_exit] destroying default context
Segmentation fault (core dumped)
```

jonasmalacofilho, 2021-11-08:
> 1.2.1 doesnt segfault! (my bad running an out of date version)

No worries...

> ```
> [...]
> [ 0.005528] [00000e6a] libusb: warning [op_init] sysfs not mounted
> [ 0.007096] [00000e6a] libusb: error [linux_udev_start_event_monitor] failed to enable the udev monitor
> [ 0.007697] [00000e6a] libusb: error [op_init] error starting hotplug event monitor
> [...]
> ```

Now you need to check whether there's something easy you can do for libusb1 to work under WSL.

I suppose you could also try libusb0 or openusb, although that's unlikely.

> ```
> 2021-11-08 17:12:15,183 DEBUG:usb.backend.libusb1:_LibUSB._finalize_object()
> [ 0.020177] [00000ea4] libusb: debug [libusb_exit]
> [ 0.020619] [00000ea4] libusb: debug [libusb_exit] destroying default context
> Segmentation fault (core dumped)
> ```

Oh, this was fixed in 1.1.0 (#232, PR #279).

q3w3e3, 2021-11-08:
afaik libusb will not work in wsl, thanks! having no segfault post PR 279 should let me fix the issues i was running into in pyvisa.

closed 2021-11-08 by q3w3e3

mcuee, 2022-02-02:
Ref: WSL does not work with libusb, at least not yet.
https://github.com/microsoft/WSL/issues/2195