pyusb/pyusb #397: Issue with reset.dev()

ahakobyan79, 2021-11-23:
Trying to create a simple listener to USB signals. Here is my code:
`import usb.core

dev=usb.core.find(idVendor=0x1235, idProduct=0x0036)
ep=dev[0].interfaces()[1].endpoints()[0]
i=dev[0].interfaces()[1].bInterfaceNumber
dev.reset()

if dev.is_kernel_driver_active(i):
    dev.detach_kernel_driver(i)

dev.set_configuration()
eaddr=ep.bEndpointAddress

r=dev.read(eaddr,1024)
print(len(r))`

When I run this, I get this error message:

`root@LaunchPad:~# python3 usbtest.py 
Traceback (most recent call last):
  File "usbtest.py", line 13, in <module>
    dev.reset()
  File "/usr/local/lib/python3.8/dist-packages/usb/core.py", line 959, in reset
    self._ctx.managed_open()
  File "/usr/local/lib/python3.8/dist-packages/usb/core.py", line 113, in wrapper
    return f(self, *args, **kwargs)
  File "/usr/local/lib/python3.8/dist-packages/usb/core.py", line 131, in managed_open
    self.handle = self.backend.open_device(self.dev)
  File "/usr/local/lib/python3.8/dist-packages/usb/backend/libusb1.py", line 804, in open_device
    return _DeviceHandle(dev)
  File "/usr/local/lib/python3.8/dist-packages/usb/backend/libusb1.py", line 652, in __init__
    _check(_lib.libusb_open(self.devid, byref(self.handle)))
  File "/usr/local/lib/python3.8/dist-packages/usb/backend/libusb1.py", line 604, in _check
    raise USBError(_strerror(ret), ret, _libusb_errno[ret])
usb.core.USBError: [Errno 19] No such device (it may have been disconnected)`

I can't figure out what's causing it. Please help.

ahakobyan79, 2021-11-23:
To provide more info:
I am running the code on a Proxmox Container with Ubuntu 20.04.
I do see the device when run: lsusb
If the device was not really connected, it would not get to the dev.reset() line.

jonasmalacofilho, 2021-11-23:
This is expected behavior for `Device.reset()`/`libusb_reset_device()`:

> If the reset fails, the descriptors change, or the previous state cannot be restored, the device will appear to be disconnected and reconnected. This means that the device handle is no longer valid (you should close it) and rediscover the device. A return code of LIBUSB_ERROR_NOT_FOUND indicates when this is the case.
> —https://libusb.sourceforge.io/api-1.0/group__libusb__dev.html#gafee9c4638f1713ca5faa867948878111

On a slight tangent, do you really need to `reset()`?

ahakobyan79, 2021-11-24:
This is the result after removing the dev.reset().

```
root@LaunchPad:~# python3 usbtest.py 
Traceback (most recent call last):
  File "usbtest.py", line 15, in <module>
    if dev.is_kernel_driver_active(i):
  File "/usr/local/lib/python3.8/dist-packages/usb/core.py", line 1107, in is_kernel_driver_active
    self._ctx.managed_open()
  File "/usr/local/lib/python3.8/dist-packages/usb/core.py", line 113, in wrapper
    return f(self, *args, **kwargs)
  File "/usr/local/lib/python3.8/dist-packages/usb/core.py", line 131, in managed_open
    self.handle = self.backend.open_device(self.dev)
  File "/usr/local/lib/python3.8/dist-packages/usb/backend/libusb1.py", line 804, in open_device
    return _DeviceHandle(dev)
  File "/usr/local/lib/python3.8/dist-packages/usb/backend/libusb1.py", line 652, in __init__
    _check(_lib.libusb_open(self.devid, byref(self.handle)))
  File "/usr/local/lib/python3.8/dist-packages/usb/backend/libusb1.py", line 604, in _check
    raise USBError(_strerror(ret), ret, _libusb_errno[ret])
usb.core.USBError: [Errno 19] No such device (it may have been disconnected)
```

jonasmalacofilho, 2021-11-24:
That's weird.  Can you provide the complete PyUSB + LibUSB debug output (run with `PYUSB_DEBUG=debug LIBUSB_DEBUG=4`)?

Also, I see that the line numbers in the tracebacks don't exactly match the snippet you posted, there are 7 missing lines.  Can you show the rest of the `usbtest.py` script?

ahakobyan79, 2021-11-24:
> 

Thank you for your interest to help with this.
In order to better understand what's going on, let me give some back story and more details:

I'm trying to read Launchpad Mini data from usb. I once used the exact same code above with a Ubuntu 20.04 Proxmox VM, added the USB device to the VM, and using the code above was able to read the inputs.
I am having to rebuild everything from scratch since my last VM crashed and I didn't have backups :(

Currently I'm trying to use a container instead, even though I tried replicating my steps from before, but I'm getting the same exact issue with a VM instead of CT.

When I run "lsusb", this is the list I get:
```
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 006 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 005 Device 002: ID 046d:c52b Logitech, Inc. Unifying Receiver
Bus 005 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
**Bus 002 Device 002: ID 1235:0036 Focusrite-Novation Launchpad Mini**
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

Here is the full code(i just removed the commented area):

```
import usb.core

dev=usb.core.find(idVendor=0x1235, idProduct=0x0036)

#print('==================================================================')
#print(dev[0].interfaces())
#print('-----------------------------')
#print(dev[0].interfaces()[1].endpoints()[0])
#print('==================================================================')

ep=dev[0].interfaces()[1].endpoints()[0]
i=dev[0].interfaces()[1].bInterfaceNumber
#dev.reset()

if dev.is_kernel_driver_active(i):
    dev.detach_kernel_driver(i)

#dev.set_configuration()
eaddr=ep.bEndpointAddress

r=dev.read(eaddr,1024)
print(len(r))
```

The only difference on proxmox between VM and CT is that CT sees all usb devices when i run "lsusb", in case of VM, I have to add USB port as a hardware piece to VM in order for it to see the USB when I run "lsusb" in the VM console.

I am VERY new to Python, and this i my first ever project using python.

I will provide the log data in my next post.

ahakobyan79, 2021-11-25:
```
root@LaunchPad:~# PYUSB_DEBUG=debug LIBUSB_DEBUG=4 python3 usbtest.py 
2021-11-25 18:29:47,145 DEBUG:usb.backend.libusb1:_LibUSB.__init__(<CDLL 'libusb-1.0.so.0', handle 29c83e0 at 0x7f932b1c9f40>)
[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.000012] [000003fa] libusb: debug [libusb_init] created default context
[ 0.000048] [000003fa] libusb: debug [libusb_init] libusb v1.0.23.11397
[ 0.000178] [000003fa] libusb: debug [find_usbfs_path] found usbfs at /dev/bus/usb
[ 0.000204] [000003fa] libusb: debug [get_kernel_version] reported kernel version is 5.4.73-1-pve
[ 0.000290] [000003fa] libusb: debug [op_init] bulk continuation flag supported
[ 0.000300] [000003fa] libusb: debug [op_init] zero length packet flag supported
[ 0.000308] [000003fa] libusb: debug [op_init] max iso packet length is (likely) 49152 bytes
[ 0.000337] [000003fa] libusb: debug [op_init] sysfs can relate devices
[ 0.000346] [000003fa] libusb: debug [op_init] sysfs has complete descriptors
[ 0.003131] [000003fc] libusb: debug [linux_udev_event_thread_main] udev event thread entering.
[ 0.010628] [000003fa] libusb: debug [linux_get_device_address] getting address for device: usb2 detached: 0
[ 0.010677] [000003fa] libusb: debug [linux_get_device_address] scan usb2
[ 0.010811] [000003fa] libusb: debug [linux_get_device_address] bus=2 dev=1
[ 0.010831] [000003fa] libusb: debug [linux_enumerate_device] busnum 2 devaddr 1 session_id 513
[ 0.010857] [000003fa] libusb: debug [linux_enumerate_device] allocating new device for 2/1 (session 513)
[ 0.011188] [000003fa] libusb: debug [linux_get_device_address] getting address for device: 2-2 detached: 0
[ 0.011207] [000003fa] libusb: debug [linux_get_device_address] scan 2-2
[ 0.011323] [000003fa] libusb: debug [linux_get_device_address] bus=2 dev=2
[ 0.011341] [000003fa] libusb: debug [linux_enumerate_device] busnum 2 devaddr 2 session_id 514
[ 0.011360] [000003fa] libusb: debug [linux_enumerate_device] allocating new device for 2/2 (session 514)
[ 0.011491] [000003fa] libusb: debug [linux_get_parent_info] Dev 0x28bcff0 (2-2) has parent 0x291a300 (usb2) port 2
[ 0.011709] [000003fa] libusb: debug [linux_get_device_address] getting address for device: usb4 detached: 0
[ 0.011729] [000003fa] libusb: debug [linux_get_device_address] scan usb4
[ 0.011826] [000003fa] libusb: debug [linux_get_device_address] bus=4 dev=1
[ 0.011850] [000003fa] libusb: debug [linux_enumerate_device] busnum 4 devaddr 1 session_id 1025
[ 0.011862] [000003fa] libusb: debug [linux_enumerate_device] allocating new device for 4/1 (session 1025)
[ 0.012151] [000003fa] libusb: debug [linux_get_device_address] getting address for device: usb5 detached: 0
[ 0.012175] [000003fa] libusb: debug [linux_get_device_address] scan usb5
[ 0.012282] [000003fa] libusb: debug [linux_get_device_address] bus=5 dev=1
[ 0.012295] [000003fa] libusb: debug [linux_enumerate_device] busnum 5 devaddr 1 session_id 1281
[ 0.012305] [000003fa] libusb: debug [linux_enumerate_device] allocating new device for 5/1 (session 1281)
[ 0.012631] [000003fa] libusb: debug [linux_get_device_address] getting address for device: 5-4 detached: 0
[ 0.012656] [000003fa] libusb: debug [linux_get_device_address] scan 5-4
[ 0.012756] [000003fa] libusb: debug [linux_get_device_address] bus=5 dev=2
[ 0.012774] [000003fa] libusb: debug [linux_enumerate_device] busnum 5 devaddr 2 session_id 1282
[ 0.012785] [000003fa] libusb: debug [linux_enumerate_device] allocating new device for 5/2 (session 1282)
[ 0.012879] [000003fa] libusb: debug [linux_get_parent_info] Dev 0x2965990 (5-4) has parent 0x29658a0 (usb5) port 4
[ 0.013136] [000003fa] libusb: debug [linux_get_device_address] getting address for device: usb1 detached: 0
[ 0.013156] [000003fa] libusb: debug [linux_get_device_address] scan usb1
[ 0.013275] [000003fa] libusb: debug [linux_get_device_address] bus=1 dev=1
[ 0.013293] [000003fa] libusb: debug [linux_enumerate_device] busnum 1 devaddr 1 session_id 257
[ 0.013315] [000003fa] libusb: debug [linux_enumerate_device] allocating new device for 1/1 (session 257)
[ 0.013604] [000003fa] libusb: debug [linux_get_device_address] getting address for device: usb6 detached: 0
[ 0.013624] [000003fa] libusb: debug [linux_get_device_address] scan usb6
[ 0.013735] [000003fa] libusb: debug [linux_get_device_address] bus=6 dev=1
[ 0.013753] [000003fa] libusb: debug [linux_enumerate_device] busnum 6 devaddr 1 session_id 1537
[ 0.013776] [000003fa] libusb: debug [linux_enumerate_device] allocating new device for 6/1 (session 1537)
[ 0.014089] [000003fa] libusb: debug [linux_get_device_address] getting address for device: usb3 detached: 0
[ 0.014104] [000003fa] libusb: debug [linux_get_device_address] scan usb3
[ 0.014200] [000003fa] libusb: debug [linux_get_device_address] bus=3 dev=1
[ 0.014221] [000003fa] libusb: debug [linux_enumerate_device] busnum 3 devaddr 1 session_id 769
[ 0.014243] [000003fa] libusb: debug [linux_enumerate_device] allocating new device for 3/1 (session 769)
[ 0.014453] [000003fa] libusb: debug [usbi_add_pollfd] add fd 6 events 1
[ 0.014493] [000003fa] libusb: debug [usbi_io_init] using timerfd for timeouts
[ 0.014505] [000003fa] libusb: debug [usbi_add_pollfd] add fd 8 events 1
2021-11-25 18:29:47,160 INFO:usb.core:find(): using backend "usb.backend.libusb1"
2021-11-25 18:29:47,160 DEBUG:usb.backend.libusb1:_LibUSB.enumerate_devices()
[... cut ...]
[ 0.047122] [000003fa] libusb: debug [libusb_unref_device] destroy device 5.2
[ 0.047131] [000003fa] libusb: debug [libusb_unref_device] destroy device 5.1
[ 0.047141] [000003fa] libusb: debug [libusb_unref_device] destroy device 4.1
[ 0.047149] [000003fa] libusb: debug [libusb_unref_device] destroy device 2.1
[ 0.047159] [000003fa] libusb: debug [usbi_remove_pollfd] remove fd 6
[ 0.047205] [000003fa] libusb: debug [usbi_remove_pollfd] remove fd 8
[ 0.047302] [000003fc] libusb: debug [linux_udev_event_thread_main] udev event thread exiting
```

jonasmalacofilho, 2021-11-26:
This appears to be an issue with libusb and/or your host's USB stack.  Can you report it to libusb?

mcuee, 2021-11-30:
Does you device work under a non-VM Linux machine? Take Proxmox out the equation first to see if that helps.

mcuee, 2021-12-11:
@ahakobyan79 As this does not look like a pyusb issue, but rather a host USB issue or maybe a libusb issue, I will close this one. Please re-open if you came back with more information. Thanks.

closed 2021-12-11 by mcuee

ahakobyan79, 2021-12-11:
> Does you device work under a non-VM Linux machine? Take Proxmox out the equation first to see if that helps.

The device works for sure. I've installed USB Analyzer on my Windows pc, and I'm able to read every key, and the data coming in is exactly the same as when it worked just fine with a VM.

mcuee, 2021-12-11:
> > Does you device work under a non-VM Linux machine? Take Proxmox out the equation first to see if that helps.
> 
> The device works for sure. I've installed USB Analyzer on my Windows pc, and I'm able to read every key, and the data coming in is exactly the same as when it worked just fine with a VM.

Good that you know the device works under your Windows PC. 
And you say it worked fine with a VM, which VM? 

> Proxmox Container with Ubuntu 20.04

So you are not using a VM under Proxmox, but rather a LXC container? If that is the case, you may want to go to libusb to create an issue there to see if people can help you. It may be a bit tricky to use libusb under a Docker or LXC container?