pyusb/pyusb #398: pyusb segfaults under macOS with devices stalling on a SET_INTERFACE request

mcuee, 2021-12-01:
https://github.com/libusb/libusb/issues/1011#issuecomment-983208629

> For reference, we can reproduce this issue with a STM32F4-Discovery board, or SparkFun Thing Plus board, using a [libopencm3 example](https://github.com/libopencm3/libopencm3-examples/tree/master/examples/stm32/f4/stm32f4-discovery/usb_cdcacm) modified with this patch, which makes it stall the pipe on a SET_INTERFACE request. [fw_set_interface_stall.patch.txt](https://github.com/libusb/libusb/files/7628519/fw_set_interface_stall.patch.txt)

Just a run log for reference using pyusb under my Mac Mini M1. Not so sure why pyusb segfaults though. I will create an issue with pyusb.

```
mcuee@mcuees-Mac-mini ~ % export LIBUSB_DEBUG=4
mcuee@mcuees-Mac-mini ~ % python3
Python 3.9.8 (main, Nov 10 2021, 03:48:35)
[Clang 13.0.0 (clang-1300.0.29.3)] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> import usb
>>> dev = usb.core.find(idVendor=0x0483)
[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.000013] [00004d43] libusb: debug [libusb_init] libusb v1.0.24.11675
...

>>> dev.set_configuration()
[ 3.254908] [000075b5] libusb: debug [libusb_get_config_descriptor] index 0
[ 3.255204] [000075b5] libusb: debug [libusb_open] open 2.20
[ 3.255456] [000075b5] libusb: debug [darwin_open] device open for access
[ 3.255489] [000075b5] libusb: debug [libusb_set_configuration] configuration 1

>>> cfg = dev.get_active_configuration()
[15.680978] [000075b5] libusb: debug [libusb_get_config_descriptor] index 0
>>> print(cfg)
[21.631005] [000075b5] libusb: debug [libusb_get_config_descriptor] index 0
[21.631410] [000075b5] libusb: debug [libusb_get_config_descriptor] index 0
[21.631596] [000075b5] libusb: debug [libusb_get_config_descriptor] index 0
[21.631676] [000075b5] libusb: debug [libusb_get_config_descriptor] index 0
[21.631780] [000075b5] libusb: debug [libusb_get_config_descriptor] index 0
[21.631976] [000075b5] libusb: debug [libusb_get_config_descriptor] index 0
[21.632184] [000075b5] libusb: debug [libusb_get_config_descriptor] index 0
  CONFIGURATION 1: 100 mA ==================================
   bLength              :    0x9 (9 bytes)
   bDescriptorType      :    0x2 Configuration
   wTotalLength         :   0x43 (67 bytes)
   bNumInterfaces       :    0x2
   bConfigurationValue  :    0x1
   iConfiguration       :    0x0
   bmAttributes         :   0x80 Bus Powered
   bMaxPower            :   0x32 (100 mA)
    INTERFACE 0: Reserved ==================================
     bLength            :    0x9 (9 bytes)
     bDescriptorType    :    0x4 Interface
     bInterfaceNumber   :    0x0
     bAlternateSetting  :    0x0
     bNumEndpoints      :    0x1
     bInterfaceClass    :    0x0 Reserved
     bInterfaceSubClass :    0x2
     bInterfaceProtocol :    0x1
     iInterface         :    0x0
      ENDPOINT 0x83: Interrupt IN ==========================
       bLength          :    0x7 (7 bytes)
       bDescriptorType  :    0x5 Endpoint
       bEndpointAddress :   0x83 IN
       bmAttributes     :    0x3 Interrupt
       wMaxPacketSize   :   0x10 (16 bytes)
       bInterval        :   0xff
    INTERFACE 1: CDC Data ==================================
     bLength            :    0x9 (9 bytes)
     bDescriptorType    :    0x4 Interface
     bInterfaceNumber   :    0x1
     bAlternateSetting  :    0x0
     bNumEndpoints      :    0x2
     bInterfaceClass    :    0xa CDC Data
     bInterfaceSubClass :    0x0
     bInterfaceProtocol :    0x0
     iInterface         :    0x0
      ENDPOINT 0x1: Bulk OUT ===============================
       bLength          :    0x7 (7 bytes)
       bDescriptorType  :    0x5 Endpoint
       bEndpointAddress :    0x1 OUT
       bmAttributes     :    0x2 Bulk
       wMaxPacketSize   :   0x40 (64 bytes)
       bInterval        :    0x1
      ENDPOINT 0x82: Bulk IN ===============================
       bLength          :    0x7 (7 bytes)
       bDescriptorType  :    0x5 Endpoint
       bEndpointAddress :   0x82 IN
       bmAttributes     :    0x2 Bulk
       wMaxPacketSize   :   0x40 (64 bytes)
       bInterval        :    0x1

>>> dev.set_interface_altsetting(interface = 0, alternate_setting = 0)
[65.169062] [000075b5] libusb: debug [libusb_get_config_descriptor] index 0
[65.169440] [000075b5] libusb: debug [libusb_get_config_descriptor] index 0
[65.169628] [000075b5] libusb: debug [libusb_claim_interface] interface 0
[65.170967] [000075b5] libusb: debug [get_endpoints] building table of
endpoints.
[65.171743] [000075b5] libusb: debug [get_endpoints] interface: 0 pipe
1: dir: 1 number: 3
[65.171813] [000075b5] libusb: debug [darwin_claim_interface] interface opened
[65.171847] [000075b5] libusb: debug
[libusb_set_interface_alt_setting] interface 0 altsetting 0
[65.176102] [000075b5] libusb: warning
[darwin_set_interface_altsetting] SetAlternateInterface: pipe is
stalled
[65.176876] [000075b5] libusb: info [darwin_claim_interface]
USBInterfaceOpen: another process has device opened for exclusive
access
[65.177150] [000075b5] libusb: error [darwin_set_interface_altsetting]
could not reclaim interface
zsh: segmentation fault  python3
```

jonasmalacofilho, 2021-12-01:
Can you get a log with `PYUSB_DEBUG=debug` set?

Also, what version of PyUSB is being used?

mcuee, 2021-12-01:
> Can you get a log with `PYUSB_DEBUG=debug` set?
Yes I will post a log this evening.

> Also, what version of PyUSB is being used?
Git head version of pyusb.

mcuee, 2021-12-01:
Full debug log with libusb git head and pyusb githead under my Mac Mini M1 (macOS 12.0.1).

```

mcuee@mcuees-Mac-mini pyusb % export PYUSB_DEBUG=debug
mcuee@mcuees-Mac-mini pyusb % export LIBUSB_DEBUG=4   
mcuee@mcuees-Mac-mini pyusb % python3
Python 3.9.8 (main, Nov 10 2021, 03:48:35) 
[Clang 13.0.0 (clang-1300.0.29.3)] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> import usb
>>> dev = usb.core.find(idVendor=0x0483)
2021-12-01 12:18:12,632 DEBUG:usb.backend.libusb1:_LibUSB.__init__(<CDLL '/opt/homebrew/lib/libusb-1.0.dylib', handle 200981ae0 at 0x1010da9d0>)
[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.000034] [00005b2c] libusb: debug [libusb_init] libusb v1.0.24.11679
[ 0.000081] [00005b2c] libusb: debug [usbi_add_event_source] add fd 3 events 1
...

>>> dev.set_interface_altsetting(interface = 0, alternate_setting = 0)
2021-12-01 12:18:28,826 DEBUG:usb.backend.libusb1:_LibUSB.open_device(<usb.backend.libusb1._Device object at 0x1010be100>)
[16.192675] [00005b2c] libusb: debug [libusb_open] open 2.11
[16.192879] [00005b2c] libusb: debug [darwin_open] device open for access
2021-12-01 12:18:28,827 DEBUG:usb.backend.libusb1:_LibUSB.get_configuration(<usb.backend.libusb1._DeviceHandle object at 0x1010d6730>)
[16.193033] [00005b2c] libusb: debug [libusb_get_configuration]  
[16.193039] [00005b2c] libusb: debug [libusb_get_configuration] active config 1
2021-12-01 12:18:28,827 DEBUG:usb.backend.libusb1:_LibUSB.get_configuration_descriptor(<usb.backend.libusb1._Device object at 0x1010be100>, 0)
[16.193200] [00005b2c] libusb: debug [libusb_get_config_descriptor] index 0
2021-12-01 12:18:28,827 DEBUG:usb.backend.libusb1:_LibUSB.get_interface_descriptor(<usb.backend.libusb1._Device object at 0x1010be100>, 0, 0, 0)
2021-12-01 12:18:28,827 DEBUG:usb.backend.libusb1:_LibUSB.get_configuration_descriptor(<usb.backend.libusb1._Device object at 0x1010be100>, 0)
[16.193574] [00005b2c] libusb: debug [libusb_get_config_descriptor] index 0
2021-12-01 12:18:28,828 DEBUG:usb.backend.libusb1:_LibUSB.claim_interface(<usb.backend.libusb1._DeviceHandle object at 0x1010d6730>, 0)
[16.193788] [00005b2c] libusb: debug [libusb_claim_interface] interface 0
[16.194961] [00005b2c] libusb: debug [get_endpoints] building table of endpoints.
[16.195839] [00005b2c] libusb: debug [get_endpoints] interface: 0 pipe 1: dir: 1 number: 3
[16.195889] [00005b2c] libusb: debug [darwin_claim_interface] interface opened
2021-12-01 12:18:28,830 DEBUG:usb.backend.libusb1:_LibUSB.set_interface_altsetting(<usb.backend.libusb1._DeviceHandle object at 0x1010d6730>, 0, 0)
[16.196016] [00005b2c] libusb: debug [libusb_set_interface_alt_setting] interface 0 altsetting 0
[16.200269] [00005b2c] libusb: warning [darwin_set_interface_altsetting] SetAlternateInterface: pipe is stalled
[16.200879] [00005b2c] libusb: info [darwin_claim_interface] USBInterfaceOpen: another process has device opened for exclusive access
[16.201137] [00005b2c] libusb: error [darwin_set_interface_altsetting] could not reclaim interface
zsh: segmentation fault  python3
```

mcuee, 2021-12-01:
Once I use libusb git head with the fix in https://github.com/libusb/libusb/pull/1031, then no more segement fault issue.

```
The run log with #1031 is okay.

```
mcuee@mcuees-Mac-mini pyusb % export PYUSB_DEBUG=debug

mcuee@mcuees-Mac-mini pyusb % export LIBUSB_DEBUG=4 

mcuee@mcuees-Mac-mini pyusb % python3
Python 3.9.8 (main, Nov 10 2021, 03:48:35) 
[Clang 13.0.0 (clang-1300.0.29.3)] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> import usb
>>> dev = usb.core.find(idVendor=0x0483)
2021-12-01 12:28:36,438 DEBUG:usb.backend.libusb1:_LibUSB.__init__(<CDLL '/opt/homebrew/lib/libusb-1.0.dylib', handle 201c59ae0 at 0x1014a2ee0>)
[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.000028] [00009b5b] libusb: debug [libusb_init] libusb v1.0.24.11680
...
>>> dev.set_interface_altsetting(interface = 0, alternate_setting = 0)
2021-12-01 12:28:46,427 DEBUG:usb.backend.libusb1:_LibUSB.open_device(<usb.backend.libusb1._Device object at 0x101a29dc0>)
[ 9.988819] [00009b5b] libusb: debug [libusb_open] open 2.11
[ 9.989057] [00009b5b] libusb: debug [darwin_open] device open for access
2021-12-01 12:28:46,427 DEBUG:usb.backend.libusb1:_LibUSB.get_configuration(<usb.backend.libusb1._DeviceHandle object at 0x1012c4190>)
[ 9.989219] [00009b5b] libusb: debug [libusb_get_configuration]  
[ 9.989226] [00009b5b] libusb: debug [libusb_get_configuration] active config 1
2021-12-01 12:28:46,427 DEBUG:usb.backend.libusb1:_LibUSB.get_configuration_descriptor(<usb.backend.libusb1._Device object at 0x101a29dc0>, 0)
[ 9.989414] [00009b5b] libusb: debug [libusb_get_config_descriptor] index 0
2021-12-01 12:28:46,428 DEBUG:usb.backend.libusb1:_LibUSB.get_interface_descriptor(<usb.backend.libusb1._Device object at 0x101a29dc0>, 0, 0, 0)
2021-12-01 12:28:46,428 DEBUG:usb.backend.libusb1:_LibUSB.get_configuration_descriptor(<usb.backend.libusb1._Device object at 0x101a29dc0>, 0)
[ 9.989885] [00009b5b] libusb: debug [libusb_get_config_descriptor] index 0
2021-12-01 12:28:46,428 DEBUG:usb.backend.libusb1:_LibUSB.claim_interface(<usb.backend.libusb1._DeviceHandle object at 0x1012c4190>, 0)
[ 9.990141] [00009b5b] libusb: debug [libusb_claim_interface] interface 0
[ 9.991458] [00009b5b] libusb: debug [get_endpoints] building table of endpoints.
[ 9.992359] [00009b5b] libusb: debug [get_endpoints] interface: 0 pipe 1: dir: 1 number: 3
[ 9.992403] [00009b5b] libusb: debug [darwin_claim_interface] interface opened
2021-12-01 12:28:46,431 DEBUG:usb.backend.libusb1:_LibUSB.set_interface_altsetting(<usb.backend.libusb1._DeviceHandle object at 0x1012c4190>, 0, 0)
[ 9.992517] [00009b5b] libusb: debug [libusb_set_interface_alt_setting] interface 0 altsetting 0
[ 9.996578] [00009b5b] libusb: warning [darwin_set_interface_altsetting] SetAlternateInterface: pipe is stalled
[ 9.996594] [00009b5b] libusb: debug [ep_to_pipeRef] converting ep address 0x83 to pipeRef and interface
[ 9.996598] [00009b5b] libusb: debug [ep_to_pipeRef] pipe 1 on interface 0 matches
```
```

tormodvolden, 2021-12-01:
Oops, the stack trace indicates the crash is in libusb-1.0.0.dylib. From that it seems more a libusb bug than pyusb (or python).

tormodvolden, 2021-12-01:
We should try to run it in a debugger (e.g. gdb) to see what happens in darwin_set_interface_altsetting(). Why I first thought that it had to do with pyusb, is that Nathan didn't get any crash when doing the same from C.

mcuee, 2021-12-01:
Close this one first since it is likely because of libusb and not pyusb. Will continue the discussion in libusb.

mcuee, 2021-12-01:
We can always reopen if later we think there are changes required for pyusb.

closed 2021-12-01 by mcuee