pyusb/pyusb #399: [capability] un/bind USB

130s, 2021-12-03:
As I asked in [stackoverflow.com#q70217666](https://stackoverflow.com/questions/70217666), I'm seeing un/bind a USB device is not possible with just a single or few commands IINM using `PyUSB`.

In the referenced SO thread I had to use convoluted and error-prone `bash` commands on Linux.

Would it make sense to add a capability to PyUSB to un/bind a given device?

jonasmalacofilho, 2021-12-04:
The use case appears to be mimicking the physical removal of USB device, by unbinding it from usb _core:_

```
DEVICE=$(grep 064f /sys/bus/usb/devices/*/idVendor | tr '/' ' ' | awk '{ print $5 }')
/bin/bash -c "echo $DEVICE >/sys/bus/usb/drivers/usb/unbind"
```

---

The reason why `usb.core.Device.detach_kernel_driver()` doesn't work for this purpose is that its use case is unbinding USB _interface_ drivers from the device, allowing LibUSB (using `usbfs`) to bind to them.  It doesn't attempt to remove the core USB device driver.

But it's not clear to me how "much" you want the device to disappear.  If PyUSB had an API that made a device disappear from itself, how would anyone later find the device to bring it back (without then resorting to `sysfs`)?

Also, with my particular kernel version (`5.15.5-arch1-1`), unbinding the `usb` driver does _not_ make the device disappear from `lsusb` or PyUSB; this seems to conflict with what you reported on SO, and may already not match your use case.

So why do you need to mimic the physical removal of the device?

P.S. You probably already know this, but if you just want to reset the device, `usb.core.Device.reset()` (which, if you're using libusb-1.0, calls [`libusb_reset_device()`](https://libusb.sourceforge.io/api-1.0/group__libusb__dev.html#gafee9c4638f1713ca5faa867948878111)) should be enough.

130s, 2021-12-04:
Thanks.

> Also, with my particular kernel version (`5.15.5-arch1-1`), unbinding the `usb` driver does _not_ make the device disappear from `lsusb` or PyUSB; this seems to conflict with what you reported on SO, and may already not match your use case.

Yeah...that seems to be the case. I just confirmed that the behavior (from a perspective of USB user who's not knowledgeable about underlining technical terms and concepts) of PyUSB's `{at/de}tach_kernel_driver` and the `bash` commands I posted on SO looks similar/the same (I think I haven't verified enough behavior of my `bash` commands) -- Both make the volume on the USB disappear and get it back while the USB remains present on `lsusb`.

So, I guess I'll try utilizing `{at/de}tach_kernel_driver` and see if that meets my need. But closing for now with a hope that the capability in the of this ticket is already present.

> So why do you need to mimic the physical removal of the device?

Because you asked, my (very niche) usecase: I have a commercial USB where the volume is encrypted with a proprietary technology so the volume is invisible from `lsblk` even though it must be mounted (I may be wrong. But another proprietary software from the same vendor can decrypt some data from this USB, which is why I said the volume must be mounted). Now, we found that this mounted volume gets disconnected in our application often, and a workaround we've found to be stable is to call `/sys/bus/usb/drivers/usb/unbind` and bind against the USB device in question during Linux' boot sequence.

closed 2021-12-04 by 130s

jonasmalacofilho, 2021-12-05:
I have to admit that I don't really understood what happens in your use case.  Anyway, I hope `usb.core.Device.(de|at)tach_kernel_driver()` or, more likely, `usb.core.Device.reset()` help you.

If not, try to find out and provide us a bit more information about the device (at the very least the output of `libusb -v -d<vid>:<pid>`) and both applications (name, libraries they use, etc), and we'll go from there.

_(Just remember that the device handle may no longer valid after a `reset()`... check the libusb doc linked before for more information)._