pyusb/pyusb #402: `ctrl_transfer` seems to behave different in Linux vs Windows

SengerM, 2021-12-20:
I don't have much experience with USB so I may be confusing something, please excuse me if this is the case. I am writing the software to talk to a USB device and have experienced that `device.ctrl_transfer` requires a different format for the data to be sent depending if we are running on Ubuntu or in Windows. This was unexpected to me, please forgive me if this is the expected behavior. In [this SO question](https://stackoverflow.com/questions/70425043/pyusb-usb-core-usberror-errno-5-input-output-error-in-windows) and [the answer](https://stackoverflow.com/a/70425131/8849755) there is more info. Let me know if you want more information on the system, backends, etc. It would be nice to unify what `data_or_wLength` expects to receive independent of the operating system.

jonasmalacofilho, 2021-12-22:
From my experience, one of the versions (sometimes, like that answer, missing padding; sometimes, with too much padding) was already wrong, but one of the operating systems was fixing it for us.

Have you confirmed that Linux really doesn't like the padded version?  (And that the amount of padding is exactly right, according to the USB and device specifications)?

SengerM, 2021-12-22:
I see. So right now I am testing the code in a second Linux machine and here I can add the padding, i.e. `packet = packet + (64-len(packet))*[0]`, and it still works. In the other machine it started working when I (randomly) removed the padding (now I doubt). To summarize my experience:
1. First Linux computer: I had to remove the padding, with padding it did not work. (Now you make me doubt, I will check and confirm this later today.)
2. Windows computer: I had to add it, otherwise it would not work. This is confirmed.
3. Second Linux computer: It is the same with or without padding. This is confirmed.

mcuee, 2021-12-27:
Do you have the "lsusb -vvv" output of the device? Do you know the communication protocol of the device?

SengerM, 2021-12-27:
This is the output of `lsusb -vvv` for this device:
```
Bus 001 Device 003: ID c251:2201 Keil Software, Inc. LASER Driver IJS
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0xc251 Keil Software, Inc.
  idProduct          0x2201 
  bcdDevice            1.00
  iManufacturer           1 LASER Driver 
  iProduct                2 LASER Driver IJS
  iSerial                 3 0001A0000000
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0022
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xc0
      Self Powered
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              4 HID
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.00
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      33
          Report Descriptor: (length is 33)
            Item(Global): Usage Page, data= [ 0x00 0xff ] 65280
                            (null)
            Item(Local ): Usage, data= [ 0x01 ] 1
                            (null)
            Item(Main  ): Collection, data= [ 0x01 ] 1
                            Application
            Item(Global): Logical Minimum, data= [ 0x00 ] 0
            Item(Global): Logical Maximum, data= [ 0xff 0x00 ] 255
            Item(Global): Report Size, data= [ 0x08 ] 8
            Item(Global): Report Count, data= [ 0x40 ] 64
            Item(Local ): Usage, data= [ 0x10 ] 16
                            (null)
            Item(Main  ): Input, data= [ 0x02 ] 2
                            Data Variable Absolute No_Wrap Linear
                            Preferred_State No_Null_Position Non_Volatile Bitfield
            Item(Global): Report Count, data= [ 0x40 ] 64
            Item(Local ): Usage, data= [ 0x10 ] 16
                            (null)
            Item(Main  ): Output, data= [ 0x02 ] 2
                            Data Variable Absolute No_Wrap Linear
                            Preferred_State No_Null_Position Non_Volatile Bitfield
            Item(Global): Report Count, data= [ 0x01 ] 1
            Item(Local ): Usage, data= [ 0x01 ] 1
                            (null)
            Item(Main  ): Feature, data= [ 0x02 ] 2
                            Data Variable Absolute No_Wrap Linear
                            Preferred_State No_Null_Position Non_Volatile Bitfield
            Item(Main  ): End Collection, data=none
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
can't get device qualifier: Resource temporarily unavailable
can't get debug descriptor: Resource temporarily unavailable
Device Status:     0x0001
  Self Powered
```
Still had no time to test with the first Linux computer.

mcuee, 2021-12-28:
Ah this is an HID device. I think you may want to use hidapi and cython-hidapi.

But take note there may be problems with Windows and you may need to use git head.
https://github.com/libusb/hidapi/pull/334

Windows libusb HID backend has one bug which is fixed in the git head but not in libusb-1.0.24 release version.
https://github.com/libusb/libusb/pull/986

mcuee, 2022-01-19:
Close the issue for now. Please re-open if necessary.

closed 2022-01-19 by mcuee