pyusb/pyusb #404: Can't import libusb using chipwhisperer on MacBook M1, Monterey

MrUndead13, 2022-01-04:
Good evening,
I'm trying to use a chipwhisperer on my MacBook M1 and it is using libusb to connect with USB.
When I try to connect to it with :

import chipwhisperer as cw
scope = cw.scope()

I get :
OSError: Could not import libusb dll. Try
pip uninstall libusb1
pip install libusb1

I tried to uninstall it and reinstall it as well as doing the same with brew and libusb but nothing changed.
I also installed pyusb but this didn't change anything.

Could someone help me please ?

MrUndead13, 2022-01-04:
I get this detailed search error : 

OSError: dlopen(/opt/homebrew/opt/libusb/lib/libusb-1.0.dylib, 0x0006): tried: '/opt/homebrew/opt/libusb/lib/libusb-1.0.dylib' (mach-o file, but is an incompatible architecture (have 'arm64', need 'x86_64')), '/usr/local/lib/libusb-1.0.dylib' (no such file), '/usr/lib/libusb-1.0.dylib' (no such file), '/opt/homebrew/Cellar/libusb/1.0.24/lib/libusb-1.0.0.dylib' (mach-o file, but is an incompatible architecture (have 'arm64', need 'x86_64')), '/usr/local/lib/libusb-1.0.0.dylib' (no such file), '/usr/lib/libusb-1.0.0.dylib' (no such file)

jonasmalacofilho, 2022-01-04:
The project you're trying to use appears to require the [`libusb1`](https://pypi.org/project/libusb1/)  Python package, not PyUSB.

closed 2022-01-04 by jonasmalacofilho

mcuee, 2022-01-05:
OP asked the question in libusb here.
https://github.com/libusb/libusb/issues/1040

Potential Answer: 
https://github.com/vpelletier/python-libusb1/issues/53
https://github.com/pyusb/pyusb/issues/355