pyusb/pyusb #405: Help: usb.core.USBError: [Errno 19] No such device (it may have been disconnected)

JartX, 2022-01-11:
Hi! 
i am sending frames to a board and it need reset when receive this frame
My problem is that the board also reset the usb connection and cant connect again without restart the application.

Any way to recover the connection? 

First try after reset usb:
File "/app/lib/python3.9/site-packages/usb/core.py", line 113, in wrapper
    return f(self, *args, **kwargs)
  File "/app/lib/python3.9/site-packages/usb/core.py", line 131, in managed_open
    self.handle = self.backend.open_device(self.dev)
  File "/app/lib/python3.9/site-packages/usb/backend/libusb1.py", line 804, in open_device
    return _DeviceHandle(dev)
  File "/app/lib/python3.9/site-packages/usb/backend/libusb1.py", line 652, in __init__
    _check(_lib.libusb_open(self.devid, byref(self.handle)))
  File "/app/lib/python3.9/site-packages/usb/backend/libusb1.py", line 604, in _check
    raise USBError(_strerror(ret), ret, _libusb_errno[ret])
usb.core.USBError: [Errno 19] No such device (it may have been disconnected)

Second try after reset usb:
Traceback (most recent call last):
  File "/app/lib/python3.9/site-packages/usbserial/Screens/SettingsExpanded/FirmwareShortcutsBox.py", line 56, in upgrade_test_firmware
    FirmwareUpdateManager().init_firmware_update()
  File "/app/lib/python3.9/site-packages/usbserial/USBFrames/FirmwareUpdateManager.py", line 19, in init_firmware_update
    self.usb_serial.init_board()
  File "/app/lib/python3.9/site-packages/usbserial/Services/USBSerialService.py", line 58, in init_board
    raise ValueError('Our device is not connected')
ValueError: Our device is not connected

Note: find result is None

mcuee, 2022-01-11:
Please post your code snippets. Supposedly you have to search for the device again after the reset.

JartX, 2022-01-12:
@mcuee many thanks for your answer :)

1: def init_firmware_update(self):
         print("Inicio Actualización")
         self.usb_serial.init_board()
         print("Versión Firmware 1")
         trama_version = self.usb_serial.send_frame(self.frames.trama_version(), True) //It send reset to board
         print(chr(trama_version[3]) + "."+ chr(trama_version[5])+"."+chr(trama_version[7]))
         #print("Trama Inicia")

         self.usb_serial.send_frame(self.frames.trama_inicia() , True)

         self.usb_serial.reset_dev()
2:  def __init__(self):
        self.trama_actual = bytearray
        self.trama_recibida = bytearray
        self.frames = OutPutFrames()
        self.dev = self.init_board_port()
        print(self.dev)

3: def init_board(self):
        if self.dev is None:
            raise ValueError('Our device is not connected')
        print(self.dev)
        if self.dev.is_kernel_driver_active(0):
            self.dev.detach_kernel_driver(0)

    def reset_dev(self):
        sleep(4) //wait to finish reset and replug on lsusb
        self.dev.reset() //Console Error: usb.core.USBError: [Errno 19] No such device (it may have been disconnected) 
        self.dev = self.init_board_port()
       
Always..: Console Error: usb.core.USBError: [Errno 19] No such device (it may have been disconnected)  only can resolve it, restarting the app

JartX, 2022-01-12:
@mcuee check that point please: 

[54.184981] [00000002] libusb: error [get_usbfs_fd] libusb couldn't open USB device **/dev/bus/usb/001/038**, errno=2
 Bus 001 **Device 038**: ID 04d8:003f Microchip Technology, Inc. (formerly SMSC) USB 2.0 Hub

After replug:
Bus 001 **Device 040**: ID 04d8:003f Microchip Technology, Inc. (formerly SMSC) USB 2.0 Hub

Device 040, but libusb need Device 038... mmmm

jonasmalacofilho, 2022-01-12:
Yes, the device id will change.  That's expected, and you should be able to use `find()` to get a handle to the new device [id], which you must use instead of the old handle.

You should also still be able to catch the exception...

jonasmalacofilho, 2022-01-12:
The odd thing is that according to the libusb1 documentation we should _normally_expect `LIBUSB_ERROR_NOT_FOUND` on `libusb_reset_device()`, but you reported getting `LIBUSB_ERROR_NO_DEVICE` instead.  What libusb version do you have, and what platform are you on?

JartX, 2022-01-12:
@jonasmalacofilho not work :S
libusb 1.0.24, ArchLinux Kernel: Linux arch 5.15.12-zen1-1-zen #1 ZEN SMP PREEMPT Wed, 29 Dec 2021 12:04:52 +0000 x86_64 GNU/Linux
Before replug
<usb.backend.libusb1._LibUSB object at 0x7f78304454c0>
[ 0.034779] [00000002] libusb: debug [libusb_get_device_list]  
[ 0.034783] [00000002] libusb: debug [discovered_devs_append] need to increase capacity
[ 0.034803] [00000002] libusb: debug [libusb_get_device_descriptor]  
[ 0.034834] [00000002] libusb: debug [libusb_get_device_descriptor]  
[ 0.034875] [00000002] libusb: debug [libusb_get_device_descriptor]  
[ 0.034909] [00000002] libusb: debug [libusb_get_device_descriptor]  
[ 0.034940] [00000002] libusb: debug [libusb_get_device_descriptor]  
[ 0.034972] [00000002] libusb: debug [libusb_get_device_descriptor]  
[ 0.035003] [00000002] libusb: debug [libusb_close]  
[ 0.035006] [00000002] libusb: debug [usbi_remove_event_source] remove fd 14
[ 0.035045] [00000002] libusb: debug [libusb_open] open 1.84
[ 0.035056] [00000002] libusb: debug [usbi_add_event_source] add fd 14 events 4
[ 0.035069] [00000002] libusb: debug [libusb_alloc_transfer] transfer 0x5590077449a8
[ 0.035072] [00000002] libusb: debug [libusb_submit_transfer] transfer 0x5590077449a8
[ 0.035075] [00000002] libusb: debug [add_to_flying_list] arm timer for timeout in 1000ms (first in line)
[ 0.035083] [00000002] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 0.035085] [00000002] libusb: debug [handle_events] event sources modified, reallocating event data
[ 0.035086] [00000002] libusb: debug [usbi_wait_for_events] poll() 3 fds with timeout in 60000ms
[ 0.040422] [00000002] libusb: debug [usbi_wait_for_events] poll() returned 1
[ 0.040427] [00000002] libusb: debug [reap_for_handle] urb type=2 status=0 transferred=4
[ 0.040430] [00000002] libusb: debug [handle_control_completion] handling completion status 0
[ 0.040432] [00000002] libusb: debug [arm_timer_for_next_timeout] no timeouts, disarming timer
[ 0.040436] [00000002] libusb: debug [usbi_handle_transfer_completion] transfer 0x5590077449a8 has callback 0x7f7830429a50
[ 0.040439] [00000002] libusb: debug [sync_transfer_cb] actual_length=4
[ 0.040443] [00000002] libusb: debug [libusb_free_transfer] transfer 0x5590077449a8
[ 0.040468] [00000002] libusb: debug [libusb_alloc_transfer] transfer 0x5590077449a8
[ 0.040470] [00000002] libusb: debug [libusb_submit_transfer] transfer 0x5590077449a8
[ 0.040471] [00000002] libusb: debug [add_to_flying_list] arm timer for timeout in 1000ms (first in line)
[ 0.040480] [00000002] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 0.040481] [00000002] libusb: debug [usbi_wait_for_events] poll() 3 fds with timeout in 60000ms
[ 0.056424] [00000002] libusb: debug [usbi_wait_for_events] poll() returned 1
[ 0.056428] [00000002] libusb: debug [reap_for_handle] urb type=2 status=0 transferred=42
[ 0.056432] [00000002] libusb: debug [handle_control_completion] handling completion status 0
[ 0.056435] [00000002] libusb: debug [arm_timer_for_next_timeout] no timeouts, disarming timer
[ 0.056438] [00000002] libusb: debug [usbi_handle_transfer_completion] transfer 0x5590077449a8 has callback 0x7f7830429a50
[ 0.056441] [00000002] libusb: debug [sync_transfer_cb] actual_length=42
[ 0.056444] [00000002] libusb: debug [libusb_free_transfer] transfer 0x5590077449a8
[ 0.056476] [00000002] libusb: debug [libusb_alloc_transfer] transfer 0x5590077449a8
[ 0.056478] [00000002] libusb: debug [libusb_submit_transfer] transfer 0x5590077449a8
[ 0.056480] [00000002] libusb: debug [add_to_flying_list] arm timer for timeout in 1000ms (first in line)
[ 0.056495] [00000002] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 0.056497] [00000002] libusb: debug [usbi_wait_for_events] poll() 3 fds with timeout in 60000ms
[ 0.066421] [00000002] libusb: debug [usbi_wait_for_events] poll() returned 1
[ 0.066429] [00000002] libusb: debug [reap_for_handle] urb type=2 status=0 transferred=20
[ 0.066431] [00000002] libusb: debug [handle_control_completion] handling completion status 0
[ 0.066432] [00000002] libusb: debug [arm_timer_for_next_timeout] no timeouts, disarming timer
[ 0.066434] [00000002] libusb: debug [usbi_handle_transfer_completion] transfer 0x5590077449a8 has callback 0x7f7830429a50
[ 0.066435] [00000002] libusb: debug [sync_transfer_cb] actual_length=20
[ 0.066439] [00000002] libusb: debug [libusb_free_transfer] transfer 0x5590077449a8
[ 0.066463] [00000002] libusb: debug [libusb_get_config_descriptor] index 0
[ 0.066504] [00000002] libusb: debug [libusb_get_config_descriptor] index 0
[ 0.066541] [00000002] libusb: debug [libusb_get_config_descriptor] index 0
[ 0.066577] [00000002] libusb: debug [libusb_get_config_descriptor] index 0
[ 0.066613] [00000002] libusb: debug [libusb_get_config_descriptor] index 0

JartX, 2022-01-12:
@jonasmalacofilho 

mmm reading backend/libusb1.py

def get_backend(find_library=None):
    global _lib, _lib_object
    try:
        if _lib_object is None:
            _lib = _load_library(find_library=find_library)
            _setup_prototypes(_lib)
            _lib_object = _LibUSB(_lib)
        return _lib_object
    except usb.libloader.LibraryException:
        # exception already logged (if any)
        _logger.error('Error loading libusb 1.0 backend', exc_info=False)
        return None
    except Exception:
        _logger.error('Error loading libusb 1.0 backend', exc_info=True)
        return None
        
       Any way to clear: "_lib_object" to force to take a new _LibUSB(_lib)

jonasmalacofilho, 2022-01-12:
> @jonasmalacofilho not work :S

Can you show me the exact program that you tried?

Additionally, can you provide the complete LIBUSB debug output (for the same program)?

JartX, 2022-01-12:
<usb.backend.libusb1._LibUSB object at 0x7f0c2c1a26a0> always have the same memory reference when get_backend() is called, never change that object that use the wrong bus id, can restart or remove the value of that object? 

def get_backend(find_library=None):
global _lib, _lib_object
try:
if _lib_object is None:
_lib = _load_library(find_library=find_library)
_setup_prototypes(_lib)
_lib_object = _LibUSB(_lib)
return _lib_object
except usb.libloader.LibraryException:
# exception already logged (if any)
_logger.error('Error loading libusb 1.0 backend', exc_info=False)
return None
except Exception:
_logger.error('Error loading libusb 1.0 backend', exc_info=True)
return None

this object: _lib_object

jonasmalacofilho, 2022-01-12:
> Code: https://pastebin.com/v1wLF5cU

Note: given that there's no call to `send_frame` in the code you provided, I'm not sure I'm looking at the right thing.  Did you physically reconnect the board between "Desconecta" and "Conecta"?

---

Anyway, can you try something simpler?

- get a valid handle to the device with `find()`
- reset the device
- check the output of `find()` every second for15 seconds

Just as one of the `find()` calls returned a stale-ish handle ("[op_open] open failed with no device, but device still attached"), because the "hotplug monitor thread hasn't processed remove event yet"[^1], it's possible that it may take a while for the attach event to the processed.

[^1]: https://github.com/libusb/libusb/blob/1170db79bdc4155c3d5a64787fcc4f8d56b26174/libusb/os/linux_usbfs.c#L1397-L1398

jonasmalacofilho, 2022-01-12:
> Any way to clear: "_lib_object" to force to take a new _LibUSB(_lib)

You could maybe try monkeypatching `_lib_object` to `None`, so that the next call to `get_backend()` creates a new context.  But it could break the cleanup of the old context, besides not being stable/guaranteed behavior...

JartX, 2022-01-12:
> > Código: https://pastebin.com/v1wLF5cU
> 
> Nota: dado que no hay una llamada `send_frame`en el código que proporcionó, no estoy seguro de estar viendo lo correcto. Volviste a conectar fisicamente la placa entre "Desconecta" y "Conecta"?
> 
> De todos modos, ¿puedes probar algo más simple?
> 
> * obtener un identificador válido para el dispositivo con `find()`
> * restablecer el dispositivo
> * verifique la salida de `find()`cada segundo durante 15 segundos
> 
> Así como una de las `find()`llamadas devolvió un identificador obsoleto ("[op_open] open falló sin dispositivo, pero el dispositivo aún está conectado"), porque el "hilo del monitor de conexión en caliente aún no ha procesado el evento de eliminación" [1](#user-content-fn-1-a565ae18bf7b3a99e79ab5c416aa9377) , es posible que puede tomar un tiempo para que el evento adjunto se procese.
> 
> ## notas al pie
> 1. https://github.com/libusb/libusb/blob/1170db79bdc4155c3d5a64787fcc4f8d56b26174/libusb/os/linux_usbfs.c#L1397-L1398 [flecha_izquierda_con_gancho](#user-content-fnref-1-a565ae18bf7b3a99e79ab5c416aa9377)

Send frame is not sended because am trying the reconnect :)

mmmm delay to try

jonasmalacofilho, 2022-01-13:
> mmmm add delays, ok, go to try it

It's not just about the delays.  I actually would like to know if the device disappears and then appears back again, and get a rough idea of how long it's taking.  That's why I asked you to "check the output of `find()` every second for 15 seconds."

jonasmalacofilho, 2022-01-18:
@JartX, did you manage to perform the test I suggested?

JartX, 2022-01-18:
@jonasmalacofilho yep, only adding delay and timeout is solved it :)

closed 2022-01-18 by JartX