pyusb/pyusb #423: How can I control two same device at the same time

Shimon-Guo, 2022-03-30:
I want to send some comamnd to 2 device at same time using pyusb, how should i do

jonasmalacofilho, 2022-03-30:
Have you tried it and hit an issue?  Or are you asking for general tips on concurrency in Python?

mcuee, 2022-03-30:
> I want to send some comamnd to 2 device at same time using pyusb, how should i do

Maybe you want to elaborate what exactly you want to achieve. Thanks.
Are you asking how to differentiate two identical devices? If the device have unique serial number, that will make things easier.

Shimon-Guo, 2022-03-30:
> > I want to send some comamnd to 2 device at same time using pyusb, how should i do
> 
> Maybe you want to elaborate what exactly you want to achieve. Thanks. Are you asking how to differentiate two identical devices? If the device have unique serial number, that will make things easier.

I use pyusb to send HID evnent to my phone, it work well on there is only one phone, but when I hava 2 same phone, I want to send HID event to both the 2 phones at same time. I try to use mutiprocess, but I find it was blocked by pyusb

jonasmalacofilho, 2022-03-31:
> I try to use mutiprocess, but I find it was blocked by pyusb

Can you elaborate?

Shimon-Guo, 2022-03-31:
> I use pyusb to send HID evnent to my phone, it work well on there is only one phone, but when I hava 2 same phone, I want to send HID event to both the 2 phones at same time. I try to use mutiprocess, but I find it was blocked by pyusb

I use pyusb to send HID evnent to my phone, it work well on there is only one phone, but when I hava 2 same phone, I want to send HID event to both the 2 phones at same time. I try to use mutiprocess, but I find it was blocked by pyusb

jonasmalacofilho, 2022-04-01:
@Shimon-Guo,

It seems you accidentally duplicated a previous comment instead of replying with more information.

mcuee, 2022-04-01:
> I use pyusb to send HID evnent to my phone, it work well on there is only one phone, but when I hava 2 same phone, I want to send HID event to both the 2 phones at same time. I try to use [...]

It is just easier to show your codes here.

jonasmalacofilho, 2022-04-01:
@mcuee,

> BTW, it is better to use HIDAPI and cython-hidapi than libusb/pyusb to talk to HID device.

I honestly curious about where the HID comes in...

Like, is the phone reporting itself as a HID?  Never saw that before...

mcuee, 2022-04-01:
@Shimon-Guo I think your English is okay, just need to write in details of what you want to achieve. Worst case you can use Chinse and I should be able to translate.

Shimon-Guo, 2022-04-02:
@mcuee Actually，I try to use pyusb to achieve GOOGLE AOA protocol (https://source.android.com/devices/accessories/aoa2),
I successful use the protocol to connect an android phone. This protocol think of my computer as an accessory to my phone like keyboard, mouse. So I can transmit a signal from the computer to the phone, for example by clicking on a certain coordinate on the screen, or entering a string of text.
And now, my questions is: it did good on one phone, but I have many of the same phone devices, I want control all of those phones at same time. But I try to conncet 2 phones use pyusb, and start two process, but I failed.

mcuee, 2022-04-02:
Show your codes how you start two processes.

Shimon-Guo, 2022-04-02:
@mcuee 
`from multiprocessing import Process
import xhid

def func_1():
    device = xhid.connect(device_id='43b64863')
    device.swipe(500, 2000, 500, 100, 1000)

def func_2():
    device = xhid.connect(device_id='5DB659YT7X4HKVWC')
    device.swipe(500, 2000, 500, 100, 1000)

process = [Process(target=func_1), Process(target=func_2)]

[p.start() for p in process]
[p.join() for p in process]`

Here is my code, but I found it work now.

Shimon-Guo, 2022-04-02:
@mcuee 
`import time
from multiprocessing import Process
import xhid

all_devices = xhid.connect_all()

def func(device):
    device.click(150, 1800)
    time.sleep(2)
    device.swipe(500, 2000, 500, 100, 1000)

process = [Process(target=func, args=(device,)) for device in all_devices]

[p.start() for p in process]
[p.join() for p in process]
`
but i tied this, it didn't work again.

When I exit, here is the messege:

Traceback (most recent call last):
  File "/home/mi/PycharmProjects/x-hid/test3.py", line 17, in <module>
    [p.join() for p in process]
  File "/home/mi/PycharmProjects/x-hid/test3.py", line 17, in <listcomp>
    [p.join() for p in process]
  File "/usr/lib/python3.6/multiprocessing/process.py", line 124, in join
    res = self._popen.wait(timeout)
  File "/usr/lib/python3.6/multiprocessing/popen_fork.py", line 50, in wait
    return self.poll(os.WNOHANG if timeout == 0.0 else 0)
  File "/usr/lib/python3.6/multiprocessing/popen_fork.py", line 28, in poll
    pid, sts = os.waitpid(self.pid, flag)
KeyboardInterrupt

Shimon-Guo, 2022-04-02:
@mcuee  I think i have found the problem, I use 'while Ture' condition is the func 'swipe', so it was blocked.
```
    def swipe(self, s_x, s_y, e_x, e_y, millis):
        """
        swipe gesture
        :param s_x:
        :param s_y:
        :param e_x:
        :param e_y:
        :param millis:
        :return:
        """
        start = datetime.datetime.now()
        dt = 0
        start_point = Point(s_x, s_y).point_scale(self.width, self.height)
        end_point = Point(e_x, e_y).point_scale(self.width, self.height)
        while dt <= millis:
            delta = datetime.datetime.now() - start
            dt = (delta.days * 24 * 60 * 60 + delta.seconds) * 1000 + delta.microseconds / 1000
            progress = dt / max(millis, 1)
            point_x = int(progress * end_point.x + (1 - progress) * start_point.x)
            point_y = int(progress * end_point.y + (1 - progress) * start_point.y)
            self._touch(touch_type='DOWN', point=BasePoint(point_x, point_y))
            # time.sleep(0.01)
        self._touch(touch_type='UP', point=end_point)
```

jonasmalacofilho, 2022-04-02:
@Shimon-Guo ,

> I think i have found the problem, I use 'while Ture' condition is the func 'swipe', so it was blocked.

I'm glad you seem to have already resolved your issue.

Here are some additional comments about your code:

- If you have any more problems, I would also take a look at `xhid.connect()`, as it's an obvious place for problems that appear when you have two identical devices;
- Having each process connect to its own device (as in you first code sample) is preferable to trying to pass PyUSB devices between processes: this is because PyUSB devices are not pickable, plus you would be sharing a single LibUSB context between different processes, which is not a way LibUSB is supposed to be used (although it can work in some scenarios);
- `KeyboardInterrupt` just means that you manually killed the process with an interrupt signal, e.g. with `Ctrl+C` on Linux.

Closing the issue as the it seems to be resolved.

P.S. In the future, please ask questions using the Discussions tab, Q&A section.

closed 2022-04-02 by jonasmalacofilho