pyusb/pyusb #438: Repeat call to find()

robert-billing, 2022-07-26:
I'm working on a utility which needs to rescan the bus to find devices several times. Doing this often results in the program exiting unexpectedly on windows. This code fragment runs correctly on Linux and Macintosh, but exits prematurely on Windows.

#! /usr/bin/python3

import usb.core

for n in range ( 5 ):
    print ( "=======", n )
    dx = usb . core . find ( find_all = True )
    dy = list ( dx )
    print ( "dy =", dy )
    for d in dy:
        print ( "{:04X} {:02X}" . format ( d . idVendor, d . idProduct ) )

jonasmalacofilho, 2022-07-26:
Hi, thanks for reporting this.

However, I can't seem to reproduce the problem.  Can you it again on your machine with both `PYUSB_DEBUG=debug` and `LIBUSB_DEBUG=4` set on your environment?

Additionally:

- what version of PyUSB are you running?
- what backend library (e.g. libusb-1.0) and version are you using?
- what architecture and version of Windows are you running on?
- what is the Python interpreter version and architecture?

mcuee, 2022-07-30:
No issue here either with libusb-1.0.26 and oyusb 1.2.1 under an virtual enviroment of Python 3.9.7 64bit, Windows 11.

mcuee, 2022-07-30:
I can only guess that ther is an issue with libusb under the paticular system.

jonasmalacofilho, 2022-09-06:
Closing this due to, sadly, a lack of feedback... it seemed that there was indeed something wrong, even if maybe not in PyUSB.

closed 2022-09-06 by jonasmalacofilho