pyusb/pyusb #441: Device._ctx.handle is None unless Device object is printed

alfiecg24, 2022-08-03:
Hi,

I am having a weird bug, where I have a Device object that is the USB device connected.` Device._ctx` is always the correct type, but `Device._ctx.handle` is None ONLY if I don't print the value of the Device object beforehand...

`def libusb1_create_ctrl_transfer(device: "Device", request, timeout):
  print(f"device: {device}")
  print(f"_ctx: {device._ctx}")
  print(f"handle: {device._ctx.handle}")
  assert usb.backend.libusb1._lib

  ptr = usb.backend.libusb1._lib.libusb_alloc_transfer(0)
  assert ptr is not None

  transfer = ptr.contents
  transfer.dev_handle = device._ctx.handle.handle
  transfer.endpoint = 0  # EP0
  transfer.type = 0  # LIBUSB_TRANSFER_TYPE_CONTROL
  transfer.timeout = timeout
  transfer.buffer = request.buffer_info()[0]  # C-pointer to request buffer
  transfer.length = len(request)
  transfer.user_data = None
  transfer.callback = usb.backend.libusb1._libusb_transfer_cb_fn_p(0)  # NULL
  transfer.flags = 1 << 1  # LIBUSB_TRANSFER_FREE_BUFFER

  return ptr`

This returns `Device._ctx.handle` as type `<usb.backend.libusb1._DeviceHandle object at 0x1046ef940>`. Whereas the following returns the error `AttributeError: 'NoneType' object has no attribute 'handle'`.

`def libusb1_create_ctrl_transfer(device: "Device", request, timeout):
  #print(f"device: {device}")
  print(f"_ctx: {device._ctx}")
  print(f"handle: {device._ctx.handle}")
  assert usb.backend.libusb1._lib

  ptr = usb.backend.libusb1._lib.libusb_alloc_transfer(0)
  assert ptr is not None

  transfer = ptr.contents
  transfer.dev_handle = device._ctx.handle.handle
  transfer.endpoint = 0  # EP0
  transfer.type = 0  # LIBUSB_TRANSFER_TYPE_CONTROL
  transfer.timeout = timeout
  transfer.buffer = request.buffer_info()[0]  # C-pointer to request buffer
  transfer.length = len(request)
  transfer.user_data = None
  transfer.callback = usb.backend.libusb1._libusb_transfer_cb_fn_p(0)  # NULL
  transfer.flags = 1 << 1  # LIBUSB_TRANSFER_FREE_BUFFER

  return ptr`

I have tried assigning the value of `device` to a dummy variable but it still won't work. Any ideas, or is this a pyusb bug? Thanks for any help.

jonasmalacofilho, 2022-08-04:
That's not a bug, PyUSB automatically manages things like getting a device handle and claiming an interface, and that's done lazily.  It's the chief purpose of the `self._ctx: _ResourceManager` context.

---

On an unrelated note, can I ask why you need to into PyUSB internals just to manually do a control transfer, instead of calling the public `device.ctrl_transfer(...)` method?

jonasmalacofilho, 2022-08-04:
> PyUSB automatically manages things like getting a device handle and claiming an interface, and that's done lazily.

For completeness, callers are not restricted to this automatic and lazy mechanism when it comes to claiming and releasing interfaces (`usb.util.claim_interface`, `usb.util.release_interface`).

alfiecg24, 2022-08-04:
> On an unrelated note, can I ask why you need to delve into PyUSB internals just to manually do a control transfer, instead of calling the public `device.ctrl_transfer(...)` method?

This isn’t my code, it’s a fork of another project and I’ve just left it how it came 😄

jonasmalacofilho, 2022-08-04:
Ah, I see.  It makes more sense now, it's trying to exploit a race in the USB DFU implementation in some devices...

By the way, you should have linked to the code in the first place, after all it's public.[^1]  Otherwise I'm left without useful information  for no reason... and it's not like many programmers (or even people in general) have an issue with iOS/iPadOS jailbreaking.[^2]

Anyway, the reason why the lazy behavior I described above doesn't affect checkm8 is that `DFUDevice.acquire_device` calls `usb.util.claim_interface`, which triggers the lazy mechanism in PyUSB to get the libusb1 handle.

Also I'm closing this issue since there's no bug and you probably now understand how it works.  But feel free to ask more questions.

[^1]: While I now know this code is from **checkm8,** I'm not sure which specific version/fork you're looking at, so I'm not going to link a random one here and possibly create more noise.
[^2]: And if I did have an issue with it, trying to get me to unknowingly help with it anyway wouldn't be a good solution.

closed 2022-08-04 by jonasmalacofilho