pyusb/pyusb #448: usb.core.find return wrong object

SharePlainCSource, 2022-08-19:
libusb：1.0.26
pyusb：1.2.1
os:windows11
python:3.9.5
driver:libusbK 3.0.7.0

Plugin USB Device->Run Program->Unplug USB Device->Plugin USB Device
[image]

jonasmalacofilho, 2022-08-19:
Hi,

I'm failing to spot what's wrong in that screenshot.  Could you point me to the issue?

SharePlainCSource, 2022-08-22:
> Hi,
> 
> I'm failing to spot what's wrong in that screenshot. Could you point me to the issue?

1.Plugin No.1 USB device to PC
2.Run Program
[image]
3.Plugin No.2 USB device,VID and PID the same as No.1 device
[image]

jonasmalacofilho, 2022-08-22:
Again, what else did you expect the second screenshot to show?

Your code searches for a device with VID:PID 1a86:5512... and you find one.  So what exactly is wrong there?

SharePlainCSource, 2022-08-22:
Sorry my mistake.
I thought the parameter find_all default vale is True.

closed 2022-08-22 by SharePlainCSource